espressif/esp-hosted #172: Any throughput improvement

alireza3137, 2022-11-15:
Is there Any throughput improvement in ESP-hosted-FG vs esp-hosted v3 in ESP32-s2?

mantriyogesh, 2022-11-16:
Also what is the SPI clock you are using right now?

alireza3137, 2022-11-16:
Actually I am using ESP-Hosted for MCU. I am using 30 MHz for SPI clock.

mantriyogesh, 2022-11-16:
Short answer:
No throughput specific improvements are added.

Long answer:
1. There are features & bugfix addition like async control path, where you can send control path requests (for example connect_ap()) asynchronously. 
2. The events like heartbeat (ESP keep alive), station disconnection and softap disconnection will be useful to add a hook to trigger specific actions.
3. Mempool at ESP32 side to minimize allocation timings
Apart from that, there are many bugfixes that have gone in ESP side, which might indirectly be helpful.

What is the current throughput you have achieved?

There would be multiple throughput improvement point I can list in ST side (easiest to hard):
1. ESP32-S2 supports till 40 MHz in SPI slave mode. Step-wise test with higher frequency might help in getting little better speed.
2. Right now, the solution is kept open for any network stack. The network stack in your implementation could be closely coupled to avoid or drastically minimize any memcpy.
4. DMA could be integrated with SPI (examples could be referred from STM Cube MX), this might improve the performace much better
5. Does your MCU support 4 bit SPI? as it is not common use case, we had not added 4 bit SPI yet in ESP-Hosted.

mantriyogesh, 2022-11-16:
If it is feasible, ESP32-C3 provides 60MHz, which also could be useful

alireza3137, 2022-11-16:
I have decribed my software and hardware in issue #27 here is its link:
https://github.com/espressif/esp-hosted/issues/27

I dont think increase SPI clock can improve throughput because I have checked both ESP32 with 10 Mhz and ESP32-S2 with 30 Mhz and no improvment in throughput was achived. 

I have explained the main bottleneck in issue #27 in specific which there is a SPI transaction with empty rx and tx. 

Without remove this transaction I dont think can get better throughput.

mantriyogesh, 2023-01-16:
Hello @alireza3137 

BTW, recent commits have gone in, with which you can test the bare SPI/SDIO throughput using 'RAW Throughout' feature.
Just thought to let you know..
This will be more transparent to measure the performance received at transport level (Without TCP/IP or network stack overhead)