espressif/esp-hosted #173: [esp-hosted-ng] No interface shown on ported Linux platform

MarsTechHAN, 2022-11-16:
Hi, I was trying to port the esp-hosted-ng to our new board, there seems to be no error compared with the official one, but no interface was shown after all the initialize was done.

Our board does not have a reset to esp32 due to some gpio controlling issue; I am unsure if the reset pin was strictly required. I attached all the logs below.

Thanks!

esp console log:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (28) boot: ESP-IDF v4.4.1-472-gc9140caf8c-dirty 2nd stage bootloader
I (28) boot: compile time 21:33:21
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 factory          factory app      00 00 00010000 00100000
I (100) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (107) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (115) boot: End of partition table
I (119) boot: Defaulting to factory image
I (124) boot_comm: chip revision: 3, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=15e00h ( 89600) map
I (172) esp_image: segment 1: paddr=00025e28 vaddr=3ffbdb60 size=0553ch ( 21820) load
I (181) esp_image: segment 2: paddr=0002b36c vaddr=40080000 size=04cach ( 19628) load
I (189) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=769dch (485852) map
I (365) esp_image: segment 4: paddr=000a6a04 vaddr=40084cac size=181d4h ( 98772) load
I (406) esp_image: segment 5: paddr=000bebe0 vaddr=50000000 size=00010h (    16) load
I (420) boot: Loaded app from partition at offset 0x10000
I (421) boot: Disabling RNG early entropy source...
I (432) cpu_start: Pro cpu up.
I (433) cpu_start: Starting app cpu, entry point is 0x4008121c
I (0) cpu_start: App cpu up.
I (449) cpu_start: Pro cpu start user code
I (449) cpu_start: cpu freq: 240000000
I (449) cpu_start: Application information:
I (453) cpu_start: Project name:     network_adapter
I (459) cpu_start: App version:      293978e
I (464) cpu_start: Compile time:     Jul 20 2022 21:33:02
I (470) cpu_start: ELF file SHA256:  573ea08fdc881889...
I (476) cpu_start: ESP-IDF:          v4.4.1-472-gc9140caf8c-dirty
I (483) heap_init: Initializing. RAM available for dynamic allocation:
I (490) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (496) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (502) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (508) heap_init: At 3FFD1498 len 0000EB68 (58 KiB): DRAM
I (514) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (521) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (527) heap_init: At 4009CE80 len 00003180 (12 KiB): IRAM
I (534) spi_flash: detected chip: generic
I (538) spi_flash: flash io: dio
W (542) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (556) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (570) FW_MAIN: *********************************************************************
I (576) FW_MAIN:                 ESP-Hosted Firmware version :: 1.0.1
I (584) FW_MAIN:                 Transport used :: SDIO
I (590) FW_MAIN: *********************************************************************
I (596) FW_MAIN: Supported features are:
I (600) FW_MAIN: - WLAN over SDIO
I (604) FW_BT: - BT/BLE
I (606) FW_BT:    - HCI Over SDIO
I (610) FW_BT:    - BT/BLE dual mode
I (612) FW_MAIN: Capabilities: 0x1d
I (626) system_api: Base MAC address is not set
I (626) system_api: read default base MAC address from EFUSE
I (630) BTDM_INIT: BT controller compile version [5688ed5]
I (634) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (962) FW_MAIN: ESP Bluetooth MAC addr: ac- b-fb-6c-ba-7e

I (962) FW_SDIO_SLAVE: Using SDIO interface
I (1964) FW_SDIO_SLAVE: last reset cause: 1x
I (18772) FW_MAIN: Start Data Path
```

dmesg log:
```
...
[    1.681453] mmc1: Unknown controller version (3). You may experience problems.
[    1.727885] mmc1: SDHCI controller on 4940000.sdhc [4940000.sdhc] using ADMA 64-bit
[    1.735758] sdhci_cdns_read_phy_reg: phy lock successful
[    1.741249] probe of 4940000.sdhc returned 1 after 65019 usecs
[    1.741482] 4950000.sdhc get flash clk:200000000Hz
[    1.746463] sdhci-cdns 4950000.sdhc: Got CD GPIO
[    1.762803] mmc2: Unknown controller version (3). You may experience problems.
[    1.771836] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[    1.784133] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[    1.792634] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[    1.800367] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    1.806041] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[    1.811712] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[    1.819149] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    1.823029] mmc2: SDHCI controller on 4950000.sdhc [4950000.sdhc] using ADMA 64-bit
[    1.824802] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[    1.832356] sdhci_cdns_read_phy_reg: phy lock successful
[    1.837998] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[    1.843266] probe of 4950000.sdhc returned 1 after 101998 usecs
[    1.848929] initcall sdhci_cdns_driver_init+0x0/0x10 returned 0 after 231189 usecs
[    1.848945] calling  hid_init+0x0/0x54 @ 1
[    1.849007] mmc1: new SDIO card at address 0001

...
[   19.032141] calling  esp_init+0x0/0x1000 [esp32_sdio] @ 1451
[   19.238097] esp_probe: ESP network device detected
[   19.243991] probe of sound returned 0 after 219 usecs
[   19.244102] probe of mmc1:0001:1 returned 1 after 6166 usecs
[   19.244251] esp_sdio: probe of mmc1:0001:2 failed with error -22
[   19.250319] probe of mmc1:0001:2 returned 0 after 6197 usecs
[   19.250471] initcall esp_init+0x0/0x1000 [esp32_sdio] returned 0 after 213129 usecs
```

ifconfig -a log:
```
/root # ifconfig -a
eth0      Link encap:Ethernet  HWaddr CE:08:A3:30:A3:49
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:53

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:224 errors:0 dropped:0 overruns:0 frame:0
          TX packets:224 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:16352 (15.9 KiB)  TX bytes:16352 (15.9 KiB)

usb0      Link encap:Ethernet  HWaddr 9E:CA:69:DC:56:A7
          inet addr:10.254.239.1  Bcast:10.255.255.255  Mask:255.0.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3222 errors:0 dropped:106 overruns:0 frame:0
          TX packets:5225 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:233212 (227.7 KiB)  TX bytes:927094 (905.3 KiB)
```

mantriyogesh, 2022-11-16:
Hello @MarsTechHAN 

Thank you for using ESP-Hosted.

> Our board does not have a reset to esp32 due to some gpio controlling issue; I am unsure if the reset pin was strictly required

Yes, the resetpin is required while reloading the driver or resetting ESP chipset.
You might have to configure one 'unused' and 'unclaimed' GPIO using Device Tree of your Linux for resetpin.

But the current problem in the logs is not yet because of resetpin. I doubt that your SDIO is not correctly set-up or configured from trace:

> mmc1: Unknown controller version (3). You may experience problems.

As I can see, CIS tuples are received for mmc1, which is good (Linux detects the ESP device connected)
The probe should not fail as in your log:

> probe of mmc1:0001:1 returned 1 after 6166 usecs

This is something wrong. The probe is failing. Not sure it it is because of,
`mmc1: Unknown controller version (3). You may experience problems.`
any debug logs if added in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/sdio/esp_sdio.c#L713 might help to understand from where the probe is failing. As such there is no return of 'value 1' from esp_probe()

Below command is Raspberry Pi specific. Can you please check similar command for your mmc1 instance and share output?
```
pi@rpi-zhf:~/esp-hosted/host/linux/host_driver/esp32 $ cat /sys/devices/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1\:0001/mmc1\:0001\:1/uevent
SDIO_CLASS=00
SDIO_ID=0092:6666
MODALIAS=sdio:c00v0092d6666
```

MarsTechHAN, 2022-11-19:
Hi, thanks for the reply. 
1. Is the GPIO strictly required? Since it's pretty hard to modify the current board, the chip is a BGA, and we don't have extra IO now, respin the board would take more than 1w. I would like to see if this has some other causes.
2. No worry about the *mmc1: Unknown controller version (3). You may experience problems.*, it is just a false warning. We use another MMC controller for the sd card, and the chip was tested with other SDIO WiFi.
3. MMC info as below
```
/root # find /sys | grep mmc1
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/device
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/subsystem
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/subsystem
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/vendor
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/device
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/subsystem
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/driver
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/class
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/modalias
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/uevent
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/type
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/vendor
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/device
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/subsystem
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/class
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/modalias
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/uevent
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/uevent
/sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/uevent
/sys/class/mmc_host/mmc1
/sys/bus/mmc/devices/mmc1:0001
/sys/bus/sdio/devices/mmc1:0001:1
/sys/bus/sdio/devices/mmc1:0001:2
/sys/bus/sdio/drivers/esp_sdio/mmc1:0001:1
/root # cat /sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/uevent
DRIVER=esp_sdio
SDIO_CLASS=00
SDIO_ID=6666:2222
MODALIAS=sdio:c00v6666d2222
/root # cat /sys/devices/platform/soc/4940000.sdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/uevent
SDIO_CLASS=02
SDIO_ID=6666:3333
MODALIAS=sdio:c02v6666d3333
```

MarsTechHAN, 2022-11-19:
Here is the modified esp_probe:
```C
static int esp_probe(struct sdio_func *func,
				  const struct sdio_device_id *id)
{
	struct esp_sdio_context *context = NULL;
	int ret = 0;

	if (func->num != 1) {
		return -EINVAL;
	}

	printk(KERN_INFO "%s: ESP network device detected\n", __func__);

	context = init_sdio_func(func);

	if (!context) {
		printk (KERN_ERR "Failed to get context\n");
		return -ENOMEM;
	}

	printk(KERN_INFO "%s: init_sdio_func success, ret: %d\n", __func__, ret);

	context->state = ESP_CONTEXT_READY;
	atomic_set(&tx_pending, 0);
	ret = init_context(context);
	if (ret) {
		printk (KERN_ERR "Failed to init context\n");
		deinit_sdio_func(func);
		return ret;
	}

	printk(KERN_INFO "%s: init_context success, ret: %d\n", __func__, ret);

	tx_thread = kthread_run(tx_process, context->adapter, "esp32_TX");

	if (!tx_thread)
		printk (KERN_ERR "Failed to create esp32_sdio TX thread\n");
	
	printk(KERN_INFO "%s: create esp32_sdio TX thread success, ret: %d\n", __func__, ret);

	context->adapter->dev = &func->dev;
	generate_slave_intr(context, BIT(ESP_OPEN_DATA_PATH));

	printk(KERN_INFO "%s: generate_slave_intr success, ret: %d\n", __func__, ret);

#ifdef CONFIG_ENABLE_MONITOR_PROCESS
	monitor_thread = kthread_run(monitor_process, context, "Monitor process");

	if (!monitor_thread)
		printk (KERN_ERR "Failed to create monitor thread\n");
	
	printk(KERN_INFO "%s: create monitor thread success, ret: %d\n", __func__, ret);
#endif

	return ret;
}
```
Log when running probe:
```
[ 1358.792526] calling  esp_init+0x0/0x1000 [esp32_sdio] @ 9369
[ 1358.998173] esp_probe: ESP network device detected
[ 1359.003261] esp_probe: init_sdio_func success, ret: 0
[ 1359.008485] esp_probe: init_context success, ret: 0
[ 1359.013989] esp_probe: create esp32_sdio TX thread success, ret: 0
[ 1359.020290] esp_probe: generate_slave_intr success, ret: 0
[ 1359.026097] esp_probe: create monitor thread success, ret: 0
[ 1359.032273] probe of sound returned 0 after 221 usecs
[ 1359.032390] probe of mmc1:0001:1 returned 1 after 34416 usecs
[ 1359.032557] esp_sdio: probe of mmc1:0001:2 failed with error -22
[ 1359.038644] probe of mmc1:0001:2 returned 0 after 6238 usecs
[ 1359.038782] initcall esp_init+0x0/0x1000 [esp32_sdio] returned 0 after 240397 usecs
```

MarsTechHAN, 2022-11-19:
I was trying to control the reset pin manually, and after some tries, I got this. It seems like resetting after enabling the driver is strictly required. Is there any chance to modify this?
```
/root # [   19.699657] sdhci-cdns 4940000.sdhc: card claims to support voltages below defined range
[   19.709966] mmc1: error -110 whilst initialising SDIO card
[   22.318131] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[   22.330672] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[   22.339293] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   22.347004] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   22.352920] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   22.358718] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   22.366245] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   22.372034] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   22.377819] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   22.383708] mmc1: new SDIO card at address 0001
[   22.388916] esp_probe: ESP network device detected
[   22.394297] esp_probe: init_sdio_func success, ret: 0
[   22.399707] esp_probe: init_context success, ret: 0
[   22.404935] esp_probe: create esp32_sdio TX thread success, ret: 0
[   22.411391] esp_probe: generate_slave_intr success, ret: 0
[   22.417431] esp_sdio: probe of mmc1:0001:2 failed with error -22
[   22.551902]
[   22.551902] Received ESP bootup event
[   22.557088] EVENT: 3
[   22.559310] ESP chipset detected [esp32]
[   22.563239] EVENT: 0
[   22.565466] ESP peripheral capabilities: 0x1d
[   22.779072] ESP Bluetooth init
[   22.782341] Capabilities: 0x1d. Features supported are:
[   22.787626]   * WLAN on SDIO
[   22.790578]   * BT/BLE
[   22.792966]     - HCI over SDIO
[   22.796151]     - BT/BLE dual mode
[   22.799784] EVENT: 1
[   22.801982] esp32: process_fw_data ESP chipset's last reset cause:
[   22.801987] POWERON_RESET
[   22.810932] esp32: ESP Firmware version: 1.0.1
...
/root # ifconfig espsta0 up
spsta0 up
/root # ifconfig
espsta0   Link encap:Ethernet  HWaddr AC:0B:FB:6C:BA:7C
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 5E:85:A1:77:AF:88
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:53

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

usb0      Link encap:Ethernet  HWaddr 82:DE:6E:1B:BA:7C
          inet addr:10.254.239.1  Bcast:10.255.255.255  Mask:255.0.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:186 errors:0 dropped:79 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:66729 (65.1 KiB)  TX bytes:1262 (1.2 KiB)
```

MarsTechHAN, 2022-11-19:
Everything looks good for now. Btw, I was wondering if there is a timeline for the AP (potentially AP+STA) mode?
```
/root # iperf3  -s
warning: this system does not seem to support IPv6 - trying IPv4
-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.86.250, port 27847
[  5] local 192.168.86.20 port 5201 connected to 192.168.86.250 port 27849
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  1015 KBytes  8.31 Mbits/sec
[  5]   1.00-2.00   sec   968 KBytes  7.92 Mbits/sec
[  5]   2.00-3.00   sec  1.04 MBytes  8.76 Mbits/sec
[  5]   3.00-4.00   sec   978 KBytes  8.02 Mbits/sec
[  5]   4.00-5.00   sec  1.03 MBytes  8.61 Mbits/sec
[  5]   5.00-6.00   sec   927 KBytes  7.59 Mbits/sec
[  5]   6.00-7.00   sec   950 KBytes  7.78 Mbits/sec
[  5]   7.00-8.00   sec   737 KBytes  6.04 Mbits/sec
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   9.00-10.00  sec   897 KBytes  7.35 Mbits/sec
[  5]  10.00-10.09  sec  77.0 KBytes  6.74 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.09  sec  8.47 MBytes  7.04 Mbits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201 (test #2)
-----------------------------------------------------------
^Ciperf3: interrupt - the server has terminated
/root # iperf3  -s^C
/root # ping 192.168.1.91^C
/root # iperf3  -c 192.168.1.91
Connecting to host 192.168.1.91, port 5201
[  5] local 192.168.86.20 port 48998 connected to 192.168.1.91 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1.30 MBytes  10.8 Mbits/sec   72   83.4 KBytes
[  5]   1.00-2.00   sec   954 KBytes  7.82 Mbits/sec    1   69.3 KBytes
[  5]   2.00-3.00   sec   954 KBytes  7.82 Mbits/sec    0   77.8 KBytes
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes
[  5]   4.00-5.00   sec   255 KBytes  2.09 Mbits/sec    3   63.6 KBytes
[  5]   5.00-6.00   sec  1.12 MBytes  9.38 Mbits/sec   11   58.0 KBytes
[  5]   6.00-7.00   sec   954 KBytes  7.82 Mbits/sec    0   69.3 KBytes
[  5]   7.00-8.00   sec   764 KBytes  6.26 Mbits/sec    1   59.4 KBytes
[  5]   8.00-9.00   sec   954 KBytes  7.82 Mbits/sec    0   67.9 KBytes
[  5]   9.00-10.00  sec   764 KBytes  6.26 Mbits/sec    0   74.9 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  7.88 MBytes  6.61 Mbits/sec   89             sender
[  5]   0.00-10.00  sec  7.43 MBytes  6.23 Mbits/sec                  receiver

iperf Done.
```

MarsTechHAN, 2022-11-20:
Also, I wondered why it was set to input when releasing the RESET. For us, the reset pin would be dragged low until the driver is loaded. Maybe changing it to set_value HIGH would be better?

```C
static void esp_reset(void)
{
	if (resetpin != HOST_GPIO_PIN_INVALID) {
		/* Check valid GPIO or not */
		if (!gpio_is_valid(resetpin)) {
			printk(KERN_WARNING "%s, ESP32: host resetpin (%d) configured is invalid GPIO\n", __func__, resetpin);
			resetpin = HOST_GPIO_PIN_INVALID;
		} else {
			gpio_request(resetpin, "sysfs");

			/* HOST's resetpin set to OUTPUT, HIGH */
			gpio_direction_output(resetpin, true);

			/* HOST's resetpin set to LOW */
			gpio_set_value(resetpin, 0);
			udelay(200);

			/* Change here to set value */
			gpio_set_value(resetpin, 1);

			printk(KERN_DEBUG "%s, ESP32: Triggering ESP reset.\n", __func__);
		}
	}
}
```

mantriyogesh, 2022-11-21:
Hello @MarsTechHAN ,

Apology for the late reply.
Yes, high -> low -> high also should work. 

WRT input pin, Intention was that this pin is no longer needed as output once driver loaded exactly one time.

closed 2022-11-22 by MarsTechHAN