espressif/esp-hosted #174: Tools for embedded linux alternative to dnsmasq

MrHaiCoder, 2022-11-18:
Environment: esp_host_fg + stm32s3
Question: To start data connection(ethap0), I can connect to the AP successfully using static method. Embedded linux has no dnsmasq tool, currently using udhpd + udhcpd.conf to do dhcp service, but found that xiaomi phone can not automatically get the ip after connection, there is an alternative lightweight tools and configuration to provide? 

without dnsmasq tools, 
sudo dnsmasq --no-daemon --no-resolv --no-poll \
		--dhcp-script=/system/bin/dhcp_announce \
		--dhcp-range=192.168.4.1,192.168.4.20,1h

mantriyogesh, 2022-11-21:
Hello @MrHaiCoder 

Thank you for using ESP-Hosted & narrowing the problem scope with verification with static IP config.

> Environment: esp_host_fg + stm32s3

I am a little confused here. But I assume from discussion, you are using ESP32-S3 with FG solution with Linux host.

> there is an alternative lightweight tools and configuration to provide?

Actually, ESP-Hosted is agnostic with networking tools/stacks used.
You can use any networking tool. I would suggest to use (uninstall-purge-reinstall-verify_default_config) stable tool available with your Linux. But overall it looks to me like some tool's config issue because of which xiaomi phone not getting IP.

It might be worthwhile to try with some other Wi-Fi station, for example other phone / laptop after re-install & re-configure?
Also, extra logging in those tools might help to understand the exact issue.

mantriyogesh, 2022-11-24:
Hello @MrHaiCoder 
Any updates?

MrHaiCoder, 2022-12-05:
Will these configurations made by dnsmasq be transferred to the esp32 side