espressif/esp-hosted #175: Problem with Spi Protocol stack

LuciaferKelvin, 2022-11-22:
hello there.
I have a problem with mcu spi Protocol stack , My mcu work as host .The project have a F429 project with RTOS.
How can I using the spi function without rtos? I had not find out any information with how to framing.
And I am sure that the spi interface is work ok ,I had using ESP_LOGI function log what am I send by mcu.
Could you tell me how to framing and what is the setting step ?

LuciaferKelvin, 2022-11-22:
Thank you very much

LuciaferKelvin, 2022-11-22:
by the way , ESP is ESP32C3-MINI-1

mantriyogesh, 2022-11-23:
@LuciaferKelvin 

Thank you for your query.
I assume F429 means STM32 board like STM32F429ZI.
You can use any RTOS for this. You might have to port the 'ioc' file using STM32CubeMX first.

> How can I using the spi function without rtos?

We have never tried this. but if you are certain, you can go ahead. SPI and GPIOs should be able to work.
Porting from RTOs to No-RTOS may take huge time (if you have not done already).

>  I had not find out any information with how to framing

I did not understand the question correctly. Are you asking about frame formation used ESP-Hosted (in between ESP and STM32)? If yes, you can check https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg#42-transport-layer-communication-protocol & esp_hosted_fg/common/include/adapter.h
https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/common/include/adapter.h#L23-L40

> what is the setting step?

I really have no clue what is meant by this. Should I assume project settings/properties to be used for STM32CubeIDE?
If so, the settings should be same. You just might have to add symbol similar to:
Right click on project, go to project properties-> C/C++ general -> Paths and symbols -> Select tab 'Symbols' -> add your STM32's debug symbol like 'STM32F469xx'. You might have to search and replace 'STM32F469xx' with your new symbol.

Rest change in settings like Station's SSID password etc can be found on the documentation as you follow, https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg#22-setup-with-mcu-host

LuciaferKelvin, 2022-11-23:
Thank you for your response.

>what is setting step?

For example ， if I want to the esp32 working in the station mode，what is the setting steps should I using spi send to esp32.
such as using the struct esp_payload_header and send frame: 
step 1:0x00 0x02 0x0c 0x00 0x00 0x00 0x00
[image]

I have not idea with how to set the commiuncation with esp32 , the frame like step 1 I had send to esp ,and I also loged the reasult,it seems work,but how should i do next.

> You might have to port the 'ioc' file using STM32CubeMX first.

I had try to open the ioc by stmcubeide ,but it only has the base project of stm469 hardware driver.Like the pic.
[image]

> SPI and GPIOs should be able to work.
I am sure that spi is work normal.
This information was loged on the esp project by uart.
[image]

LuciaferKelvin, 2022-11-23:
> ,but how should i do next.
like how to set up the MAC,how to get the SSID,then how to set the SSID,how to set the password and so on.

mantriyogesh, 2022-11-23:
I think better to read up the documentation first. start from https://github.com/espressif/esp-hosted and try to explore esp_hosted_fg and MCU based readme first.
Actually we tried to explain in stepwise manner how is the -
1. Design
2. Architecture
3. Code organization
4. Project settings & tuning
5. Demo arping application

I know it is little time taking, but I assure it is worthwhile to get aligned with documentation.
Then you start to port your solution for your STM32.
We will be here to help you out

LuciaferKelvin, 2022-11-23:
OK，Maybe some important information I had miss, I will read it again.Thank you.

mantriyogesh, 2022-11-24:
Hello @LuciaferKelvin 

Do connect if you face any issues.
Just to brief a little, the hooks are available to integrate your network stack. you can start from a demo task of arp-ping https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/host/stm32/app/app_main.c#L341-L363
and dig down how the packets are sent or received.

The framing and all underlying packing, handling of messages are already taken care. When you just implement the hooks, you can use your network stack for your use case.

There are multiple users who have integrated their own stacks easily, and you can even find references in this project's issues or PRs.

Let us know if you are unsure of any thing, we will definitely try to get you a solution.

LuciaferKelvin, 2022-11-24:
Fine ,Thank you very much.

LuciaferKelvin, 2022-12-01:
hello@mantriyogesh
I had add my function to send the package to ESP.
But ESP always reboot when I send one package.
I used get_ap_scan_list() function in control_utils.c to make the msg package.And I using the thirdpart Lwip,so I just need the real msg for netdev. I have not use the ARP ex. in the demo.
The debug msg as show:
Actual data as follow:
[image]
ESP accpt msg as follow:
[image]
ESP error log as follow:
[image]
ESP reboot log as follow:
[image]

when I log the msg as character: I saw "crltResp",but ESP log that "No registered endpoint for CtrlResp",

Is that means the command I am use is wrong ? I have no idea with thar.But I haven`t change anything when the msg packed.
It is all form the demo.

mantriyogesh, 2022-12-01:
Okay.
1. Can you please show the code you have added a packet in detail?
2. Are you using ESP binaries ? If yes, which version?  In that case, are you using the source code tar bundle available with release at Linux side building?

LuciaferKelvin, 2022-12-01:
1. The code detaill I can`t show here,I can tell the packeting process:
get_ap_scan_list()->test_get_available_wifi()->wifi_app_scan_list(req)->CTRL_SEND_REQ(CTRL_REQ_GET_AP_SCAN_LIST)
->transport_pserial_send(tx_data, tx_len)->serial_drv_write(serial_handle, write_buf, count, &count)->serial_ll_if_g->fops
->write(serial_ll_if_g, buf, in_count)->ESP_SendToSlave(serial_ll_hdl->if_type,serial_ll_hdl->if_num, wbuffer, wlen)
all of the funciton base on demo except ESP_SendToSlave，but that funciton same with demo,I just change the hardware send function.
2.I use VSCODE ESP-IDF to compling the esp project(esp-hosted-release-v0.4),it from https://github.com/espressif/esp-hosted/tree/release/v0.4#22-getting-started-with-mcu-host

mantriyogesh, 2022-12-01:
@ajita02 
Do you think that host is on > 0.4 and ESP on 0.4?
I strongly suspect that because 'ctrlResp' I think is not present in release/v0.4.

mantriyogesh, 2022-12-01:
@LuciaferKelvin 
Can you please let us know output of command `$ git commit` when ran at ESP-Hosted repo where MCU code is built?

LuciaferKelvin, 2022-12-01:
I cant ,cause I download it with zip,not for git clone,Is there any other way to show the inforamtion ?
by the way , ESP base I using the  https://github.com/espressif/esp-hosted/tree/release/v0.4#22-getting-started-with-mcu-host but the mcu base I using the  https://github.com/espressif/esp-hosted which you had told me ,Is there any 
possibility may be cause this problem？

LuciaferKelvin, 2022-12-01:
My network is not very well , it always lost when using "git clone" command ,so I download it with zip. QAQ

closed 2022-12-14 by LuciaferKelvin

LuciaferKelvin, 2022-12-14:
hello there,I had rebuild my project.Now I can commiucate with the espc3.
but I had meet with some new problems now.
I try to start the ap service.
fist I sent command to get the mac,
then I sent the command to set ap config ,
when last command send I saw esp log that ,
[image]
what the ap stop automatically.
Is there any msg need I feed back to esp?

LuciaferKelvin, 2022-12-14:
send success and recive success log is add by me in order to comfirm spi transacation is work .

LuciaferKelvin, 2022-12-14:
[image]
hello there , I had rebuild my project.
Now I can using the normal command to commiuncate with esp.
but when I start the softap service ,the ap sevice has stop automatically like the pic.
Is there any reponse need I feed back to esp?

mantriyogesh, 2022-12-14:
Can you please send:

1. Complete logs at both sides:
ESP - ESP log (textual)
Host - (1) dmesg log or /var/log/kern.log (2) test run log
2. Is it possible to share the application code, to understand if there are any problems (you can use minimal demo application as well, to showcase what you want to do)
3. Is it possible to use FG release [0.0.5](https://github.com/espressif/esp-hosted/releases/tag/release%2Ffg-v0.0.5) -> source code for both ESP flashing and host driver building

LuciaferKelvin, 2022-12-14:
1. here is the esp log .
[esp-log.txt](https://github.com/espressif/esp-hosted/files/10227862/esp-log.txt)
2. no ,it isn`t.for esp host is  parts of the project , i can`t make it become a independent project.
3. time will not allowed to transplant thrid times

LuciaferKelvin, 2022-12-14:
this function I saw that just using two commands to start the ap sevice,one is get the mac replace host save mac.another is start the ap service .
[image]

mantriyogesh, 2022-12-14:
As you are using old code, we need to understand exact code base you are referring to. We have solved many issues in this code part.

Requesting you to share the above three things.

LuciaferKelvin, 2022-12-15:
Thanks for you patient.I had solved this problem.It haven`t any wrong in fact.The antenna is not connected into the esp when I test it.Although the log report "soft ap stop handle stop",but the sevice also start up successfully.

mantriyogesh, 2022-12-15:
@LuciaferKelvin 

Okay. Thanks for the update.
In case you face any issues in future, let us know in this thread.

LuciaferKelvin, 2023-01-04:
hello there, I have meet with a problem.First this problem have nothing to do with ESP.I have no idea to fixed that ,So I want to ask your guys have ever meet with the same problem.
The problem is  PC client can`t reicve the msg  from esp , but I can get the msg by wireshake .In that msg ,all the paramter is right ,and wireshake could distinguish that msg is the second times hand shake msg for PC client,but the PC client can`t recive it .
The esp could recive the first times hand shake msg.
In this pic ,the 192.168.33.1 is the mcu host ,192.168.33.10 is the pc client.
[image]
Do your guys has any idea about this question? Thanks a lot .I am sure that the fire wall is not effect the connection ,it has not intercept the msg.

mantriyogesh, 2023-01-05:
retransmissions look problem.
Is it possible to localize if the packets even reach to ESP (& till what point in host)?

LuciaferKelvin, 2023-01-05:
In the wireshark , I can sure that the ESP recive the first times hand shake msg when PC send that.
then mcu send the second times hand shake msg to the PC by ESP .I am also sure ESP send that msg success because wireshark had capture the second tims msg.But I am not sure if the msg reach in the client, seems the client had not get the second times hand shark msg ,then both of PC and mcu retransmission the first times hand shake msg and the second times hand shake msg.
I do not known the reason why the wireshark can capture the msg but the client can not get the second times msg.
I have no idea to location the msg was lost in which step.

mantriyogesh, 2023-02-07:
Some questions:
1. Where is the wireshark running? Is it sniffer capture?
2. Can you please explain through some block diagram, what you are trying to exactly do?

Once we understand the scenario better, can help in localising the problem.

LuciaferKelvin, 2023-02-07:
1. wireshark is running in the pc .yes .
2. I had solved with the last question which talk to you in the issues.
The new question is in make a tcp/ip connection.
pc is the client, mcu is the server. 
like this 
[image]
them can`t make a tcp/ip connection.

mantriyogesh, 2023-02-07:
Lets us take step by step.

Actually I wanted to know how the PC is connected or communicating to MCU.

I assume,
```
PC <-- wifi -->  MCU host (exposing ESP softAP)
```
i.e.
```
PC <-- wifi --> ESP softAP <== ESP-Hosted (SPI slave) ==> MCU host (SPI master)
```
Is this correct?

LuciaferKelvin, 2023-02-07:
sure ,it is right

mantriyogesh, 2023-02-07:
Now as I understand, MCU host is up with ESP-Hosted.
1. So, control API have started the softAP at MCU host. Can you please get me log at MCU host?
2. How are the IPs assigned at PC and MCU host (static or dhcp)?
3. Are you able to ping 192.168.33.1 (MCU host) from 192.168.33.10 (PC)?

mantriyogesh, 2023-02-07:
4. Have you integrated the network stack at MCU host?

5. I think, use the project as is first, check if `arping` are reached to & fro.
This would confirm if ESP-Hosted is ported and functioning correctly.

6. if step (5) is pass and Integrate the network stack. Test ping first.

LuciaferKelvin, 2023-02-07:
1.Actually I had using the lwip to manager the softap ,what kind of log do you want ?
2.I had add the dhcp server to mcu host , the pc ip is assign by mcu host.
3.yes ,but it must open the lwip cfg
[image]

LuciaferKelvin, 2023-02-07:
and the mcu host work normal, it is the ip assign by mcu host
[image]

mantriyogesh, 2023-02-07:
# Hosted scope
If you see, https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/stm32/app/app_main.c#L318-L332
here we receive packet and manually check if received packet is destined for us? if yes, we just send arping response.

Now your network stack will replace this functionality for handling layer 3 packets.
So if you are able to send/receive ping even with checksum disabled as you pointed, the ESP-Hosted is confirmed to work **fine**.

The network stack porting is kept outside scope as every network stack has some adaptations and limitation.

# Outside Hosted scope (network stack porting)
Caution: A little technical discussion ahead & take it slow to understand.
- Worth to pay attention to alignments used and the packets received are aligned.
- Please pay attention if the network stack wherever discards header and access remaining packet from next byte as as start pointer, should **always be aligned**
  - For example, check network stack, if when ethernet header of 14 bytes is processed & when payload is going to be accessed, it should do adjustments such that it should align to 4 bytes alignment.
  - **Little out of scope for lwip, as this is netx/threadx** , but relatable. Please see https://github.com/azure-rtos/netxduo/issues/17
    - In this case (just for your understanding), in `process_rx_task()`, 

    ```sh
    packet_ptr -> nx_packet_prepend_ptr += 2;
    packet_ptr->nx_packet_append_ptr += 2;
    ```
    could have helped. Anyway, this is one of way to do it, it is upto user how they want to port the network stack

Coming to CHECKSUM disabled problems, you can only say that network stack is ported correctly when you are able to work with all such checksums both, enabled & disabled.

Porting the network stack usually take some time & **expected to face issues like this**. Nonetheless, please think from processor's point of view when accessing the packet at any layer, which might help.

Please do let us know how thing proceed for you.

LuciaferKelvin, 2023-02-07:
ok，thanks for your patient.I am sure than esp is working fine. All this question is base on net protocol stack.
To be honest,It's a little difficult.

mantriyogesh, 2023-02-07:
I understand the pain.
We plan to integrate the sample network stack for users reference. But it would definitely not in very near future because of other priority features & issues.

LuciaferKelvin, 2023-03-03:
[mantriyogesh](https://github.com/mantriyogesh)，hello there, I had done with that project. But this has some point I don`t konw how it works.
1. When esp in sta mode connect into an ap , the ap decide the  authentication mode. If the ap support WEP only and hidden, does the esp could connect to it successful?
2. in the hosted ex. code,When using soft ap mode, the esp only support four authentication mode :ESP_HOSTED_ENCRYPTION_MODE__Type_Open,
ESP_HOSTED_ENCRYPTION_MODE__Type_WPA_PSK,ESP_HOSTED_ENCRYPTION_MODE__Type_WPA2_PSK,ESP_HOSTED_ENCRYPTION_MODE__Type_WPA_WPA2_PSK, Do others authentication mode not support?
3. what are the means about (is_wpa3_supported) and (has_is_wpa3_supported) in _EspHostedCmdSetAPConfig struct ? Does it means ESP could connected ap with all of the  authentication mode,but the soft ap only support four authentication mode when it start server?
 by the way , the esp firmware I have used is Relase V0.0.4.

mantriyogesh, 2023-03-03:
1. External AP setup with WEP and hidden SSID & ESP in sta mode with correct authmode should get connected.
Are you facing any issues? Logs and steps to reproduce will be helpful

2. In softAP mode, ESP supports open and wpa/wpa2/wpa3 in PSK. But as you are on old release, 0.4, you will be on old IDF (mostly 4.0).
Can you please give ESP side full log? I will print IDF version used.

For softAP, idf release/v5.0 should work with open and wpa/wpa2/wpa3 in PSK authmodes. but the [API](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#115-ctrl_cmd_t--wifi_start_softapctrl_cmd_t-req) is not yet updated. I will update it. authmode values can be found: https://github.com/espressif/esp-idf/blob/release/v5.0/components/esp_wifi/include/esp_wifi_types.h#L52-L64

3. Clearly naming, `is_wpa3_supported` to be changed to `use_wpa3`. I couldn't find 'has_is_wpa3_supported' in 0.0.5, may be cleaned up.

Fundamentally SoftAP doesn't support all the values of authmode.
open/wpa/wpa2/wpa3 with PSK combination are only possible with SoftAP.
For station, no such requirement.

By the way, idf 5.0 comes up with minimum or threshold auth mode concept, to avoid TKIP and use CCMP for WPA2 (by default)

mantriyogesh, 2023-03-03:
Want to get notified for any station disconnection happens?
0.0.5 can get you the notifications which you had subscribed to, as they trigger.

We also strongly suggest to migrate to 0.0.5, where many bugfixes are fixed.
Control path design is changed for betterment there.

LuciaferKelvin, 2023-03-03:
1. I haven`t meet with that scene, That just I support if I meet with the scene how can I slove that.
2. that is the esp log
> ESP-ROM:esp32c3-api1-20210207
> Build:Feb  7 2021
> rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fcd6100,len:0x16cc
> load:0x403ce000,len:0x930
> load:0x403d0000,len:0x2d40
> entry 0x403ce000
> [0;32mI (30) boot: ESP-IDF v4.4-rc1-8-g0b46ac1732-dirty 2nd stage bootloader[0m
> [0;32mI (30) boot: compile time 19:43:58[0m
> [0;32mI (30) boot: chip revision: 3[0m
> [0;32mI (34) boot.esp32c3: SPI Speed      : 80MHz[0m
> [0;32mI (39) boot.esp32c3: SPI Mode       : DIO[0m
> [0;32mI (43) boot.esp32c3: SPI Flash Size : 4MB[0m
> [0;32mI (48) boot: Enabling RNG early entropy source...[0m
> [0;32mI (53) boot: Partition Table:[0m
> [0;32mI (57) boot: ## Label            Usage          Type ST Offset   Length[0m
> [0;32mI (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
> [0;32mI (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
> [0;32mI (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
> [0;32mI (87) boot:  3 factory          factory app      00 00 00010000 00100000[0m
> [0;32mI (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
> [0;32mI (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
> [0;32mI (109) boot: End of partition table[0m
> [0;32mI (113) boot: Defaulting to factory image[0m
> [0;32mI (118) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=1d180h (119168) map[0m
> [0;32mI (145) esp_image: segment 1: paddr=0002d1a8 vaddr=3fc90800 size=02e70h ( 11888) load[0m
> [0;32mI (148) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=93eb0h (605872) map[0m
> [0;32mI (246) esp_image: segment 3: paddr=000c3ed8 vaddr=3fc93670 size=00544h (  1348) load[0m
> [0;32mI (246) esp_image: segment 4: paddr=000c4424 vaddr=40380000 size=10624h ( 67108) load[0m
> [0;32mI (264) esp_image: segment 5: paddr=000d4a50 vaddr=50000010 size=00010h (    16) load[0m
> [0;32mI (269) boot: Loaded app from partition at offset 0x10000[0m
> [0;32mI (269) boot: Disabling RNG early entropy source...[0m
> [0;32mI (284) cpu_start: Pro cpu up.[0m
> [0;32mI (293) cpu_start: Pro cpu start user code[0m
> [0;32mI (293) cpu_start: cpu freq: 160000000[0m
> [0;32mI (293) cpu_start: Application information:[0m
> [0;32mI (295) cpu_start: Project name:     network_adapter[0m
> [0;32mI (301) cpu_start: App version:      release0.4-60-g22180f7-dirty[0m
> [0;32mI (308) cpu_start: Compile time:     Jan 24 2022 19:43:50[0m
> [0;32mI (314) cpu_start: ELF file SHA256:  15a671cd67cbaa8d...[0m
> [0;32mI (320) cpu_start: ESP-IDF:          v4.4-rc1-8-g0b46ac1732-dirty[0m
> [0;32mI (327) heap_init: Initializing. RAM available for dynamic allocation:[0m
> [0;32mI (334) heap_init: At 3FC98DA0 len 00027260 (156 KiB): DRAM[0m
> [0;32mI (340) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM[0m
> [0;32mI (347) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM[0m
> [0;32mI (354) spi_flash: detected chip: generic[0m
> [0;32mI (358) spi_flash: flash io: dio[0m
> [0;32mI (362) sleep: Configure to isolate all GPIO pins in sleep state[0m
> [0;32mI (369) sleep: Enable automatic switching of GPIO sleep configuration[0m
> [0;32mI (376) coexist: coexist rom version 9387209[0m
> [0;32mI (381) cpu_start: Starting scheduler.[0m
> [0;32mI (385) NETWORK_ADAPTER: *********************************************************************[0m
> [0;32mI (385) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4                        [0m
> [0;32mI (395) NETWORK_ADAPTER:                 Transport used :: SPI                           [0m
> [0;32mI (405) NETWORK_ADAPTER: *********************************************************************[0m
> [0;32mI (415) NETWORK_ADAPTER: Supported features are:[0m
> [0;32mI (425) NETWORK_ADAPTER: - WLAN over SPI[0m
> [0;32mI (425) ESP_BT: - BT/BLE[0m
> [0;32mI (435) ESP_BT:    - HCI Over UART[0m
> [0;32mI (435) ESP_BT:    - BLE only[0m
> [0;32mI (435) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (445) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (455) gpio: GPIO[8]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (465) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (475) ESP_BT: baud rate for HCI uart :: 921600 
> [0m
> [0;32mI (485) BTDM_INIT: BT controller compile version [6881869][0m
> [0;32mI (485) phy_init: phy_version 907,3369105-dirty,Dec  3 2021,14:55:12[0m
> [0;32mI (545) system_api: Base MAC address is not set[0m
> [0;32mI (545) system_api: read default base MAC address from EFUSE[0m
> [0;32mI (545) BTDM_INIT: Bluetooth MAC: 34:b4:72:f3:f5:62
> [0m
> [0;32mI (545) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 34-b4-72-f3-f5-62
> [0m
> [0;32mI (555) SPI_DRIVER: Using SPI interface[0m
> [0;32mI (555) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (565) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (1575) pp: pp rom version: 9387209[0m
> [0;32mI (1575) net80211: net80211 rom version: 9387209[0m
3. the version of pdf will be update at furtrue, but not now.

LuciaferKelvin, 2023-03-03:
when esp at sta mode , the authmode will be automatic set or need program set it ? I don`t find out any struct value to set it but has_is_wpa3_supported and is_wpa3_supported.
```
struct defintion is that
struct  _EspHostedCmdSetAPConfig
{
  ProtobufCMessage base;
  char *ssid;
  char *pwd;
  char *bssid;
  protobuf_c_boolean has_is_wpa3_supported;
  protobuf_c_boolean is_wpa3_supported;
  protobuf_c_boolean has_listen_interval;
  int32_t listen_interval;
};
```

mantriyogesh, 2023-03-03:
Single line answer: Not needed. If you are using wpa3 authmode, you still need to set `is_wpa3_supported`.

I got the definitions:

- Sta mode:
  - 0.4
    - https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/common/proto/esp_hosted_config.proto#L76-L81

  - 0.0.5:
    - https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/common/proto/esp_hosted_config.proto#L218-L224

- SoftAP
  - 0.4
    - https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/common/proto/esp_hosted_config.proto#L102-L110
  - 0.0.5
    - https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/common/proto/esp_hosted_config.proto#L245-L253
    
    
But you are referring to protobuf layer (lower layer), which is accessed by higher layer user space applications.

The authmode is not required to pass in station mode.

Additional information for you:
IDF 5.0 keeps a provision, of threshold of authmode (in easier words, do not connect to AP even matching SSID, password but lower security) enforcement 
[wifi_sta_config_t](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32c3/api-reference/network/esp_wifi.html#_CPPv417wifi_sta_config_t) -> [wifi_scan_threshold_t](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32c3/api-reference/network/esp_wifi.html#_CPPv421wifi_scan_threshold_t) threshold.
This parameter would be supported in upcoming releases of ESP-Hosted.

LuciaferKelvin, 2023-03-24:
hello yogesh.
I have some question with ota api.
The firmware i used is release v0.0.4.
1.I should use (int esp_ota_begin()) first ,then use (int esp_ota_write(uint8_t* ota_data, uint32_t ota_data_len))to transmit data, finally use (int esp_ota_end()) when I want to upgrade the firmware in esp, Is that right?
2. when using  (int esp_ota_write(uint8_t* ota_data, uint32_t ota_data_len)), what is the start address ?Is the address start on 0x0000?
3. Should I must using combined firmware ?(combined firmware include bootloader ,partition-table, ota data initial,and firmware)Or need I transmit the four bin files time by time?

mantriyogesh, 2023-03-24:
Hello @LuciaferKelvin 

Are you using MCU with 0.4?
With Linux, you can simply use https://github.com/espressif/esp-hosted/blob/81778f63a9d84fa46f8a907078f9e2681b391bf5/esp_hosted_fg/host/linux/host_control/c_support/test.c#L100-L103

For MCU, there are two cases.
1. If you are able to hold the network_adapter.bin fully in MCU, you can use same approach.
2. If size constraints, you can act actually transfer your image and flash part by part. you can refer python application
https://github.com/espressif/esp-hosted/blob/81778f63a9d84fa46f8a907078f9e2681b391bf5/esp_hosted_fg/host/linux/host_control/python_support/py_parse/cmds.py#L299-L314
where http client library is used. For MCU with less memory or when unsure how to hold the image, it is better to go with this approach. You would want to do similar stuff in C if python not supported in MCU (with your existing HTTP client)

mantriyogesh, 2023-03-24:
Just for additional information about OTA: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html

Hosted has done some customisation so that OTA could be done over Wi-Fi itself (with image being transferred part by part over HTTP on existing ESP-Hosted transport (say SPI)). This actually refers to point (2) in last comment.

LuciaferKelvin, 2023-05-29:
hallo, @mantriyogesh ,
I had meet with new problem with firmware v0.4,
In sta mode,
There are probabilities esp can not work normally when the esp power on first time.
It could commiucate with route or other node by arp,but can not make any connection with tcp/ip,or never reponse icmp.
the esp log show it connect route successful.
I had fixed that by reconnect route when esp first time power on ,
there have three step:
1.connected the route
2.send cmd disconnect the route
3.connected the route
it could fixed that problem , but I have no idea about what is the reason make this sence.

mantriyogesh, 2023-05-29:
Hello @LuciaferKelvin 

Can you please explain the scenario with logs (ESP, Host side commands, their output, snippet of commands/ codes you are trying to make work)?
I am still trying to understand at what layer, what exactly is the problem.

LuciaferKelvin, 2023-06-08:
@mantriyogesh 
hello , last question need some time to collect the info .
I have a new questions.
how long will the esp entry the save power mode since recive last message form route in station mode ？
the esp default save power mode is （WIFI_PS_MIN_MODEM）， I found that when i using icmp to ping the sta, the delay time like this .
[image]
but if I do a lot of transmision when using icmp to ping esp , the delay time like this.
[image]
so I think maybe the save power cause that problem , because the icmp ping will send the package every one second.

mantriyogesh, 2023-06-08:
1. Is host system a Linux or microcontroller?
2. Can you change https://github.com/espressif/esp-hosted/blob/8dc9280339e1539ba4a5bae65a77b3fe99c9ce14/esp_hosted_fg/host/linux/host_control/c_support/ctrl_config.h#L61 to `3` (This will still allow modem sleep with WIFI_PS_MIN_MODEM enabled but more frequent wakeups)
3. Disable Min modem sleep, which means modem will always be alert. Please note, ESP on boot enters into WIFI_PS_MIN_MODEM. To disable this behaviour:
- Are you using Bluetooth? If not, you can change the power save value to [WIFI_PS_NONE ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv4N14wifi_ps_type_t12WIFI_PS_NONEE) . ESP-Hosted right now doesn't allow it to be set to None as ESP-Hosted by default allows BT+Wi-Fi, which needs at least WIFI_PS_MIN_MODEM (to switch in between radio freq when either Bluetooth or Wi-Fi is sleeping)
- You might have to manually mandate `esp_wifi_set_ps(WIFI_PS_NONE);` at the end of [app_main.c](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L787)
- Do not change the power save later from host side.
4. If you use microcontroller, also you need to check your network stack, if it behaves fine.

LuciaferKelvin, 2023-06-08:
1. I using microcontroller.
2. not linux
3. not . only using wifi mode . that paras only have three mode:
typedef enum {
    WIFI_PS_MIN_MODEM = 1,
    WIFI_PS_MAX_MODEM,
    WIFI_PS_INVALID,
} wifi_ps_type_t;
4. icmp frame .I recive till I send just cost 2ms .I have test that.
And I want to know How long is the time will the esp entry the save power mode when recive last data from route.
by the way the firmware is release v0.4

mantriyogesh, 2023-06-08:
1. okay
2. 
```
typedef enum {
WIFI_PS_MIN_MODEM = 1,
WIFI_PS_MAX_MODEM,
WIFI_PS_INVALID,
} wifi_ps_type_t;
```

is hosted defined structure (which you can ignore). we are re-structring to re-use this structure from esp-idf .
Anyway at slave side code should have visibility of 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv414wifi_ps_type_t.
from esp-idf code.

just follow point (3) from https://github.com/espressif/esp-hosted/issues/175#issuecomment-1581853363

4. Power save is re-used from ESP-IDF as is. Details could be found from ESP-IDF documentation, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#station-sleep

LuciaferKelvin, 2023-06-08:
ok thanks a lot

LuciaferKelvin, 2023-06-08:
- Call esp_wifi_set_ps(WIFI_PS_NONE) to disable Modem-sleep entirely. This has much higher power consumption, but provides minimum latency for receiving Wi-Fi data in real time. When Modem-sleep is enabled, received Wi-Fi data can be delayed for as long as the DTIM period (minimum power-saving mode) or the listen interval (maximum power-saving mode). Disabling Modem-sleep entirely is not possible for Wi-Fi and Bluetooth coexist mode.

Does WIFI_PS_NONE is WIFI_PS_INVALID?

LuciaferKelvin, 2023-06-08:
is that means it must set at esp firmware code ?

mantriyogesh, 2023-06-08:
yes in case of 0.4, at https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/app_main.c#L755

LuciaferKelvin, 2023-06-08:
it was crashed.
`ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806ba
riscv32-esp-elf-addr2line -pfiaC -e d:\Program\ESP32\esp-hosted-release-v0.4\esp\esp_driver\network_adapter\build\network_adapter.elf 0x403806ba: [WinError 2] ?????????????????????????????????
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16bc
load:0x403cc710,len:0x930
riscv32-esp-elf-addr2line -pfiaC -e d:\Program\ESP32\esp-hosted-release-v0.4\esp\esp_driver\network_adapter\build\network_adapter.elf 0x403cc710: [WinError 2] ?????????????????????????????????
load:0x403ce710,len:0x2d28
riscv32-esp-elf-addr2line -pfiaC -e d:\Program\ESP32\esp-hosted-release-v0.4\esp\esp_driver\network_adapter\build\network_adapter.elf 0x403ce710: [WinError 2] ?????????????????????????????????
entry 0x403cc710
riscv32-esp-elf-addr2line -pfiaC -e d:\Program\ESP32\esp-hosted-release-v0.4\esp\esp_driver\network_adapter\build\network_adapter.elf 0x403cc710: [WinError 2] ?????????????????????????????????
I (35) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (35) boot: compile time 16:21:57
I (35) boot: chip revision: 4
I (37) boot_comm: chip revision: 4, min. bootloader chip revision: 3
I (44) boot.esp32c3: SPI Speed      : 80MHz
I (49) boot.esp32c3: SPI Mode       : DIO
I (54) boot.esp32c3: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (82) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (90) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  3 factory          factory app      00 00 00010000 00100000
I (105) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (112) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (120) boot: End of partition table
I (124) boot: Defaulting to factory image
I (129) boot_comm: chip revision: 4, min. application chip revision: 3
I (136) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=1c908h (117000) map
I (163) esp_image: segment 1: paddr=0002c930 vaddr=3fc92a00 size=03328h ( 13096) load
I (165) esp_image: segment 2: paddr=0002fc60 vaddr=40380000 size=003b8h (   952) load
I (169) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=95980h (612736) map
I (273) esp_image: segment 4: paddr=000c59a8 vaddr=403803b8 size=12510h ( 75024) load
I (287) esp_image: segment 5: paddr=000d7ec0 vaddr=50000010 size=00010h (    16) load
I (293) boot: Loaded app from partition at offset 0x10000
I (293) boot: Disabling RNG early entropy source...
I (307) cpu_start: Pro cpu up.
I (315) cpu_start: Pro cpu start user code
I (316) cpu_start: cpu freq: 160000000
I (316) cpu_start: Application information:
I (318) cpu_start: Project name:     network_adapter
I (324) cpu_start: App version:      1
I (328) cpu_start: Compile time:     Jun  8 2023 16:20:30
I (334) cpu_start: ELF file SHA256:  bc34575f258c9b20...
I (340) cpu_start: ESP-IDF:          v4.4.3-dirty
I (346) heap_init: Initializing. RAM available for dynamic allocation:
I (353) heap_init: At 3FC9B020 len 000416F0 (261 KiB): DRAM
I (359) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (366) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (373) spi_flash: detected chip: generic
I (377) spi_flash: flash io: dio
I (381) sleep: Configure to isolate all GPIO pins in sleep state
I (388) sleep: Enable automatic switching of GPIO sleep configuration
I (395) coexist: coexist rom version 9387209
I (400) cpu_start: Starting scheduler.
I (404) NETWORK_ADAPTER: *********************************************************************
I (404) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4
I (414) NETWORK_ADAPTER:                 Transport used :: SPI
I (424) NETWORK_ADAPTER: *********************************************************************
I (434) NETWORK_ADAPTER: Supported features are:
I (444) NETWORK_ADAPTER: - WLAN over SPI
I (444) ESP_BT: - BT/BLE
I (454) ESP_BT:    - HCI Over SPI
I (454) ESP_BT:    - BLE only
[... 52 lines cut ...]
3fcb2d90: 0x3fc9ba40 0x3fc9baa8 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
3fcb2db0: 0x4208217c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
riscv32-esp-elf-addr2line -pfiaC -e d:\Program\ESP32\esp-hosted-release-v0.4\esp\esp_driver\network_adapter\build\network_adapter.elf 0x4208217c: [WinError 2] ?????????????????????????????????
3fcb2dd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2df0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2e10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2e30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2e50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2e70: 0x00000000 0x00000000 0x00000000 0x3f000000 0x00000054 0x3fcb2e84 0x3fcb2e84 0x3fcb2e84
3fcb2e90: 0x3fcb2e84 0x00000000 0x3fcb2e9c 0xffffffff 0x3fcb2e9c 0x3fcb2e9c 0x00000000 0x3fcb2eb0
3fcb2eb0: 0xffffffff 0x3fcb2eb0 0x3fcb2eb0 0x00000000 0x00000001 0x00000000 0x6600ffff 0x00000000
3fcb2ed0: 0xb33fffff 0x00000000 0x00000820 0x6f6d706f 0x00006564 0x2e617473 0x64697373 0x00000000
[... cut ...]
3fcb2f70: 0x00000001 0x3fc98ce2 0x2e617473 0x64777370 0x00000000 0x2e617473 0x006b6d70 0x2e617473
3fcb2f90: 0x6e616863 0x00000000 0x00410603 0x00000000 0x00000000 0x00000000 0x3fc98ce3 0x2e617473
3fcb2fb0: 0x006b6d70 0x2e617473 0x6e616863 0x00000000 0x6f747561 0x6e6f632e 0x0000006e 0x00200704
3fcb2fd0: 0x00000000 0x00000000 0x00000000 0x3fc98d24 0x2e617473 0x6e616863 0x00000000 0x6f747561
3fcb2ff0: 0x6e6f632e 0x0000006e 0x69737362 0x65732e64 0x00010005 0x00000000 0x0000000e 0x00000000

ELF file SHA256: bc34575f258c9b20

Rebooting...`

LuciaferKelvin, 2023-06-08:
[image]

[44 more replies]