espressif/esp-hosted #177: esp_hosted_ng: insmod esp32_sdio.ko No network card is generated

panbijub63, 2022-11-23:
Now the module can be loaded, but the network card has not been generated. Help with the analysis. thank you
[image]

mantriyogesh, 2022-11-23:
Please port https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/rpi_init.sh for your Linux.
resetpin is mandatory.

Can you please send the texual logs:
- Linux:
  - /var/log/kern.log
  - terminal (running or testing script) log
- ESP
  - idf.py flash monitor log (if using surces)
  - minicom if using binary

If in case you are using release (say 1.0.1), Please make sure that you use source code zip from the release for Linux building.

mantriyogesh, 2022-11-23:
1. rpi_init.sh will not be found on your platform. It is created for ESP-Hosted project only.
It is named like that, to understand that it is to be changed as per your Linux.
Please port this file. It has to be adjusted as per your platform.

2. resetpin has to be configured for your platform and used in the rpi_init.sh accordingly.

3. Porting guide could be used from: 
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md and
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md

4. external SD card pull up registers might also be required. Check : https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

panbijub63, 2022-11-26:
This is used esp_hosted_fg, but in the end I couldn't find the driver entrance. I have no problem using the previous driver, but the wifi speed will slow down in a few days. So I want to try a new driver now, but I can't find the driver entrance now.

1.Driving results:
 [image] 

2.The old driver was successfully connected to wifi
[image]
[image]

[Old drive-esp32.zip](https://github.com/espressif/esp-hosted/files/10095552/Old.drive-esp32.zip)

mantriyogesh, 2022-11-26:
1. Can you please what is git commit at both, ESP & host, when you use (a) old driver (b) new driver?
It is imp that ESP and host point to same git commit in either case.

2. Can you please add the ESP Logs in these both scenarios?

3. While running both scenarios (and every new run), is the ESP manually reset (as there is no resetpin in commands)?

panbijub63, 2022-11-26:
Which pin is the reset pin connected to the esp

mantriyogesh, 2022-11-27:
[image]

panbijub63, 2022-11-28:
Hello, is the speed about 500KB/s normal

mantriyogesh, 2022-11-28:
Hello @panbijub63 ,

Great that the ESP-Hosted is up at your place.

As you are using ESP32 over SDIO transport, 500K**B**ps meaning 4Mbps, looks much less speed.
Can you please attach the logs in my first response? In addition, please attach the iperf test runs (including iperf test commands at both ends) as well.

mantriyogesh, 2022-11-28:
Speed totally depends upon the SDIO freq you use.
What is the output of:
$ cat /sys/kernel/debug/mmc2/ios

panbijub63, 2022-11-28:
Hello, I don't know how to extract esp32 logs

cat /sys/kernel/debug/mmc2/ios
clock:          50000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    2 (sd high-speed)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
[iperf .txt](https://github.com/espressif/esp-hosted/files/10101891/iperf.txt)

mantriyogesh, 2022-11-28:
The freq is set fine.

What is your throughput requirement?

As I understand, you have tested this over the air. The throughput is shaky in the run log. At times, it had crossed even 14Mbps. 
Can you please not include '-w' option in iperf commands?

The performance numbers we measure inside the shilded box (no interference):
https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg#5-throughput-performance

In case you do not have shield box, some things to consider while taking performance test:

1. Place Station and AP kept reasonable close (say 3-5 feet)
2. Prefer home environment than office environment as office environments are generally very noisy for Wi-Fi
3. Try to limit any interference from Bluetooth, AP or Microwave. Try to be little distant from these technologies
4. Prefer AP than Mobile hotspot
5. You can set 'auto' in Wi-Fi channel so that AP will automatically use the least noisy channel. You might have to login in your AP to set these configs.
6. 5MHz bandwidth is better option than to use 2.4MHz. It will avoid much noise.

panbijub63, 2022-12-01:
I replaced the firmware：（https://github.com/espressif/esp-hosted/releases）, but the module was not recognized.

panbijub63, 2022-12-01:
[image]
Watch the released video. There is ESP-Hosted-NG-raloase_*.tgz firmware. I can't find it here

panbijub63, 2022-12-01:
I used the ESP-Hosted-FG-raloase_*.tgz firmware to load the esp32_sdio_ng.ko driver, and the following errors occurred

insmod esp32_sdio.ko :
 esp_probe: ESP network device detected
 esp_handle_isr: CMD53 read/write error at 98
 esp_sdio: probe of mmc2:0001:2 failed with error -22

Mr-Bossman, 2022-12-01:
hi i also get this same -EINVAL issue using the latest it seems to work on rasbian but when I compile it with Buildroot it breaks. It may be a dep of the kernel as the code is the same and it runs fine on rasbian for me.
```
[  617.601619] esp_sdio: probe of mmc1:0001:1 failed with error -110
[  617.610029] esp_sdio: probe of mmc1:0001:2 failed with error -110
[  618.331945] mmc1: card 0001 removed
[  618.454866] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  618.473162] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  618.485908] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  618.496622] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  618.504858] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  618.512454] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  618.523142] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  618.530487] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  618.538482] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  618.546012] mmc1: new SDIO card at address 0001
[  618.553794] esp_probe: ESP network device detected
[  618.568256] end of esp_probe
[  618.573236] esp_sdio: probe of mmc1:0001:2 failed with error -22
```

mantriyogesh, 2022-12-01:
Hello @panbijub63 

I am a little confused here.

In general, there are two types of solution. First generation & Next Generation.

If you use firmware binaries from First generation (For example, **0**.0.5 release), you need to use mapping source code from release for Linux building.

Similarly, if you use Binaries from Next generation (for example release **1**.0.1), please use mapping source code from the release for Linux building.

First digit in release say `X` in 'X.Y.Z'  differentiates the Generation. If X is 0, It is first generation. If X is 1, it is next generation.

Regarding the image you had posted, For NG release 1.0.1, please go to releases page and scroll down to tag [ESP-Hosted-NG Release 1.0.1](https://github.com/espressif/esp-hosted/releases/tag/release%2Fv1.0.1) just open the fold of Assets, you can find the tarballs there. [ESP-Hosted-NG-release_1.0.1.tgz](https://github.com/espressif/esp-hosted/releases/download/release%2Fv1.0.1/ESP-Hosted-NG-release_1.0.1.tgz) refers to the ESP binaries to flash & mapped [Source code (tar.tgz)](https://github.com/espressif/esp-hosted/archive/refs/tags/release/v1.0.1.tar.gz) to be used for Linux building.

**FG** firmware is not supported with **NG** kernel driver and vice versa.

Hope this helps.

Mr-Bossman, 2022-12-01:
> Hello @panbijub63
> 
> I am a little confused here.
> 
> In general, there are two types of solution. First generation & Next Generation.
I think this is a typo
> If you use firmware binaries from First generation (For example, **0**.0.5 release), you need to use mapping source code from release for Linux building.
>
> Similarly, if you use Binaries from Next [...]

mantriyogesh, 2022-12-01:
> hi i also get this same -EINVAL issue using the latest it seems to work on rasbian but when I compile it with Buildroot it breaks. It may be a dep of the kernel as the code is the same and it runs [...]

Hello @Mr-Bossman ,
Actually, you are getting **-110** error. Can you please let us know if you have connected the pull ups as mentioned in initial posts of this thread?
Also can you please let me know what the kernel version you are using?

> but when I compile it with Buildroot it breaks

Can you please let me know the steps, I can try the same, if it helps.

>  esp_sdio: probe of mmc1:0001**:2** failed with error -22

The `:2` says it is 'function 2'. There are two functions, function 1 and function 2.
Function 2 we are not using, so please ignore it.
For Function 1, you get issue of -110. Which is [Timeout](https://elixir.bootlin.com/u-boot/latest/source/include/linux/errno.h#L118). Please check SD pull ups. If you are using jumper cables to connect ESP and Host, please use good quality & very small wires (<6cm) as otherwise SDIO protocol faces signal integrity issues.

Mr-Bossman, 2022-12-02:
> Hello @Mr-Bossman , Actually, you are getting **-110** error. Can you please let us know if you have connected the pull ups as mentioned in initial posts of this thread? Also can you please let me know what the kernel version you are using?

I am using 5.10 tag.
The pullups should be on the esp32-wrover-e as far as I'm aware as it works fine in rasbian.
https://github.com/Mr-Bossman/buildroot/tree/dev/add-esp-hosted-ng

> 
> > but when I compile it with Buildroot it breaks
> 
> Can you please let me know the steps, I can try the same, if it helps.

```bash
git clone -b dev/add-esp-hosted-ng https://github.com/Mr-Bossman/buildroot
cd buildroot
make raspberrypi0w_defconfig
make -j$(($(nproc)/2))
sudo dd if=output/images/sdcard.img of=/dev/sdX oflag=sync status=progress bs=4M; sync 
```
> 
> > esp_sdio: probe of mmc1:0001**:2** failed with error -22
> 
> The `:2` says it is 'function 2'. There are two functions, function 1 and function 2. Function 2 we are not using, so please ignore it. 

what does this mean where are their declarations?

> For Function 1, you get issue of -110. Which is [Timeout](https://elixir.bootlin.com/u-boot/latest/source/include/linux/errno.h#L118). Please check SD pull ups. If you are using jumper cables to [...]

mantriyogesh, 2022-12-02:
@Mr-Bossman 

Thank you for all the details.

1. Are you using the PCB or wires (<6cm)?
2. Can you please if it ends for function 1 at the end of probe?
3. Can you please add complete dmesg?
4. As earlier pointed, you are able to run this directly on Raspbian, but when you use the same with buildroot, it doesn't work. Correct?
5. Is there kernel version change in these two? If yes, is it possible to test same kernel version at both ends (for a clearer comparison)?

mantriyogesh, 2022-12-02:
Actually, If it is getting only issues in ISR, it means probe is complete. But then why you get, 
```sh
esp_sdio: probe of mmc1:0001:1 failed with error -110
```
sometimes. When it got -110, card removed and tried to probe again by Linux. Mostly it boils to the connection instability or signal integrity issues. Wires got loose or something like that?

> The pullups should be on the esp32-wrover-e as far as I'm aware as it works fine in rasbian.

I referred SD lines pull up requirements. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html where there was no listing of your device, WROVER-E. I am cross-checking this with driver team internally, if external pull-ups required or not. Have you added the pull-ups in your setup?

Further, it would be helpful for us to understand difference in setups.
Is it possible to try to build root with downgraded kernel version, 5.10 as you had on Raspbian (I assume on Raspbian, **same** setup works fine till getting station up and ping working fine)?

Mr-Bossman, 2022-12-02:
> @Mr-Bossman
> 
> Thank you for all the details.
> 
> 1. Are you using the PCB or wires (<6cm)?

I'm using jumper cables they are 8cm. I will make them shorter if need be.
> 2. Can you please if it ends for function 1 at the end of probe?

what is  function 1? I put a print at esp_hosted_ng/host/sdio/esp_sdio.c#L755 and it exits 0 and gets called once
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/sdio/esp_sdio.c#L755
> 3. Can you please add complete dmesg?

> 4. As earlier pointed, you are able to run this directly on Raspbian, but when you use the same with buildroot, it doesn't work. Correct?

yes, with all the same HW. This is why I believe the esp32-wrover-e has the resistors
> 5. Is there kernel version change in these two? If yes, is it possible to test same kernel version at both ends (for a clearer comparison)?
> Further, it would be helpful for us to understand difference in setups. Is it possible to try to build root with downgraded kernel version, 5.10 as you had on Raspbian (I assume on Raspbian, **same** setup works fine till getting station up and ping working fine)?

yes, I can ping using rasbian. Originally I had the error on 5.10, but then used 5.15 because rasbian uses 5.15.61+

> I referred SD lines pull up requirements. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html where there was no listing of your device, [...]

I will add them.

Mr-Bossman, 2022-12-02:
[image]

Mr-Bossman, 2022-12-03:
New issue!!! any ideas

[image]

mantriyogesh, 2022-12-03:
We will check this issue on priority and let you know.

Can you check if wpa_supplicant instance is running or non running before?

Mr-Bossman, 2022-12-03:
it is not
[image]
[build.zip](https://github.com/espressif/esp-hosted/files/10145486/build.zip)

mantriyogesh, 2022-12-03:
This again works on Raspberry Pi and not on build root. Correct?

Mr-Bossman, 2022-12-03:
> This again works on Raspberry Pi and not on build root. Correct?
yes
[image]

Mr-Bossman, 2022-12-04:
the new issue above also hapens on the rasbian too some times
[image]

panbijub63, 2022-12-06:
I am based on c_ Support generates the (conn wifi) bin file。
Some errors occurred when opening wifi。
[rk3399.txt](https://github.com/espressif/esp-hosted/files/10163486/rk3399.txt)
[esp32.txt](https://github.com/espressif/esp-hosted/files/10163485/esp32.txt)

mantriyogesh, 2022-12-06:
What is the esp32 are you using?
Have you connected the pull ups?

panbijub63, 2022-12-06:
[image]
[JLD109_V1.2.pdf](https://github.com/espressif/esp-hosted/files/10163629/JLD109_V1.2.pdf)
The pull-up is already connected

panbijub63, 2022-12-06:
Hello, do you have any suggestions

ajita02, 2022-12-06:
Hi @panbijub63,
From your attached logs two issues has observed.
> 1.  [   39.051844] read_packet: Failed to read data - -84 [0 - 20]

add below changes on esp side and test it. Use esp-idf release/v5.0 branch.
> diff --git a/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c b/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c
index 7585021..81eea0f 100644
--- a/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c
+++ b/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c
@@ -187,6 +187,8 @@ static interface_handle_t * sdio_init(void)
                   */
                //.flags              = SDIO_SLAVE_FLAG_INTERNAL_PULLUP,
                .flags              = SDIO_SLAVE_FLAG_DEFAULT_SPEED,
 +       //      .timing = SDIO_SLAVE_TIMING_PSEND_PSAMPLE,
 +               .timing = SDIO_SLAVE_TIMING_NSEND_PSAMPLE,
        };
        sdio_slave_buf_handle_t handle;
 

> 2.  Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
wlan0: Failed to initialize driver interface

you should not use wlan0 in "conn-wifi". wlan0 is not up in ifconfig.

mantriyogesh, 2022-12-06:
Thought to point out, that I'm really not sure the effect of attaching pull up on CLK pin.

Mr-Bossman, 2022-12-06:
> Thought to point out, that I'm really not sure the effect of attaching pull up on CLK pin.

It doesn't change anything.

mantriyogesh, 2022-12-07:
What is the /var/log/kern.log after insmod?
Is the capability event received?

panbijub63, 2022-12-07:
Hello, my buildroot system does not have this file, and all my logs can be printed directly
[image]

[log.txt](https://github.com/espressif/esp-hosted/files/10172471/log.txt)

Mr-Bossman, 2022-12-07:
Please `cat /var/log/kern.log`
Please also provide the speed you are running the sdio device at. Or provide the device tree.

panbijub63, 2022-12-07:
[image]
My sdio speed seems wrong

mantriyogesh, 2022-12-07:
To change frequency less than that of max set in device tree (25MHz in your case) right from within the ESP kernel module,

Just add, 
```sh
struct mmc_host *host = func->card->host;

host->ios.clock = 1* 1000000; /*1MHz*/
host->ops->set_ios(host, &host->ios);
```
after init_sdio_func() is successful at https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L748

Credits: @Mr-Bossman for code from  https://github.com/espressif/esp-hosted/pull/181

mantriyogesh, 2022-12-07:
1. 
> Hello, my buildroot system does not have this file, and all my logs can be printed directly [image]
> 
> [log.txt](https://github.com/espressif/esp-hosted/files/10172471/log.txt)

Missed this earlier.
As I can understand from your log, First (capability) event has not received.
First event should look something like this:

```sh
[   77.596173] mmc2: card 0001 removed
[   77.649578] mmc2: queuing unknown CIS tuple 0x01 (3 bytes)
[   77.657019] mmc2: queuing unknown CIS tuple 0x1a (5 bytes)
[   77.660243] mmc2: queuing unknown CIS tuple 0x1b (8 bytes)
[   77.662448] mmc2: queuing unknown CIS tuple 0x80 (1 bytes)
[   77.662545] mmc2: queuing unknown CIS tuple 0x81 (1 bytes)
[   77.662643] mmc2: queuing unknown CIS tuple 0x82 (1 bytes)
[   77.664445] mmc2: queuing unknown CIS tuple 0x80 (1 bytes)
[   77.664541] mmc2: queuing unknown CIS tuple 0x81 (1 bytes)
[   77.664637] mmc2: queuing unknown CIS tuple 0x82 (1 bytes)
[   77.664832] mmc2: new SDIO card at address 0001
[   77.665287] esp_probe: ESP network device detected
[   77.877892] Features supported are:
[   77.877901] 	 * WLAN
[   77.877906] 	 * BT/BLE
[   77.877911] 	   - HCI over UART
[   77.877916] 	   - BT/BLE dual mode
[   78.096179] esp_sdio: probe of mmc2:0001:2 failed with error -22
```

2. 
From this thread,

> I am based on c_ Support generates the (conn wifi) bin file。
> Some errors occurred when opening wifi。
> [rk3399.txt](https://github.com/espressif/esp-hosted/files/10163486/rk3399.txt)
> [esp32.txt](https://github.com/espressif/esp-hosted/files/10163485/esp32.txt)

Log in esp shows,

>  I (7179) NETWORK_ADAPTER: Start Data Path

means the scratch registers are getting set fine.

Can you please wrap

https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L741

Like this:
```sh
#if CONFIG_ESP_SPI_HOST_INTERFACE
    datapath = 1;
#endif
```
and try again?

mantriyogesh, 2022-12-07:
Have you added reset pin? In dt and in script?

panbijub63, 2022-12-07:
[image]

复位是硬件进行的，没有在软件上进行复位，不知道这样合不合理
The reset is performed by hardware, but not by software. I don't know whether this is reasonable

Mr-Bossman, 2022-12-07:
it should be done by software. the sdio finds the device via hotplug. so if there is a large delay the esp won't respond.

mantriyogesh, 2022-12-07:
> [image] [test.txt](https://github.com/espressif/esp-hosted/files/10173584/test.txt) It's already working. I'll further test it. I changed the sdio speed to 50MHz

That's great news ! Let us know should you face any issues.

> The reset is performed by hardware, but not by software. I don't know whether this is reasonable

Yes, that's indeed a limitation. We are working to avoid extra 'resetpin' and use internal reset APIs for SDIO. But by the time, we have to use this way.

mantriyogesh, 2022-12-07:
Hello @Mr-Bossman, yes you are correct.

> 
> if (!request->ssids) {
>        printk (KERN_ERR "esp32: %s: Failed skbn is null", __func__);
> }
> 
> scan_req = (struct scan_request *) (cmd_node->cmd_skb->data +
>                 sizeof(struct esp_payload_header));
> 
> /* TODO: Handle case of multiple SSIDs or channels */
> if (**request->ssids &&** request->ssids[0].ssid_len) {
> // **OR**
> // if (**request->n_ssids &&** request->ssids[0].ssid_len) {
>         memcpy(scan_req->ssid, request->ssids[0].ssid, MAX_SSID_LEN);
> 
> }

Should get it working

We can safely remove 
```
if (!request->ssids) {
        printk (KERN_ERR "esp32: %s: Failed skbn is null", __func__);
}
```

mantriyogesh, 2022-12-07:
> the new issue above also hapens on the rasbian too some times [image]

This looks a different issue, although.
https://elixir.bootlin.com/linux/v5.15.76/source/net/wireless/sme.c#L756

panbijub63, 2022-12-08:
@mantriyogesh I just set the device to the appropriate sdio speed. 25Mhz doesn't work for me. Setting sdio speed to 50Mhz works.
> @panbijub63 For educational purpose, can you please share what made it work?

Mr-Bossman, 2022-12-08:
> @mantriyogesh I just set the device to the appropriate sdio speed. 25Mhz doesn't work for me. Setting sdio speed to 50Mhz works.
That's very funny because it's usually decreasing the clock fixes it because of signal integrity. But nice to know that my fix also fixed your problem too. I had to lower my clock rate to 20Mhz for it to work. I used the dt-overlay first to fix it.

panbijub63, 2022-12-08:
[image]
Another problem is when the esp updates the firmware. I need to pull the bottom of IO0 pin and reset it every time. Can I remove this step?

mantriyogesh, 2022-12-09:
Hello @panbijub63 

Understood the pain.

For ESP32 case, GPIO0 defines the boot mode (https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#gpio0). Is this some custom hardware? If you could follow our devkit schematic for strap pin settings, then you should not face this problem.

closed 2022-12-13 by panbijub63

mantriyogesh, 2022-12-13:
Was the link useful @panbijub63 ?

panbijub63, 2022-12-14:
@mantriyogesh  It's useful, but now the hardware is ready, and it can only be changed next time

Mr-Bossman, 2022-12-14:
You can unload the esp32 module and use the gpio subsystem to pull it low then program it.

mantriyogesh, 2023-02-06:
> [image] Another problem is when the esp updates the firmware. I need to pull the bottom of IO0 pin and reset it every time. Can I remove this step?

Hello @panbijub63 
I just had some additional info.

I am not sure by `pull the bottom of IO0 pin` , you mean hardware pull-up. but **if** there is no hardware pull-up as such, then you can use simply configure using https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html for your expected boot seq.

if above step doesn't solve, you can think of changing the handshake pin (GPIO2) to some other pin (obviously hw & sw, both), it will definitely solve your problem.