espressif/esp-hosted #178: Which kernel is recommended?

rimb05, 2022-11-28:
Hi, I’m starting to build an esp_hosted-fg project on an Allwinner v3s platform. Which Linux kernel is currently recommended, combined with which hosted_fg release?  I tried Linux 5.2x with fg release 0.0.5 and ran into some issues with crashing when it calls ether_addr_copy.

I’m using fg because I need AP functionality which ng doesn’t have. 

Would love to get some feedback about a workingo firmware/kernel combo to try.  I could use either SPI or SDIO. 

Thanks in advance!
Rim

rimb05, 2022-11-28:
Some more info on the crash I get.  This is using Linux 5.2 on a v3s board (Licheepi Zero).  I'm using SDIO mode. The firmware on the ESP is 0.0.5, and I built the esp32_sdio.ko module using the source included in the firmware zip. I run:

modprobe bluetooth
insmod esp32_sdio.ko resetpin=197

The crash happens in esp_init_net_dev when it tries to call:
ether_addr_copy(ndev->dev_addr, priv->mac_address);

I would really appreciate any insights

dmesg output:
[   28.497905] esp32_sdio: loading out-of-tree module taints kernel.
[   28.505584] esp_reset, ESP32: Resetpin of Host is 65
[   28.505666] sun8i-v3s-pinctrl 1c20800.pinctrl: 1c20800.pinctrl supply vcc-pc not found, using dummy regulator
[   28.516085] esp_reset, ESP32: Triggering ESP reset.
[   28.517905] esp_sdio: probe of mmc1:0001:1 failed with error -110
[   29.442267] mmc1: card 0001 removed
[   30.610752] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[   30.623985] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[   30.632924] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   30.639785] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   30.646068] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   30.651766] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   30.658141] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   30.664454] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   30.670041] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   30.676337] mmc1: new SDIO card at address 0001
[   30.681381] esp_probe: ESP network device detected
[   30.690887] Unable to handle kernel NULL pointer dereference at virtual address 00000001
[   30.699082] pgd = (ptrval)
[   30.701822] [00000001] *pgd=00000000
[   30.705409] Internal error: Oops: 805 [#1] SMP ARM
[   30.710200] Modules linked in: esp32_sdio(O)
[   30.714481] CPU: 0 PID: 5 Comm: kworker/0:0 Tainted: G           O      5.2.0 #2
[   30.721873] Hardware name: Allwinner sun8i Family
[   30.726592] Workqueue: events_freezable mmc_rescan
[   30.731404] PC is at esp_add_interface.constprop.0+0x110/0x180 [esp32_sdio]
[   30.738369] LR is at esp_add_interface.constprop.0+0xfc/0x180 [esp32_sdio]
[   30.745240] pc : [<bf000664>]    lr : [<bf000650>]    psr: 60000013
[   30.751504] sp : c3845d70  ip : 00000007  fp : 00000001
[   30.756726] r10: 00000010  r9 : bf0042d8  r8 : c3093e00
[   30.761949] r7 : bf003070  r6 : 00000000  r5 : c30ad000  r4 : 00000001
[   30.768472] r3 : 00000001  r2 : 00000569  r1 : 00000000  r0 : bf003490
[   30.774998] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   30.782129] Control: 10c5387d  Table: 430f406a  DAC: 00000051
[   30.787874] Process kworker/0:0 (pid: 5, stack limit = 0x(ptrval))
[   30.794051] Stack: (0xc3845d70 to 0xc3846000)
[   30.798410] 5d60:                                     00000001 00000001 00000000 b9abbfd9
[   30.806589] 5d80: bf0042c8 c30d6e00 bf004280 00000001 bf0042c8 bf000e40 c30d6e00 c3902840
[   30.814767] 5da0: 00000001 bf001940 00000001 00000010 00000001 00000000 c30d6e08 bf003264
[   30.822945] 5dc0: c30d6e00 bf004024 bf004024 c05a02fc c30d6e08 c0ba79e0 00000000 c0ba79e4
[   30.831124] 5de0: 00000000 c047e5e4 00000000 c30d6e08 bf004024 c047ecb8 c30ac808 00000000
[   30.839302] 5e00: 00000000 00000002 00000001 c047eab0 bf004024 c047ed30 00000000 c3845e54
[   30.847480] 5e20: c047ecb8 c30ac808 00000000 c047c60c 00000000 c397966c c30a7838 b9abbfd9
[   30.855659] 5e40: c30d6e08 00000001 c30d6e4c c047e250 c30a7980 c30d6e08 00000001 b9abbfd9
[   30.863837] 5e60: c30d6e08 c0b5f77c c30d6e08 c047d754 c30d6e08 00000000 c0ba79bc c047a058
[   30.872015] 5e80: c30d6e00 c3b67400 00000001 c05955d4 c30d6e00 b9abbfd9 00061a80 c30d6e00
[   30.880193] 5ea0: c30d6e08 00061a80 c30ac804 c3b67400 c30ac800 c05a0618 00000000 00000002
[   30.888372] 5ec0: 00061a80 c059f978 00000000 c3845edf 00000000 00000000 00000000 02000000
[   30.896550] 5ee0: 20ffff00 b9abbfd9 ffffff92 c3b6768c c3b67400 00061a80 c086b0ac 00000000
[   30.904728] 5f00: c086b0b8 00000000 c3fde880 c05963f8 c3b6768c c380d680 c3fde880 c3fe1c00
[   30.912906] 5f20: 00000000 00000000 00000000 c01372f4 c3fde898 c0b03d00 c380d680 c3fde880
[   30.921085] 5f40: c380d694 c3fde898 c0b03d00 00000008 c3844000 c0137a3c c3844000 c0b6a068
[   30.929263] 5f60: ffffe000 c3816980 c38168c0 00000000 c3844000 c380d680 c01379e0 c381699c
[   30.937441] 5f80: c3835ebc c013ce0c 00000001 c38168c0 c013ccf0 00000000 00000000 00000000
[   30.945619] 5fa0: 00000000 00000000 00000000 c01010e8 00000000 00000000 00000000 00000000
[   30.953797] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   30.961975] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[   30.970175] [<bf000664>] (esp_add_interface.constprop.0 [esp32_sdio]) from [<bf000e40>] (esp_add_card+0x28/0x9c [esp32_sdio])
[   30.981488] [<bf000e40>] (esp_add_card [esp32_sdio]) from [<bf001940>] (esp_probe+0x21c/0x264 [esp32_sdio])
[... 73 lines cut ...]
I (596) heap_init: At 3FFCCA38 len 000135C8 (77 KiB): DRAM
I (603) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (609) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (615) heap_init: At 4009DF1C len 000020E4 (8 KiB): IRAM
I (623) spi_flash: detected chip: generic
I (626) spi_flash: flash io: dio
W (630) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (644) cpu_start: Starting scheduler on PRO CPU.
[... cut ...]
I (692) NETWORK_ADAPTER: - WLAN over SDIO
I (696) ESP_BT: - BT/BLE
I (698) ESP_BT:    - HCI Over SDIO
I (700) ESP_BT:    - BT/BLE dual mode
I (704) NETWORK_ADAPTER: capabilities: 0x1d
I (716) BTDM_INIT: BT controller compile version [8020d24]
I (718) system_api: Base MAC address is not set
I (718) system_api: read default base MAC address from EFUSE
I (724) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1050) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 34:94:54:39:c6:6a
I (1052) SDIO_SLAVE: Using SDIO interface

mantriyogesh, 2022-11-28:
I am digesting your complete story, by the time check below:

>  I have 10K pullups on all the pins except reset and CMD

Did you mean resetpin & CLK? Please check, 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#sd-pull-up-requirements
```sh
the CMD and DATA (DAT0 - DAT3) lines of the SD bus must be pulled up by 10 kOhm resistors. 
SD cards and SDIO devices should also have pull-ups 
on all above-mentioned lines (regardless of whether these lines are connected to the host) 
in order to prevent them from entering a wrong state.
```

rimb05, 2022-11-28:
Yes, sorry, I meant except CLK.  I have pullups on CMD, DAT0, DAT1, DAT2, DAT3

mantriyogesh, 2022-11-28:
Apologies for the delayed response.

1. Does it crash if you manually use definition from
https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/host/linux/host_driver/esp32/esp_kernel_port.h#L34-L44

2. Basically, define a different function, say ether_addr_copy_temp() using above definition and call it instead.
Interested to understand if unaligned access. (most likely not the case)

3. Also, can you please localize if anyone is NULL from these two:
(1) ndev->dev_addr
(2) priv->mac_address

rimb05, 2022-11-30:
Hi Yogesh,

I put that kernel on hold and tried working with an older one.  I have gotten much further progress using 4.10.15.  This is a kernel that was patched specifically for this chip, so I thought I might have better luck here.  

I  was able to get both SDIO and SPI to work, however I am having trouble with the control mechanism.  I built the c_support code (test.out), and created the character device using:

mknod /dev/esps0 c 221 0
chmod 666 /dev/esps0

I checked, and on my system 221 is not used by any other drivers.

When I try to use test.out, for example with: "test.out scan" it times out.  However I can see that the ESP module received the command because on its log I see all the access points being discovered.  So it's receiving commands, but the communications seem to only be in one direction.  

I am using the same source as the firmware version (from the same release).  I tried 0.0.5, and I also tried 0.4, and I get the same result.  

Any ideas?

mantriyogesh, 2022-12-01:
Can you please send both side logs to understand exactly?

rimb05, 2022-12-01:
Sure, here are the log from both sides.  The strange thing is the same problem occurs with both SDIO or SPI.  So it doesn't seem like it's a hardware problem.   The communication seems to only be one direction (host -> ESP).  

The following logs are after I ran these commands:

modprobe bluetooth
insmod esp32_spi.ko resetpin=197
mknod /dev/esps0 c 221 0
chmod 666 /dev/esps0
./test.out scan

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:12
load:0x3fff0028,len:7432
load:0x40078000,len:15100
ho 0 tail 12 room 4
load:0x40080400,len:3904
entry 0x400806b8
I (80) boot: Chip Revision: 3
I (81) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (44) boot: ESP-IDF v4.0.3-215-gbe7df8bce9-dirty 2nd stage bootloader
I (44) boot: compile time 17:19:23
I (45) boot: Enabling RNG early entropy source...
I (50) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : DIO
I (58) boot: SPI Flash Size : 4MB
I (63) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (81) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  3 factory          factory app      00 00 00010000 00100000
I (103) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (111) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (118) boot: End of partition table
I (123) boot: Defaulting to factory image
I (127) boot_comm: chip revision: 3, min. application chip revision: 0
I (135) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x202a8 (131752) map
I (191) esp_image: segment 1: paddr=0x000302d0 vaddr=0x3ffbdb60 size=0x03d40 ( 15680) load
I (197) esp_image: segment 2: paddr=0x00034018 vaddr=0x40080000 size=0x00400 (  1024) load
I (198) esp_image: segment 3: paddr=0x00034420 vaddr=0x40080400 size=0x0bbf0 ( 48112) load
I (227) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8eab8 (584376) map
I (436) esp_image: segment 5: paddr=0x000cead8 vaddr=0x4008bff0 size=0x123b0 ( 74672) load
I (486) boot: Loaded app from partition at offset 0x10000
I (486) boot: Disabling RNG early entropy source...
I (486) cpu_start: Pro cpu up.
I (490) cpu_start: Application information:
I (495) cpu_start: Project name:     network_adapter
I (500) cpu_start: App version:      release0.4-60-g22180f7
I (506) cpu_start: Compile time:     Jan 24 2022 17:19:16
I (513) cpu_start: ELF file SHA256:  0f66794e8494e214...
I (519) cpu_start: ESP-IDF:          v4.0.3-215-gbe7df8bce9-dirty
I (525) cpu_start: Starting app cpu, entry point is 0x400812fc
I (0) cpu_start: App cpu up.
I (536) heap_init: Initializing. RAM available for dynamic allocation:
I (543) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (549) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (555) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (561) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (567) heap_init: At 3FFC80E0 len 00017F20 (95 KiB): DRAM
I (573) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (579) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (586) heap_init: At 4009E3A0 len 00001C60 (7 KiB): IRAM
I (592) cpu_start: Pro cpu start user code
I (610) spi_flash: detected chip: generic
I (611) spi_flash: flash io: dio
W (611) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image h.
[... 286 lines cut ...]
[    1.134079] usb usb2: Manufacturer: Linux 4.10.15-licheepi-zero+ ohci_hcd
[    1.140878] usb usb2: SerialNumber: 1c1a400.usb
[    1.146083] hub 2-0:1.0: USB hub found
[    1.149940] hub 2-0:1.0: 1 port detected
[    1.154505] sun6i-rtc 1c20400.rtc: setting system clock to 1970-01-01 00:00:03 UTC (3)
[    1.162685] vcc5v0: disabling
[    1.168923] EXT4-fs (mmcblk0p2): couldn't mount as ext3 due to feature incompatibilities
[    1.179875] EXT4-fs (mmcblk0p2): couldn't mount as ext2 due to feature incompatibilities
[    1.206094] random: fast init done
[    1.230299] EXT4-fs (mmcblk0p2): recovery complete
[    1.236551] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    1.244760] VFS: Mounted root (ext4 filesystem) on device 179:2.
[    1.253395] devtmpfs: mounted
[    1.257689] Freeing unused kernel memory: 1024K
[    1.407217] EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered
[    1.974505] Bluetooth: Core ver 2.22
[    1.978189] NET: Registered protocol family 31
[    1.982723] Bluetooth: HCI device and connection manager initialized
[    1.989098] Bluetooth: HCI socket layer initialized
[    1.994006] Bluetooth: L2CAP socket layer initialized
[    1.999111] Bluetooth: SCO socket layer initialized
[   22.041838] esp32_spi: loading out-of-tree module taints kernel.
[   22.048753] esp_reset, ESP32: Resetpin of Host is 197
[   22.048898] esp_reset, ESP32: Triggering ESP reset.
[   22.049456] ESP32 peripheral is registered to SPI bus [32766],chip select [0], SPI Clock [5]`

mantriyogesh, 2023-01-16:
Apologies for the late reply. Did you get it working?
It looks like you have some issues with Basic SPI and GPIO connections.

Please look reference here: https://github.com/espressif/esp-hosted/issues/192#issuecomment-1382656599

In general, please remember exact same git commit to be used with ESP code and Host kernel building.  In case you are using the released binary, please use 'Source code (zip)' from that release for host side kernel module building.

The next thing is to configure the SPI config in your Linux Device Tree (the bus should be usable) & configuring the Handshake, Data_Ready & Reset GPIO pins configured and getting them working. In your very recent log I see that initial transaction has not received. Unless the initial transaction is working, Any data pushed toward ESP (using say test.out), would be non useful.

mantriyogesh, 2024-04-01:
Closing due to inactivity.

closed 2024-04-01 by mantriyogesh