espressif/esp-hosted #179: When inserting linux spi driver as a module, failure to connect to a spi controller results in a crash

drosen-taranawireless, 2022-12-02:
The problem seems to be in `spi_exit` which does not check for incomplete initialization

```
# modprobe $ESP32_MODULE
[  139.994557] Failed to obtain SPI master handle
[  140.431813] ------------[ cut here ]------------
[  140.436403] WARNING: CPU: 3 PID: 2510 at lib/kobject.c:705 kobject_put+0x104/0x214
[  140.444058] kobject: '(null)' (14c26720): is not initialized, yet kobject_put() is being called.
[  140.452767] Modules linked in: esp32netdev(+) ip6table_mangle xt_DSCP xt_conntrack ip6table_filter ip6_tables iptable_filter bpfilter majic_zsp_remoteproc at24 regmap_i2c uio_pdrv_genirq
[  140.469201] CPU: 3 PID: 2510 Comm: modprobe Tainted: G           O      4.19.94-rt39+ #403
[  140.469207] Hardware name: Majic Platform
[  140.469240] [<c011265c>] (unwind_backtrace) from [<c010d8c4>] (show_stack+0x20/0x24)
[  140.469258] [<c010d8c4>] (show_stack) from [<c0860804>] (dump_stack+0x88/0x9c)
[  140.469274] [<c0860804>] (dump_stack) from [<c011f83c>] (__warn.part.0+0xd0/0xec)
[  140.469286] [<c011f83c>] (__warn.part.0) from [<c011f9cc>] (warn_slowpath_fmt+0x74/0x9c)
[  140.469299] [<c011f9cc>] (warn_slowpath_fmt) from [<c0865dec>] (kobject_put+0x104/0x214)
[  140.469314] [<c0865dec>] (kobject_put) from [<c02a9048>] (cdev_del+0x38/0x3c)
[  140.469354] [<c02a9048>] (cdev_del) from [<bf0e402c>] (esp_serial_cleanup+0x24/0x4c [esp32netdev])
[  140.469408] [<bf0e402c>] (esp_serial_cleanup [esp32netdev]) from [<bf0e3260>] (spi_exit+0x90/0xd4 [esp32netdev])
[  140.469447] [<bf0e3260>] (spi_exit [esp32netdev]) from [<bf0e3ae4>] (esp_init_interface_layer+0x1b0/0x1d8 [esp32netdev])
[  140.469486] [<bf0e3ae4>] (esp_init_interface_layer [esp32netdev]) from [<bf0e9180>] (esp_init+0x180/0x1000 [esp32netdev])
[  140.469514] [<bf0e9180>] (esp_init [esp32netdev]) from [<c01029f8>] (do_one_initcall+0x64/0x2bc)
[  140.469529] [<c01029f8>] (do_one_initcall) from [<c01c1548>] (do_init_module+0x74/0x2d4)
[  140.469540] [<c01c1548>] (do_init_module) from [<c01c3e08>] (load_module+0x25f0/0x2b20)
[  140.469550] [<c01c3e08>] (load_module) from [<c01c4604>] (sys_finit_module+0xf4/0x130)
[  140.469560] [<c01c4604>] (sys_finit_module) from [<c0101000>] (ret_fast_syscall+0x0/0x28)
[  140.469566] Exception stack(0xed341fa8 to 0xed341ff0)
[  140.469576] 1fa0:                   000daa60 80000000 00000003 000daa60 00000000 000daa18
[  140.469586] 1fc0: 000daa60 80000000 000da918 0000017b 000daa18 00000000 00000001 000d9ca4
[  140.469593] 1fe0: bed4daf0 bed4dae0 00031acc b6e917b0
[  140.469598] ---[ end trace 0000000000000002 ]---
[  140.638430] ------------[ cut here ]------------
```

mantriyogesh, 2022-12-02:
Hello @drosen-taranawireless ,

Noted. We have resolved this problem in NG. we will propogate the fix in FG.

however, it looks like the SPI bus is unclaimable with your Linux.
May we know what platform  &  Linux kernel veraion do you use?
Most likely DT is not configured correct or spidev is not disabled.

mantriyogesh, 2022-12-02:
Can you please have a look at https://github.com/espressif/esp-hosted/issues/165, where similar issue of spi bus unclaimable was observed?

drosen-taranawireless, 2022-12-02:
Hi @mantriyogesh,

I found this running on a prototype of a custom embedded system, using a
proprietary SPI controller that is under development. In this instance
there was a bug in our FPGA that caused the SPI controller to bail. We've
since fixed the FPGA and put in a safeguard in the SPI controller. But this
fix would have allowed the Espressif SPI protocol driver to exit
gracefully. I don't know the kernel version off hand; I can get you that
tomorrow.

Regards,
David Rosen

drosen-taranawireless, 2022-12-02:
Hi @mantriyogesh,
In my previous email, when I wrote "this fix" I was referring to pull
request #180, which I just submitted.

Regards,
David Rosen

mantriyogesh, 2022-12-02:
Yes.. Missed it.. Thank you @drosen-taranawireless for taking time and efforts for correction in the code. We appreciate your inputs.