espressif/esp-hosted #182: hosted fg crash after esp32 reboot

liyijunsir, 2022-12-20:
First of all, thanks a lot for this nice project.

We were able to run esp32 hosted (FG release version 0.0.5) on an Linux processor running linux kernel 4.9, with esp32-c3 chip , with spi connections. 

we found a crash when host send connect ap cmd and have not receive the connect result, at this time, we manual reset esp32c3.

after host wait connect result timeout, host reveive esp32 INIT EVENT，and then crash.

wo found the crash piont in platform_wrapper.c timer_ll_callback func, in line "timer_arg->timer_cb(timer_arg->arg)"。

did you have some idea to hlep me?

best regards!

mantriyogesh, 2022-12-20:
Hello @liyijunsir ,

Thank you for using ESP-Hosted & rasing this issue.
As I understand, you face crash in Linux user-space.

Some information would help to understand the issue:
1. Can you please send the running log?
2. The application code or part of code how you are using this part of code?
Why, Because, these are async APIs you are using. Is your application closing too soon before getting the response?
This case, Can you try the demo app as is, please confirm if it also crashes at same point.
3. Async APIs are something you can do parallel work while waiting for the response. but if your expectation is not to wait like that, you just change:

https://github.com/espressif/esp-hosted/blob/29208b6c156be06dfd90dadc6dd43acd4d98d748/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L705-L722

to 
```c
int test_station_mode_connect(void)
{
	/* implemented Synchronous */
	ctrl_cmd_t req = CTRL_CMD_DEFAULT_REQ();
	ctrl_cmd_t *resp = NULL;

	strcpy((char *)&req.u.wifi_ap_config.ssid, STATION_MODE_SSID);
	strcpy((char *)&req.u.wifi_ap_config.pwd, STATION_MODE_PWD);
	strcpy((char *)&req.u.wifi_ap_config.bssid, STATION_MODE_BSSID);
	req.u.wifi_ap_config.is_wpa3_supported = STATION_MODE_IS_WPA3_SUPPORTED;
	req.u.wifi_ap_config.listen_interval = STATION_MODE_LISTEN_INTERVAL;

	resp = wifi_connect_ap(req);

	return ctrl_app_resp_callback(resp);
}
```
This synchronous change should work for you. But anyway, we will try to correct if there is something wrong in async path.

liyijunsir, 2022-12-20:
thanks for your solution and suggestion @mantriyogesh 
I try to decrease DEFAULT_CTRL_RESP_TIMEOUT from 30  to 1s, and find linux host will crash always. 
After I analyzed the code logic, i modify the code:
[https://github.com/espressif/esp-hosted/blob/29208b6c156be06dfd90dadc6dd43acd4d98d748/esp_hosted_fg/host/linux/port/src/platform_wrapper.c#L348](url)
to
static struct timer_arg_t timer_arg;

After this modification, it did not crash, but there is no prompt for timeout err, but"Msg type is not event[0]".
[https://github.com/espressif/esp-hosted/blob/29208b6c156be06dfd90dadc6dd43acd4d98d748/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L94](url)

it is because  no valid value assigned to the variable app_resp->msg_id in func ctrl_async_timeout_handler.

so, i think there is something wrong in async path.

best regards!

mantriyogesh, 2023-01-16:
okay. I will further look in this async path. Did you get it working with sync path at least to get unblocked?

mantriyogesh, 2023-03-13:
Yes
The code should be modified as:
```c
static void ctrl_async_timeout_handler(void *arg)
{
    ctrl_cmd_t *app_req = (ctrl_cmd_t *)arg; //new

    if (!app_req || !app_req->ctrl_resp_cb) {
      if (!app_req)
        printf("%s:%u NULL app_req\n",__func__, __LINE__);

      if (!app_req->ctrl_resp_cb)
        printf("%s:%u NULL app_req->resp_cb\n",__func__, __LINE__);
      return;
    }

    ctrl_resp_cb_t func = app_req->ctrl_resp_cb; //new
    ctrl_cmd_t *app_resp = NULL; 
    app_resp = (ctrl_cmd_t *)g_h.funcs->_h_malloc(sizeof(ctrl_cmd_t));
    if (!app_resp) {
        printf("Failed to allocate app_resp\n");
        return;
    }
    app_resp->msg_id = app_req->msg_id - CTRL_MSG_ID__Req_Base + CTRL_MSG_ID__Resp_Base; //new
    app_resp->msg_type = CTRL_RESP;
    app_resp->resp_event_status = CTRL_ERR_REQUEST_TIMEOUT;

    /* call func pointer to notify failure */
    func(app_resp);

    /* Unlock semaphore in negative case */
    g_h.funcs->_h_post_semaphore(ctrl_req_sem);
}
```

caller should be changed too from `app_req->ctrl_resp_cb` to `app_req`

```c
    if (app_req->ctrl_resp_cb) {
        async_timer_handle = g_h.funcs->_h_timer_start(app_req->cmd_timeout_sec, CTRL__TIMER_ONESHOT,
                ctrl_async_timeout_handler, app_req);
        if (!async_timer_handle) {
            printf("Failed to start async resp timer\n");
            goto fail_req;
        }
    }
```