espressif/esp-hosted #184: kernel 4.14 and version 1.0.2

micheledarold, 2022-12-21:
Good morning trying to compile the version 1.0.2 we found a issue, with kernel 4.14

[0001-esp_cmd-fix_old_kernel.patch](https://github.com/espressif/esp-hosted/files/10276584/0001-esp_cmd-fix_old_kernel.patch)

mantriyogesh, 2022-12-21:
Hello @micheledarold 

Thank you for using ESP-Hosted and providing fix to make it better.
Appreciate the time & efforts for making this project better.

mantriyogesh, 2022-12-22:
I hope rest is working fine for you..

clwe, 2023-02-23:
Thanks. I had the same issue with a 4.19 kernel.

clwe, 2023-02-24:
I'm running into a problem with this new version. I'm not sure if it could be related to the kernel version or the fix provided. The dmesg log says it detects esp32 bluetooth but not the network interface. 

```
[   38.463295] esp_reset, ESP32: Triggering ESP reset.
[   38.668848] ESP32 peripheral is registered to SPI bus [2],chip select [1], SPI Clock [10]
[   40.919976] 
               Received ESP bootup event
[   40.919999] EVENT: 3
[   40.920006] EVENT: 2
[   40.920013] EVENT: 0
[   40.920018] EVENT: 4
[   40.920024] EVENT: 1
[   40.920036] esp32: process_fw_data ESP chipset's last reset cause: 
[   40.920043] POWERON_RESET
[   40.920051] esp32: ESP Firmware version: 1.0.2
[   40.920058] ESP chipset detected [esp32]
[   46.048063] esp32: Command[1] timed out
[   46.052004] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   46.060191] esp32: add_network_iface() failure, ret: -1
[   46.065574] network iterface init failed
[   46.072815] ESP peripheral capabilities: 0xf8
[   46.280027] ESP Bluetooth init
[   46.280632] Capabilities: 0xf8. Features supported are:
[   46.280643]   * WLAN on SPI
[   46.280648]   * BT/BLE
[   46.280655]     - HCI over SPI
[   46.280661]     - BT/BLE dual mode
[   46.616161] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   46.616179] Bluetooth: BNEP filters: protocol multicast
[   46.616215] Bluetooth: BNEP socket layer initialized
```
Also the output of hciconfig doesn't show a mac address:

```
# hciconfig 
hci0:   Type: Primary  Bus: SPI
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN 
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:3 acl:0 sco:0 commands:0 errors:0
```
Any ideas what could is going on? Could the above patch be a problem?

mantriyogesh, 2023-02-24:
Hello @clwe ,

Can you please try commenting these two lines
https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_ng/host/spi/esp_spi.c#L216-L217

And try again, if it works or not?

clwe, 2023-02-27:
Thanks for your reply. Unfortunately this doesn't fix it for me. Do you think the SPI clock was set too fast? 10MHz should work, right?

mantriyogesh, 2023-02-27:
@clwe 
Worth to check any SPI clock related issue.
Please check https://github.com/espressif/esp-hosted/issues/200

mantriyogesh, 2023-03-10:
@clwe 
Any update with reference to above?

clwe, 2023-03-11:
In the meantime I switched to esp-hosted-fg. I had similar connection problems there and found it was caused by a faulty USB cable and/or an additional ground connection between my embedded board (pocketbeagle) and the esp32. Possibly a ground loop problem. The patch made it compile and run the NG firmware, but I couldn't check SPI connectivity yet. I would need to switch back to NG firmware first. Thanks a lot for your help!

mantriyogesh, 2023-03-12:
That's great. Thank you for sharing the possible reasons for futute issues.
Do let us know should you face any issue while using ESP-Hosted-FG as well.