espressif/esp-hosted #186: ESP32 as HOST MCU

alberk8, 2023-01-02:
Is it possible to have an example implementation where ESP32 is use as the HOST and ESP32-C3 as the client for both Wifi and BLE with SPI and/or Uart.  Thank you.

ajita02, 2023-01-02:
Hi @alberk8, 
Its definitely possible. Please follow this [documentation](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/Getting_started.md). You need to port MCU example according to ESP32.

alberk8, 2023-01-02:
> Hi @alberk8, Its definitely possible. Please follow this [documentation](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/Getting_started.md). You need to port MCU example according to ESP32.

Thank you for the prompt reply.  A working example would be much appreciated.

ajita02, 2023-01-02:
Hi @alberk8,

Configuring ESP as host and client doesn't make helpful use case for end users, as ESP already have WiFi-BT functionality. You can take help of MCU implementation for porting application to ESP32.

Thanks,
Ajita

alberk8, 2023-01-02:
I might looked not useful but it gives a testable scenario before moving to implementing on other MCU.  I would like to call it a reference implementation.

copercini, 2023-01-06:
I think it's interesting to use an ESP32 with an ESP32-C6 for example, then it's possible to use Ethernet from the standard ESP32 and Wi-Fi 5GHz from ESP32-C6

alberk8, 2023-01-07:
> I think it's interesting to use an ESP32 with an ESP32-C6 for example, then it's possible to use Ethernet from the standard ESP32 and Wi-Fi 5GHz from ESP32-C6

There is a use case that I want to use.  On an ESP32 I would setup an Soft AP accepting client connection and on a C3 where it connects to an STA to push data up to a cloud.   Well you might say that this already works with one ESP32.  Yes but you cannot have high volume traffic when both STA and AP is running on one hardware.

mantriyogesh, 2023-01-07:
We can do it if you feel the use case helps you.
Spare us two week time to implement and test, considering current priority items.

alberk8, 2023-01-07:
> We can do it if you feel the use case helps you. Spare us two week time to implement and test, considering current priority items.

Hi @mantriyogesh ,  Thank you for the prompt reply.  Please proceed with the reference implementation.  Thank you again.

mantriyogesh, 2023-01-31:
Hello @alberk8 

I have done PoC for this & arping is working fine.
Right now I have added high debug prints and some in place changes for porting from cmsis to freertos (ESP operated on idf based freertos).
If impatient & If you want to try the solution (to try), let me know.

I am going to add this in supported host MCU list itself, once I cleanup and add some more code.

alberk8, 2023-01-31:
Thanks for the update.  I can wait for the final product.

mantriyogesh, 2023-02-14:
Hello @alberk8 

We are parking this activity for some time, because of another high priority items to take.
Anyway, the ESP32 as host is working fine. The station and softAP should work with ESP32 as Host (on SPI transport)

We used ESP32 as host and ESP32-C3 as slave. You can change as per your need.

1. Connections:

|  Function | ESP32 as Host  | ESP32-C3 as slave  |
|---|---|---|
| CS0  | IO14  | IO10  | 
|  SCLK | IO15  | IO06  |
| MISO  | IO13  | IO02  | 
| MOSI  |  IO12 | IO07  | 
| GND  | GND  | GND  | 
| Handshake  | IO02  | IO03  | 
| Data ready  | IO04  | IO04  | 
| Slave reset  | IO05  | RST  | 

2. Files to note:
 esp_hosted_fg/host/microcontrollers/port/esp_hosted_config.h - contains the common config

3. Building
a. Apply patch  [ESP32_as_Host_over.patch](https://github.com/espressif/esp-hosted/files/10729778/ESP32_as_Host_over.patch) on git commit, fe0b10584417629908cee8141699c2f35ea05a25
b. setup idf release/v5.0 on your laptop
c. build and flash esp32 as host
```
cd esp_hosted_fg/host/microcontrollers/port/esp/esp32
idf.py set-target esp32
idf.py flash monitor
```
4. Known issues (will **solve these once resume this activity**)

a. Flashing either chipset when both chipsets are connected in usb is not possible.
This mostly would be due to using bootstrap pin GPIO02 as handshake.
workaround: flash slave (C3) first and detach, flash host ESP32 with `idf.py flash monitor` & then while flashing in progress, attach C3
correct fix: try changing HANDSHAKE Pin from GPIO02 to some other [non bootstrapping pin](https://www.esp32.com/viewtopic.php?t=5970)

b. reset pin is not working
slave reset (c3 reset) is not happening. Should be small issue to fix, but did not get the time to look into this.

c. Linux and STM32 is broken to build. Basically, the changes are done from system and design point of view. But in meantime, I have broken STM32 an Linux build system due to change of directory structure.
ESP32 as host will not face any issue although for building.

5. Limitations - We have not yet tried to integrate SDIO transport

Please do try and let us know, if the things work at your place.

alberk8, 2023-02-18:
Thank you for the detailed update.

mantriyogesh, 2023-03-30:
With fe0b10584417629908cee8141699c2f35ea05a25 as base ESP-Hosted commit,

ESP chipset as host some more updates:
[lwip_ported_ESP32_as_Host_SerialMsgs_changed.patch](https://github.com/espressif/esp-hosted/files/11111271/lwip_ported_ESP32_as_Host_SerialMsgs_changed.patch)

This patch working lwip stack working with station and softap mode. memcpy etc could be optimised ahead.

Please note:
1. this has changed the base Serial messages and not compatible with the Hosted master.
What it means, you need to use slave and host code both from this patch.
2. host/microcontrollers/drivers/netif will showcase the lwip port according to [ESP-IDF standards](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif_driver.html)

idf.py monitor can be used to change the SPI (and all ESP-Hosted) pins at host ESP.

closed 2023-07-07 by mantriyogesh

alberk8, 2023-07-08:
@mantriyogesh,  Where can I find the completed ESP32 as a host example?  Thank you.

mantriyogesh, 2023-07-08:
Did you try https://github.com/espressif/esp-hosted/issues/186#issuecomment-1490201090?

mantriyogesh, 2023-07-08:
You can refer to the upto date code base.
esp_as_slave+esp_as_host: [h1.8jul23.tgz](https://github.com/espressif/esp-hosted/files/11988513/h1.8jul23.tgz)

Pins for ESP32 <- SPI -> ESP32
Slave & Host ESP32:
MOSI : 13
MISO: 12
CLK: 14
CS: 15
HS: 26
DR : 4
GND

Host pin 5 <--> Slave EN

Other esp chipsets are also supported. pins could be found from idf.py menuconfig at both slave and host.

idf supported: release/v5.1
ESP chipsets, esp32, esp32-c2/c3/c6/s2/s3 should be supported by this.
1. disconnect both slave & host
2. **connect slave** esp
```sh
$ cd esp_hosted_fg/esp
$ idf.py -p <serial_port_slave> menuconfig 
# Navigate to  'Example Configuration'. please check pins to be used at slave, hosted config, spi config etc using this command.
$ idf.py -p <serial_port_slave> flash monitor
```
3. **disconnect** **slave** esp from laptop 
4.  **connect host** esp to laptop
5. in new terminal window,
```sh
$ cd esp_hosted_fg/host/microcontrollers/port/esp/examples/wifi/iperf
$ idf.py -p <serial_port_host> menuconfig
# Navigate to  'Example Configuration'. please check pins to be used at host esp, hosted config, spi config etc using this command.
$ idf.py -p <serial_port_host> flash monitor
```
5. connect back slave esp. logs should start in slave terminal. if not started, run
```sh
$ idf.py -p <serial_port_slave> monitor
```

Some points to note:
1. LWIP performance is not great, looking into this issue
2. SPI as transport is supported till now

mantriyogesh, 2023-07-10:
GPIO for ESP as slave and ESP as Host with SPI transport to work with https://github.com/espressif/esp-hosted/issues/186#issuecomment-1626859260

| ESP32 | ESP32-S2/S3 | ESP32-C2/C3/C6 | Function |
:--------:|:-------:|:-------------:|:------------------:|
| 13 | 11 | 7 | MOSI |
| 12 | 13 | 2 | MISO | 
| 14 | 12 | 6 | CLK | 
| 15 | 10 | 10 | CS | 
| 26 | 17 | 3 | Handshake | 
| 4 | 4 | 4 | Data Ready | 
| GND | GND | GND | Ground |
| 5 | 5 | 5 | Host GPIO<br/>Connect this to slave EN/RST |

mantriyogesh, 2023-07-18:
Latest fixes for stable LWIP :

On https://github.com/espressif/esp-hosted/commit/fe0b10584417629908cee8141699c2f35ea05a25 , apply [ESP_as_Host_with_stable_LWIP.patch](https://github.com/espressif/esp-hosted/files/12075460/ESP_as_Host_with_stable_LWIP.patch)