espressif/esp-hosted #187: ESP32-C3 SSID scan issue with hosted version 1.0.2

vijay-yaga, 2023-01-03:
Hi mantriyogesh

we have successfully compiled and compiled and loaded the hosted application V1.0.2 and binaries for ESP32-C3
we are able to communicate the ESP32-C3 module using spi communication
we are able to do insert the esp32_spi.ko and ESP32-C3 module is detecting and communicating
but while scanning the WIFI SSID using (iwlist)  command we are getting error as shown below
        **iwlist espsta0 scan
        esp_cfg80211_scan
        espsta0   No scan results**
[ESP32_C3_LOG.txt](https://github.com/espressif/esp-hosted/files/10334983/ESP32_C3_LOG.txt)
[Hosted_driver_LOG.txt](https://github.com/espressif/esp-hosted/files/10334985/Hosted_driver_LOG.txt)

mantriyogesh, 2023-01-16:
Hello @vijay-yaga 

Did you get it working or still having issues here?

mantriyogesh, 2023-01-17:
Hello @vijay-yaga 
So you had been using 1.0.1 As I remember, which was working for you.
Is that correct assumption?

mantriyogesh, 2023-01-18:
@vijay-yaga any updates please?

vijay-yaga, 2023-01-18:
Hi @mantriyogesh 
NO we are using FG with call back events its working fine
with NG v 1.0.2 code its having the issue

can you provide me the solution for this

Regards
vijay