espressif/esp-hosted #188: USB Support ETA

uschmelmer, 2023-01-04:
Hi,
in your EDC22 Day 2 Talk4 ([https://www.youtube.com/watch?v=g14aEjnjRLw](https://www.youtube.com/watch?v=g14aEjnjRLw)) you mentioned USB Connection to Host as planned feature.
Is there any ETA when this feature will be released?
Will USB interfacing also support [Host Sleep](https://github.com/espressif/esp-hosted/blob/29208b6c156be06dfd90dadc6dd43acd4d98d748/esp_hosted_ng/docs/host_sleep.md) and wakeup via GPIO?

We are currently planning our hardware architecture and USB would save us some IOs and also a voltage level translator (Host 1.8V vs ESP 3.0V).

Kind Regards,
Udo Schmelmer

mantriyogesh, 2023-01-05:
Hello @uschmelmer ,
Thank you for  taking a look at video.

The USB feature might take some time. I can comment correctly about possible time to deliver USB in the upcoming week, after deciding upon some current issues.
As per my understanding, ESP32-S3 would be the candidate for USB.

diego0815, 2025-11-15:
Any update on this after more than 2 years passed? Has there been any development in the meantime?

mantriyogesh, 2025-11-15:
Hello @diego0815,

Because USB capability of high speed was not available for slave devices, this got lower priority.
Recently we have added usb support for esp-hosted-ng, it is in review state, however, if you wish to evaluate, can you please send me test mail on my mail handle from github?

mantriyogesh, 2025-11-17:
sent through mail..