espressif/esp-hosted #192: esp_ hosted_ fg esp_spi Interrupt pin

MrHaiCoder, 2023-01-12:
I am using esp_ hosted_ fg，host based linux4.9 ,chip is esp32s3;
Use/ test.out get_ sta_ mac_ addr; Get a mac without considering the heartbeat,
Discover ESP_SPI interrupt count is 3, ESP_ SPI_ DATA_ READY is 1, 
you can explain ESP in detail_ Is the reason for disconnection in SPI 3? It is better to be illustrated

mantriyogesh, 2023-01-12:
Can you please help me understand what do you mean by 'reason for disconnection in SPI 3'?
What is getting disconnected exactly?

mantriyogesh, 2023-01-12:
Also can you please provide:
1. git commit you are using at both sides?
2. Complete textual logs at ESP (minicom or idf.py monitor logs) and Host (/var/log/kern.log)
3. Details when you run the commands and running log of those

MrHaiCoder, 2023-01-14:
> see /proc/interrupts , ESP_SPI interrupt count was treble the ESP_SPI_DATA_READY interrupt count

[root@Zeratul:mnt]# ./test_2.out 
[root@Zeratul:mnt]# 
[root@Zeratul:mnt]# cat /proc/interrupts 
           CPU0       CPU1       
  2:       7287       6591   XBurst2   2  xburst2-intc
  3:        226       3328   XBurst2   3  jz-mailbox
  4:       2715       1041   XBurst2   4  core_timerevent
  8:          0          0  XBurst2-irqchip   0  13280000.jpeg0
 14:          0          0  XBurst2-irqchip   6  ipu
 15:         41          0  XBurst2-irqchip   7  13440000.sfc0
 17:          0          0  XBurst2-irqchip   9  10043000.spi0
 18:          0          0  XBurst2-irqchip  10  pdma
 20:          0          0  XBurst2-irqchip  12  lcdc0
 21:          0          0  XBurst2-irqchip  13  isp-ivdc
 38:        243       2118  XBurst2-irqchip  30  isp-w02
 39:       4578        483  XBurst2-irqchip  31  isp-m0
 45:         21         57  XBurst2-irqchip  37  mmc0
 46:          0          0  XBurst2-irqchip  38  

                                                   47:          0          0  XBurst2-irqchip  39  

                                                                                                     48:          0          0  XBurst2-irqchip  40  

 50:          0          0  XBurst2-irqchip  42  dbox
 53:          0          0  XBurst2-irqchip  45  i2d
 58:        345          0  XBurst2-irqchip  50  uart1
 67:          0          0  XBurst2-irqchip  59  i2c1
 68:       2059       3933  XBurst2-irqchip  60  i2c0
 69:          0          0  XBurst2-irqchip  61  pdmam
 70:          0          0  XBurst2-irqchip  62  avpu.0
 72:          0          0      GPIO  21  13060000.msc
 73:          0          0      GPIO   6  ESP_SPI
 74:          0          0      GPIO   7  ESP_SPI_DATA_READY
ERR:          0
[root@Zeratul:mnt]# 
[root@Zeratul:mnt]# 
[root@Zeratul:mnt]# ./test_2.out get_sta_mac_addr
Enable heartbeat with duration 20
Heartbeat operation successful
mac address is f4:12:fa:51:4b:4c
^C
Clean-up and exit
Disable Heartbeat
Heartbeat operation successful
[root@Zeratul:mnt]# cat /proc/interrupts 
           CPU0       CPU1       
  2:       9380       9609   XBurst2   2  xburst2-intc
  3:        242       4323   XBurst2   3  jz-mailbox
  4:       2732       1328   XBurst2   4  core_timerevent
  8:          0          0  XBurst2-irqchip   0  13280000.jpeg0
 14:          0          0  XBurst2-irqchip   6  ipu
 15:         41          0  XBurst2-irqchip   7  13440000.sfc0
 17:          0          0  XBurst2-irqchip   9  10043000.spi0
 18:         10          0  XBurst2-irqchip  10  pdma
 20:          0          0  XBurst2-irqchip  12  lcdc0
 21:          0          0  XBurst2-irqchip  13  isp-ivdc
 38:        253       3026  XBurst2-irqchip  30  isp-w02
 39:       6503        483  XBurst2-irqchip  31  isp-m0
 45:         21         61  XBurst2-irqchip  37  mmc0
 46:          0          0  XBurst2-irqchip  38  

                                                   47:          0          0  XBurst2-irqchip  39  

                                                                                                     48:          0          0  XBurst2-irqchip  40  

 50:          0          0  XBurst2-irqchip  42  dbox
 53:          0          0  XBurst2-irqchip  45  i2d
 58:        481          0  XBurst2-irqchip  50  uart1
 67:          0          0  XBurst2-irqchip  59  i2c1
 68:       2059       6039  XBurst2-irqchip  60  i2c0
 69:          0          0  XBurst2-irqchip  61  pdmam
 70:          0          0  XBurst2-irqchip  62  avpu.0
 72:          0          0      GPIO  21  13060000.msc
 73:          9          0      GPIO   6  ESP_SPI
 74:          3          0      GPIO   7  ESP_SPI_DATA_READY
ERR:          0

mantriyogesh, 2023-01-14:
Thank you for investingating at your side.

First transaction
---
Are you getting first transaction from ESP to Host?
/var/log/kern.log will confirm that. This will verify if SPI pins (CLK, MOSI, CS).

If you get first transaction, proceed ahead

GPIOs
---
Next thing to confirm is GPIOs. Below steps will help verify if the GOIOs are working or not:

Three GPIOs needed -
1. Handshake pin:
Please add a KERN_ERR log in
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/spi/esp_spi.c#L76

 
2. Data Ready from ESP to Host:
Similarly, please add a KER_ERR log in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/spi/esp_spi.c#L67

3. ESP Reset pin:
This pin will make sure ESP is reset on every fresh insert of kernel module esp32_spi.ko

mantriyogesh, 2023-01-18:
Any updates @MrHaiCoder ?

mantriyogesh, 2023-01-20:
Any updates @MrHaiCoder ??