espressif/esp-hosted #194: esp_ hosted_ NG esp_spi Interrupt pin

zhlftest, 2023-01-17:
I am using esp_ hosted_ ng，host based linux4.4, chip is esp32c3, spi only
The esp32c3 use https://github.com/espressif/esp-hosted/releases/tag/release%2Fng-v1.0.2,and the linux drivers use https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng;
[image]
I check the \\esp-hosted-master\esp_hosted_ng\docs\setup.mk,the Handshake pin used the esp32c3 IO3,but in my hardware design,the IO3 has used to other function,can you help to build another firmware,change the Handshake pin to the esp32c3 IO5;

mantriyogesh, 2023-01-17:
I understood your concern. Let me create firmware for you.
Is this the only pin you want to change?

zhlftest, 2023-01-17:
yes,only need to change IO3 to IO5,thanks

mantriyogesh, 2023-01-17:
[NG_ESP32C3_17jan23_GH194_Handshake_IO5.tgz](https://github.com/espressif/esp-hosted/files/10433566/NG_ESP32C3_17jan23_GH194_Handshake_IO5.tgz)

Use the firmware bins from `30MHz` directory.

Please capture the logs while running:
Host: /va/log/kern.log
ESP: minicom log with 112500 baud rate

mantriyogesh, 2023-01-18:
Please let us know once you are unblocked..

zhlftest, 2023-01-18:
The driver insmod successfully, but linux didn't get signal from ESP32C3, Now, we were measuring the signal for handshake and Data ready with an oscilloscope
[log(1).txt](https://github.com/espressif/esp-hosted/files/10441569/log.1.txt)

mantriyogesh, 2023-01-18:
By the way resetpin is mandatory. 
We need to reset ESP on every loading of Kernel module.

I hope the length of all the jumper wires is similar & ~5-6cm each.

Alternative to oscilloscope, you can also add KERN_ERR logs in https://github.com/espressif/esp-hosted/blob/29208b6c156be06dfd90dadc6dd43acd4d98d748/esp_hosted_ng/host/spi/esp_spi.c#L67-L74
and 
https://github.com/espressif/esp-hosted/blob/29208b6c156be06dfd90dadc6dd43acd4d98d748/esp_hosted_ng/host/spi/esp_spi.c#L76-L83

You need to get the first (bootup) event, from ESP to Host. Worth checking if Connections are correct. (First Milestone)

Which Linux are using? 

Are you using Linux other than Raspberry Pi? If so, check SPI config in Device Tree (DT) is correct. Next check if spidev is working fine after SPI is configured. Further disable the spidev after verification & then load the esp32_spi.ko driver using `rpi_init.sh`.

zhlftest, 2023-01-18:
Linux version 4.4.289
CPU: ARM926EJ-S
the en/handshake/data-ready pin is work fine，but the spi host does not communication with the esp32c3;
when the handshake/data-ready pin pull up,they keep high level all the way,is it correct?
[image]
[image]

the spi host config as below:
[image]

mantriyogesh, 2023-01-18:
what is output of `ls /dev/spidev*`

zhlftest, 2023-01-18:
[image]

It is normal to wire spi host mosi and miso together and test the device with software.

# ./spi -D /dev/spidev2.0 -v
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
TX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@.....................
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@.....................

mantriyogesh, 2023-01-18:
Now please disable the spidev. see reference: https://github.com/espressif/esp-hosted/issues/165#issuecomment-1275877729

Once you disable spidev, the user-space will not claim the spi. Only then the esp32_spi.ko can claim the free SPI bus.

zhlftest, 2023-01-18:
when the handshake/data-ready pin pull up,they keep high level all the way,is it correct?

zhlftest, 2023-01-18:
[err.txt](https://github.com/espressif/esp-hosted/files/10444242/err.txt)

mantriyogesh, 2023-01-18:
Are using jumper cables? is the length <= 5 to 6cm?

mantriyogesh, 2023-01-18:
Hello @zhlftest ,

Can you please use this debug binary for flashing and the host side code patch for ESP32-C3 testing (includes IO 05 changes)?

[2023.01.18_NG_ESP32C3_gh194_debug.tgz](https://github.com/espressif/esp-hosted/files/10445060/2023.01.18_NG_ESP32C3_gh194_debug.tgz)

zhlftest, 2023-01-19:
> Are using jumper cables? is the length <= 5 to 6cm?

no,I designed the hardware. I put chips on the hardware board

mantriyogesh, 2023-01-19:
Okay..

mantriyogesh, 2023-01-19:
No problem. Let us know once you are able to test it again.

emmmmlb, 2023-01-29:
The patch has been added, and the result is the same as before. It seems that the program does not go into the added debugging information.

The log is as follows：
Jan 29 08:54:19 buildroot kern.info kernel: nf_conntrack: automatic helper assignment is deprecated and it will be removed soon. Use the iptables CT target to attach helpers instead.
Jan 29 08:54:22 buildroot kern.info kernel: esp_reset, ESP32: Triggering ESP reset.
Jan 29 08:54:22 buildroot kern.info kernel: ESP32 peripheral is registered to SPI bus [2],chip select [1], SPI Clock [10]
Jan 29 08:54:23 buildroot kern.info kernel: esp_init, init esp32, ret = 0
Jan 29 08:54:23 buildroot auth.info sshd[1237]: WARNING: /usr/local/etc/moduli does not exist, using fixed modulus
Jan 29 08:54:26 buildroot auth.info sshd[1237]: Accepted password for root from 192.168.1.200 port 50211 ssh2

emmmmlb, 2023-01-29:
@mantriyogesh Please take a look at this question, thank you

mantriyogesh, 2023-01-30:
Have you checked,
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#124-peripheral-configurations from porting guide?

I am curious to know why are you using chip select as `1` at 
https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_ng/host/spi/esp_spi.c#L469-L470

Considering /dev/spidev2.0 was tested, you can make `esp_board.bus_num = 2;` and **`esp_board.chip_select = 0;`**

mantriyogesh, 2023-01-30:
Did you get a chance to look at this?

emmmmlb, 2023-01-30:
**/dev/spidev2.0**  has been occupied by the system. Modules cannot use the same spi device

emmmmlb, 2023-01-30:
[image]

mantriyogesh, 2023-01-30:
By `/dev/spidev2.0 has been occupied by the system`,
I interprete that you want to use but the system is not leaving spi instance.
Most likely it would be because spidev is claiming the device.
If you intent to use, need to disable spidev https://github.com/espressif/esp-hosted/issues/194#issuecomment-1386573333

Once spidev is disabled for this spi instance, you should be able to use it.

emmmmlb, 2023-01-30:
This Linux system does not use the device tree function. If I disable spidev in the Linux system, the loading module will report this error (Failed to observe SPI master handle)

mantriyogesh, 2023-01-30:
By the way, getting Linux platform up is outside scope of ESP-Hosted.

I might need some details to get to the problem:
1. Which is the Linux system you are using?
2. Can you descibe the steps, how are you disabling the spidev?
3. Also can you share complete /var/log/kern.log at host?

emmmmlb, 2023-01-31:
1.root@buildroot:/# uname -a
Linux buildroot 4.4.289 #111 PREEMPT Wed Oct 12 14:06:03 CST 2022 armv5tejl GNU/Linux

2.My configuration is in the blue square
[image]

3.There is a problem with the prototype designed before. We have modified the circuit. At present, the new circuit board has not been completed. We can continue testing this weekend.

mantriyogesh, 2023-02-01:
2. So worth to disable 'User mode SPI device driver support', to verify if spidev will not start/claim spi bus? 
3. Sure. Thanks for the update. Will wait for the outcome.
4. In case different spi instance and chip select has to be used, worth to test spidev on that first and then disable before loading the esp32 module.

emmmmlb, 2023-02-03:
2.After disabling 'User mode SPI device driver support'，When I load "insmod esp32_spi. ko resetpin=206", the error :
root@buildroot:/usr/modules# insmod esp32_spi.ko resetpin=206
Failed to obtain SPI master handle
Failed Init SPI device
insmod: can't insert 'esp32_spi.ko': No such device
system log:
RX nuc980_spi1_probe: dma0chan0 module removed
TX nuc980_spi1_probe: dma0chan1 module removed
nuc980-spi1 nuc980-spi1.0: No platform data supplied
nuc980-spi1: probe of nuc980-spi1.0 failed with error -2

mantriyogesh, 2023-02-03:
`insmod: can't insert 'esp32_spi.ko': No such device` this is very worrisome.

Can you please do 
```bash
$ file esp32_spi.ko
```
and check if it is expected for your platform?

What is output of **before** and **after** you disable 'User mode SPI device driver support' ?
```sh
$ ls /dev/spidev*
```

emmmmlb, 2023-02-03:
1. user@ubuntu:/nuc980bsp/linux-4.4.y/drivers/net/wireless/esp_hosted_ng$ file esp32_spi.ko 
esp32_spi.ko: ELF 32-bit LSB relocatable, ARM, EABI5 version 1 (SYSV), BuildID[sha1]=30c749e0d44df9215229d00a244b285befa1fc4b, not stripped
user@ubuntu:/nuc980bsp/linux-4.4.y/drivers/net/wireless/esp_hosted_ng$ 
2 ls: cannot access '/dev/spidev*': No such file or directory

mantriyogesh, 2023-02-03:
Okay.
From log, `Failed to obtain SPI master handle`, the spi instance-chip select mentioned in  was not 
In general, we should first get the spidev working on some specific (spi instance and chip select), then disable it only for user space.
This way when the driver tries to capture the bus, as user space will not claiming, the kernel module is able to claim it.
I am not really sure how can you do through your linux configuration, but this is the way it has to be.

So ideally, when you were able to test spidev in user space (https://github.com/espressif/esp-hosted/issues/194#issuecomment-1386567343), you might have to check how can you selectively only disable 'spidev'. once this step is done, you should be able to use spi bus.

emmmmlb, 2023-02-03:
The error has been resolved, and the current status is as follows：
root@buildroot:/usr/modules# insmod esp32_spi.ko resetpin=206
esp_reset, ESP32: Triggering ESP reset.
ESP32 peripheral is registered to SPI bus [2],chip select [1], SPI Clock [10]
esp_init, init esp32, ret = 0
root@buildroot:/usr/modules# ls /dev/spidev*
/dev/spidev2.0  /dev/spidev2.1
root@buildroot:/usr/modules#

mantriyogesh, 2023-02-03:
Great!!
I am not sure how but now you need to just disable spidev. make sure you just disable spidev only. not the original spi instance.

emmmmlb, 2023-02-03:
I first need to disable  /dev/spidev2.0 ？ then "insmod esp32_spi.ko resetpin=206"?   
Does this configuration need to be modified?
[image]

mantriyogesh, 2023-02-03:
1. 

> I first need to disable /dev/spidev2.0 ？ then "insmod esp32_spi.ko resetpin=206"?

yes. you might need to study how it is done for your Linux system. But you can find something after checking other peripherals / configurations.

2. 
> Does this configuration need to be modified?

modalias just a name, Better you do not name it spidev as it might confuse you later point of time. You only want to change last two lines bus_num=2 and chip_select=0.

Note: after doing only step (1), you should check if /dev/spidevX.Y are no more seen. Only then you move to step (2) and module building and loading.

emmmmlb, 2023-02-03:
If there is no/dev/spidevX. Y, when I execute "insmod esp32_spi. ko reset=206", the error "Failed to obtain SPI master handle, Failed Init SPI device" should appear.

mantriyogesh, 2023-02-03:
1. Have you tried both bus=2, cs-0 & bus =2, cs =1?
2. I still suspect if you are disabling the SPI instance itself.
you might want to disable only spidev only and not the SPI altogether.

For reference of Device Tree, see this:
[image]
You might have to figure out how this can be translated to your Linux configuration system.

In above image, SPI instance is enabled but only spidev is disabled for Chip Select 0.

3. Although in the image it says spi5, sometimes it may take some other number in spidev (say spidev0.Y)
So, before doing above changes, can you please try changing code as, (trial1) bus=0, cs=0 and (trial2) bus=0, cs=1

emmmmlb, 2023-02-07:
Sorry, there is no progress for the time being. It may not be tested in the near future.

mantriyogesh, 2023-02-07:
Okay.
Anyway, can you please close the issue by the time?
You can re-open when you test again?

mantriyogesh, 2023-02-09:
Please do re-open when you come back.

closed 2023-02-09 by mantriyogesh