espressif/esp-hosted #195: insmod esp32_sdio.ko error (ESP32 SDIO)

x893, 2023-01-20:
Hello,
i see error after

## insmod esp32_sdio.ko
esp32_sdio: loading out-of-tree module taints kernel.
esp_probe: ESP network device detected
Rx Pre ====== 0
esp_handle_isr: CMD53 read/write error at 111
Rx Pos ======  38
Tx Pre ======  0
Tx Pos ======  10
esp_sdio: probe of mmc0:0001:2 failed with error -22

linux 5.15.91 and latest esp32 firmwar.
Use only SDIO pins with 10K pull-ups on (D0-D3, CMD) but without RESET/EN pin.

Best wishes
Mitek

mantriyogesh, 2023-01-20:
You can ignore SDIO function 2 errors: `esp_sdio: probe of mmc0:0001:2 failed with error -22` as we are not using function 2 & only using function 1 of SDIO.

The real problem is `esp_handle_isr: CMD53 read/write error at 111`

Can you please let us know:
1. git commit at host and ESP firmware release you had used?
2. photo of setup to understand the connections and length of wires

x893, 2023-01-20:
Hi,
thanks for answer.
1. Firmware from ESP-Hosted-NG_release_v1.0.2.tgz  folder sdio_only

[image]

3. Latest esp-hosted repository. Build for Zynq XC7Z010. Linux 5.15 with disabled hci_... interface.

[image]

I try esp-gateway (previous name for esp-iot-bridge) with the same configuration and it works. But was other problems with esp-iot-bridge so i decide check esp-hosted.

mantriyogesh, 2023-01-20:
Can you try with 0.0.5 release (ESP-Hosted-FG) by the time (if you are okay not to use wpa_supplicant) ?

please use the correct source code for Linux building as shown:
[image]

https://github.com/espressif/esp-hosted/releases/tag/release%2Ffg-v0.0.5

mantriyogesh, 2023-01-20:
Also it is worth to reduce your wire length. It seems they are around 10cm. Recommended of <5-6 cm. Smaller the better. Why? SDIO is very sensitive protocol and this is needed for signal integrity.

mantriyogesh, 2023-01-20:
If the shortening of wires still doesn't work on either 1.0.2 or 0.0.5, I will create a patch firmware, which you can try.

x893, 2023-01-20:
Thank you very much
I will try to make your recommendations and write about the results.

mantriyogesh, 2023-01-20:
> Can you try with 0.0.5 release (ESP-Hosted-FG) by the time (if you are okay not to use wpa_supplicant) ?
> 
> please use the correct source code for Linux building as shown: [image]
> 
> https://github.com/espressif/esp-hosted/releases/tag/release%2Ffg-v0.0.5

For FG, the code is also **open source**. You can burn firmware on your own as well (explained in FG documentation), which should be ease to try. This will help in long run to localise issue as patch firmware sharing would take bigger cycle to identify and fix the issue.

One possible timing issue I suspect for your SDIO, which can be changed enabling
https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L201

x893, 2023-01-20:
Hi,
first result about 1.0.2 version (0.0.5 later, need rebuild driver)
esp32 stop after scan command

## 1. Log from esp32
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (27) boot: ESP-IDF v4.4.1-584-g91dc99a3ce-dirty 2nd stage bootloader
I (27) boot: compile time 12:09:04
I (28) boot: chip revision: 0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (76) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 factory          factory app      00 00 00010000 00100000
I (91) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (99) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (106) boot: End of partition table
I (110) boot: Defaulting to factory image
I (115) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=17950h ( 96592) map
I (159) esp_image: segment 1: paddr=00027978 vaddr=3ffbdb60 size=05574h ( 21876) load
I (167) esp_image: segment 2: paddr=0002cef4 vaddr=40080000 size=03124h ( 12580) load
I (173) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=88d8ch (560524) map
I (378) esp_image: segment 4: paddr=000b8db4 vaddr=40083124 size=1a43ch (107580) load
I (422) esp_image: segment 5: paddr=000d31f8 vaddr=50000000 size=00010h (    16) load
I (437) boot: Loaded app from partition at offset 0x10000
I (437) boot: Disabling RNG early entropy source...
I (449) cpu_start: Pro cpu up.
I (449) cpu_start: Starting app cpu, entry point is 0x40081288
I (0) cpu_start: App cpu up.
I (465) cpu_start: Pro cpu start user code
I (465) cpu_start: cpu freq: 240000000
I (465) cpu_start: Application information:
I (470) cpu_start: Project name:     network_adapter
I (475) cpu_start: App version:      2027593
I (480) cpu_start: Compile time:     Dec 19 2022 12:08:55
I (486) cpu_start: ELF file SHA256:  76d958924cb99ea3...
I (492) cpu_start: ESP-IDF:          v4.4.1-584-g91dc99a3ce-dirty
I (499) heap_init: Initializing. RAM available for dynamic allocation:
I (506) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (512) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (519) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (525) heap_init: At 3FFD1A40 len 0000E5C0 (57 KiB): DRAM
I (531) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (537) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (544) heap_init: At 4009D560 len 00002AA0 (10 KiB): IRAM
I (551) spi_flash: detected chip: winbond
I (555) spi_flash: flash io: dio
I (560) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (573) stats: *********************************************************************
I (579) stats:                 ESP-Hosted Firmware version :: 1.0.2                        
I (587) stats:                 Transport used :: SDIO only                     
I (593) stats: *********************************************************************
I (599) FW_MAIN: Supported features are:
I (603) FW_MAIN: - WLAN over SDIO
I (607) FW_BT: - BT/BLE
[... 66 lines cut ...]
EVENT: 1
# esp32: process_fw_data ESP chipset's last reset cause: 
RTCWDT_RTC_RESET
esp32: ESP Firmware version: 1.0.2
esp_inetaddr_event: NETDEV_UP interface espsta0 ip changed to  169.254.085.010 

# ifcondfig 0-a
espsta0   Link encap:Ethernet  HWaddr 30:AE:A4:03:A0:BC  
          inet addr:169.254.85.10  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::9a9b:e553:f3f4:4ae1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:18 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

# iw dev espsta0 scan
esp_cfg80211_scan
read_packet: Failed to read data - -84 [0 - 366]

^C
# 
# iw dev espsta0 scan
command failed: Device or resource busy (-16)
#

mantriyogesh, 2023-01-20:
This time it went ahead till some point, but anyway I strongly think the timing is little odd. How long the wires are now?

Anyway, I had tweaked the timing setting. can you please check on NG setup itself below:

[2023.01.20_NG_ESP32_195_Timing_NP_SDIO.tgz](https://github.com/espressif/esp-hosted/files/10466701/2023.01.20_NG_ESP32_195_Timing_NP_SDIO.tgz)

Please refer readme.txt from tarball.

x893, 2023-01-20:
Thanks for resplay.
I reduce SDIO frequency to 5 MHz in DTS. Reduce wires to 5 cm. With long wires and sdio frequency > 5 (10, 15, ...) no espsta0 exists (created).
May be problem deal with ESP32 chip ? (see below)

esptool.py v4.4
Serial port COM3
Connecting...........
### Chip is ESP32-D0WDQ6 (revision v0.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:03:a0:bc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB

I flash above version to ESP32 and see
`
> insmod esp32-hosted.ko
esp32_hosted: loading out-of-tree module taints kernel.
esp_probe: ESP network device detected
esp_sdio: probe of mmc0:0001:1 failed with error -84
esp_sdio: probe of mmc0:0001:2 failed with error -22
>
`

I check esp-iot-bridge repository. Adapter created but other problems presents.

mantriyogesh, 2023-01-20:
I think worth to check ESP-Hosted-FG and correct the issues on that first. If it works there, It must work on NG as well.
I hope you are using resetpin?

BTW, can you please tell the hardware details (host), if we have it here, I can check that as well.

x893, 2023-01-20:
I can check FG version later.
Don't use reset pin (need a time to add).
Host: Zynq XC7Z010 with sdio connection. Linux 5.15.67
If you have a time then i can make remote connection (via anydesk) and you can see all in details.

mantriyogesh, 2023-01-20:
Okay. Here it is holiday till 24th. But anyway, can check.
What's your time zone mine is Singapore TZ at the moment.

x893, 2023-01-20:
No problem. We can test after 24th.
I'm in GMT+3

x893, 2023-01-20:
Hello again,
i check on other board and i have some progress now.
1. Log from host

## insmod esp32-hosted.ko 
esp_probe: ESP network device detected

Received ESP bootup event
EVENT: 3
Rx Pre ====== 38
ESP chipset detected [esp32]
Rx Pos ======  38
EVENT: 0
ESP peripheral capabilities: 0x1d
Tx Pre ======  0
Capabilities: 0x1d. Features supported are:
Tx Pos ======  10
 * WLAN on SDIO
 * BT/BLE
   - HCI over SDIO
   - BT/BLE dual mode
esp_sdio: probe of mmc1:0001:2 failed with error -22
EVENT: 4
EVENT: 1
`#` esp32: process_fw_data ESP chipset's last reset cause: 
POWERON_RESET
esp32: ESP Firmware version: 1.0.2

## ifconfig espsta0 up
## iw dev espsta0 scan
esp_cfg80211_scan
BSS 90:ef:68:7d:2b:7c(on espsta0)
TSF: 66130870693 usec (0d, 18:22:10)
freq: 2462
beacon interval: 100 TUs
capability: ESS ShortPreamble ShortSlotTime (0x0421)
signal: 0.-66 dBm
last seen: 490 ms ago
SSID: ZyXEL7D2B7C
Supported rates: 1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 
DS Parameter set: channel 11
Country: CHEnvironment: Indoor/Outdoor
Channels [1 - 13] @ 20 dBm
ERP: <no flags>
Extended supported rates: 24.0 36.0 48.0 54.0 
HT capabilities:
Capabilities: 0x1ad
RX LDPC
HT20
SM Power Save disabled
RX HT20 SGI
TX STBC
RX STBC 1-stream
Max AMSDU length: 3839 bytes
No DSSS/CCK HT40
Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
Minimum RX AMPDU time spacing: No restriction (0x00)
HT TX/RX MCS rate indexes supported: 0-15
HT operation:
 * primary channel: 11
 * secondary channel offset: no secondary
 * STA channel width: 20 MHz
 * RIFS: 1
 * HT protection: no
 * non-GF present: 1
 * OBSS non-GF present: 0
 * dual beacon: 0
 * dual CTS protection: 0
 * STBC beacon: 0
 * L-SIG TXOP Prot: 0
[... 177 lines cut ...]
I (131470) FW_CMD: Connecting to ZyXEL7D2B7C, channel: 11 [0]
I (131470) FW_CMD: AUTH Commit

I (131476) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (131478) wifi:state: init -> auth (b0)
I (131482) FW_CMD: handle_wpa_sta_rx_mgmt:627 Auth[11] recvd

I (131528) FW_MAIN: Assoc request

I (131530) wifi:state: auth -> assoc (0)
I (131534) FW_CMD: handle_wpa_sta_rx_mgmt:635 ASSOC Resp[1] recvd

I (131534) FW_CMD: STA connect event [channel 11]

I (131536) BSSID: 0x3ffe8ab6   90 ef 68 7d 2b 7c                                 |..h}+||
I (131544) wifi:state: assoc -> run (10)
I (131546) wifi:connected with ZyXEL7D2B7C, aid = 8, channel 11, BW20, bssid = 90:ef:68:7d:2b:7c
I (131552) wifi:security: Open Auth, phy: bgn, rssi: -64
I (131558) wifi:pm start, type: 1

I (131560) FW_CMD: Wifi Station Connected event!! 

I (131568) wifi:AP's beacon interval = 102400 us, DTIM period = 1

Hope you can help me with problem.

x893, 2023-01-20:
With WPA2 also not complete

espsta0: SME: Trying to authenticate with 36:ce:ef:af:ca:0c (SSIDesp_cfg80211_authenticate='iPhone' freq=2462 MHz)
Authentication request: 36:ce:ef:af:ca:0c 11 0 0 0
espsta0: Trying to associate with 36:ce:ef:af:ca:0c (SSID='iPhoneesp_cfg80211_associate
Dima' freq=2462 MHz)
Association request: 36:ce:ef:af:ca:0c 11 40
Connection status: 0
espsta0: Associated with 36:ce:ef:af:ca:0c
espsta0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
esp_cfg80211_add_key
esp_cfg80211_add_key
espsta0: WPA: Key negotiation completed with 36:ce:ef:af:ca:0c [PTK=CCMP GTK=CCMP]
espsta0: CTRL-EVENT-CONNECTED - Connection to 36:ce:ef:af:ca:0c completed [id=0 id_str=]

also stay here, if i press CTRL-C
^Cesp_cfg80211_deauth
espsta0: CTRL-EVENT-DISCONNECTED bssid=36:ce:ef:af:ca:0c reason=3esp_cfg80211_del_k  ey
 locally_generated=1
Disconnect event for ssid iPhone [reason:3]
esp_cfg80211_del_key
esp_cfg80211_del_key
esp_cfg80211_del_key
esp_cfg80211_del_key
esp_cfg80211_del_key
esp_cfg80211_del_key
espsta0: CTRL-EVENT-DSCP-POLICY clear_all
espsta0: CTRL-EVENT-DSCP-POLICY clear_all
nl80211: deinit ifname=espsta0 disabled_11b_rates=0
espsta0: CTRL-EVENT-TERMINATING

and esp32 log

I (1422858) FW_MAIN: Scan request

I (1424272) FW_MAIN: Auth request

I (1424374) FW_CMD: Connecting to iPhoneDima, channel: 11 [3]
I (1424374) FW_CMD: AUTH Commit

I (1424380) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (1424380) wifi:state: init -> auth (b0)
I (1424384) FW_CMD: handle_wpa_sta_rx_mgmt:627 Auth[11] recvd

I (1424416) FW_MAIN: Assoc request

I (1424416) wifi:state: auth -> assoc (0)
I (1424422) FW_CMD: handle_wpa_sta_rx_mgmt:635 ASSOC Resp[1] recvd

I (1424422) FW_CMD: STA connect event [channel 11]

I (1424424) BSSID: 0x3ffe8b5a   36 ce ef af ca 0c                                 |6.....|
I (1424432) wifi:state: assoc -> run (10)
I (1425208) FW_MAIN: Add key request
I (1425208) FW_CMD: process_add_key:1561
I (1425256) FW_MAIN: Add key request
I (1425256) FW_CMD: process_add_key:1561
I (1425256) FW_CMD: Setting GTK [1]
I (1425258) FW_CMD: process_add_key:1622 auth done
I (1425258) wifi:connected with iPhone, aid = 1, channel 11, BW20, bssid = 36:ce:ef:af:ca:0c
I (1425268) wifi:security: WPA2-PSK, phy: bgn, rssi: -32
I (1425272) wifi:pm start, type: 1

I (1425276) FW_CMD: Wifi Station Connected event!!

I (1425328) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1537488) FW_MAIN: STA disconnect request

I (1537488) FW_CMD: STA Disconnect request: reason [3]

I (1537490) wifi:state: run -> init (3c0)
I (1537492) wifi:pm stop, total sleep time: 131615669 us / 140270494 us
I (1537496) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (1537502) FW_CMD: STA disconnected [3]
I (1537506) FW_CMD: STA Disconnect event: 3
I (1537510) FW_CMD: cleanup_ap_bssid

mantriyogesh, 2023-01-21:
Great. Once connected, Do not press ctrl c. Wpa supplicant need to be running in background.
Check iwconfig, dhcp and ping/data on another terminal

mantriyogesh, 2023-01-21:
What's difference in these two boards? And which firmware worked the release firmware or patch with timing adjustments?

x893, 2023-01-21:
First with open wifi AP, seconds with WPA2

Host always Zynq XC7Z010
Firmware 1.02 Release (from site).
I need change to wpa_sup... in background.
I can prepare my changes for host driver (remove hci in main.c and ..._sdio.c) if you want add to main branch.

mantriyogesh, 2023-01-21:
Yes definitely, please raise Pull Request. Will review and merge.
Btw, are you talking about the disabling Bluetooth? I suppose there is open pr, pending for merge

x893, 2023-01-21:
Yes, disable bluetooth (esp_bt) and add contions for
CONFIG_SUPPORT_ESP_SERIAL
as in esp-iot-bridge (aka esp-gateway)
I prepare pull later.

Thanks for your help

mantriyogesh, 2023-01-22:
do you mean https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L409-L422 etc? they are part of ESP-Hosted-FG solution.

will wait for your changes.

x893, 2023-01-22:
Hi, i change init.d scripts to start wpa-sup... at startup.
Also i connect RESET line to ESP32 and need increase
esp_reset()
msleep( ... )
now learn hostapd with esp32 hosted.
Thanks for help
P.S. I prepare changes in human readable format :-) and place in my git repo

mantriyogesh, 2023-01-30:
` I prepare changes in human readable format` great to know. I hope the ESP-Hosted documentation was easy to grasp? Any comments that can help are welcome.
Can you please help me pointing your git repo with all the changes?

mantriyogesh, 2023-03-15:
Hello @x893 

Can we close this issue, if you got the solution you were seeking?

static-void, 2024-04-22:
> 3\. reset pin is mandatory as if now. We have some code changes which makes resetpin not needed for SDIO. but they are still in progress. by the time you can add one unused GPIO for resetpin?

Hello, is the reset pin still mandatory? I have prepared an ESP32-C6 for connecting to an SD socket, will it work without reset?

[image]

mantriyogesh, 2024-04-24:
It should work without reset pin as such for sdio.