espressif/esp-hosted #196: compile issue with esp-hosted

zrs5532, 2023-01-28:
on this page: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/Getting_started.md
when configuring stm32+spi, there are a couple of issues.
1. in project properties, resources, linked resoures, "CODE_BASE" path is missing a colon.
2. under project folder, /common/protobuf-c, missing protobuf-c.c and protobuf-c.h files.

mantriyogesh, 2023-01-28:
Hello @zrs5532 

Thank you for using ESP-Hosted for MCU based solution.
Responses:
1. Which platform are you using? (Linux/Windows/Mac/others), some details please? Have you used the prepare_project script ?
2. The code base needs submodules as well to be checked out. Please see https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/MCU_based_readme.md#113-esp-hosted-code-repository:

[image]

zrs5532, 2023-01-29:
yeah, question 1, I am on windows 10, with stm32cubeide version 1.11.0 and I used the script with windows version.

so I didn't load the submodule like you suggested there: git submodule update --init --recursive.
now it is all solved. 

Thanks!

mantriyogesh, 2023-01-30:
How did you get the git submodules update in windows?
We will try to add that step in documentation.

Any other changes you had to do apart from these?

mantriyogesh, 2023-01-30:
In case the issue is solved, help us to close it.