espressif/esp-hosted #197: esp_hosted_fg esp_spi_deinit device reboot

hulinfeng-github, 2023-01-30:
I am developing on ESP32S3 chip and need to integrate the function of auto light sleep and esp32_host_fg. I need to call the interface of SPI initialization and deinit repeatedly to achieve this function. After calling deinit interface, the device appears to reboot and the following content appears in the log: "rst:0x10 ( RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)". Now the initialization and deinitialization of SPI cannot be called repeatedly in the code, so I need to make some changes to the source code. Do you have any suggestions for modification?

mantriyogesh, 2023-01-30:
I understand that you need to make light sleep.
This is host side sleep or ESP side sleep?

As you said you have tried yourself the change, if it is feasible, can you get patch and logs? I'll try to reproduce what you mean.

hulinfeng-github, 2023-01-31:
## Problem Description
The design logic is:After the ESP goes into a light sleep state, it powers down the host, and when a wakeup event is triggered, the ESP powers up the host. The problem with the current modified code is that the EPS is restarted by the RTC watchdog after the ESP is added to the SPI de-initialization logic.

## Log
[09:44:02]E (73998) NETWORK_ADAPTER: spi_driver_deinit driver
[09:44:02]
[09:44:02]I (73998) mpool: 0x3fcad098: num_ret: 63
[09:44:02]I (74008) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[09:44:02]I (74018) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[09:44:02]I (74018) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[09:44:02]I (74028) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[09:44:02]I (74038) mpool: 0x3fcacebc: num_reuse: 60
[09:44:02]I (74048) mpool: 0x3fcad098: num_reuse: 61
[09:44:02]I (74048) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[09:44:02]I (74058) mpool: 0x3fcacebc: num_reuse: 62
[09:44:02]I (74068) mpool: 0x3fcacebc: num_ret: 64
[09:44:02]I (74068) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[09:44:02]I (74078) mpool: 0x3fcacebc: num_ret: 65
[09:44:02]I (74088) mpool: 0x3fcad098: num_ret: 66
[09:44:03]I (75088) mpool: Destroy mempool 0x3fcacebc
[09:44:03]I (75088) SPI_DRIVER: mempool_destroy(buf_mp_g)
[09:44:03]I (75088) mpool: Destroy mempool 0x3fcad098
[09:44:03]I (75088) SPI_DRIVER: mempool_destroy(trans_mp_g)
[09:44:03]E (75088) SPI_DRIVER: spi_slave_free(ESP_SPI_CONTROLLER)
[09:44:03]
[09:44:03]I (75158) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED
[09:44:03]I (75168) sleep: Enable automatic switching of GPIO sleep configuration
[09:44:03]I (75178) sleep: Code start at 42000020, total 829.28 KiB, data start at 3c0d0020, total 190.50 KiB
[09:44:04]ESP-ROM:esp32s3-20210327
[09:44:04]Build:Mar 27 2021
[09:44:04]rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[09:44:04]SPIWP:0xee
[09:44:04]mode:DIO, clock div:2
[09:44:04]load:0x3fce3808,len:0x1654
[09:44:04]load:0x403c9700,len:0xbc8
[09:44:04]load:0x403cc700,len:0x2fb8
[09:44:04]SHA-256 comparison failed:
[09:44:04]Calculated: 73346833c3df666de923f1499af7fcb0214d5051de7a4071e0d3a4b50b9eddd4
[09:44:04]Expected: 9f89bcc59d88e86f82eb1b9f6230dce16bceaf6f73ce1576304298799a7cae2a
[09:44:04]Attempting to boot anyway...
[09:44:04]entry 0x403c9954
[09:44:04]I (48) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
[09:44:04]I (48) boot: compile time 10:04:45
[09:44:04]I (48) boot: chip revision: 0
[09:44:04]I (50) boot.esp32s3: Boot SPI Speed : 40MHz
[09:44:04]I (55) boot.esp32s3: SPI Mode       : DIO
[09:44:04]I (60) boot.esp32s3: SPI Flash Size : 16MB
[09:44:04]I (64) boot: Enabling RNG early entropy source...
[09:44:04]I (70) boot: Partition Table:
[09:44:04]I (73) boot: ## Label            Usage          Type ST Offset   Length
[09:44:04]I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[09:44:04]I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
[09:44:04]I (96) boot:  2 factory          factory app      00 00 00010000 00177000
[09:44:04]I (103) boot: End of partition table

mantriyogesh, 2023-02-03:
It is getting some time, I will check on this and let you know

mantriyogesh, 2023-03-15:
Hello @hulinfeng-github 

I can try this at my end, if you still face the problem.
I am not very clear why deinit & init you had to invoke explicitly. Once ESP comes out of sleep, it will show you similar log to kind of esp is reset but with different reset cause.

I am still trying to understand exact issue.

hulinfeng-github, 2023-04-04:
I have found the cause of the crash: the chip status of the host is abnormal, causing the handshake interrupt to be triggered all the time, and the spi_transaction_post_process_task() thread keeps running, causing other threads to be unscheduled and causing the program to crash. 

In addition, I need ESP32S3 to enter the automatic lightsleep mode. If the SPI does not have deinit, it cannot enter the sleep mode, and the power consumption is about 36mA. After I call the spi_slave_free() interface, the power consumption is about 3mA, and then I hope to re-enable the SPI but find that the SPI cannot communicate with the host normally.
I kept the esp_spi_init() at the beginning, hoping to implement the SPI function switch through the esp_spi_reinit() and esp_spi_deinit() interfaces, but the following program did not achieve the result I expected. Can you provide some suggestions to help me realize this function? Thank you very much for your support.

`
static interface_handle_t * esp_spi_reinit(void)
{
	esp_err_t ret = ESP_OK;
	uint32_t flags = 0;

	spi_bus_config_t buscfg={
		.mosi_io_num=GPIO_MOSI,
		.miso_io_num=GPIO_MISO,
		.sclk_io_num=GPIO_SCLK,
		.quadwp_io_num = -1,
		.quadhd_io_num = -1,
		.max_transfer_sz = SPI_BUFFER_SIZE,
	};

	spi_slave_interface_config_t slvcfg={
		.mode=SPI_MODE_2,
		.spics_io_num=GPIO_CS,
		.queue_size=SPI_QUEUE_SIZE,
		.flags=0,
		.post_setup_cb=spi_post_setup_cb,
		.post_trans_cb=spi_post_trans_cb
	};

	spi_slave_initialize(ESP_SPI_CONTROLLER, &buscfg, &slvcfg, DMA_CHAN);//deinit test
	set_handshake_gpio()；
	memset(&if_handle_g, 0, sizeof(if_handle_g));
	if_handle_g.state = INIT;
	is_exit = 0;
	
	return &if_handle_g;
}

static void esp_spi_deinit(interface_handle_t *handle)
{
	int ret = 0;
	is_exit = 1;

	ret = spi_slave_free(ESP_SPI_CONTROLLER);
	if (ESP_OK != ret) {
		ESP_LOGE(TAG, "spi_slave_free free failed\n");
		return;
	}

	reset_handshake_gpio();
	reset_dataready_gpio();
	gpio_config_t config = {
		.pin_bit_mask = BIT(GPIO_MOSI)|BIT(GPIO_MISO)|BIT(GPIO_SCLK),
		.mode = GPIO_MODE_OUTPUT_OD,
		.pull_down_en = false,
		.pull_up_en = false,
		.intr_type = GPIO_INTR_DISABLE,
    };
    
    gpio_config(&config);
	gpio_set_level(GPIO_MOSI, 0);
	gpio_set_level(GPIO_MISO, 0);
	gpio_set_level(GPIO_SCLK, 0);
}
`

hulinfeng-github, 2023-07-20:
Replace the initialization and resource release of spi slave by means of pm lock

closed 2023-07-20 by hulinfeng-github