espressif/esp-hosted #198: esp-hosted-ng version 1.0.2 compile error

gauthamikosanam, 2023-02-01:
May I know the solution for this
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=arm CROSS_COMPILE=/opt/toolchains/gcc-linaro-7.3.1-2018.05-i686_arm-linux-gnueabihf/bin -C /lib/modules/5.14.0-1056-oem/build M=/home/gauthami/Desktop/esp-hosted-release-ng-v1.0.2(1)/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host clean
/bin/sh: 1: Syntax error: "(" unexpected
make: *** [Makefile:55: clean] Error 2

mantriyogesh, 2023-02-01:
Hello @gauthamikosanam ,

Thank you very much for using ESP-Hosted-NG.
Can you please either of:
1. Rename `esp-hosted-release-ng-v1.0.2(1)` to `esp-hosted-release-ng-v1.0.2` and retry building <- preferred solution
2. OR escape the `(`

gauthamikosanam, 2023-02-01:
Thanks for the reply...I renamed and compiled the code....getting the below error....could you please help me out
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-mrecord-mcount’

mantriyogesh, 2023-02-01:
Can you please send output of:

$ cat /etc/*elease
$ uname -a
$ arm-linux-gnueabihf-gcc -v

I suspect you are using incompatible toolchain to build.

Alternatively, if your expected host has development environment, you can build directly there (cross compilation not needed that case then)

gauthamikosanam, 2023-02-01:
Thanks for your reply....May I know the toolchain to build?
gauthami@gauthami-Latitude-3420:~/Documents/esp-hosted/esp_hosted_ng/host$ cat /etc/*elease
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.4 LTS"
NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.4 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

gauthami@gauthami-Latitude-3420:~/Documents/esp-hosted/esp_hosted_ng/host$ uname -a
Linux gauthami-Latitude-3420 5.14.0-1056-oem #63-Ubuntu SMP Fri Dec 16 14:32:59 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

gauthami@gauthami-Latitude-3420:~/Documents/esp-hosted/esp_hosted_ng/host$ arm-linux-gnueabihf-gcc -v
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc-cross/arm-linux-gnueabihf/9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.4.0-1ubuntu1~20.04.1' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --without-target-system-zlib --enable-libpth-m2 --enable-multiarch --enable-multilib --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --disable-werror --enable-multilib --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=arm-linux-gnueabihf --program-prefix=arm-linux-gnueabihf- --includedir=/usr/arm-linux-gnueabihf/include
Thread model: posix
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1)

gauthamikosanam, 2023-02-01:
I am using atmel a5d2x processor.So in my board having esp32-wroom32ue wifi module.To install the packages am using yocto environment version 2.5.So am loading the boot images into the board and running the application.So I want to know how to load this wifi module driver in yocto......?Can you please briefly explain me......

mantriyogesh, 2023-02-01:
destination machine, for which you are building is 32bit arm? (Reason I am asking, you are cross compiling for 32bit arm).

Anyway, worth to check, https://www.youtube.com/watch?v=gm7lEJS3IuY
Use 'hf' if you know what you are doing.

gauthamikosanam, 2023-02-01:
In usr/bin path arm-linux-gnueabihf is there.....but still getting error....Can you please provide some solution for this issue....
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /lib/modules/5.14.0-1056-oem/build M=/home/gauthami/Documents/esp-hosted/esp_hosted_ng/host clean
make[1]: Entering directory '/usr/src/linux-headers-5.14.0-1056-oem'
make[1]: Leaving directory '/usr/src/linux-headers-5.14.0-1056-oem'
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /lib/modules/5.14.0-1056-oem/build M=/home/gauthami/Documents/esp-hosted/esp_hosted_ng/host modules
make[1]: Entering directory '/usr/src/linux-headers-5.14.0-1056-oem'
  CC [M]  /home/gauthami/Documents/esp-hosted/esp_hosted_ng/host/esp_bt.o
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-mrecord-mcount’

mantriyogesh, 2023-02-01:
Uninstall hf and install non hf version. Uninstall and install with some correct reference.
Did you check the video mentioned above?

mantriyogesh, 2023-02-01:
https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

mantriyogesh, 2023-02-01:
Should cross compilation for target happen through /lib/modules/5.14.0-1056-oem/build?

In general, while using yocto on target, we build kernel with bitbake under yocto environment. As you said, you have worked with yocto, I'm assuming you already know some alternative way, how you are currently doing it.

gauthamikosanam, 2023-02-02:
For cross compiler we are using /opt/toolchains/gcc-linaro-7.3.1-2018.05-i686_arm-linux-gnueabihf/bin.Till now for any packages we need then with the help of yocto using bitbake we are adding the packages and taking the boot images and loading it to the target.So for wifi esp32 wroom 32ue module I want to add this driver in yocto and want to use it in my application.May I know the procedure for that?

gauthamikosanam, 2023-02-02:
How to load the esp-hosted-ng driver and compile in the kernel for atmel Rugged Board A5d2x?May I know the process for that?

mantriyogesh, 2023-02-03:
Hello @gauthamikosanam ,

Apologies, but we are no experts on build systems. You can use the compiler supported for your device. Getting the platform up is outside scope of ESP-Hosted.

You can have a look at  
https://developer.ruggedboard.com/g5-system-development-guide/yocto-compilation-for-rugged-board-a5d2x
https://github.com/linux4sam/meta-atmel

We do not have specific platform instructions as the platforms vary from user to user and there are very high number of platforms present.
For porting the ESP-Hosted to your platform, please check porting guide https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md and https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md

In this doc, you would be asked to change rpi_init.sh with your own KERNEL, ARCH, CROSS_COMPILE, CC, etc variables. You also need to configure the desired peripheral (transport we call it in ESP-Hosted such as SPI, SDIO, UART) on your board , which will be usable by ESP-Hosted kernel module.

gauthamikosanam, 2023-02-08:
Hello@mantriyogesh
I have taken the ver 0.2 and compiled with kernel version 4.9,generated esp32_sdio.ko.when I try to do insmod in my board getting error.

root@rugged-board-a5d2x-sd1:/data# insmod esp32_sdio.ko 
esp32_sdio: version magic '4.9.151-linux4sam_5.8+-04825-g17ec695 mod_unload ARMv7 p2v8 ' should be '4.9.151-linux4sam_5.8+-04826-gef5e9ce04-d'
esp32_sdio: version magic '4.9.151-linux4sam_5.8+-04825-g17ec695 mod_unload ARMv7 p2v8 ' should be '4.9.151-linux4sam_5.8+-04826-gef5e9ce04-d'
insmod: can't insert 'esp32_sdio.ko': invalid module format

Can you please help me out on this...

mantriyogesh, 2023-02-08:
You do not have correct linux headers installed against expected linux kernel version.
As you are building on target linux, you can check the `modinfo` for module built and `uname -a` should match

mantriyogesh, 2023-02-08:
This may happen some times as SoCs do not always maintain this matching headers. You can search how to install exact matching linux kernel headers for your SoC.

For example, raspberry also might have similar issue, there is one way to match kernel headers using `rpi-source`.  
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Linux_based_readme.md#211-raspberry-pi-specific-setup

Anyway above link is just for reference, do not use on your board. You would need to find similar alternative for your SoC.

Basically, you need following command working:

```
ls /lib/modules/$(uname -r)/build/
```
build directory here will be populated once you have correct linux headers installed

gauthamikosanam, 2023-02-09:
Hi
I have inserted the kernel module in my board.Whether it is properly loaded ?
modprobe esp32_sdio.ko 
root@rugged-board-a5d2x-sd1:/data# lsmod
    Not tainted
esp32_sdio 11736 0 - Live 0xbf000000
If it is properly loaded then I am not able to scan the espsta0.May I know the reason.
ifconfig espsta0 up
ifconfig: SIOCGIFFLAGS: No such device
root@rugged-board-a5d2x-sd1:/data# ifconfig espsta0
ifconfig: espsta0: error fetching interface information: Device not found
Can I know the solution for this?

mantriyogesh, 2023-02-09:
Please port rpi_init.sh for your platform.
You can omit/comment the things you do not need. but the sequence of building/loading kernel module should be as close as rpi_init.sh.

You need reset pin and pull-up. By the way, these all information are already captured in porting guide.

mantriyogesh, 2023-02-09:
Hello @gauthamikosanam 

Caution: little longer but detailed reply ahead.

Check the log as mentioned in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SDIO_setup.md#3-checking-the-setup-for-sdio

You should ideally get
(1)
```

[   69.245969] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[   69.253368] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[   69.256622] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   69.258842] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   69.258939] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   69.259035] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   69.260840] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   69.260939] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   69.261040] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
```
Basically CIS info u should get, if the sdio is detected. In your case the SDIO card is not probed.

It might be either of
A. sdio is not correctly configured or instance is not created
B. Wires are either loose or longer than 6cm
C. Device Tree might have have very high sdio freq set (you can set it to 25 or 50MHz)
D. the sdhci or sdhost support is not enabled in your Linux. You might want to check how to enable sdio or sdmmc or sdhci support in your Linux SoC. Generally it will be through Linux config like .config or similar. you might want to change the config and rebuild and load Linux.

(2)
```
[   76.892073] esp32: loading out-of-tree module taints kernel.
[   76.893083] esp32: Resetpin of Host is 6
[   76.893202] esp32: Triggering ESP reset.
```
This log will tell you that sdio kernel module is inserted (just inserted. Whether working or not will be next section)

E. Check command `$ lsmod | grep -I esp` if the kernel module inserted or not. This is independent of step (1) above. In your log I can't see any esp32 trace. So doubtful if the module has inserted or not

(3)
```
[   77.665287] esp_probe: ESP network device detected
[   77.877892] Features supported are:
[   77.877901] 	 * WLAN
[   77.877906] 	 * BT/BLE
[   77.877911] 	   - HCI over SDIO
[   77.877916] 	   - BT/BLE dual mode
[   78.096179] esp_sdio: probe of mmc1:0001:2 failed with error -22
```

E. This will mean that the bootup or init event is received from ESP (first transaction from ESP to Host worked)

Note:
* unfortunately, unless the platform is up & configured, you cannot expect the ESP-Hosted to work. Scan (or any other operation is expected to fail unless you reach minimal to step (3) above)
* porting guide mentioned in above message is important to go through
* whenever you are producing the log always provide both side logs ( host & esp). Without both logs, it is impossible to understand if there is any issue

gauthamikosanam, 2023-02-10:
Hello Sir,
In the device tree, I have added the below lines.
sdmmc0: sdio-host@a0000000 {
                        compatible = "espressif,esp32_sdio";
                        resetpin = <6>;
                        clock-rate = <1000000>;
                        bus-width = <4>;
                        pinctrl-names = "default";
                        pinctrl-0 = <&pinctrl_sdmmc0_default>;
                        non-removable;
                        status = "okay";
                };
But unable to see the log of esp32 related when I boot the kernel in my soc atmel sam a5d2x target board.
Shall we connect the anydesk sir if possible.

mantriyogesh, 2023-02-10:
1. Hello, Can you please help me get complete device tree 

A)
* Before your changes
AND
* After your changes

Zip the all possible of Device Tree where dts, dtc and dtsi are available. This is easier to understand.

B. Last resort if step (1A) is not possible.
create dts from device dtb booted (basically reverse of compiling dtb to dts: This is too much info, hard to debug
https://unix.stackexchange.com/questions/289563/how-to-list-the-kernel-device-tree

Disclaimer: 
A. Please keep in mind, that getting the platform up is **outside** the scope of ESP-Hosted, we are just trying to help you out.
B. Anyway if you loose the data or device not booting or stuck in bootloop, we will not be held responsible for that.
C. We are not master of Device Tree configs & DT varies for all the boards/SoCs.

2. Possible problem in DT
```
resetpin = <6>;
```
Looks very odd in this config
Also clock is much small, some times this also creates problem. Use 50M Clock

3. Possibly sdmmc0 is in use.
In general, you always try to use the peripheral, which is not used. Basically enable the disabled peripheral, say sdmmc1 (if available)

gauthamikosanam, 2023-02-10:
Hello Sir,
[a5d2x-rugged_board.zip](https://github.com/espressif/esp-hosted/files/10706056/a5d2x-rugged_board.zip)
Yes am using cross compilation.

mantriyogesh, 2023-02-10:
> Hello Sir, [a5d2x-rugged_board.zip](https://github.com/espressif/esp-hosted/files/10706056/a5d2x-rugged_board.zip) Yes am using cross compilation given by a5d2x processor. https://github.com/rugged-board/poky

You can call me Yogesh !
These are current configs. Can you please send the original configs, to identify the differences you had made?

gauthamikosanam, 2023-02-10:
Please find it below.
[a5d2x-rugged_board.zip](https://github.com/espressif/esp-hosted/files/10706368/a5d2x-rugged_board.zip)

mantriyogesh, 2023-02-10:
> Please find it below. [a5d2x-rugged_board.zip](https://github.com/espressif/esp-hosted/files/10706368/a5d2x-rugged_board.zip)

orig dts,dtsi? dtb is binary format.

gauthamikosanam, 2023-02-10:
previously we used to take the boot images from support team...we didnt changed the source code....now i installed the yocto and kernel source from rugged board a5d2x and started editing the source code.and compiled.

gauthamikosanam, 2023-02-13:
Hi yogesh,
I am getting this trace .
esp32: Initialization.
esp_reset function
reset pin 3 GPIOPIN INVALID -1
esp32: Resetpin of Host .
esp32: Resetpin of Host is 3
esp32: Triggering ESP reset.
root@rugged-board-a5d2x-sd1:/lib/modules/4.9.151-linux4sam_5.8+-04826-gef5e9ce04-dirty/kernel/drivers/mmc/host/esp32# lsmod | grep -I esp
esp32_sdio 11928 0 - Live 0xbf000000

But still not getting mmc1 and esp_probe: ESP network device detected logs in my code....
Is there any changes I need to do?

mantriyogesh, 2023-02-13:
### Kernel module inserted (step(2))
> esp32: Initialization.
> esp_reset function
> reset pin 3 GPIOPIN INVALID -1
> esp32: Resetpin of Host .
> esp32: Resetpin of Host is 3
> esp32: Triggering ESP reset.

This really means that the kernel module is inserted fine (Step (2) above).
Although it seems the `resetpin` used is incorrect. You might want to find out how the pin numbers are assigned for your board from the your SoC's datasheet.
Links might be useful to check which GPIOs are in use.
https://stackoverflow.com/questions/16033748/how-do-can-i-find-out-which-linux-driver-is-hogging-my-gpio
https://unix.stackexchange.com/questions/485324/how-to-identify-gpio-pin-numbers-is-gpio-driver-loaded
https://www.thegoodpenguin.co.uk/blog/stop-using-sys-class-gpio-its-deprecated/

### Next things
For the kernel module to work correctly, `esp_probe()` function should be automatically called.
Before that, It should sense that there is SD type of device connected.
Basically, Next thing you really need to focus on how to get step (1) working
`mmc1: queuing unknown CIS tuple` in kern.log as discussed in [prior](https://github.com/espressif/esp-hosted/issues/198#issuecomment-1424282645) comment.

Possible reasons are listed, can you please try one by one?
Also I hope you have attached the SDIO pull ups (depends upon which ESP32 chipset you use.) https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

There are some ESP32 chipset types, where in built pull-ups are also provided like ESP32-WROVER-Kit. For the chip you use, please refer above link.

mantriyogesh, 2023-02-16:
Hello @gauthamikosanam 

Any thing got progressed ahead?

gauthamikosanam, 2023-02-16:
Hi Yogesh,
Thanks for your concern to know the status of mine. I have checked that we are using PD5 gpio for enable pin (3rd pin). But how to write in dts file I am checking....because am not that much familiar of handling dts files.
esp _probe function is not automatically called in my case....how to proceed further could youp please let me know? This is my board having wifi module.
[image]

mantriyogesh, 2023-02-16:
Okay.
Have you changed & loaded the sdio DT as provide earlier?
Take a backup of your existing DT and logs such as /var/log/kern/log.log before doing the DT changes.
When new DT in place, at least you should be able to see the CIS traces (check step (1) above).

Also it is worth to check if ESP out and MMC input pin are correctly connected for each function. You can use multimeter to check this.
I am not sure if you had added the expected pull-up registers (check prior posts above) already or not. If not, this circuitry will not work.

gauthamikosanam, 2023-02-16:
I have changed and loaded the previous DT file but am stucking when it is booting.
Hit any key to stop autoboot:  0 
=> pri
arch=arm
baudrate=115200
board=rugged_board_a5d2x
board_name=rugged_board_a5d2x
bootargs=console=ttyS0,115200 console=tty0 earlyprintk root=/dev/mmcblk1p2 rw rootwait
bootcmd=fatload mmc 1 0x21000000 a5d2x-rugged_board.dtb; fatload mmc 1 0x22000000 zImage; bootz 0x22000000 - 0x21000000
bootdelay=3
cpu=armv7
ethaddr=04:91:62:d3:7a:c6
fdtcontroladdr=23b56288
soc=at91
stderr=serial@f8020000
stdin=serial@f8020000
stdout=serial@f8020000
vendor=atmel

Environment size: 480/16380 bytes
=> boot
33551 bytes read in 20 ms (1.6 MiB/s)
4126424 bytes read in 266 ms (14.8 MiB/s)
## Flattened Device Tree blob at 21000000
   Booting using the fdt blob at 0x21000000
   Loading Device Tree to 23b49000, end 23b5430e ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.151-linux4sam_5.8+-04826-gef5e9ce04-dirty (root@0d4c4bb56531) (gcc version 7.3.0 (GCC) ) #4 Thu Feb 9 07:22:22 UTC 2023
CPU: ARMv7 Processor [410fc051] revision 1 (ARMv7), cr=10c53c7d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: RUGGED BOARD A5D2X
cma: Failed to reserve 64 MiB
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 16240
Kernel command line: console=ttyS0,115200 console=tty0 earlyprintk root=/dev/mmcblk1p2 rw rootwait
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 54004K/65536K available (7168K kernel code, 215K rwdata, 1184K rodata, 1024K init, 220K bss, 11532K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0800000   (8160 kB)
      .init : 0xc0a00000 - 0xc0b00000   (1024 kB)
      .data : 0xc0b00000 - 0xc0b35fc8   ( 216 kB)
       .bss : 0xc0b35fc8 - 0xc0b6d208   ( 221 kB)
NR_IRQS:16 nr_irqs:16 16
L2C-310 ID prefetch enabled, offset 2 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 128 kB
L2C-310: CACHE_ID 0x410000c9, AUX_CTRL 0x36020000
clocksource: pit: mask: 0x7ffffff max_cycles: 0x7ffffff, max_idle_ns: 11654027029 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Console: colour dummy device 80x30
console [tty0] enabled
Calibrating delay loop... 326.86 BogoMIPS (lpj=1634304)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x20100000 - 0x20100058
devtmpfs: initialized
VFP support v0.3: implementor 41 architecture 2 part 30 variant 5 rev 1
[... 139 lines cut ...]
sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
NET: Registered protocol family 17
can: controller area network core (rev 20120528 abi 9)
NET: Registered protocol family 29
can: raw protocol (rev 20120528)
can: broadcast manager protocol (rev 20161123 t)
can: netlink gateway (rev 20130117) max_hops=1
Bluetooth: HIDP (Human Interface Emulation) ver 1.2
Bluetooth: HIDP socket layer initialized
at91_i2c f8028000.i2c: can't get DMA channel, continue without DMA support
at91_i2c f8028000.i2c: Using FIFO (16 data)
at24 0-0050: 256 byte 24c02 EEPROM, writable, 8 bytes/write
at91_i2c f8028000.i2c: AT91 i2c bus driver (hw version: 0x704).
at91_rtc f80480b0.rtc: setting system clock to 2012-01-01 00:01:17 UTC (1325376077)
ALSA device list:
  No soundcards found.
Waiting for root device /dev/mmcblk1p2...

I am not added any pull up registers in my board.

In driver I have added printf statement after init module  like ep32 initialization.

when I modprobe the module again If I do dmesg am getting that statement .

after espreset function not able to  get the messages because resetpin is not configured in dts file.

mantriyogesh, 2023-02-16:
If this boot is stuck at,
```
Waiting for root device /dev/mmcblk1p2...
```
I think all the files expected and not copied to the correct expected place. I think your boot-loader loads from location ` /dev/mmcblk1p2`. Do you have prior backup of it? Just check if dtb or any things to be copied are missed while copying in the expected path.

> In driver I have added printf statement after init module like ep32 initialization.
> when I modprobe the module again If I do dmesg am getting that statement .
> after espreset function not able to get the messages because resetpin is not configured in dts file.

Can you please give output of 
```
$ git log --oneline | head -1
$ git diff
```
at esp-hosted repo?

mantriyogesh, 2023-02-16:
```
root@0d4c4bb56531:/home/kernel_source/linux-rba5d2x/drivers/mmc/host/esp32# ls
adapter.h      esp32_sdio.mod.c  esp_api.h     esp_bt.o  esp_rb.c  esp_serial.c  LICENSE  Makefile         Module.symvers
built-in.o     esp32_sdio.mod.o  esp_bt_api.h  esp.h     esp_rb.h  esp_serial.h  main.c   Makefile_backup  sdio
esp32_sdio.ko  esp32_sdio.o      esp_bt.c      esp_if.h  esp_rb.o  esp_serial.o  main.o   modules.order    spi

root@0d4c4bb56531:/home/kernel_source/linux-rba5d2x/drivers/mmc/host/esp32# git log --oneline | head -1
ef5e9ce04 added PPP and QMI default support in rugged board defconfig

root@0d4c4bb56531:/home/kernel_source/linux-rba5d2x/drivers/mmc/host/esp32# git diff
diff --git a/arch/arm/boot/dts/a5d2x-rugged_board_common.dtsi b/arch/arm/boot/dts/a5d2x-rugged_board_common.dtsi
index 7b854d5..7853acc 100644
--- a/arch/arm/boot/dts/a5d2x-rugged_board_common.dtsi
+++ b/arch/arm/boot/dts/a5d2x-rugged_board_common.dtsi
@@ -91,11 +91,23 @@
                };
 
                sdmmc0: sdio-host@a0000000 {
-                       bus-width = <8>;
+                       bus-width = <4>;
                        pinctrl-names = "default";
                        pinctrl-0 = <&pinctrl_sdmmc0_default>;
                        non-removable;
                        status = "okay";
+                       #address-cells = <1>;
+                       #size-cells = <0>;
+                       esp32: esp32_sdio@0 {
+                               reg = <0>;
+                               #address-cells = <1>;
+                               #size-cells = <0>;
+                               status = "okay";
+                               sdio_drv_name = "esp_sdio";
+                               compatible = "espressif,esp32_sdio";
+                               resetpin = <3>;
+                               //probe = <&esp_probe>;
+                       };
                };
 
                sdmmc1: sdio-host@b0000000 {
@@ -178,7 +190,7 @@
                        uart2: serial@f8024000 {
                                pinctrl-names = "default";
                                pinctrl-0 = <&pinctrl_uart2_default>;
-                               rtsgpio = <&pioA PIN_PC21 GPIO_ACTIVE_HIGH>;
+                       ///     rtsgpio = <&pioA PIN_PC21 GPIO_ACTIVE_HIGH>;
```

mantriyogesh, 2023-02-16:
The images built with this driver tree is booting or not booting?

gauthamikosanam, 2023-02-16:
Its booting yogesh.

mantriyogesh, 2023-02-16:
As one of prior reply, https://github.com/espressif/esp-hosted/issues/198#issuecomment-1425397661 resetpin placement is wrong.

> I am not added any pull up registers in my board

If sd pull up link shared earlier says that your board needs pull-up, iit simply needs pull-up, otherwise even if you get sdio CIS (at some point ahead), the kernel driver will not work.

If you not able add now or don't wish to alter the board, you can move to spi solution.
As asked in earlier post, did you verify the pin connections say using multimeter?

mantriyogesh, 2023-02-16:
There are already many questions open this post. I am still waiting to get the ESP firmware log (if it is even starting and loading correctly?)

mantriyogesh, 2023-02-16:
I also pointed to https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md which explain all the requirements for kernel driver.

mantriyogesh, 2023-02-16:
Also sdmmc might be getting used by system. try using sdmmc1. you can refer to the dts modified changes provided earlier.
In any case, do not proceed with SDIO if you do not have pull-ups (if they are required for your ESP).

mantriyogesh, 2023-02-16:
for reset pin, please refer how others do it:
https://github.com/espressif/esp-hosted/issues/200#issuecomment-1418383327

Please do not copy paste, as it will not work. Refer to your SoC reference manual or datasheet for correct values.

gauthamikosanam, 2023-02-22:
Hi Yogesh
This is my .config file.
[config.zip](https://github.com/espressif/esp-hosted/files/10802201/config.zip)
CONFIG_MMC_SDHCI=y ,this was enabled.
I am using esp ver 0.2.
modprobe esp32_sdio resetpin=101
getting logs of esp32
esp32: Resetpin of Host is 101
esp32: Triggering ESP reset.
lsmod
esp32_sdio 11864 0 - Live 0xbf000000
I am using sdmmc1 is for sdcard and sdmmc0 is using for wifi device purpose.
I am using esp32 wroom 32E module.
I didnt understand the sd pull up requirements.Could you please let me know so that I can move forward.

mantriyogesh, 2023-02-25:
https://github.com/espressif/esp-hosted/issues/198#issuecomment-1427732622

gauthamikosanam, 2023-02-27:
Hi Yogesh,
I have added sd pullup requirements. Added 10kohm resistor to ClK,CMD,DAT0-DAT3 pins of esp32 wroom 32E module.
I have run the code of ver0.2.
Unable to enter into interface esp_init_interface_layer function.Could you please let me know to proceed further.
esp32: Resetpin of Host is 101
esp32: Triggering ESP reset.
ESP Adapter function

gauthamikosanam, 2023-02-28:
Hi Yogesh,
I have made some changes in the config file....Added the esp32 support in Kconfig file but still not able to receive the sdio logs in kernel.
[config.tar.gz](https://github.com/espressif/esp-hosted/files/10848813/config.tar.gz)

mantriyogesh, 2023-03-01:
Get ESP log

gauthamikosanam, 2023-03-01:
Hi Yogesh,
These are the logs of config and esp log.
[esp_log.txt](https://github.com/espressif/esp-hosted/files/10857098/esp_log.txt)
[config.tar.gz](https://github.com/espressif/esp-hosted/files/10857100/config.tar.gz)
I have added sd pull ups for clk,cmd,dat0-dat3 of 10kohm resistor.
unable to detect the sdio.Please let me know where I am doing wrong.

mantriyogesh, 2023-03-01:
esp_log.txt contains host side log.
Need ESP side log. you can use minicom or idf.py monitor for the same.

gauthamikosanam, 2023-03-01:
Hi Yogesh,
I am getting this error when I was connected with minicom.
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

mantriyogesh, 2023-03-01:
Exactly, you might have saved a lot time, if you would have provided ESP log [earlier](https://github.com/espressif/esp-hosted/issues/198#issuecomment-1423958267).

Try re-flashing the ESP32 with 1.0.2 ESP32 SDIO only release binaries.
Please get full log of the flashing, in case it doesn't work.

gauthamikosanam, 2023-03-01:
Hi Yogesh,
could you please tell me the procedure for this.
Try re-flashing the ESP32 with 1.0.2 ESP32 SDIO only release binaries.

mantriyogesh, 2023-03-01:
Please check the ESP-Hosted documentation in detail for steps.
https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng

gauthamikosanam, 2023-03-01:
Hi Yogesh,
If I do pip install esptool ,I am getting like this.
gauthami@gauthami-Latitude-3420:~/ng-v1.0.2/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host$ pip install esptool
Requirement already satisfied: esptool in /home/gauthami/.local/lib/python3.8/site-packages (4.4)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in /home/gauthami/.local/lib/python3.8/site-packages (from esptool) (1.5.4)
Requirement already satisfied: ecdsa>=0.16.0 in /home/gauthami/.local/lib/python3.8/site-packages (from esptool) (0.18.0)
Requirement already satisfied: cryptography>=2.1.4 in /usr/lib/python3/dist-packages (from esptool) (2.8)
Requirement already satisfied: pyserial>=3.0 in /home/gauthami/.local/lib/python3.8/site-packages (from esptool) (3.5)
Requirement already satisfied: bitstring<4,>=3.1.6 in /home/gauthami/.local/lib/python3.8/site-packages (from esptool) (3.1.9)
Requirement already satisfied: six>=1.9.0 in /usr/lib/python3/dist-packages (from ecdsa>=0.16.0->esptool) (1.14.0)
I have downloaded the ver 1.0.2 .
What is the procedure to flash the sdio binaries?

mantriyogesh, 2023-03-01:
Hello @gauthamikosanam 
Please check documentation first. I think your questions should be automatically answered.
These are some basic steps you want to follow from documentation.

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md

mantriyogesh, 2023-03-02:
I hope you would have figured out the binary flashing by now.
[image]

gauthamikosanam, 2023-03-02:
Yes Yogesh I found and downloaded the latest release.Found the flashing_cmd.txt file in the folder.
But I am facing this issue.
dmesg
[249741.907974] ftdi_sio 3-7:1.0: FTDI USB Serial Device converter detected
[249741.908017] usb 3-7: Detected FT232RL
[249741.908657] usb 3-7: FTDI USB Serial Device converter now attached to ttyUSB0
gauthami@gauthami-Latitude-3420:~/ESP-Hosted-NG_release_v1.0.2/ESP-Hosted-NG_release_v1.0.2/esp32/sdio_only$ esptool.py -p /dev/ttyUSB0 -b 115200 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 network_adapter.bin
esptool.py v4.4
Serial port /dev/ttyUSB0

A fatal error occurred: Could not open /dev/ttyUSB0, the port doesn't exist

Checking on that.

mantriyogesh, 2023-03-02:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html

gauthamikosanam, 2023-03-02:
Hi Yogesh 
I have done flashing but getting this error.
[image]

mantriyogesh, 2023-03-02:
I suggest you to follow the documentation slowly and correctly.
Some times even google can help.

Please understand the issues you are reporting are not in the scope of ESP-Hosted.
Those are just basic issues to get the platform or environment up (much before ESP-Hosted can work).

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html#adding-user-to-dialout-on-linux

gauthamikosanam, 2023-03-03:
Hi Yogesh,
I have compiled the hello world program and If do flashing into the board getting error. Could you please help me out on this.
This is my schematic of esp32 wroom 32E wifi module...
[Error log.zip](https://github.com/espressif/esp-hosted/files/10878437/Error.log.zip)

[image]

[90 more replies]

closed 2023-07-12 by mantriyogesh