espressif/esp-hosted #199: ESP32c3 doesn't work when docker image is there in yocto

meera1310, 2023-02-03:
ESP32c3 was working fine in my yocto image. When I added docker service in my image then as soon as I run this command insmod esp32_spi.ko resetpin=297
my board goes into kernel panic mode

mantriyogesh, 2023-02-03:
Okay. Thank you for explaining scenario.

1. Can you please run tail -F /var/log/kern.log in seperate terminal and load the module in other terminal and get back both the logs?
2. Same time, esp logs
3. What is the libux platform/board you are using? Kernel version?

meera1310, 2023-02-06:
Processor: smarc-rzg2lc
Kernel: 5.10.83-cip1+gitAUTOINC+e52825e45b
[esp32c3 side logs.odt](https://github.com/espressif/esp-hosted/files/10659852/esp32c3.side.logs.odt)
[module insert logs.odt](https://github.com/espressif/esp-hosted/files/10659854/module.insert.logs.odt)

mantriyogesh, 2023-02-06:
1. Please let us know output of:
```sh
$ cat init_script.sh
```
init_script here should be very similar to rpi_init.sh.

2. Also, I am not understanding if the crash is in minicom or in the module. Can you please run
```sh
$ tail -F /var/log/kern.log &
```
in separate terminal (different from console window) and minicom in another (again different than console)
and get us all 4 window outputs (esp minicom, /var/log/kern.log, commands run terminal log, console log)

meera1310, 2023-02-06:
root@oldsilicon:~# cat esp32c3/init_script.sh 
mkdir /sys/kernel/config/device-tree/overlays/spi_driver
# Disable default spidev driver
dtc spidev_disabler.dts -O dtb > spidev_disabler.dtbo
cat spidev_disabler.dtbo > /sys/kernel/config/device-tree/overlays/spi_driver/dtbo
insmod esp32_spi.ko resetpin=297

mknod /dev/esps0 c 221 0
chmod 666 /dev/esps0

root@oldsilicon:~# tail: cannot open `/var/log/kern.log' for reading: No such file or directory

mantriyogesh, 2023-02-06:
Just rewriting the response with correct formatting:

```sh
root@oldsilicon:~# cat esp32c3/init_script.sh 
mkdir /sys/kernel/config/device-tree/overlays/spi_driver
# Disable default spidev driver
dtc spidev_disabler.dts -O dtb > spidev_disabler.dtbo
cat spidev_disabler.dtbo > /sys/kernel/config/device-tree/overlays/spi_driver/dtbo
insmod esp32_spi.ko resetpin=297

mknod /dev/esps0 c 221 0
chmod 666 /dev/esps0
```

mantriyogesh, 2023-02-06:
1. Where do you do steps mentioned in rpi_init.sh?

Like 
1. removing existing module, serial driver https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L46-L53

2. 
https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L169-L172

and remaining others.

Please port rpi_init.sh correctly for your platform.
You will end up in issues similar to this if you bypass.

2. Any of /var/log/messages or /var/log/syslog or /var/log/kern.log
If nothing there, clean console log will be appreciated..
Need to get clear log, otherwise cannot understand the scenario for crash

meera1310, 2023-02-06:
root@oldsilicon:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.10.83-cip1-yocto-standard (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 8.3.0, GNU ld (GNU Binutils) 2.31.1) #1 SMP PREEMPT Sat Feb 27 02:21:18 UTC 2021
[    0.000000] Machine model: Renesas SMARC EVK based on r9a07g044c2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000058000000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma@58000000, compatible id shared-dma-pool
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000068000000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,multimedia, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000048000000-0x00000000bfffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000048000000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000048000000-0x00000000bfffffff]
[    0.000000] On node 0 totalpages: 491520
[    0.000000]   DMA zone: 7680 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 491520 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 23 pages/cpu s55576 r8192 d30440 u94208
[    0.000000] pcpu-alloc: s55576 r8192 d30440 u94208 alloc=23*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 483840
[    0.000000] Kernel command line: root=/dev/mmcblk1p2 rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1511420K/1966080K available (13632K kernel code, 1112K rwdata, 3876K rodata, 2176K init, 431K bss, 61444K reserved, 393216K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=2.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 480 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x0000000011940000
[    0.000000] random: get_random_bytes called from start_kernel+0x318/0x4c8 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000004] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000130] Console: colour dummy device 80x25
[    0.000512] printk: console [tty0] enabled
[    0.000549] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.000573] pid_max: default: 32768 minimum: 301
[    0.000710] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000734] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.002294] rcu: Hierarchical SRCU implementation.
[    0.002851] Detected Renesas RZ/G2L r9a07g044 
[    0.003079] EFI services will not be available.
[    0.003328] smp: Bringing up secondary CPUs ...
[    0.003797] Detected VIPT I-cache on CPU1
[    0.003838] GICv3: CPU1: found redistributor 100 region 0:0x0000000011960000
[    0.003890] CPU1: Booted secondary processor 0x0000000100 [0x412fd050]
[    0.004100] smp: Brought up 1 node, 2 CPUs
[    0.004113] SMP: Total of 2 processors activated.
[... 379 lines cut ...]
[    5.544774] systemd[1]: Mounted POSIX Message Queue File System.
[    5.553641] systemd[1]: Mounted Kernel Debug File System.
[    5.561828] systemd[1]: Mounted Temporary Directory (/tmp).
[    5.655932] mmngr: loading out-of-tree module taints kernel.
[    5.663403] rgnmm_drv mmngr: assigned reserved memory node linux,multimedia
[    5.768894] systemd[1]: Started Journal Service.
[    5.915687] EXT4-fs (mmcblk1p2): re-mounted. Opts: (null)
[    6.027283] systemd-journald[161]: Received client request to flush runtime journal.
[    8.054741] random: crng init done
[    8.058164] random: 7 urandom warning(s) missed due to ratelimiting
[... cut ...]
[  133.986393] audit: type=1325 audit(1675674229.896:11): table=filter family=2 entries=12 op=xt_replace pid=338 comm="iptables"
[  134.050745] audit: type=1325 audit(1675674229.976:12): table=nat family=2 entries=7 op=xt_replace pid=344 comm="iptables"
[  134.071766] audit: type=1325 audit(1675674230.000:13): table=nat family=2 entries=8 op=xt_replace pid=346 comm="iptables"
[  134.093222] audit: type=1325 audit(1675674230.020:14): table=filter family=2 entries=12 op=xt_replace pid=349 comm="iptables"
[  134.902944] process 'docker/tmp/qemu-check140870943/check' started with executable stack

mantriyogesh, 2023-02-06:
I did not see any esp32 kernel module prints in this log.
Nevertheless, worth noting that kernel does point you that the DT you have is not correct.
From your log:
```
 /soc/spi@1004b000/spidev@1: buggy DT: spidev listed directly in DT
 ```
 spidev has to be be disabled 'correctly' from you device tree. unless that done, both spidev and esp32 kernel module will crash for unexpected reasons.

meera1310, 2023-02-06:
This whole functionality was working since 2 months before I haven't added docker in my image. This wifi feature we have tested so many times, along with that all the functionalities are working fine. As soon as I added docker my esp32c3 stopped working.

mantriyogesh, 2023-02-06:
I understand the pain. 
Let's go stepwise.

1. Docker setup - I am trying to understand, how the docker is relevant & what it could change from device tree / config.
Can you state why the docker was needed and how it was related for building, or it was altogether for some other software dependency and you needed for that (Non ESP-Hosted user space application) 

2. Do you have any full dmsg log saved that time, when it was working?
It will be easy hint to compare what went wrong.

3. I still think that spidev should be properly disabled.
in your case, spidev is trying to probe, it is failing.
Did it start to probe after your docker installation (and was not invoking before docker installation)?
https://elixir.bootlin.com/linux/v5.10.83/source/drivers/spi/spidev.c#L736 
says it is mostly because your 'compatible' string is incorrect, should be small thing to fix.

4. Also, please change your shell script such that it will 'remove/delete the nodes which it intend to create'.
https://github.com/espressif/esp-hosted/issues/199#issuecomment-1418619623

5. Once these things are good, let us try to module and capture the logs.

Do not worry, You had it already working earlier, means you fairly know the setup and eventually you will get it working.

meera1310, 2023-02-06:
(1) I have another image without docker and in that esp32c3 working fine: 
root@oldsilicon:~# docker
-sh: docker: command not found
root@oldsilicon:~# cd esp32c3/
esp32_spi.ko                  spidev_disabler.dts
esp32c3-wifi-ap-mode-init.sh  test.out
init_script.sh                wifi-ap-control.log
pspl-dcu-kill-dnsmasq.py      wifi-ap-dnsmasq.log
spidev_disabler.dtbo          
root@oldsilicon:~# cd esp32c3/
root@oldsilicon:~/esp32c3# sh init_script.sh 
[   88.529226] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@1004b000/spidev@1/status
[   88.568698] ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
[   88.921658] usb 4-1.1: new full-speed USB device number 5 using ohci-platform
root@oldsilicon:~/esp32c3# [   90.121987] 
[   90.121987] Received INIT event from ESP32 peripheral
[   90.122002] EVENT: 2
[   90.130806] EVENT: 1
[   90.132987] EVENT: 0
[   90.135218] ESP peripheral capabilities: 0xe8
[   90.346532] EVENT: 3

(2) Also I have added this patch but /soc/spi@1004b000/spidev@1: buggy DT: spidev listed directly in DT this line remained same:

`static const struct of_device_id spidev_dt_ids[] = {
        { .compatible = "lwn,bk4" },
        { .compatible = "dh,dhcom-board" },
        { .compatible = "menlo,m53cpld" },
       +{ .compatible = "spidev" }, 
        {},
 };
 MODULE_DEVICE_TABLE(of, spidev_dt_ids);

(3) root@oldsilicon:~/esp32c3# cat init_script.sh 

mkdir /sys/kernel/config/device-tree/overlays/spi_driver
# Disable default spidev driver
dtc spidev_disabler.dts -O dtb > spidev_disabler.dtbo
cat spidev_disabler.dtbo > /sys/kernel/config/device-tree/overlays/spi_driver/dtbo

if [ `lsmod | grep esp32 | wc -l` != "0" ]; then
	sudo rm /dev/esps0
	if [ `lsmod | grep esp32_sdio | wc -l` != "0" ]; then
		sudo rmmod esp32_sdio &> /dev/null
		else
		sudo rmmod esp32_spi &> /dev/null
        fi
fi

insmod esp32_spi.ko resetpin=297

mknod /dev/esps0 c 221 0
chmod 666 /dev/esps0
[without_docker.odt](https://github.com/espressif/esp-hosted/files/10662053/without_docker.odt)

mantriyogesh, 2023-02-06:
Please use github formatting. It helps to read better.

mantriyogesh, 2023-02-06:
Anyway I am checking the logs.

mantriyogesh, 2023-02-06:
https://github.com/espressif/esp-hosted/issues/199#issuecomment-1418742257 cannot be reference as failed log, as the module seems not loaded.
Can you get the logs (non working setup)?

Please always get ESP side, dmesg logs and run log together. that way no need to repeat test for another log.

mantriyogesh, 2023-02-10:
@meera1310 
Did you get chance to test?

meera1310, 2023-02-10:
[failure_logs.odt](https://github.com/espressif/esp-hosted/files/10703941/failure_logs.odt)

mantriyogesh, 2023-02-10:
I am not sure, why I have to repeat this again, but
Please always get **ESP side, dmesg logs and run log** together.

Anyway, with current log, I cannot deduce where the kernel crash originated.
Please add entry and exit logs and try to localise which till which line the kernel crash is not occurred.

Add this definition to common header file like adapter.h
```
#define HOSTED_DBG printk(KERN_ERR "%s:%u \n", __func__, __LINE__)
```
and call it in all affected areas to point out which like is kernel crash.

mantriyogesh, 2023-03-08:
any updates ahead?

mantriyogesh, 2023-03-15:
Any updates ahead @meera1310 ?

meera1310, 2023-03-15:
It worked with fresh VLP image. I think so there might be some problems related to network configuration files. Thanks

mantriyogesh, 2023-03-15:
Great.
Thanks for the response.

Could we please close this thread if you no more face issue?

closed 2023-03-16 by mantriyogesh