espressif/esp-hosted #200: ESP32C3 not respond when porting to imx6 platform

yruigg, 2023-02-04:
Hi,
I have successfully used the NG repository on RaspberryPi 4B+ platform and tested the Wifi and Blutooth function. I am running esp32-c3 in spi mode for both wifi and bluetooth, everything is working well. dmesg shown in screenshot below. 
[image]

But things turn bad when I try to port this to imx6ull platform. My host has debian os. We have modified our dtb and driver code matching with our hardware. I can load driver esp32_spi.ko at host successfully, but it stuck afther register SPI bus.  Here are some message we collected.
 ## dmesg from imx6ull
```
[  202.288873] esp32_spi: loading out-of-tree module taints kernel. 
[  202.304461] esp_reset, ESP32: Triggering ESP reset.  
[  202.537600] ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [5]  
[  203.911984] data_ready_irq!  
[  203.915001] spi_irq!  
[  203.919120] spi_irq!  
[  203.921434] spi_irq!  
[  203.923728] spi_irq!  
```
 ## uart console from esp32c3
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16cc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
[0;32mI (30) boot: ESP-IDF v4.4.1-584-g91dc99a3ce-dirty 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 17:17:27[0m
[0;32mI (30) boot: chip revision: 3[0m
[0;32mI (34) boot.esp32c3: SPI Speed      : 80MHz[0m
[0;32mI (39) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (43) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Enabling RNG early entropy source...[0m
[0;32mI (53) boot: Partition Table:[0m
[0;32mI (57) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (87) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (109) boot: End of partition table[0m
[0;32mI (113) boot: Defaulting to factory image[0m
[0;32mI (118) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=16f48h ( 94024) map[0m
[0;32mI (141) esp_image: segment 1: paddr=00026f70 vaddr=3fc90a00 size=032d0h ( 13008) load[0m
[0;32mI (144) esp_image: segment 2: paddr=0002a248 vaddr=40380000 size=05dd0h ( 24016) load[0m
[0;32mI (152) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=8af44h (569156) map[0m
[0;32mI (244) esp_image: segment 4: paddr=000baf6c vaddr=40385dd0 size=0aa30h ( 43568) load[0m
[0;32mI (253) esp_image: segment 5: paddr=000c59a4 vaddr=50000010 size=00010h (    16) load[0m
[0;32mI (258) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (258) boot: Disabling RNG early entropy source...[0m
[0;32mI (273) cpu_start: Pro cpu up.[0m
[0;32mI (282) cpu_start: Pro cpu start user code[0m
[0;32mI (282) cpu_start: cpu freq: 160000000[0m
[0;32mI (282) cpu_start: Application information:[0m
[0;32mI (285) cpu_start: Project name:     network_adapter[0m
[0;32mI (291) cpu_start: App version:      2027593[0m
[0;32mI (296) cpu_start: Compile time:     Dec 16 2022 17:17:16[0m
[0;32mI (302) cpu_start: ELF file SHA256:  4771564a015efb36...[0m
[0;32mI (308) cpu_start: ESP-IDF:          v4.4.1-584-g91dc99a3ce-dirty[0m
[0;32mI (315) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (322) heap_init: At 3FC97E80 len 00028180 (160 KiB): DRAM[0m
[0;32mI (328) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM[0m
[0;32mI (335) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM[0m
[0;32mI (341) spi_flash: detected chip: generic[0m
[0;32mI (346) spi_flash: flash io: dio[0m
[0;32mI (350) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (356) sleep: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (364) coexist: coexist rom version 9387209[0m
[0;32mI (369) cpu_start: Starting scheduler.[0m
[0;32mI (373) stats: *********************************************************************[0m
[0;32mI (373) stats:                 ESP-Hosted Firmware version :: 1.0.2                        [0m
[0;32mI (383) stats:                 Transport used :: SPI only                      [0m
[0;32mI (393) stats: *********************************************************************[0m
[0;32mI (403) FW_MAIN: Supported features are:[0m
[0;32mI (403) FW_MAIN: - WLAN over SPI[0m
[0;32mI (413) FW_BT: - BT/BLE[0m
[0;32mI (413) FW_BT:    - HCI Over SPI[0m
[0;32mI (413) FW_BT:    - BLE only[0m
[0;32mI (423) FW_MAIN: Capabilities: 0xe8[0m
[0;32mI (433) pp: pp rom version: 9387209[0m
[0;32mI (433) net80211: net80211 rom version: 9387209[0m
I (443) wifi:wifi driver task: 3fc9e49c, prio:23, stack:6656, core=0
[0;32mI (443) system_api: Base MAC address is not set[0m
[0;32mI (443) system_api: read default base MAC address from EFUSE[0m
I (453) wifi:wifi firmware version: 9f5e450
I (453) wifi:wifi certification version: v7.0
I (463) wifi:config NVS flash: disabled
I (463) wifi:config nano formating: disabled
I (463) wifi:Init data frame dynamic rx buffer num: 32
I (473) wifi:Init management frame dynamic rx buffer num: 32
I (473) wifi:Init management short buffer num: 32
I (483) wifi:Init dynamic tx buffer num: 32
I (483) wifi:Init static tx FG buffer num: 2
I (493) wifi:Init static rx buffer size: 1600
I (493) wifi:Init static rx buffer num: 10
I (503) wifi:Init dynamic rx buffer num: 32
[0;32mI (503) wifi_init: rx ba win: 6[0m
[0;32mI (503) wifi_init: tcpip mbox: 32[0m
[0;32mI (513) wifi_init: udp mbox: 6[0m
[0;32mI (513) wifi_init: tcp mbox: 6[0m
[0;32mI (513) wifi_init: tcp tx win: 5744[0m
[0;32mI (523) wifi_init: tcp rx win: 5744[0m
[0;32mI (523) wifi_init: tcp mss: 1440[0m
[0;32mI (533) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (533) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (543) BTDM_INIT: BT controller compile version [05195c9][0m
[0;32mI (543) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07[0m
[0;32mI (583) BTDM_INIT: Bluetooth MAC: 34:b4:72:49:48:7e
[0m
[0;32mI (583) FW_MAIN: ESP Bluetooth MAC addr: 34-b4-72-49-48-7e
[0m
[0;32mI (583) FW_SPI: Using SPI interface[0m
[0;32mI (583) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (593) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1603) FW_MAIN: Initial set up done[0m
```
I read some issues from this repository and make some attempts, here are something i know: 
- it seems the host does not get bootup event from esp32c3
- this same esp32c3(firmware also same) works well with RaspberryPi 4B+.
- i have changed the SPI_MODE to 0, 1, 2, 3  but no help.
- i test the imx6ull spi interface(SPI3) with other spi device, and it works well.
- i add message print in interrupt_handler functions in esp_spi.c(visible in dmesg), it seems the handshake and dataready pin is working
- i am using PCB hardware so no need to worry jump wires.
---
Now I am stuck and confused. Should we change SPI_MODE both in host and in esp32 firmware?  But we do not have the NG esp firmware source code and do not know how to modify.  Please help.

mantriyogesh, 2023-02-06:
Thank you @yruigg for much detailed debug.

Can you please add some points for did you port your solution?
Also when you say, you have received the interrupt(s), these are manully induced or com I ng from ESP-Hosted

mantriyogesh, 2023-02-06:
As you say rpi worked fine, I assume, you have used the same code base to build the kernel module at the imx.

mantriyogesh, 2023-02-06:
Also, any chance you hold logic analyzer?

yruigg, 2023-02-06:
1. We change the pin define and spi bus num in host source code. these mostly refer to the patch file in your review in issue #137
2. Here is the DTS overlay
```
#include "../imx6ul-pinfunc.h"

/dts-v1/;
/plugin/;

/ {
		fragment@0 {
			target = <&ecspi3>;
			__overlay__ {
				fsl,spi-num-chipselects = <1>;
				cs-gpios = <&gpio1 20 0>;
				pinctrl-names = "default";
				pinctrl-0 = <&pinctrl_ecspi3>;
				status = "okay";
				#address-cells = <1>;
				#size-cells = <0>; 
				spidev@0 {
					status = "disable";
					compatible = "spidev";
					spi-max-frequency = <10000000>;
					reg = <0>;
				};
			};
		};
		
		fragment@1 {
			target = <&iomuxc>;
			__overlay__ {
				pinctrl_ecspi3:ecspi3grp {
					fsl,pins = <
						MX6UL_PAD_UART2_TX_DATA__ECSPI3_SS0     	0x10b0
						MX6UL_PAD_UART2_RX_DATA__ECSPI3_SCLK		0x10b0
						MX6UL_PAD_UART2_CTS_B__ECSPI3_MOSI			0x10b0
						MX6UL_PAD_UART2_RTS_B__ECSPI3_MISO			0x10b0
						MX6UL_PAD_GPIO1_IO00__GPIO1_IO00			0x10b0/* handshake */
						MX6UL_PAD_GPIO1_IO02__GPIO1_IO02			0x10b0/* data ready */
						MX6UL_PAD_CSI_DATA01__GPIO4_IO22			0x10b0/* esp32 reset */
					>;
				};						
			};
		};
};
```
3. no logic analyzer nearby currently, can be used after several days. 
4.  is it necessary to change SPI_MODE in ESP32C3 firmware? could you provide the NG version 1.02 firmware which change the SPI_MODE for me to test?

mantriyogesh, 2023-02-06:
I can give you the spi mode changed firmware, but issue doesn't seem to be for spi transaction.

If the interrpts are coming from the ESP-Hosted pins,
https://github.com/espressif/esp-hosted/blob/29208b6c156be06dfd90dadc6dd43acd4d98d748/esp_hosted_ng/host/spi/esp_spi.c#L67-L74

And 
https://github.com/espressif/esp-hosted/blob/29208b6c156be06dfd90dadc6dd43acd4d98d748/esp_hosted_ng/host/spi/esp_spi.c#L76-L83

Then I think base spi working. And only might be timing issue.

Although, sometimes, this  could also be false sense some times if the received data is bit shifted (due to spi timing)

Anyway, I will add some debug prints at spi tx and rx along with spi mode changed firmware s, which would point the issue.

yruigg, 2023-02-06:
we add print in two interrupt_handler funcitons as blow
```
static irqreturn_t spi_data_ready_interrupt_handler(int irq, void * dev)
{
	/* ESP peripheral has queued buffer for transmission */
	printk("data_ready_irq!\n");
 	if (spi_context.spi_workqueue)
 		queue_work(spi_context.spi_workqueue, &spi_context.spi_work);

 	return IRQ_HANDLED;
 }

static irqreturn_t spi_interrupt_handler(int irq, void * dev)
{
	/* ESP peripheral is ready for next SPI transaction */
	printk("spi_irq!\n");
	if (spi_context.spi_workqueue)
		queue_work(spi_context.spi_workqueue, &spi_context.spi_work);

	return IRQ_HANDLED;
}
```
we do insmod command with esp32_spi.ko and get dmesg
```
[  202.288873] esp32_spi: loading out-of-tree module taints kernel. 
[  202.304461] esp_reset, ESP32: Triggering ESP reset.  
[  202.537600] ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [5]  
[  203.911984] data_ready_irq!  
[  203.915001] spi_irq!  
[  203.919120] spi_irq!  
[  203.921434] spi_irq!  
[  203.923728] spi_irq! 
```

mantriyogesh, 2023-02-06:
Yes. this is great.

This most likely this looks timing issue.

We will add debug logs in ESP firmware & Host for TX and RX to understand the scenario.
We will be able to send the patch & firmware bin on or before tomorrow morning.
Is that okay?

yruigg, 2023-02-06:
OK. Thanks.

yruigg, 2023-02-09:
We get waveform in SPI_MODE 2 . 
[esp32_capture.zip](https://github.com/espressif/esp-hosted/files/10695171/esp32_capture.zip)

Seems that bytes at MOSI are all zeros. And MISO has a strange rise.

mantriyogesh, 2023-02-09:
I think it is better to do some home work, before jumping directly to the project.

SPI transport:
1. Can you please verify if spidev works first? Tx & RX (This verifies SPI all pins - CLK, MISO, MOSI, CS)
2. (After step (1) pass,) Add 3 more pins (which are **free** & **not used** by any other peripheral)
  2.1. Resetpin Pin - check if GPIO low/high works as expected
  2.2. Data Ready Pin- Check if GPIO works with Positive edge interrupt some ISR function is invoked
  2.3. Handshake Pin- Check if GPIO works with Positive edge interrupt some ISR function is invoked
3. (After steps (2.1) (2.2) (2.3) pass,) Disable spidev for bus `X` and ChipSelect `Y`. Verify `ls /dev/spidevX.Y` is not more visible in Linux
4. Port the ESP-Hosted code (Which you have already done I believe). Porting instructions: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md & https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md (including rpi_init.sh script porting, code adaptation for Pins etc
5. Connect hardware wires in ESP-Hosted expected way (jumper wire and their lengths). You already have done this with PCB. We always recommend to try it with small wires first and then go for PCB.
6. Build and load the kernel module using ported rpi_init.sh. Keep an eye on `tail -F /ver/log/kern.log` or `tail -F /var/log/messages`
7. Build and load ESP binaries/ build & flash ESP through source. Keep an eye on ESP logs using either`idf.py monitor` or using minicom

yruigg, 2023-02-13:
Now I can get data at MISO, but still no bootup event. I compare imx6 waveform with rpi, seems there is one bit shift at MISO. Below are waveforms.
[image]
[image]

mantriyogesh, 2023-02-13:
oh okay.
What was the mode used 2 or 3 (both the cases)?
Can you try other one?
Or else mode mismatch?

Please find the readme.txt from the tarball:
[2023.02.13_NG_B1.0.2_GH200_SPI_Modes.tgz](https://github.com/espressif/esp-hosted/files/10722815/2023.02.13_NG_B1.0.2_GH200_SPI_Modes.tgz)

This tarball contains firmware binaries for esp32-c3 for SPI modes (1, 2 & 3).

mantriyogesh, 2023-02-13:
The SPI debug logs of tx & rx at host can be changed:

1. SPI_Tx: Uncomment: https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_ng/host/spi/esp_spi.c#L417

2. SPI_Rx: Add line 
```
    /*printk(KERN_INFO "next rx pkt->\n");
    print_hex_dump(KERN_ERR, "spi_rx: ", DUMP_PREFIX_ADDRESS, 16, 1, skb->data, skb->len, 1);*/
```
at line 612, just above `skb_pull(skb, offset);`

https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_ng/host/main.c#L611-L616

mantriyogesh, 2023-02-13:
Freezing the SPI speed from Host:

Also, for time being, **comment** these two line:
https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_ng/host/spi/esp_spi.c#L216-L217

And
Change https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_ng/host/spi/esp_spi.c#L29
to your expected frequency.

As you are aware, with the so much debugs prints enabled, this firmware could be used for debugging only(unfit for production level).
We will try to fix the timing first and then I can float the firmware without such debug prints enabled.

yruigg, 2023-02-15:
Hi. Now ESP host is working at imx6. I use SPI_MODE_0 for imx6 host, and SPI_MODE_3 for esp32c3 firmware. I did some tests with espsta0 and everything works fine. Thanks for you help.

[image]

```
debian@npi:~/EmbedFire/esp32_host$ ifconfig
espsta0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.171.103  netmask 255.255.255.0  broadcast 192.168.171.255
        inet6 240e:40e:2300:2c0:36b4:72ff:fe49:487c  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::36b4:72ff:fe49:487c  prefixlen 64  scopeid 0x20<link>
        ether 34:b4:72:49:48:7c  txqueuelen 1000  (Ethernet)
        RX packets 20  bytes 9087 (8.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 28  bytes 4812 (4.6 KiB)
        TX errors 0  dropped 16 overruns 0  carrier 0  collisions 0

```

mantriyogesh, 2023-02-15:
Okay. Great!!
Thank you @yruigg for the confirmation.
The interop - timing issues are sometimes tricky and eat some time sometimes.

mantriyogesh, 2023-02-15:
[2023.02.13_NG_B1.0.2_GH200_SPIMode3_NoDebug.tgz](https://github.com/espressif/esp-hosted/files/10739634/2023.02.13_NG_B1.0.2_GH200_SPIMode3_NoDebug.tgz)
Check once.
This is ESP32-C3 firmware with SPI mode 3 (based on 1.0.2 release)

mantriyogesh, 2023-02-15:
Please do let me know once you are okay to close this thread.

yruigg, 2023-02-16:
I am okay to close this thread. And could you provide different SPI_MODE firmware when new version release?

mantriyogesh, 2023-02-16:
Yes, I understand the problem (We are working on this config problem also).

Anyway, Just add message here once you decide to upgrade in future. I will get what you need exactly.

closed 2023-02-16 by yruigg