espressif/esp-hosted #201: write failed: Broken pipe

MrHaiCoder, 2023-02-08:
I am use esp_hosted_fg, use spi transfer；
test espsta mode，use iperf tcp test, often  occur  write failed: Broken pipe;
What causes this situation？
[image]

MrHaiCoder, 2023-02-08:
use esp32s3

mantriyogesh, 2023-02-08:
Can't judge only from iperf log.
I suspect stream getting closed (due to.timeout or something)

Can you share:
1. Which MCU/Linux?
2. Logs at host, esp

mantriyogesh, 2023-02-08:
Any errors popped up in either logs above will give some hint.

MrHaiCoder, 2023-02-08:
> 在上面的任一日志中弹出的任何错误都会给出一些提示。
use linux4.9.4 version;
to save a pin,we donot use host reset esp32s3 pin, 
esp32s3 will sleep  wait a time，to sure host recv init event ;
detail log
[20230208-173146.log](https://github.com/espressif/esp-hosted/files/10684526/20230208-173146.log)

mantriyogesh, 2023-02-08:
All logs are mixed up and cannot understand.
Still I checked, no noticeable error found in log.

Can you please open a new terminal and do
$ tail -F /var/log/kern.log
If you don't have /var/log/kern.log, you can use /var/log/messages.
Now, In the **same terminal**, get to connection up (test_2.out etc ... Till ping working) and run iperf there, to reproduce the issue.

Also please collect **esp side logs** at the same time.
- If you are flashing source code with idf.py, you can add 'monitor' in the command
- if you are flashing release binary, you can do minicom

mantriyogesh, 2023-02-08:
Are you receiving `Received INIT event from ESP32 peripheral` when the issue occurs?
If yes, that hits that ESP is getting reset, we need to check ESP logs in that case.

MrHaiCoder, 2023-02-09:
in Office environment test ，There will be iperf broken pipe on your side ?

mantriyogesh, 2023-02-09:
Nope. It doesn't.
Please send us the above logs, we can definitely check if something is wrong or not.

MrHaiCoder, 2023-02-09:
What does the kernel log mainly focus on? There is no such file in our system. The log mcu colleagues on the esp side are collecting data. I first used tcpdump to capture the data packets. You can use wireshark to view

192.168.0.1 is ethap0 ip

[target.zip](https://github.com/espressif/esp-hosted/files/10694449/target.zip)

MrHaiCoder, 2023-02-09:
Hi，mantriyogesh，
In the space environment without other wifi ap, use iperf to conduct multiple tests for a long time, and there is no broken pipe

mantriyogesh, 2023-02-09:
> What does the kernel log mainly focus on? There is no such file in our system. The log mcu colleagues on the esp side are collecting data. I first used tcpdump to capture the data packets. You can [...]

Two parts for ESP-Hosted:
1. Host
- As you said, you are using Linux system, it must have some log, where kernel dumps the log. It could be either `/var/log/kern.log` or /var/log/messages`
Nothing works out, run `dmesg` to get current kernel log.
Our kernel driver will log some information onto kernel log (as it is running in kernel space).
- In addition, Get us the run log (commands you have fired and their output)

2. ESP
Get us ESP log
- If you are using firmware release binary you can get ESP side log with minicom.
- If you are building and flashing manually with code, even adding 'monitor' in existing idf.py command will get you ESP side log. e.g. `idf.py flash monitor`

---

> Hi，mantriyogesh，
> In the space environment without other wifi ap, use iperf to conduct multiple tests for a long time, and there is no broken pipe

Good to see you are trying it with other AP (to verify if it is AP problem itself).

BTW, which iperf version you are using?

mantriyogesh, 2023-02-09:
By the time I will check you iperf tcp wireshark also. I am no expert in TCP, but as I understand, TCP re-transmission and error control is part of TCP protocol.

MrHaiCoder, 2023-02-09:
use iperf2.0.8  test

mantriyogesh, 2023-02-09:
Can you please send `ifconfig -a`?

Also, I see no FIN in the log.
There are too many factors affecting the OOO or duplicate ACKs like line of sight, distance, radio conditions, noisy environment (say microwave or Bluetooth device is nearby the AP or station etc, etc).

I would suggest you to try with with another channel number and different bandwidth say 40MHz for softAP and re-assess the connection.

mantriyogesh, 2023-02-09:
Alternatively, to assess is to use these two (ESP-SoftAP and AP) in quieter environment and have same (exactly same, like channel, BW, 2.4GHz etc) radio configs (switch on one at a time) and compare if issue observed.
Do let us know the finding, we can know how to progress then.

MrHaiCoder, 2023-02-09:
[image]
./test_2.out  softap_start 
ifconfig ethap0 192.168.0.1 netmask 255.255.255.0
route add default gw 192.168.0.1 dev ethap0
./busybox udhcpd udhcpd.conf &

mantriyogesh, 2023-02-09:
ethap0 doesn't show the dropped packets.

Anyway, share the logs as mentioned in ealier comment to verify is something wrong at ESP-Hosted.

MrHaiCoder, 2023-02-26:
> ethap0 doesn't show the dropped packets.
> 
> Anyway, share the logs as mentioned in ealier comment to verify is something wrong at ESP-Hosted.

There is no problem with the version of iperf2.0.9. It should be the version of iperf；
 wifi/iperf/main/iperf_example_main.c also no this problem.

mantriyogesh, 2023-02-26:
https://github.com/espressif/esp-hosted/issues/201#issuecomment-1424031200

Please change (improve) radio conditions & espap0 radio configs as discussed earlier.
If you see the code, the ESP-Hosted just relays the packet without even processing it.

MrHaiCoder, 2023-02-26:
> [#201 (comment)](https://github.com/espressif/esp-hosted/issues/201#issuecomment-1424031200)
> 
> Please change (improve) radio conditions & espap0 radio configs as discussed earlier. If you see the code, the ESP-Hosted just relays the packet without even processing it.

Yeah,I understand. Linux host code is really only used to add data headers and transmit them.

I've encountered another problem on my side. It is also linux 4.9+esp32s3, Fg solution, linux is running on dual core mips chip. The esp32s3 acts as an ap, the pc goes to connect to the ap and uses tcp to transfer data, after running for about 5min, the network latency becomes very high.
I added the log to read the value of tx_pending after atomic_inc(&tx_pending) and also added the log to read the value of tx_pending after atomic_dec(&tx_pending), which is a producer-consumer relationship, and I found that the producer is very slow at the back, i.e. the network is not sending out. But I turned off a cpu core in going to run the network, and there was no network latency getting higher

MrHaiCoder, 2023-02-26:
[image]
See this image, should this { be moved to the top? Because tx_pending is used to control the
PRIO_Q_OTHER queue

mantriyogesh, 2023-02-26:
Okay, let me review the code in the morning, get back to you..

mantriyogesh, 2023-02-26:
Btw thank you for checking this issue and all observations.
If any problem, definitely, we should address.

MrHaiCoder, 2023-02-27:
> Btw thank you for checking this issue and all observations. If any problem, definitely, we should address.

[image]
[image]

[image]

CPU production package becomes very slow.

mantriyogesh, 2023-02-27:
Actually, producer can wait if the current packets are not fully processed.

[image]

mantriyogesh, 2023-02-27:
Okay, let me check your last post, it seems I also added the post same time.

MrHaiCoder, 2023-02-27:
> Actually, producer can wait if the current packets are not fully processed.
> 
> [image]

[image]

mantriyogesh, 2023-02-27:
[image]

We should only decrement the tx_pending when we have actually processed the packet.
If the tx_skb is NULL for PRIO_Q_OTHERS, we simply come out, because there is no packet pending in the queue.

Can you please go through below trial list and review and/or test:
[TRY 1]
We can check the tx_pending first and if `tx_pending` is non zero, try to fetch the tx_skb out.
If `tx_pending` is zero, we safely assume, there is no packet pending for PRIO_Q_OTHERS. 
```c
283     if (trans_ready) {
284         if (data_path) {
285             tx_skb = skb_dequeue(&spi_context.tx_q[PRIO_Q_SERIAL]);
286             if (!tx_skb)
287                 tx_skb = skb_dequeue(&spi_context.tx_q[PRIO_Q_BT]);
288             if (!tx_skb) {
289                 if (atomic_read(&tx_pending)) {
290                     tx_skb = skb_dequeue(&spi_context.tx_q[PRIO_Q_OTHERS]);
291                     if (tx_skb) {
292                         atomic_dec(&tx_pending);
293
294                         if (atomic_read(&tx_pending) < TX_RESUME_THRESHOLD) {
295                             esp_tx_resume();
296 #if TEST_RAW_TP
297                             esp_raw_tp_queue_resume();
298 #endif
299                         }
300                     }
301                 }
302             }
303         }
```

[Try 2] From the above snippet, **remove** part
```c
294                         if (atomic_read(&tx_pending) < TX_RESUME_THRESHOLD) {
295                             esp_tx_resume();
296 #if TEST_RAW_TP
297                             esp_raw_tp_queue_resume();
298 #endif
299                         }
```
and add it at just before
```
360     mutex_unlock(&spi_lock);
```

[TRY 3]
As I said in earlier post, if we out-running the capacity. we can find out if we are hitting 
[image]
Please revert all TRY 1 / 2 changes so that we do not complicate the findings.

[TRY 4]
As from your earlier observation for SMP you are facing the issue and not facing issue when it is uniprocessor.
Scheduling problems?

I had thought if we can replace `mutex_lock(&spi_lock); & mutex_unlock(&spi_lock);` with spinlock like `spin_lock_irqsave & spin_unlock_irqrestore` or down_interruptible semaphore.
Please revert all TRY 1 / 2 changes so that we do not complicate the findings.

MrHaiCoder, 2023-02-27:
I will try it as soon as possible. This problem will not occur when using a core CPU.
BTW, why is there no lock at the place where you join the queue?

mantriyogesh, 2023-02-27:
> why is there no lock at the place where you join the queue?

Can you please point me the exact code line(s)?

MrHaiCoder, 2023-02-27:
> > why is there no lock at the place where you join the queue?
> 
> Can you please point me the exact code line(s)?

[image]

mantriyogesh, 2023-02-27:
I am not sure what are you pointing me at.
Anyway, Queues are by default protected. All operations are expected to be serialised when we do operations over queue.
So we do not want any additional locks for queues.

mantriyogesh, 2023-02-27:
But then what are you running on the Linux in user space?
Is there any packet sent from Linux to ESP?
Can you share the complete patch & logs together?

tx pause happens when the packets are flowing more than capacity, to hold the queue for any further inputs.
tx resume happens when there is sufficient room to hold new packets (capacity is available)

I cannot understand from your last images.

For small number of packets, queue will not be paused & resumed, as it will be well within the capacity.
tx_pending will increment after placing item into tx_queue. Similarly decrement once it is dequeued for processing.

So it is better to share the code patch and logs at the same time to derive correct conclusion (and for shared understanding)

mantriyogesh, 2023-02-27:
Packet will only be sent if user space sends something (say iperf traffic or some packet by higher layer like arp, DHCP etc).
If nothing is being sent from user-space, nothing the kernel module will do.
It will only act if user space have sent something.

MrHaiCoder, 2023-02-27:
[src_log.tar.gz](https://github.com/espressif/esp-hosted/files/10837127/src_log.tar.gz)

Above is the test program and the corresponding kernel driver log, the user space program is sending network packets without interruption. In the log you can focus on the time before and after the empty line

mantriyogesh, 2023-02-27:
tcp_server.c -> What am I expected to understand from this file?

I need the code change you have with ESP-Hosted repo.
You can make patch of it.

also please capture the running log (commands you fire in user-space and output of them)
Also get ESP log, just to be sure nothing else is going wrong underneath.

mantriyogesh, 2023-02-27:
Also please help point me the problem in the log may be using timestamp or line number

MrHaiCoder, 2023-02-27:
tcp_server.c is my test demo，what I want to say is that this program does not stop sending data.

[image]

mantriyogesh, 2023-02-27:
https://github.com/espressif/esp-hosted/issues/201#issuecomment-1445848967

MrHaiCoder, 2023-02-27:
I need another colleague to help me with the esp32 log, but I can confirm that the data is being sent through the spi very quickly. Is it convenient to set up a test environment on your side?

mantriyogesh, 2023-02-27:
I can, but it might take some time to test, I am in between priority work.
As you had been in problem since long, I am trying to respond quickly.

Can you send the ESP-Hosted code change patch and tcp client side logs?

MrHaiCoder, 2023-02-27:
[patch.zip](https://github.com/espressif/esp-hosted/files/10837486/patch.zip)
NetAssist can used tcp client

MrHaiCoder, 2023-02-27:
> I can, but it might take some time to test, I am in between priority work. As you had been in problem since long, I am trying to respond quickly.
> 
> Can you send the ESP-Hosted code change patch and tcp client side logs?

Please support as soon as possible, thank you very much!

mantriyogesh, 2023-02-27:
Yes, Anyway, I will test with iperf2, when I reproduce this issue.
Is that okay for you?

I might take some time to test this.

MrHaiCoder, 2023-02-27:
If you can use iperf, it is recommended that you run the ping program every other time and check the ttl;
Because my git repository no longer contains the previous. git hidden file, it is not convenient to generate a patch. I just add print to the two functions without logical modification

MrHaiCoder, 2023-02-28:
If you are ready to test, it is recommended to use only two cpu cores for verification

mantriyogesh, 2023-03-01:
I will be able to do this tomorrow.
Apologies for the delay..

mantriyogesh, 2023-03-02:
If I add cs_change for esp32s3 on Raspberry Pi 4, the init/bootup event is also not received.
It is dependent on the host as said earlier, please **remove** it if things work with your host.

mantriyogesh, 2023-03-02:
[2023.03.92_FG_NotReproduced_broken_pipe.tgz](https://github.com/espressif/esp-hosted/files/10870853/2023.03.92_FG_NotReproduced_broken_pipe.tgz)

Broken pipe not seen,
Also the traffic did not stuck in between for me (3/4 sec delay kind of issue for tx process)
Tried both, with Rx & Tx. (Tx makes more sense, still tried Rx as well)

MrHaiCoder, 2023-03-03:
Because without this code （trans.cs_change = 1）, the host can't get the mac by calling the command,   
./test.out set_sta_mac_addr
so add

MrHaiCoder, 2023-03-03:
Have you matched the wifi antenna of the Raspberry Pi you are using?
The high ping ttl latency from previous feedback may be related to the esp32s3, which our colleague added to print. Then after removing host and esp32s3 newly added debug logs.
We then tested inside the wireless shield box, the bandwidth can be up to 15Mbps, but then inside the office congested wifi channel, the bandwidth is very low, as follows.

[image]

Can you provide any suggestions on how to improve wifi bandwidth stability?

mantriyogesh, 2023-03-03:
Anyway It should not reduce to ZERO.
Can you check the routes and ifconfig if traffic routed incorrect IP?

Regarding Wi-Fi stability, I also test in office environment only, where signals are crowded, the effect will be seen with fluctuations, not dead zero.
Can you try another AP, with automatic channel selection and bandwidth to HT40 from HT20?

I have not done any changes to RF. (just bare metal)

mantriyogesh, 2023-03-03:
[image]

mantriyogesh, 2023-03-03:
Can you also try on spi mode 3 at both ESP & RPi

MrHaiCoder, 2023-03-03:
Thank you very much for trying!
I don't have a Raspberry Pi board, but what you said I will try。

Here is my network configuration, because our product has fewer resources for cost, the choice is udhcpd and his configuration as a dhcp service.

ifconfig ethap0 192.168.0.1 netmask 255.255.255.0 &&  route add default gw 192.168.0.1 dev ethap0 && ./busybox udhcpd udhcpd.conf & 
[udhcpd.zip](https://github.com/espressif/esp-hosted/files/10882550/udhcpd.zip)

mantriyogesh, 2023-03-03:
I can't verify the udhcpd configs as such. But if you can get the ip address, should be fine.

It is still puzzling me whay I could go 15mbps  and other place going down to 0.
Can you help to figure out what could be different?
Hardwares used like specific chipset or soc having issue/software like low ram or cpu or kernel or linux change/ quality or length of jumper wires used/ power cables/power adapters/ specific place in office/debug prints / change in tools used for testing

mantriyogesh, 2023-03-03:
$ ifconfig -a
$ route -n

mantriyogesh, 2023-03-03:
Can you capture wifi sniffer logs ?
May give some hint.

MrHaiCoder, 2023-03-04:
There are plenty of 2.4G wifi cameras in the office, tested on the negative ground floor without wifi interference with the office, using the same hardware and firmware program. 
esp and host are on the same pcb, not convenient to use dupont wire to replace the device
I can   capture wifi sniffer logs .  Available next Monday

mantriyogesh, 2023-03-04:
Carrie other logs as well at the same time when you test

[3 more replies]

closed 2023-03-20 by MrHaiCoder