espressif/esp-hosted #202: Station and AP simultaneously on Linux

philip-ro, 2023-02-09:
Hi all,

I'd like to use the ESP32 on Linux, providing both interfaces simultaneously, one as station and one as access point.

Reading the docs I see the drawback of both ESP-hosted versions: FG is not recommended for Linux and NG does not support the softAP mode.

Is my use case possible? How would you recommend to proceed?

Best regards,
Philip

philip-ro, 2023-02-09:
Hi @mantriyogesh,

I see. Thank you for the quick reply.

It sounds like the softAP support will come eventually for NG. Is there a time frame of when this could be expected to happen?

mantriyogesh, 2023-02-09:
softAP support will come to NG, but it will definitely take much time. No schedules as if now I have to share.