espressif/esp-hosted #203: what's the behavior after "rmmod esp32_sdio.ko" in esp32 firmware?

Ncerzzk, 2023-02-11:
Hi, Thank you for the great work of this project.

For now, I use the sdio interface of esp-hosted-ng in my linux board.
There are other 2.4G device on this board, so I want to shutdown the esp-netcard when necessary to prevent interference and save power.

Can I just simplify `rmmode esp32_sdio.ko` to shut it down? Will the firmware of esp32 then shut down its wifi interface? Or I need to pull down the reset pin to make sure the firmware is not running?

Thank you.

mantriyogesh, 2023-02-13:
Hello @Ncerzzk 

Thank you for using ESP-Hosted & raising this query.
If I understood correctly what you are asking is `host sleep` feature.
It is available in NG release 1.0.2.

Please note, host sleep feature will only make sense if host is sleep-able (some host aren't). Also how they sleep (commands to be fired to make them sleep) varies from platform to platform (Soc).

In host sleep feature, your host will sleep on some script or trigger (which you can easily handle or run it). The sleeping host will be awaken by ESP using a GPIO. This specific GPIO need to be configured in DeviceTree that it is not being used (free) by any other peripheral in specific way.

All such details are captured in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/host_sleep.md. Please take a look.

Ncerzzk, 2023-02-13:
> 

Thank you @mantriyogesh.

But I'm sorry that make your misunderstand. What I want to ask is : 

Can `rmmod esp32_sdio.ko` shutdown the esp32? ( the 'shutdown' means esp32 would not send or recv 2.4G signals and I don't care whether the firmware is still running or not)

to shutdown the esp-netcard now, I control the reset pin to low. But I want to know whether I can omit this step(pull reset pin down) or not.

mantriyogesh, 2023-02-14:
Oh understood now.
In short,
Yes! In fact, that would be lot easier for you.

Alternatively, you can do 2 step process (applicable only if want to save the ESP side reset):
Send a station disconnect using 
https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_ng/host/include/adapter.h#L110
which works similar to [esp_wifi_disconnect()](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_disconnectv)

Obviously the ESP doesn't need the reset pin circuitry in this case but you would have to send connect message again to kickstart the station connection.

Ncerzzk, 2023-02-14:
Thank you. I got it.
now I can omit the pin control code ^ ^

closed 2023-02-14 by Ncerzzk