espressif/esp-hosted #204: no rssi and quality udpdated 1.0.2

micheledarold, 2023-02-13:
Good morning with the command 
 iwlist espsta0 scan |grep 'ESSID\|Channel\|Encryption\|level'

I got
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=0 dBm  
                    Encryption key:off
                    ESSID:"ESP-0DAE84"
                    Channel:1

Quality is always 70/70 and Signal level = 0 dBm
the connection is working but these value in never updated. I'm using a wrong command? Or this feature in not supported?

Regards
Michele

mantriyogesh, 2023-02-13:
Very good morning @micheledarold & thank you for tetsing ESP-Hosted.
I regret that you had to face such issue.

Anyway, I have a cached command output of release 1.0.2 : `sudo iw dev espsta0 scan`
shows me the variations in signal quality (see below)

If you need specific command, I can try that and let you know so.
But it is all communicated to kernel. if one command works, other also should work, unless there are any limitation at application side.

pi@raspberrypi:~/h2/esp_hosted_ng/host$ sudo iw dev espsta0 scan

> Dec 30 11:33:07 raspberrypi kernel: [  627.316839] esp_cfg80211_scan
> Dec 30 11:33:07 raspberrypi kernel: [  627.316839] esp_cfg80211_scan
> BSS 00:0a:f5:14:33:5c(on espsta0)
> 	TSF: 17251569937 usec (0d, 04:47:31)
> 	freq: 2412
> 	beacon interval: 100 TUs
> 	capability: ESS Privacy ShortPreamble ShortSlotTime (0x0431)
> 	**signal: 0.-56 dBm**
> 	last seen: 1890 ms ago
> 	**SSID: SaveEarth**
> 	Supported rates: 1.0* 2.0* 5.5* 11.0*
> 	DS Parameter set: channel 1
> 	ERP: <no flags>
> 	Extended supported rates: 6.0 9.0 12.0 18.0 24.0 36.0 48.0 54.0
> 	RSN:	 * Version: 1
> 		 * Group cipher: CCMP
> 		 * Pairwise ciphers: CCMP
> 		 * Authentication suites: PSK
> 		 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
> 	HT capabilities:
> 		Capabilities: 0x12c
> 			HT20
> 			SM Power Save disabled
> 			RX HT20 SGI
> 			RX STBC 1-stream
> 			Max AMSDU length: 3839 bytes
> 			No DSSS/CCK HT40
> 		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
> 		Minimum RX AMPDU time spacing: No restriction (0x00)
> 		HT RX MCS rate indexes supported: 0-7
> 		HT TX MCS rate indexes are undefined
> 	HT operation:
> 		 * primary channel: 1
> 		 * secondary channel offset: no secondary
> 		 * STA channel width: 20 MHz
> 		 * RIFS: 0
> 		 * HT protection: nonmember
> 		 * non-GF present: 0
> 		 * OBSS non-GF present: 1
> 		 * dual beacon: 0
> 		 * dual CTS protection: 0
> 		 * STBC beacon: 0
> 		 * L-SIG TXOP Prot: 1
> 		 * PCO active: 0
> 		 * PCO phase: 0
> 	WMM:	 * Parameter version 1
> 		 * u-APSD
> 		 * BE: CW 15-1023, AIFSN 3
> 		 * BK: CW 15-1023, AIFSN 7
> 		 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
> 		 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec
> BSS e4:a7:c5:e2:44:42(on espsta0)
> 	TSF: 23297870983 usec (0d, 06:28:17)
> 	freq: 2432
> 	beacon interval: 100 TUs
> 	capability: ESS Privacy ShortPreamble QoS ShortSlotTime (0x0631)
> 	**signal: 0.-48 dBm**
> 	last seen: 1310 ms ago
> 	**SSID: Bubblesnake**
> 	Supported rates: 1.0* 2.0* 5.5* 6.0 9.0 11.0* 12.0 18.0
> 	DS Parameter set: channel 5
> 	ERP: <no flags>
> 	RSN:	 * Version: 1
> 		 * Group cipher: TKIP
> 		 * Pairwise ciphers: CCMP TKIP
> 		 * Authentication suites: PSK
> 		 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC MFP-capable (0x008c)
> 		 * 0 PMKIDs
> 		 * Group mgmt cipher suite: AES-128-CMAC
> 	WPA:	 * Version: 1
> 		 * Group cipher: TKIP
> 		 * Pairwise ciphers: CCMP TKIP
> 		 * Authentication suites: PSK
> 	WMM:	 * Parameter version 1
> 		 * u-APSD
> 		 * BE: CW 15-1023, AIFSN 3
> 		 * BK: CW 15-1023, AIFSN 7
> 		 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
> 		 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec
> 	Extended supported rates: 24.0 36.0 48.0 54.0
> 	BSS Load:
> 		 * station count: 1
> 		 * channel utilisation: 0/255
> 		 * available admission capacity: 0 [*32us]
> 	HT capabilities:
> 		Capabilities: 0xd1ad
> 			RX LDPC
> 			HT20
> 			SM Power Save disabled
> 			RX HT20 SGI
> 			TX STBC
> 			RX STBC 1-stream
> 			Max AMSDU length: 3839 bytes
> 			DSSS/CCK HT40
> 			40 MHz Intolerant
> 			L-SIG TXOP protection
> 		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
> 		Minimum RX AMPDU time spacing: 4 usec (0x05)
> 		HT TX/RX MCS rate indexes supported: 0-15
> 	HT operation:
> 		 * primary channel: 5
> 		 * secondary channel offset: no secondary
> 		 * STA channel width: 20 MHz
> 		 * RIFS: 1
> 		 * HT protection: no
> 		 * non-GF present: 1
> 		 * OBSS non-GF present: 0
> 		 * dual beacon: 0
> 		 * dual CTS protection: 0
> 		 * STBC beacon: 0
> 		 * L-SIG TXOP Prot: 0
> 		 * PCO active: 0
> 		 * PCO phase: 0
> 	Extended capabilities:
> 	WPS:	 * Version: 1.0
> 		 * Wi-Fi Protected Setup State: 2 (Configured)
> 		 * Response Type: 3 (AP)
> 		 * UUID: c3675f49-a006-5fec-9317-2402d395f53d
> 		 * Manufacturer: HUAWEI
> 		 * Model: E5
> 		 * Model Number: 123
> 		 * Serial Number: 12345
> 		 * Primary Device Type: 6-0050f204-1
> 		 * Device name: E5
> 		 * Config methods: Label, PBC
> 		 * Unknown TLV (0x1049, 6 bytes): 00 37 2a 00 01 20
>

micheledarold, 2023-02-13:
hi @mantriyogesh the command you suggest me is working,
maybe the problem could be related to the command iwlist (we are on a embedded system no raspberry based on yocto and busybox)?

mantriyogesh, 2023-02-13:
Oh, okay.
For yocto, you might want to check any/all depending packages are installed.

In any case, I'll verify your expected command tomorrow on Raspberry Pi and check for consistent result. Should that be okay?

micheledarold, 2023-02-13:
yes sure let me know the result!

mantriyogesh, 2023-03-02:
Apologies for delayed response.

I tried myself and can reproduce this issue.

```sh
pi@raspberrypi:~/h3/esp_hosted_ng/host $ iwlist espsta0 scan |grep 'ESSID|Channel|Encryption|level'
pi@raspberrypi:~/h3/esp_hosted_ng/host $ iwlist espsta0 scan
espsta0   Scan completed :
          Cell 01 - Address: 74:11:B2:74:1E:A1
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
                    Quality=70/70  Signal level=0 dBm
                    Encryption key:on
                    ESSID:"ESP_Guest"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000005761254a30
                    Extra: Last beacon: 1120ms ago
                    IE: Unknown: 00094553505F4775657374
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 030106
                    IE: Unknown: 0706534704010D17
                    IE: Unknown: 200100
                    IE: Unknown: 2A0102
                    IE: Unknown: 32040C121860
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 0B0500005F8D5B
                    IE: Unknown: 46053300000000
                    IE: Unknown: 3603AC3400
                    IE: Unknown: 2D1AAD0917FFFFFFFF00000000000000000000000000000000000000
                    IE: Unknown: 3D1606000300000000000000000000000000000000000000
                    IE: Unknown: 7F0404100804
                    IE: Unknown: FF2023010008120010202002C00F438518000C00AAFFAAFF3B1CC7711CC7711CC771
                    IE: Unknown: FF072404000081FCFF
                    IE: Unknown: FF0E260800A40820A408404308603208
                    IE: Unknown: DD050040960B89
                    IE: Unknown: DD050040960305
                    IE: Unknown: DD050040961400
                    IE: Unknown: DD050040962C0E
                    IE: Unknown: DD180050F2020101880003A4000027A4000042435E0062322F00
          Cell 02 - Address: 74:11:B2:C9:5E:40
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
                    Quality=70/70  Signal level=0 dBm
                    Encryption key:on
                    ESSID:"ESP_1"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=000000b27dd50e4f
                    Extra: Last beacon: 520ms ago
                    IE: Unknown: 00054553505F31
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 03010B
                    IE: Unknown: 0706534704010D17
                    IE: Unknown: 200100
                    IE: Unknown: 2A0104
                    IE: Unknown: 32040C121860
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 0B0501003A8D5B
                    IE: Unknown: 46053300000000
[... 683 lines cut ...]
		 * secondary channel offset: no secondary
		 * STA channel width: 20 MHz
		 * RIFS: 1
		 * HT protection: nonmember
		 * non-GF present: 0
		 * OBSS non-GF present: 1
		 * dual beacon: 0
		 * dual CTS protection: 0
		 * STBC beacon: 0
		 * L-SIG TXOP Prot: 0
		 * PCO active: 0
		 * PCO phase: 0
	Extended capabilities:
		 * Extended Channel Switching
		 * Proxy ARP Service
		 * BSS Transition
		 * DMS
	WMM:	 * Parameter version 1
		 * u-APSD
		 * BE: CW 15-1023, AIFSN 3
		 * BK: CW 15-1023, AIFSN 7
		 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
		 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec

```

mantriyogesh, 2023-03-02:
I will check further on this and get back.

For now, Can 
```sh
$ sudo iw dev espsta0 scan
```
workaround for you, till I find the root cause?

micheledarold, 2023-03-02:
Yes we can use the workaround.
Thank you

kapilkedawat, 2023-05-11:
@micheledarold this is fixed in https://github.com/espressif/esp-hosted/commit/c13c46a9329b539b09c22738ff758901a06923ba

renyeguo, 2023-05-12:
@kapilkedawat this modify fix it

kapilkedawat, 2023-06-12:
@mantriyogesh please close this issue.

closed 2023-06-12 by mantriyogesh