espressif/esp-hosted #205: esp_hosted_ng: esp32-c3: firmware source

ricoheller, 2023-02-13:
Hello,

I am using the NG solution.

Where can I find the source code of the ESP firmware?
What is the maximum number of concurrent connections for ESP32-C3?
I did some simple tests but could not get more than 4 concurrent connections with the available firmware. Is it possible to increase that number?

Thanks

mantriyogesh, 2023-02-13:
Hello & Welcome back @ricoheller ,

Thanks for using ESP-Hosted-FG & thinking to use NG.
The source code for NG is not yet open sourced & it will be like that at least for some time.

> I did some simple tests but could not get more than 4 concurrent BLE connections with the available firmware. Is it possible to increase that number?

Its puzzling me if you have experienced any difference in FG & NG wrt Bluetooth?
Actually the Bluetooth code remains just the same.

If you can just showcase the issue with appropriate & complete logs of scenario, I can check it with Bluetooth team.
- Host side
  - [ ] Run bluez with debug logs enabled
  - [ ] `sudo btmon`
  - [ ] `hciconfig -a`
  - [ ] Any configs you have used/altered
  - [ ] run logs (commands you had executed (in say `sudo bluetoothctl`)
  - [ ] `/var/log/kern.log` or `dmesg` or `/var/log/messages`
  
- ESP32-C3
  - [ ] firmware (minicom) logs

mantriyogesh, 2023-02-14:
hello @ricoheller 

I think your issue is similar to one for classic BT, https://github.com/espressif/esp-hosted/issues/206
And yes, I meant the configuration seems to be issue for your BLE.

If So, Give me some time to check on this & get back to you.
Meantime, sit back and relax !!

mantriyogesh, 2023-02-14:
Hello @ricoheller 

Total of max 10 connections (including connections, scan , sync and adv) are allowed as I check for **ESP32-C3** **BLE** controller.
BT_CTRL_BLE_MAX_ACT is set to 10.

Is there any limiting value set through Bluez configuration side (In Linux)?

Question still valid:
1. Do you have any difference in the FG and NG wrt max BLE concurrent connections? If so, please get us the above logs
2. Worth to check if your scan/sync/adv etc connections are closed or not while making connection

ricoheller, 2023-02-14:
Hi @mantriyogesh 

I haven't tested the FG release yet because I have only compiled the NG version for my target board. Do you think there will be any difference? If so I can try to compile the FG version.

I believe my BlueZ stack is not limiting the number of devices.
I have tested with BLE radios from other manufacturers using the same BlueZ stack and configuration and could achieve more than 30 simultaneous connections.

When you say "Total of max 10 connections (including connections, scan , sync and adv)" you mean that the Scan, Sync and Adv count as connections as well? 

I noticed that sometimes I get more than 4 connections for a very short period of time (up to 6 so far). But it is not stable. It only gets stable with 4 connections.

mantriyogesh, 2023-02-14:
Oh okay.
Understood. I have raised this concern to concerned (Bluetooth) team.

I thought you had used ESP-Hosted-FG earlier from your old issues.
This is good observation that you are saying it is not stable for 6 connections.

Can you please recreate the scenario with above logs captured?

> When you say "Total of max 10 connections (including connections, scan , sync and adv)" you mean that the Scan, Sync and Adv count as connections as well?

I think so, but I will come back on this with confirmation.

SatishSolankeEsp, 2023-02-14:
Hi @ricoheller ,
When you say "Total of max 10 connections (including connections, scan, sync, and adv)" do you mean that the Scan, Sync, and Adv count as connections as well?
>BT_CTRL_BLE_MAX_ACT needs to be increased, the default value is 10, which can support up to nine connections, and others do not need to be configured (scan, sync, and adv).
Thanks,
Satish

ricoheller, 2023-02-14:
Hi @SatishSolankeEsp

As @mantriyogesh mentioned earlier in this post, you only provide binaries for the firmware. Can you provide binaries with higher value?

Thanks

mantriyogesh, 2023-02-14:
Hello @ricoheller ,

I just checked internally with @SatishSolankeEsp.

This value, 10 for `BT_CTRL_BLE_MAX_ACT`, is already at the highest limit permitted from Bluetooth controller code as per current design. 

However, should you experience any issue with active 9 such connections (considering all other activities like scan, sync etc are not simultaneously present), we will definitely look into the problem.

mantriyogesh, 2023-03-08:
Do you still experience issue  @ricoheller ?
Or it is working for you (with limitation of 10 max active BLE activities)

ricoheller, 2023-03-08:
> Do you still experience issue @ricoheller ? Or it is working for you (with limitation of 10 max active BLE activities)

Hi @mantriyogesh 

I still have the issue. I was unable to reliably keep more than 4 simultaneous BLE connections with ESP32-C3.
In rare occasions it can reach 5 or 6 simultaneous connections, but is not stable and eventually some of those connections are dropped.

mantriyogesh, 2023-03-08:
Hello @ricoheller 

These are the logs from Bluetooth team tested 1 Master and 9 slave BLE connections.
Please see attached log :

[9th_slave.txt](https://github.com/espressif/esp-hosted/files/10920735/9th_slave.txt)
[master.txt](https://github.com/espressif/esp-hosted/files/10920736/master.txt)

Master initiating connections and the 9th slave which got connected to esp32c3.
Can you please share OTA logs to further check the issue?
Please collect `btmon` logs and ESP, host side logs as well, when you recreate the issue..

mantriyogesh, 2023-03-15:
@ricoheller 

Were you able to check this further?