espressif/esp-hosted #206: Using WROOM32-E with more than 2 BT classic SSP clients

franziskanerda, 2023-02-13:
I am using ESP hosted NG version, firmware 1.0.2.

I want to use the WROOM32-E to communicate via BT classic as master with up to 6 SPP clients. Currently communication only works with 2 clients, when I try to connect to the third station, I get the message 'Can't connect RFCOMM socket: Too many links'. I have seen that in the directory '\ESP-Hosted-NG_release_v1.0.2\esp32\sdio_only\' there is a file 'sdkconfig' with the lines 'CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN=2' and 'CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN_EFF=2'. Unfortunately I can't change these settings and recompile the firmware because there are no sources available.

Is there a possibility to operate the WROOM32-E with more than 2 clients? Can you compile a special version for me? Thank you very much for your help!

mantriyogesh, 2023-02-14:
Hello @franziskanerda 

Thank you for using ESP-Hosted. Understood your issue.

If you want to use only Bluetooth, I think you can easily use ESP-Hosted-FG, where the source code is available and you can try all expected configs. FG and NG are not different as far as Bluetooth is concerned.

FG (although named as First Generation) is in active development. 

Regarding original question about NG,
Give me some time to check this with Bluetooth team internally and get back to you with your expected binaries.

mantriyogesh, 2023-02-14:
Hello @franziskanerda 

Can you please let me know which transport combination do you use?
I will change and build the binary for you.

franziskanerda, 2023-02-14:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hello Yogesh,

I’m using SDIO only (for BT classic and WiFi). Firmware is ESP hosted-NG Release 1.0.2.

Best regards and thanks a lot for your help,
Frank

[Fluke Reliability logo]        Frank Votteler
Senior Hardware Engineer

Phone: +498999616289

Email: ***@***.***
Homepage: www.pruftechnik.com<http://www.pruftechnik.com/>

[Fluke familyl logos]

Fluke Deutschland GmbH Freisinger Str. 34, 85737 Ismaning, Germany

Sitz Ismaning / HRB 705 046, Freiburg i. Br.
Managing Directors: Ankush Malhotra, Dr. Edwin Becker, Sigrid Hanna Teich, Jeroen van de Wiel

Please be advised that this email may contain confidential information. If you are not the intended recipient, please notify us by email by replying to the sender and delete this message.

mantriyogesh, 2023-02-14:
Added both (SDIO only) & (SPI only) firmwares with the change.
PTAL at `readme.txt` from 
[2023.02.14_NG_MaxACLConn_ClassicBT.tgz](https://github.com/espressif/esp-hosted/files/10731391/2023.02.14_NG_MaxACLConn_ClassicBT.tgz)

franziskanerda, 2023-02-14:
Thanks a lot. We will test the new version this week. Best regards, Frank!

franziskanerda, 2023-02-16:
Hello Yogesh, 
we have tested the customized firmware. It works with more than two sensors (we tested it with three sensors), but it makes the Wi-Fi communication very slow. Our idea would be to test the FG version with different settings to find the best setting. Then we would come back to you to get the NG firmware with the found settings. Is this procedure OK for you? 
Best regards, Frank

franziskanerda, 2023-02-16:
Sorry for the ambiguity. By sensors I mean Bluetooth clients. We need the connection to more than two Bluetooth clients and the Wi-Fi connection. Would it help if we disconnect the WLAN connection before establishing Bluetooth connections and vice versa? Probably we will not need WLAN and Bluetooth simultaneously.

mantriyogesh, 2023-02-16:
If you plan to do A2DP or something (where BT data is involved) **along** with Wi-Fi, you would be needed special co-ex enabled firmware (I can give that if you need it, but may take 2 days).

Please note, the RF is shared on runtime in between Bluetooth and Wi-Fi.
In general, We always give higher preference to BT packets in co-ex (obviously to avoid any jittery experience say like while music streaming on bt headphones).

It might help for you to disconnect WLAN, but it is not mandated approach. When Bluetooth + Wi-Fi case is present, while picking/handling the packets Bluetooth packet is given higher preference (Reason again being, WLAN or say TCP can handle the retransmissions, but say music streaming experience cannot). What it means, while BT packets are ongoing, you would experience little bit slower Wi-Fi.

Preference of Bluetooth over Wi-Fi also can be changed also (but mostly your case might not be like that).

franziskanerda, 2023-02-16:
Our BT clients are sensors, each transmitting data via SPP (BT classic) at 115200 baud. We do not use A2DP. If it would be possible to run Bluetooth and WLAN at the same time using a co-existence mode, that would help us a lot. Normally only one or two BT clients are used, but in some applications we use more than two (up to 6) clients.

mantriyogesh, 2023-02-16:
Co-Ex is already present by default in each firmware, It just that we might need tuning of the params **only** in case of loss sensitive applications (say A2DP or music streaming).
Otherwise, In general, there is no need to such tuning and default BT+Wi-Fi should just work fine.
In this case also, Bluetooth (be it any type of BT packet)  gets priority over Wi-Fi.

franziskanerda, 2023-02-23:
My colleague got the FG version up and running. It is now running stable with 6 BT classic SPP connections and WiFi at same time. I send you the config file 'esp_hosted_fg_working_OK_sdkconfig' we have been using for this. My colleague has compiled the driver and firmware with ESP-IDF release v5.0.1. Probably the most important difference to the original firmware was the omission of the BLE function as we don't need it and it disturbs us. Please compile a NG version with these settings for us. We hope that the NG version works as well as the FG version, because it can be implemented with less effort on our system.
[esp_hosted_fg_working_OK_sdkconfig.zip](https://github.com/espressif/esp-hosted/files/10813969/esp_hosted_fg_working_OK_sdkconfig.zip)

mantriyogesh, 2023-02-23:
Please try these.
sdkconfig orig and new, merged with [esp_hosted_fg_working_OK_sdkconfig.zip](https://github.com/espressif/esp-hosted/files/10813969/esp_hosted_fg_working_OK_sdkconfig.zip) is present for your reference.

[esp_hosted_fg_working_OK_sdkconfig.zip](https://github.com/espressif/esp-hosted/files/10814743/esp_hosted_fg_working_OK_sdkconfig.zip)

franziskanerda, 2023-02-23:
Thanks a lot. We will test this version immediately.

franziskanerda, 2023-02-24:
Thanks a lot for your help. We could solve the issue!

closed 2023-02-24 by franziskanerda