espressif/esp-hosted #207: ESP32-C3 SPI Chip Enable pin selection

MooseGmaing, 2023-02-17:
Hello,
I am using ESP-Hosted NG over SPI with an ESP32-C3. It worked fine, but I needed to use CE0 for a different SPI slave and now (when the ESP is connected to CE1) it does not work. Does the code Require a specific Chip Enable pin, or is it selectable/autodetected? Looking through the code I didn't find anything specifying the CE pin, but I'm not very good with C so I might have overlooked something.

mantriyogesh, 2023-02-17:
Hello @MooseGmaing ,

Thanks for using ESP-Hosted.
I'm little confused, as CE0 or CE1.

If I understand correctly, your host had two ChipSelects. If you use CE0 with ESP32-C3, it works but if you move ESP32-C3 to other CHIPSelect CE1 **of Host**, the ESP32-C3 is not working. Is this correct understanding?

Also can you please help with:
1. Which platform (SoC) and kernel version?
2. Host /var/log/kern.log or /var/log/messages or dmesg log
3. Esp minicom log

MooseGmaing, 2023-02-17:
1. I'm using a Raspberry Pi and I'm talking about its Chip Select Pins (CE0, CE1). My kernel version is `5.10.103-v7+` (RetroPie is my operating system). 
2. This is my dmesg log:
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 5.10.103-v7+ (dom@buildbot) (arm-linux-gnueabihf-gc                                                                                                                                                                                                                                             c-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34)                                                                                                                                                                                                                                              #1529 SMP Tue Mar 8 12:21:37 GMT 2022
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instructio                                                                                                                                                                                                                                             n cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x2ac00000, size 64 M                                                                                                                                                                                                                                             iB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id share                                                                                                                                                                                                                                             d-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000] On node 0 totalpages: 196608
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_a                                                                                                                                                                                                                                             lsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=320 bcm2708_fb.fbheight=240 b                                                                                                                                                                                                                                             cm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  consol                                                                                                                                                                                                                                             e=ttyAMA0,115200 console=tty1 root=PARTUUID=dfbf6d4c-02 rootfstype=ext4 fsck.rep                                                                                                                                                                                                                                             air=yes rootwait loglevel=3 consoleblank=0 plymouth.enable=0
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes,                                                                                                                                                                                                                                              linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, li                                                                                                                                                                                                                                             near)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 696112K/786432K available (10240K kernel code, 1312K rwda                                                                                                                                                                                                                                             ta, 2952K rodata, 1024K init, 862K bss, 24784K reserved, 65536K cma-reserved)
[... cut ...]
[   26.681638] OF: overlay: WARNING: memory leak will occur if overlay removed,                                                                                                                                                                                                                                              property: /soc/spi@7e204000/spidev@0/status
3. I'm sorry, but I don't know what an ESP minicom log is.
For some additional information, I do have the SPI bus on, as well as power for the ESP32.

mantriyogesh, 2023-02-18:
Please check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#124-peripheral-configurations : 1.2.4 Peripheral configurations -> `SPI Bus instance and Chip select number`

Additional reference:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md

MooseGmaing, 2023-02-18:
Ok, so I just need to change the Chip Select number in the code from 0 to 1, correct?

mantriyogesh, 2023-02-18:
Yes

MooseGmaing, 2023-02-18:
In line 470 of esp-spi.c I changed `esp_board.chip_select = 0;` to `esp_board.chip_select = 1;`. This is what I get when running rpi_init.sh:
`Building for spi protocol
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /lib/modules/5.10.103-v7+/build M=/home/pi/esp-hosted/esp_hosted_ng/host clean
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/linux-headers-5.10.103-v7+'
  CLEAN   /home/pi/esp-hosted/esp_hosted_ng/host/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-5.10.103-v7+'
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /lib/modules/5.10.103-v7+/build M=/home/pi/esp-hosted/esp_hosted_ng/host modules
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/linux-headers-5.10.103-v7+'
  CC [M]  /home/pi/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/pi/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/pi/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/pi/esp-hosted/esp_hosted_ng/host/esp_wpa_utils.o
  CC [M]  /home/pi/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/pi/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/pi/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  LD [M]  /home/pi/esp-hosted/esp_hosted_ng/host/esp32_spi.o
  MODPOST /home/pi/esp-hosted/esp_hosted_ng/host/Module.symvers
  CC [M]  /home/pi/esp-hosted/esp_hosted_ng/host/esp32_spi.mod.o
  LD [M]  /home/pi/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.10.103-v7+'
insmod: ERROR: could not insert module esp32_spi.ko: No such device
`
I don't know why it does not recognize the ESP32. Do I need to have spi enabled? The tutorial said that esp-hosted uses its own spi driver.

mantriyogesh, 2023-02-19:
1. Some command outpu of below cmds required:
```
modinfo esp32_spi.ko
uname -a
cat /boot/config.txt
```
2. have you changed SPI GPIO (s) hardware connection as per one in you're interest?
You may refer pin configs from https://pinout.xyz/
3. Worth checking https://blog.stabel.family/raspberry-pi-4-multiple-spis-and-the-device-tree/

MooseGmaing, 2023-02-19:
This is what I get when running `modinfo esp32_spi.ko`: 

`
filename:       /home/pi/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
version:        0.1
description:    Wifi driver for ESP-Hosted solution
author:         Yogesh Mantri <yogesh.mantri@espressif.com>
author:         Mangesh Malusare <mangesh.malusare@espressif.com>
author:         Amey Inamdar <amey.inamdar@espressif.com>
license:        GPL
srcversion:     E58581E8221234024F10B3E
depends:        cfg80211,bluetooth
name:           esp32_spi
vermagic:       5.10.103-v7+ SMP mod_unload modversions ARMv7 p2v8
parm:           resetpin:Host's GPIO pin number which is connected to ESP32's EN to reset ESP32 device (int)
`

My kernel version is 5.10.103-v7+ (RetroPie is my operating system).
Here is my `/boot/config.txt`: 

#For more options and information see
#http://rpf.io/configtxt
#Some settings may impact device functionality. See link above for details

#uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

#uncomment this if your display has a black border of unused pixels visible
#and your display can output without overscan
disable_overscan=1

#uncomment the following to adjust overscan. Use positive numbers if console
#goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

#uncomment to force a console size. By default it will be display's size minus
#overscan.
framebuffer_width=320
framebuffer_height=240

#uncomment if hdmi display is not detected and composite is being output
hdmi_group=2
hdmi_mode=87
hdmi_cvt=320 240 60 1 0 0 0
hdmi_force_hotplug=1

#uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

#uncomment to force a HDMI mode rather than DVI. This can make audio work in
#DMT (computer monitor) modes
#hdmi_drive=2

#uncomment to increase signal to HDMI, if you have interference, blanking, or
#no display
#config_hdmi_boost=4

#uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800
core_freq=250
core_freq_min=250

#Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
dtoverlay=spi0-cs

#Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

#Additional overlays and parameters are documented /boot/overlays/README

#Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
#Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
#overscan_scale=1

I have every pin like in the ESP-Hosted hardware spi setup, the only difference is that instead of CE0 I have CE1

mantriyogesh, 2023-02-20:
1. Please uncomment from your device tree
```
dtparam=spi=on
```
And reboot and verify if /dev/spidev0.1 is created or not by using
```
ls /dev/spidev*
```

Once it is visible, run
```
cd esp_hosted_fg/host/linux/host_control
bash rpi_init.sh spi
```
After running this command, /dev/spidev0.1 should no more be visible and get back the dmesg log

2. You would have done hardware connection using one of esp peripheral from https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#11-hardware-setup
Obviously CE1 pin is different from CE0. You would have to change the gpio from pin24 (bcm pin 8) to pin 26(bcm pin 7) connected

3. As per I think, cs0 & cs1 should be by default enabled, considering there is no mention of `spi0-2cs`.
Remove setting 
```
dtoverlay=spi0-cs
```

4. Further, spi1-2cs is to change the default cs pins. If you wish the same, please mention the pins
See https://raspberrypi.stackexchange.com/questions/121478/re-configure-spi-cs-as-gpio-while-using-spi

Please check that pin numbers are expected to be bcm pin numbers (bcm pins not exactly match the physical numbers)

5. Reference
https://github.com/raspberrypi/linux/blob/04c8e47067d4873c584395e5cb260b4f170a99ea/arch/arm/boot/dts/overlays/README#L1154-L1158

MooseGmaing, 2023-02-22:
Hello, sorry for the wait, but I didn't find time to troubleshoot this. I did what you said and this is what happens:
`pi@retropie~/esp-hosted/esp_hosted_fg/host/linux/host_control $ bash rpi_init.s                                                                                                                                                                                                                                             h spi
Building for spi protocol
rm: cannot remove 'spidev_disabler.dtbo': No such file or directory
rm -f *.out *.o
gcc -C -Wall --sysroot= -I/home/pi/esp-hosted/esp_hosted_fg/host/linux/host_cont                                                                                                                                                                                                                                             rol/c_support/../../../../common/protobuf-c -I/home/pi/esp-hosted/esp_hosted_fg/                                                                                                                                                                                                                                             host/linux/host_control/c_support/../../../../common/include -I/home/pi/esp-host                                                                                                                                                                                                                                             ed/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/include                                                                                                                                                                                                                                              -I/home/pi/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../c                                                                                                                                                                                                                                             ontrol_lib/src/include -I/home/pi/esp-hosted/esp_hosted_fg/host/linux/host_contr                                                                                                                                                                                                                                             ol/c_support/../../../virtual_serial_if/include -I/home/pi/esp-hosted/esp_hosted                                                                                                                                                                                                                                             _fg/host/linux/host_control/c_support/../../../components/include -I/home/pi/esp                                                                                                                                                                                                                                             -hosted/esp_hosted_fg/host/linux/host_control/c_support/../../port/include -I. /                                                                                                                                                                                                                                             home/pi/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../c                                                                                                                                                                                                                                             ommon/protobuf-c/protobuf-c/protobuf-c.c /home/pi/esp-hosted/esp_hosted_fg/host/                                                                                                                                                                                                                                             linux/host_control/c_support/../../../../common/esp_hosted_config.pb-c.c /home/p                                                                                                                                                                                                                                             i/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../control_li                                                                                                                                                                                                                                             b/src/ctrl_core.c /home/pi/esp-hosted/esp_hosted_fg/host/linux/host_control/c_su                                                                                                                                                                                                                                             pport/../../../control_lib/src/ctrl_api.c /home/pi/esp-hosted/esp_hosted_fg/host                                                                                                                                                                                                                                             /linux/host_control/c_support/../../../virtual_serial_if/src/serial_if.c /home/p                                                                                                                   [... cut ...]
pi@retropie:~/esp-hosted/esp_hosted_fg/host/linux/host_control $
`

Even though it says that RPi init successfully completed I don't see WiFi or Bluetooth. I don't understand why ESP Hosted-NG doesn't recognize the ESP32. Oh, and one **very bad thing**, my other SPI slave (the one connected to CE0) is a 3,2 inch ili9341 display. It uses the GitHub driver `fbcp-ili9341` and it says to disable `dtparam=spi=on` and with `dtparam=spi=on` on the display does not work. Does this mean there is no chance for this to work?

mantriyogesh, 2023-02-22:
gcc: error: /home/pi/esp-hosted/esp_hosted_fg/host/linux/host_control/python_sup port/../../../../common/protobuf-c/protobuf-c/protobuf-c.c: No such file or dire ctory
[image]

mantriyogesh, 2023-02-23:
Question regarding `dtparam=spi=on`,
The raspberry pi documentation says to enqble it to use spi. Check: https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/

If you are getting `/dev/spidev0.1` listed, with or without ` /dev/spidev0.1` line , it should be okay.

mantriyogesh, 2023-02-23:
As NG driver works on cs0, it should just work with cs1, unless system doesn't expose/configured cs1.

As you have another device working with cs0, can just you try to use cs1 with that, if that passes?

MooseGmaing, 2023-02-23:
I updated the repo and turned SPI back on and it still doesn't work. I unfortunately can't use CS1 with my other device, as it is a TFT display using the `fbcp-ili9341` GitHub driver and it doesn't support CS1. I opened an issue there for adding CS1 support but they haven't responded yet.

MooseGmaing, 2023-02-24:
Hello, I reinstalled ESP Hosted NG and finished the SPI-only tutorial for the ESP32-C3. I ran the script and it says

`esp32 module inserted,
ESP32 host init successfully completed
` 

But when I run `ifconfig` and `hciconfig` I don't see any ESP device. This is what I get in the `dmesg` log:

`esp_reset, ESP32: Triggering ESP reset.
[  131.750606] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[  302.030957] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
`

I have no idea why it doesn't look like the example log.

mantriyogesh, 2023-02-24:
The same worked earlier.. I mean c3 with ng, with ce0.
Am I correct?
Any changes you had done from earlier working setup?
Like code changes, config changes?

MooseGmaing, 2023-02-24:
Yes, you're right this is exactly what I had earlier. The only thing I changed is I installed the driver, and that's when it started to stop working. I removed it but as you can see it doesn't work. I think I'll use a new OS image and start from zero.

mantriyogesh, 2023-02-24:
I suppose you might already have taken care,  to revert the chip select and config changes.

mantriyogesh, 2023-02-24:
If you can send me that logs,I can check if anything wrong.

Host - /var/log/kern.log, /boot/config.txt, commands (run) log
Esp - minicom

Can you try to comment two lines,
https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_ng/host/spi/esp_spi.c#L216-L217

And try again? Please get the logs, will be able to understand something.
I still think you might not need to re-flash the os.

MooseGmaing, 2023-02-24:
This is the kern.log:

`Feb 24 16:55:07 retropie kernel: [   61.435651] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
Feb 24 16:55:08 retropie kernel: [   61.689992] Bluetooth: Core ver 2.22
Feb 24 16:55:08 retropie kernel: [   61.690201] NET: Registered protocol family 31
Feb 24 16:55:08 retropie kernel: [   61.690217] Bluetooth: HCI device and connection manager initialized
Feb 24 16:55:08 retropie kernel: [   61.690262] Bluetooth: HCI socket layer initialized
Feb 24 16:55:08 retropie kernel: [   61.690283] Bluetooth: L2CAP socket layer initialized
Feb 24 16:55:08 retropie kernel: [   61.690336] Bluetooth: SCO socket layer initialized
Feb 24 16:56:18 retropie kernel: [  132.243851] esp32_spi: loading out-of-tree module taints kernel.
Feb 24 16:56:18 retropie kernel: [  132.248712] esp_reset, ESP32: Triggering ESP reset.
Feb 24 16:56:18 retropie kernel: [  132.469666] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
`

Here is my `config.txt`:

#For more options and information see
#http://rpf.io/configtxt
#Some settings may impact device functionality. See link above for details

#uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

#uncomment this if your display has a black border of unused pixels visible
#and your display can output without overscan
#disable_overscan=1

#uncomment the following to adjust overscan. Use positive numbers if console
#goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

#uncomment to force a console size. By default it will be display's size minus
#overscan.
#framebuffer_width=1280
#framebuffer_height=720

#uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

#uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

#uncomment to force a HDMI mode rather than DVI. This can make audio work in
#DMT (computer monitor) modes
#hdmi_drive=2

#uncomment to increase signal to HDMI, if you have interference, blanking, or
#no display
#config_hdmi_boost=4

#uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800
core_freq=250
core_freq_min=250

#Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

#Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

#Additional overlays and parameters are documented /boot/overlays/README

#Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
#Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1

I unfortunately can't give you the ESP minicom log as I am using a ESP32-C3-32S (basically WROOM) this means I don't have a USB port.

mantriyogesh, 2023-02-24:
Have you used resetpin?
What is the command run log?

MooseGmaing, 2023-02-24:
I have it connected to GPIO/BCM 6 on the Raspberry Pi and to EN(Enable/Reset) on the ESP. Could you specify what a command run log is and how to get it?

mantriyogesh, 2023-02-24:
It's okay, i got it.
$ git diff # in esp-hosted folder

The GPIO connection is moved back to CE0? (and all connections are firm?)
also can you please reboot and try again (if you had changed /boot/config.txt, reboot is needed to take changes in place)

MooseGmaing, 2023-02-24:
When I run `git diff #` it doesn't show me anything.
Yes, I have the GPIO on CE0 and all connections are firm and the ESP is hot so it is getting power.
After reboot still no difference.

mantriyogesh, 2023-02-24:
Can you re-flash the ESP **NG** binary ? Are you using **1.0.2** ESP firmware and host is checked out at https://github.com/espressif/esp-hosted/commit/29208b6c156be06dfd90dadc6dd43acd4d98d748 or later?

Actually ESP32 log will be much useful. You don't need USB as such. just find out which device (device for example, /dev/ttyUSB0) is ESP32 exposed to  (from the same machine, where you were flashing ESP32) & run `minicom -D /dev/ttyUSB0`

command log I was referring to all the commands you had run (like bash rpi_init.sh) and output of those commands

MooseGmaing, 2023-02-24:
[image]
This is what I was using to flash my ESP, I don't have it connected now. I have all the latest firmware.

mantriyogesh, 2023-02-24:
Okay. I was just reconfirming that it has NG firmware. FG firmware and NG host are not compatible (and vice versa).

MooseGmaing, 2023-02-24:
Yes, It has NG firmware.

mantriyogesh, 2023-02-24:
Now, I think left out with https://github.com/espressif/esp-hosted/issues/207#issuecomment-1443912393

MooseGmaing, 2023-02-24:
I found this in the dmesg log:

`OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
`

mantriyogesh, 2023-02-24:
Not a prob.

MooseGmaing, 2023-02-24:
Ok, so what is the problem here?

mantriyogesh, 2023-02-24:
Oh, I am not sure.

Can you please send setup image?

There are two ways, I will go in debug mode and give you binaries to flash (long cut), but even in this case I would need esp logs (otherwise there is no way to cross check message transfer)
Other way, As you had it working first time, still reflashing would be helpful. Including flashing esp, disconnect hw connection and reconnect.

MooseGmaing, 2023-02-24:
My setup image is the latest RetroPie image for  the Raspberry Pi 2B

mantriyogesh, 2023-02-24:
`OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status ` will come in positive case also.

mantriyogesh, 2023-02-24:
Well, we had tested on RPi 3 & 4. But that should not be problem.

mantriyogesh, 2023-02-24:
By Image I meant the picture of setup

MooseGmaing, 2023-02-25:
I have a new OS, so I don't have that driver and everything is just like it was when it worked.

mantriyogesh, 2023-02-26:
So now you have CE0 working?

MooseGmaing, 2023-02-26:
No, maybe the ESP32 is broken somehow. Perhaps something is shorted. I'll try re-flashing the binaries.

mantriyogesh, 2023-03-02:
Hello @MooseGmaing 

I got it worked myself on Raspberry Pi 4 with spi bus 0 chip select 1.
Changes:
1. Hardware connection: Change the connections as:
[image]

2. Code changes:
```diff
diff --git a/esp_hosted_fg/host/linux/host_control/spidev_disabler.dts b/esp_hosted_fg/host/linux/host_control/spidev_disabler.dts
index 664ecb9..b6de380 100644
--- a/esp_hosted_fg/host/linux/host_control/spidev_disabler.dts
+++ b/esp_hosted_fg/host/linux/host_control/spidev_disabler.dts
@@ -5,7 +5,7 @@
     compatible = "brcm,bcm2708";

     fragment@0 {
-        target = <&spidev0>;
+        target = <&spidev1>;

         __overlay__ {
             status = "disabled";
diff --git a/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c b/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c
index a320a09..9293a7e 100644
--- a/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c
+++ b/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c
@@ -366,7 +366,7 @@ static int spi_dev_init(int spi_clk_mhz)
        esp_board.mode = SPI_MODE_2;
        esp_board.max_speed_hz = spi_clk_mhz * NUMBER_1M;
        esp_board.bus_num = 0;
-       esp_board.chip_select = 0;
+       esp_board.chip_select = 1;

        master = spi_busnum_to_master(esp_board.bus_num);
        if (!master) {
```

3. /boot/config used:
<details>
  <summary>$ cat /boot/config.txt</summary>

```sh
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=2
[... 991 lines cut ...]
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

##dtoverlay=sdio,poll_once=off,sdio_overclock=5,max-frequency=5,overclock_50=5
##dtoverlay=sdio,poll_once=off,max-frequency=10000000,sdio_overclock=10
#force_turbo=1

#dtoverlay=sdio,poll_once=off
#dtoverlay=disable-bt

dtparam=spi=on
dtoverlay=disable-bt
core_freq=250
core_freq_min=250

enable_uart=1
dtoverlay=disable-bt

pi@raspberrypi:~/h3/esp_hosted_fg/host/linux/host_control/c_support $ cat /boot/cmdline.txt
console=tty1 root=PARTUUID=c9650e41-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait net.ifnames=0 biosdevname=0 fbcon=scrollback:1024k
pi@raspberrypi:~/h3/esp_hosted_fg/host/linux/host_control/c_support $
```  
<details>

mantriyogesh, 2023-03-07:
Hello @MooseGmaing 

Did you get chance to test this?

mantriyogesh, 2023-03-09:
@MooseGmaing any updates please?

MooseGmaing, 2023-03-09:
Sorry, but I won't be able to test this until next weekend. I'll let you know if it works.

mantriyogesh, 2023-03-09:
Okay.

mantriyogesh, 2023-03-21:
Any updates @MooseGmaing ??

mantriyogesh, 2024-07-31:
Closing due to inactivity, please re-open this issue any further updates or issues you observe @MooseGmaing .

closed 2024-07-31 by mantriyogesh