espressif/esp-hosted #208: Phone connection AP disconnected

hulinfeng-github, 2023-02-23:
I use the chip is s3, esp-idf version is 4.4.2, the program is using fg. currently encountered two problems, a problem is in the case of no access to the network cable, enable AP mode, the phone connected to the AP after a period of time, the phone and the AP connection will automatically disconnect, the time is about a few tens of seconds different phone models may vary time. But I run the softap routine in idf and the phone will not be disconnected. Another question is to ask the current demo DHCP service is placed on the host to run, DHCP service can be run on S3?

mantriyogesh, 2023-02-23:
Hello @hulinfeng-github,

Thank you for using ESP-Hosted.
Two issues, as I summarise,
1. You are using softAP on ESP-Hosted-FG solution and you face the disconnect from softAP when you connect your phone to ESP SoftAP.
- Can you please give logs when you face this issue:
  - ESP : idf.py monitor or minicom log
  - Host : 
    - dmesg or /var/log/kern.log or /var/log/messages
    - run log (basically commands you had run and their output)
2. If you want just softAP, you can actually use the ESP-IDF softAP example directly. ESP-Hosted is way to get the Wi-Fi and Bluetooth capability **on host**. So running DHCP server on host makes sense when using ESP-Hosted.
In case of ESP-Hosted, The packets are destined to & expected to end in host through ESP chipset. So DHCP server running on ESP32-S3 doesn't make much sense in ESP-Hosted context.

hulinfeng-github, 2023-02-24:
Thank you very much for your reply. From the logs, we can only see that the station has left, but we can’t tell why it should have left. The mobile phone model is Redmi K50, and it will be disconnected within a short time after being connected. It will take a long time for the iphone to connect to this AP.The following is the serial port log of ESP32S3

[15:33:48.368][20230224_15:33:48]
[15:33:58.441]ESP-ROM:esp32s3-20210327
[15:33:58.441]Build:Mar 27 2021
[15:33:58.442]rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
[15:33:58.442]SPIWP:0xee
[15:33:58.442]mode:DIO, clock div:2
[15:33:58.442]load:0x3fce3808,len:0x1650
[15:33:58.442]load:0x403c9700,len:0xb54
[15:33:58.442]load:0x403cc700,len:0x2f78
[15:33:58.442]SHA-256 comparison failed:
[15:33:58.442]Calculated: 5c7404766df1ef8c2e78fd74b493d4e76fe1c5c585e1a0396bc878dcd10886a1
[15:33:58.442]Expected: a4d2bd60a5a33ed1685da2ebdad7100b5489180ce86550c995fc533a7bbbd5c5
[15:33:58.442]Attempting to boot anyway...
[15:33:58.442]entry 0x403c9938
[15:33:58.442]I (48) boot: ESP-IDF v4.4.2 2nd stage bootloader
[15:33:58.442]I (48) boot: compile time 18:48:57
[15:33:58.442]I (48) boot: chip revision: 0
[15:33:58.442]I (49) boot.esp32s3: Boot SPI Speed : 40MHz
[15:33:58.442]I (54) boot.esp32s3: SPI Mode       : DIO
[15:33:58.449]I (59) boot.esp32s3: SPI Flash Size : 16MB
[15:33:58.449]I (64) boot: Enabling RNG early entropy source...
[15:33:58.449]I (69) boot: Partition Table:
[15:33:58.449]I (73) boot: ## Label            Usage          Type ST Offset   Length
[15:33:58.449]I (80) boot:  0 nvs              WiFi data        01 02 00009000 00004000
[15:33:58.449]I (87) boot:  1 otadata          OTA data         01 00 0000d000 00002000
[15:33:58.449]I (95) boot:  2 phy_init         RF data          01 01 0000f000 00001000
[15:33:58.450]I (102) boot:  3 factory          factory app      00 00 00010000 00100000
[15:33:58.450]I (110) boot:  4 ota_0            OTA app          00 10 00110000 00100000
[15:33:58.450]I (117) boot:  5 ota_1            OTA app          00 11 00210000 00100000
[15:33:58.450]I (125) boot: End of partition table
[15:33:58.451]I (129) boot: Defaulting to factory image
[15:33:58.451]I (134) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=1c008h (114696) map
[15:33:58.475]I (171) esp_image: segment 1: paddr=0002c030 vaddr=3fc95330 size=039b8h ( 14776) load
[15:33:58.475]I (175) esp_image: segment 2: paddr=0002f9f0 vaddr=40374000 size=00628h (  1576) load
[15:33:58.487]I (177) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=7c608h (509448) map
[15:33:58.625]I (312) esp_image: segment 4: paddr=000ac630 vaddr=40374628 size=10d08h ( 68872) load
[15:33:58.756]I (332) esp_image: segment 5: paddr=000bd340 vaddr=50000000 size=00010h (    16) load
[15:33:58.757]I (339) boot: Loaded app from partition at offset 0x10000
[15:33:58.758]I (339) boot: Disabling RNG early entropy source...
[15:33:58.758]I (352) cpu_start: Pro cpu up.
[15:33:58.758]I (352) cpu_start: Single core mode
[15:33:58.758]I (362) cpu_start: Pro cpu start user code
[15:33:58.758]I (362) cpu_start: cpu freq: 160000000
[15:33:58.758]I (362) cpu_start: Application information:
[15:33:58.758]I (364) cpu_start: Project name:     network_adapter
[15:33:58.758]I (370) cpu_start: App version:      release/ng-v1.0.2-10-gfe0b105-d
[15:33:58.758]I (377) cpu_start: Compile time:     Feb 20 2023 09:45:50
[15:33:58.758]I (383) cpu_start: ELF file SHA256:  e8b88b73aeff5c25...
[15:33:58.758]I (389) cpu_start: ESP-IDF:          v4.4.2
[15:33:58.758]I (394) heap_init: Initializing. RAM available for dynamic allocation:
[15:33:58.758]I (401) heap_init: At 3FC9D880 len 00042780 (265 KiB): D/IRAM
[15:33:58.758]I (408) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
[15:33:58.758]I (414) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
[15:33:58.758]I (420) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
[15:33:58.759]I (428) spi_flash: detected chip: generic
[15:33:58.759]I (432) spi_flash: flash io: dio
[15:33:58.759]I (436) sleep: Configure to isolate all GPIO pins in sleep state
[15:33:58.759]I (442) sleep: Enable automatic switching of GPIO sleep configuration
[15:33:58.759]I (450) cpu_start: Starting scheduler on PRO CPU.
[15:33:58.771]I (455) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
[15:34:08.960]I (10455) NETWORK_ADAPTER: *********************************************************************
[15:34:08.961]I (10455) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5                        
[15:34:08.961]I (10455) NETWORK_ADAPTER:                 Transport used :: SPI only                      
[15:34:08.961]I (10465) NETWORK_ADAPTER: *********************************************************************
[15:34:08.962]I (10475) NETWORK_ADAPTER: Supported features are:
[15:34:08.962]I (10485) NETWORK_ADAPTER: - WLAN over SPI
[15:34:08.962]I (10485) NETWORK_ADAPTER: capabilities: 0xa0
[15:34:08.962]I (10495) SPI_DRIVER: Using SPI interface
[... 46 lines cut ...]
[15:34:38.771]I (40435) slave_ctrl: ssid nova7sh1 pwd Hai123456 authmode 3 ssid_hidden 0 max_conn 4 channel 1
[15:34:38.771]I (40445) slave_ctrl: ESP32 SoftAP is avaliable 
[15:34:38.772]I (40445) NETWORK_ADAPTER: Get softap mac address
[15:34:38.772]I (40455) NETWORK_ADAPTER: AP mac [f4:12:fa:52:ad:ed] 
[15:34:38.772]I (40455) slave_ctrl: softap stop handler stop
[15:34:38.793]I (40465) NETWORK_ADAPTER: Get softap mac address
[... cut ...]
[15:36:04.820]I (126515) wifi:station: 82:98:90:b7:87:7e leave, AID = 1, bss_flags is 658546, bss:0x3fcf62d0
[15:36:04.834]I (126515) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
[15:36:06.563]I (128265) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
[15:36:06.584]I (128265) wifi:station: 82:98:90:b7:87:7e join, AID=1, bgn, 20
[15:36:06.645]I (128325) slave_ctrl: station 82:98:90:b7:87:7e join, AID=1
[15:36:08.284]W (129965) wifi:<ba-add>idx:2 (ifx:1, 82:98:90:b7:87:7e), tid:0, ssn:19, winSize:64

mantriyogesh, 2023-02-24:
Please see summary of problems and replies:

> Reason of disconnection of station from ESP softAP is not present

Yes, there is no reason currently populated at Wi-Fi driver (ESP-IDF WiFi component), which is why ESP-Hosted doesn't know what is the reason & doesn't populate the reason. Anyway, we will look into it.

> iPhone is connecting slower than Redmi

\- There is no preference as such. How are you connecting? Are you using static IP or DHCP server?
\- Can you stop DHCP server and manually configure static IPs to these phones and try again?
\- Get the logs (ESP, Host run log & Wi-Fi sniffer log), which will be easier for us to get to the issue.
\- May be iPhone is taking some additional time time for something before actual request triggers out?

In general, these things depends upon your radio conditions a lot. Try to use radio silent environment to rule out possibilities.
- Prefer to run in less wifi crowded environment
- Try not to be very close to bluetooth / microwave
- Optimal distance in between these stations and SoftAP
- Please try different channel and possibly 40 MHz bandwidth. Check the softAP API [here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#115-ctrl_cmd_t--wifi_start_softapctrl_cmd_t-req)

https://github.com/espressif/esp-hosted/blob/fe0b10584417629908cee8141699c2f35ea05a25/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L757-L776

hulinfeng-github, 2023-02-24:
In the way of dhcp, the iphone is slow to connect to the AP, and the speed will be much faster in static mode. There is no special log on the host side after sending softap_start. The following is the network packet of redmi k50 connecting to AP until disconnection.
[redmi_disconnect.zip](https://github.com/espressif/esp-hosted/files/10822309/redmi_disconnect.zip)

closed 2023-02-27 by hulinfeng-github

hulinfeng-github, 2023-02-27:
I found that this phenomenon is caused by the different configuration of udhcpd.conf, if the IP of the dns configuration in udhcpd.conf and the assigned IP are not in the same network segment it will cause this phenomenon.Here is what has been modified.
```shell
start 		192.168.0.20	#default: 192.168.0.20
end		192.168.0.254	#default: 192.168.0.254
```
``` shell
#Examles
opt	dns	192.168.0.1
option	subnet	255.255.255.0
opt	router	192.168.0.1
opt	wins	192.168.0.1
```

mantriyogesh, 2023-02-27:
Thank you @hulinfeng-github for letting us know. This will help for users facing similar problems.