espressif/esp-hosted #210: ESP32 not responding

take-42, 2023-03-06:
I am currently trying to use the TI SK-AM62 board with the ESP32-WROOM-32E connected via SPI, but it is not responding and I need help.

I have made the following changes to release/fg-v0.0.5.

I have edited the board's dts to enable SPI0 and confirmed that /dev/spidev3.0 appears and I can echo strings from there.
At this time, I also enabled GPIO for HandShake, DataReady, and ESP32Reset.
Then I changed SPI from dts to disabled as an alternative to spidev_disabler.

Changed HANDSHAKE_PIN and SPI_DATA_READY_PIN in esp_spi.h to the corresponding numbers for this board.

Tried all combinations of SPI_MODE_X set in esp_spi.c and spi_slave_api.c.

Made the following changes to rpi_init.sh
Commented out spidev_disabler.
Set ARCH and CROSS_COMPILE for make command.

In this state . /rpi_init.sh and use the demo app, you will get the following state.

```
root@am62xx-evm:~/esp-hosted/esp_hosted_fg/host/linux/host_control# ./rpi_init.sh spi resetppin=432
esp32 module inserted
/dev/esps0 device created
RPi init successfully completed
root@am62xx-evm:~/esp-hosted/esp_hosted_fg/host/linux/host_control# cd c_support/
root@am62xx-evm:~/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support# ./test.out get_softap_mac_addr
Enable heatbeat with duration 20
Control response timed out after 30sec
Response not received
Control response timed out after 30sec
Response not received

Requested operation complete
Sleeping for some time just to showcase heatbeat
```

/dev/esps0, ethap0, and ethsta0 are present.

mantriyogesh, 2023-03-06:
Hello @take-42 ,

Thank you for using the ESP-Hosted & following [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md) & checking other issues for possible repetition.

Can you please get the 
- [ ] ESP side logs 
  - [ ] minicom (uart log with 115200 baud rate)
- [ ] Host side log
  - [ ] /var/log/kern.log OR /var/log/messages OR dmesg

take-42, 2023-03-07:
Hi, @mantriyogesh. Thanks for the reply.

The logs are as follows

---
### ESP Side
<details>
<summary>minicom</summary>

```
I (0) cpu_start: App cpu up.
I (494) cpu_start: Pro cpu start user code
I (494) cpu_start: cpu freq: 240000000
I (494) cpu_start: Application information:
I (499) cpu_start: Project name:     network_adapter
I (504) cpu_start: App version:      release/ng-v1.0.2-10-gfe0b105-d
I (511) cpu_start: Compile time:     Mar  6 2023 13:34:51
I (518) cpu_start: ELF file SHA256:  08fab8356147d960...
I (523) cpu_start: ESP-IDF:          v4.4.1-dirty
I (529) heap_init: Initializing. RAM available for dynamic allocation:
I (536) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (542) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (548) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (554) heap_init: At 3FFCC5C0 len 00013A40 (78 KiB): DRAM
I (561) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (567) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (573) heap_init: At 4009D4EC len 00002B14 (10 KiB): IRAM
I (581) spi_flash: detected chip: generic
I (584) spi_flash: flash io: dio
I (589) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (603) NETWORK_ADAPTER: *********************************************************************
I (609) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5
I (617) NETWORK_ADAPTER:                 Transport used :: SDIO only
I (625) NETWORK_ADAPTER: *********************************************************************
I (633) NETWORK_ADAPTER: Supported features are:
I (637) NETWORK_ADAPTER: - WLAN over SDIO
I (641) ESP_BT: - BT/BLE
I (643) ESP_BT:    - HCI Over SDIO
I (645) ESP_BT:    - BT/BLE dual mode
I (649) NETWORK_ADAPTER: capabilities: 0x1d
I (661) BTDM_INIT: BT controller compile version [5688ed5]
I (663) system_api: Base MAC address is not set
I (663) system_api: read default base MAC address from EFUSE
I (669) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (993) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 8c:4b:14:15:a4:22
I (993) SDIO_SLAVE: Using SDIO interface
I (997) wifi:wifi driver task: 3ffdae28, prio:23, stack:6656, core=1
I (999) wifi:wifi firmware version: 63017e0
I (1001) wifi:wifi certification version: v7.0
I (1005) wifi:config NVS flash: disabled
I (1007) wifi:config nano formating: disabled
I (1011) wifi:Init data frame dynamic rx buffer num: 32
I (1015) wifi:Init management frame dynamic rx buffer num: 32
I (1019) wifi:Init management short buffer num: 32
I (1023) wifi:Init dynamic tx buffer num: 64
I (1027) wifi:Init static rx buffer size: 1600
I (1029) wifi:Init static rx buffer num: 16
I (1033) wifi:Init dynamic rx buffer num: 32
I (1037) wifi_init: rx ba win: 6
I (1039) wifi_init: tcpip mbox: 32
I (1043) wifi_init: udp mbox: 6
I (1045) wifi_init: tcp mbox: 6
I (1049) wifi_init: tcp tx win: 5744
I (1051) wifi_init: tcp rx win: 5744
I (1055) wifi_init: tcp mss: 1440
I (1059) wifi_init: WiFi IRAM OP enabled
I (1063) wifi_init: WiFi RX IRAM OP enabled
I (1069) wifi:mode : null
I (1069) NETWORK_ADAPTER: Initial set up done
[... 871 lines cut ...]
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 496  bytes 43480 (42.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500  metric 1
        ether e0:ff:f1:38:4b:01  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500  metric 1
        inet 192.168.43.1  netmask 255.255.255.0  broadcast 192.168.43.255
        inet6 fe80::e2ff:f1ff:fe38:4b02  prefixlen 64  scopeid 0x20<link>
        ether e0:ff:f1:38:4b:02  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 43  bytes 5946 (5.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
</details>

---
When observed with an oscilloscope, the reset pin goes low for a moment when loaded, and it seems that the reset is actually done, but there is no response after that.

mantriyogesh, 2023-03-07:
Hello @take-42 

Thank you for using ESP-Hosted.

Getting bit confused, if you intend to use SDIO or SPI?
You are using 0.0.5 with FG solution with 'SDIO only' transport combination as per your firmware log.
SDIO firmware cannot work with SPI host driver.

- SDIO
  - I hope you had used the necessary pull-ups applied? Please check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md (it applies for FG also, we are already going to unify this doc with FG porting guide).
  - 
  
- SPI
  - If you want SPI, I would suggest to use use SPI firmware binaries or build and flash SPI firmware (check documentation)
  - Please keep same SPI mode at both the sides (possibly revert all other changes than those needed for the porting)

take-42, 2023-03-07:
Hello, @mantriyogesh.  Thank you for pointing this out.
It seems I made an elementary mistake.
I changed the following to assume SPI and flashed again to run the test, but the "Response not received" remained.
  

1.   build and flash SPI firmware
$ idf.py menuconfig
Example Configuration -> Transport layer -> [X] SPI interface

1. keep same SPI mode at both the sides
SPI_MODE is changed to the same value as the initial (both SPI_MODE_2)

mantriyogesh, 2023-03-07:
No issues @take-42 

Can you please provide the logs at both ends?

mantriyogesh, 2023-03-07:
esp `$ idf.py flash monitor` log and 
dmesg or /var/log/messages or /var/log/kern/log at host

mantriyogesh, 2023-03-07:
By the time, I checked your other logs.
Please disable config for `wlan0` and `wlan1`, once your get the `ethsta0` working from host driver. We recommend to do it such that on boot always it is disabled.

I am not sure if `spidev_disabler.dts` changes will be usable by your Linux (it is Raspberry Pi specific). You might have to see the device tree changes. Where SPI bus is enabled and spidev for that bus and chip select is disabled.

Ideally it could be done in following steps:
1. Make sure the length of jumper cables is small, typically < 6cm, all same length. Smaller the better. PCB the best.
Connect the correct connections and make sure they are firm.
2. Enable Device tree, do not disable spidev
3. `$ ls /dev/spidev*` should list available SPI devices. Test /dev/spidevX.Y where X is 3 and Y is 0 in your case with Rx & Tx. This will confirm the PSI pins loopback are working fine or not.
4. Disable the spidev from device tree. you should now will **not** see output for your disabled device(s). Basically in your case once spidev disabled, /dev/spidev3.0 should be gone.
5. Now Verify your Data read, handshake and reset pin are working or now. I think you have already made the changes in the esp_spi.h for this. But make sure if you get the prints in interrupt handlers for data ready and handshake (I noticed you have added logs in interrupt handlers already) You can change these logs like below, which will make sure that log will be printed always

> printk(**KERN_ERR** "%s:%u\n", __func__,__LINE__);
6. Verify resetpin (ESP should reset on loading of kernel module) (I think you have done this already?)

Lets get synced till this point, then will move ahead.

mantriyogesh, 2023-03-08:
any updates please?

take-42, 2023-03-09:
Hello, @mantriyogesh 
Sorry for the delay in replying. I will not be able to work for a few days due to other work, but here is what I can confirm at this time.

All jump wires are 10 cm.

Device tree SPI so if status is okay, /dev/spidev3.0 appears and can be sent/received manually. if status is disabled, it will not appear.

GPIO pins used for DataReady, HandShake, and Reset are enabled, and reset works when the module is loaded (observe low for a moment on the oscilloscope, and the startup log is played)

mantriyogesh, 2023-03-09:
Hello @take-42 ,

The current logs (Esp and host) will be useful to find out if any obvious misbehavior.

If you are going to take some time, we can close this issue and re-open once you are back.

take-42, 2023-03-10:
ESP and dmsesg logs. The following is a list of the logs.

Since the printk statement you gave me would result in an undeclared error, I typed out the function name in solid type as follows
`printk( KERN_ERR "spi_data_ready_interupt_handler\n" );`

<details><summary>ESP</summary>

```
I (0) cpu_start: App cpu up.
I (498) cpu_start: Pro cpu start user code
I (498) cpu_start: cpu freq: 240000000
I (498) cpu_start: Application information:
I (503) cpu_start: Project name:     network_adapter
I (508) cpu_start: App version:      release/ng-v1.0.2-10-gfe0b105-d
I (515) cpu_start: Compile time:     Mar  7 2023 12:53:05
I (522) cpu_start: ELF file SHA256:  f4469ae4d8f6acb2...
I (527) cpu_start: ESP-IDF:          v4.4.1-dirty
I (533) heap_init: Initializing. RAM available for dynamic allocation:
I (540) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (546) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (552) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (558) heap_init: At 3FFC89F0 len 00017610 (93 KiB): DRAM
I (565) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (571) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (577) heap_init: At 4009DBB0 len 00002450 (9 KiB): IRAM
I (585) spi_flash: detected chip: generic
I (588) spi_flash: flash io: dio
I (593) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (607) NETWORK_ADAPTER: *********************************************************************
I (613) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5
I (621) NETWORK_ADAPTER:                 Transport used :: SPI only
I (629) NETWORK_ADAPTER: *********************************************************************
I (637) NETWORK_ADAPTER: Supported features are:
I (641) NETWORK_ADAPTER: - WLAN over SPI
I (645) ESP_BT: - BT/BLE
I (647) ESP_BT:    - HCI Over SPI
I (649) ESP_BT:    - BT/BLE dual mode
I (653) NETWORK_ADAPTER: capabilities: 0xf8
I (665) BTDM_INIT: BT controller compile version [5688ed5]
I (667) system_api: Base MAC address is not set
I (667) system_api: read default base MAC address from EFUSE
I (673) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (999) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 8c:4b:14:15:a4:22
I (1001) SPI_DRIVER: Using SPI interface
I (1001) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1009) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1015) SPI_DRIVER: SPI Ctrl:1 mode: 2, InitFreq: 10MHz, ReqFreq: 10MHz
GPIOs: MOSI: 13, MISO: 12, CS: 15, CLK: 14 HS: 2 DR: 4

I (1025) SPI_DRIVER: Hosted SPI queue size: Tx:10 Rx:10
I (1035) wifi:wifi driver task: 3ffd7ba4, prio:23, stack:6656, core=1
I (1037) wifi:wifi firmware version: 63017e0
I (1039) wifi:wifi certification version: v7.0
I (1041) wifi:config NVS flash: disabled
I (1045) wifi:config nano formating: disabled
I (1047) wifi:Init data frame dynamic rx buffer num: 32
I (1051) wifi:Init management frame dynamic rx buffer num: 32
I (1057) wifi:Init management short buffer num: 32
I (1059) wifi:Init dynamic tx buffer num: 64
I (1063) wifi:Init static rx buffer size: 1600
I (1067) wifi:Init static rx buffer num: 16
I (1069) wifi:Init dynamic rx buffer num: 32
I (1073) wifi_init: rx ba win: 6
I (1075) wifi_init: tcpip mbox: 32
I (1079) wifi_init: udp mbox: 6
I (1081) wifi_init: tcp mbox: 6
I (1085) wifi_init: tcp tx win: 5744
I (1089) wifi_init: tcp rx win: 5744
I (1091) wifi_init: tcp mss: 1440
[... 428 lines cut ...]
[  422.700491] am65-cpsw-nuss 8000000.ethernet eth0: Link is Down
[  422.764004] am65-cpsw-nuss 8000000.ethernet: down msc_sl f0000000 tmo 0
[  422.776469] am65-cpsw-nuss 8000000.ethernet eth0: PHY [8000f00.mdio:00] driver [TI DP83867] (irq=POLL)
[  422.785805] am65-cpsw-nuss 8000000.ethernet eth0: configuring for phy/rgmii-rxid link mode
[  425.853882] am65-cpsw-nuss 8000000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[  425.862746] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  475.011398] am65-cpsw-nuss 8000000.ethernet eth0: Link is Down
[  475.075425] am65-cpsw-nuss 8000000.ethernet: down msc_sl f0000000 tmo 0
[  475.087846] am65-cpsw-nuss 8000000.ethernet eth0: PHY [8000f00.mdio:00] driver [TI DP83867] (irq=POLL)
[  475.097190] am65-cpsw-nuss 8000000.ethernet eth0: configuring for phy/rgmii-rxid link mode
[  478.173878] am65-cpsw-nuss 8000000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[  478.182736] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  641.625516] Bluetooth: Core ver 2.22
[  641.629328] NET: Registered protocol family 31
[  641.633821] Bluetooth: HCI device and connection manager initialized
[  641.640213] Bluetooth: HCI socket layer initialized
[  641.645104] Bluetooth: L2CAP socket layer initialized
[  641.650193] Bluetooth: SCO socket layer initialized
[  908.163956] esp_reset, ESP32: Resetpin of Host is 432
[  908.164188] esp_reset, ESP32: Triggering ESP reset.
[  908.164964] ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [10]
[  909.299920] spi_data_ready_interrupt_handler

```
</details>

take-42, 2023-03-10:
I tried switching SPI_MODE in esp_spi.c and spi_slave_api.c from 0 to 3 (16 patterns in total), but no change was observed.

mantriyogesh, 2023-03-10:
This is surprising, that none worked, considering that spidev test was success.

Can you please test simple spi slave program from esp-idf first?
Flash https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_slave  receiver to esp chipset and mimic sender in host using some user space program (you might have to reenable spidev for this)

I think once spi (all pins, including miso & mosi) is working fine, esp-hosted would smoothly work.

closed 2023-03-13 by take-42

mantriyogesh, 2023-03-13:
Did the issue resolve for you?

take-42, 2023-03-13:
No, I have not even been able to conduct a test of the spi slave program yet.
When the time is right to work on it, I would like to re-open it.