espressif/esp-hosted #212: Kernel5.4.3, ESP32 WRoom 32UE ,use esp-hosted-release-v0.3  wifi

shihui2014, 2023-03-11:
insmod esp32_sdio.ko,can not run esp_probe function
i have update dtsi
 &usdhc2 {
    #address-cells = <1>;
    #size-cells = <0>;
    pinctrl-names = "default";
    max-frequency = <50000000>;
    pinctrl-0 = <&pinctrl_usdhc2_alt &pinctrl_wifi>;// &pinctrl_wifi
    bus-width = <4>;
   // no-1-8-v;   /* force 3.3V VIO */
    non-removable;
    pm-ignore-notify;
    cap-power-off-card;
//    /delete-property/ wakeup-source;
    status = "okay";	
    esp32: esp32@1 {
        reg = <1>;
        compatible = "esp32_sdio";
		gpios = <&gpio3 4 GPIO_ACTIVE_LOW>;
    };
};
 pinctrl_wifi: wifigrp {
        fsl,pins = <
            MX6UL_PAD_LCD_RESET__GPIO3_IO04    		0x1b0b0

        >;
    };
pinctrl_usdhc2_alt: usdhc2grp_alt {
        fsl,pins = <
            MX6UL_PAD_LCD_DATA18__USDHC2_CMD  0x17051
            MX6UL_PAD_LCD_DATA19__USDHC2_CLK  0x17051
            MX6UL_PAD_LCD_DATA20__USDHC2_DATA0  0x17051
            MX6UL_PAD_LCD_DATA21__USDHC2_DATA1  0x17051
            MX6UL_PAD_LCD_DATA22__USDHC2_DATA2  0x17051
            MX6UL_PAD_LCD_DATA23__USDHC2_DATA3  0x17051
        >;
    };
and menuconfig MMC/SD/SDIO card support (MMC [=y]) 

insmod esp32_sio.ko can restart esp32 ,because i have config static int resetpin = 68;
the esp32 log is 
I (647) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (757) phy: phy_version: 4390, 6b3c1f2, Sep 10 2020, 15:09:07, 0, 0
I (1227) SDIO_SLAVE: Using SDIO interface
I (1247) wifi:wifi driver task: 3ffde798, prio:23, stack:6656, core=0
I (1247) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1247) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1257) wifi:wifi firmware version: 6e1f9e4
I (1257) wifi:config NVS flash: disabled
I (1267) wifi:config nano formating: disabled
I (1267) wifi:Init data frame dynamic rx buffer num: 32
I (1277) wifi:Init management frame dynamic rx buffer num: 32
I (1277) wifi:Init management short buffer num: 32
I (1287) wifi:Init dynamic tx buffer num: 32
I (1287) wifi:Init static rx buffer size: 1600
I (1287) wifi:Init static rx buffer num: 10
I (1297) wifi:Init dynamic rx buffer num: 32
I (1307) wifi:mode : null
I (1307) NETWORK_ADAPTER: Initial set up done
kernel log
dmesg |grep usdh
[2.313463] mmcl: SDHCI ontroller on 2194000.usdhc [2194000.usdhc] using ADMA
and /sys/bus/sdio/devices nothing

mantriyogesh, 2023-03-11:
Hello  @shihui2014 ,

Thank you for using ESP-HOSTED.

As I see you are using 0.3, I recommend using latest release, 0.0.5.

Can you please confirm the source code git commit you are using for 0.3, at ESP side and Host side?
Host and ESP side code is expected to be on same commit.
If in case you are using released ESP binaries, 'source code.zip' from assets on same page is expected to be used for host side kernel module building.

shihui2014, 2023-03-11:
i have use same verion, esp_hosted_firmware_esp32_sdio_uart_v0.3 and esp-hosted-release-v0.3,
 al the pins mast  10K pullups?

shihui2014, 2023-03-11:
Can you tell me kernel ,esp32 sdio work time sequence?

mantriyogesh, 2023-03-12:
Okay.
In your case, pull ups are required (mandatory) for sdio. See the [details)[https://docs.espressif.com/projects/esp-idf/en/v4.1/api-reference/peripherals/sd_pullup_requirements.html).

Can you help to provide logs at host (**/var/log/kern.log** or dmesg or /var/log/messages) and commands run at host and their output.

Kernel 5.4.3 should be okay.
Can you please elaborate 'sdio work time sequence'?

Some milestones:
1. Most imp: porting guide. please check both, https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md
And 
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md
1. Getting sdio configured on your platform using say device tree
2. Add external pull ups
3. Connect esp to host using sdio lines
4. Burn esp with sdio binaries
6. While bootup of host, you should get CIS info printed (esp identified as sdio) with matching sdio instance as configured (in device tree)
7. Resetpin should be verified working
8. Load the kernel driver
9. Get the bootup / init event from ESP

CIS and bootup logs can be referred from https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SDIO_setup.md#3-checking-the-setup-for-sdio

Getting platform configured is prerequisite for esp hosted kernel driver to work.
we can help you in the process, but it largely remains outsides the scope of esp_hosted.

If you are no position to add pull ups, you can use alternative transport, spi, where pull ups are not required.

shihui2014, 2023-03-13:
thanks for your reply,Below is the kernel5.4.3 log
and we  have add external pull ups
Loading Environment from NAND... *** Warning - bad CRC, using default environment

Display: MYIR-LCD-7-800x480 (800x480)
Video: 800x480x16
In:    serial
Out:   serial
Err:   serial
Net:
Warning: ethernet@02188000 (eth0) using random MAC address - 66:63:3a:14:e0:9f
eth0: ethernet@02188000 [PRIME]
Fastboot: Normal
Normal Boot
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0x600000, size 0xa00000
 10485760 bytes read: OK

NAND read: device 0 offset 0x1000000, size 0x100000
 1048576 bytes read: OK
Kernel image @ 0x80800000 [ 0x000000 - 0x83ee60 ]
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
   Using Device Tree in place at 83000000, end 8300bca9
Modify /soc/aips-bus@2200000/epdc@228c000:status disabled
ft_system_setup for mx6

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.3-g9c2490a-dirty (wt@ubuntu) (gcc version 9.2.0 (GCC)) #1 SMP PREEMPT Thu Jun 9 18:15:48 PDT 2022
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Freescale i.MX6 ULL 14x14 MYiR Board
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 32 MiB at 0x8e000000
[    0.000000] percpu: Embedded 15 pages/cpu s32076 r8192 d21172 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: console=ttymxc0,115200 ubi.mtd=4 root=ubi0:rootfs rootfstype=ubifs mtdparts=gpmi-nand:5m(boot),1m(env),10m(kernel),1m(dtb),-(rootfs)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 208284K/262144K available (11264K kernel code, 727K rwdata, 4060K rodata, 1024K init, 430K bss, 21092K reserved, 32768K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=1.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2c8/0x468 with crng_init=0
[    0.000000] Switching to timer-based delay loop, resolution 41ns
[    0.000016] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000055] clocksource: mxc_timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.003077] Console: colour dummy device 80x30
[    0.003148] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.003184] pid_max: default: 32768 minimum: 301
[    0.003483] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.003520] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.005051] CPU: Testing write buffer coherency: ok
[    0.005734] CPU0: update cpu_capacity 1024
[    0.005765] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.060581] Setting up static identity map for 0x80100000 - 0x80100060
[    0.080553] rcu: Hierarchical SRCU implementation.
[    0.120722] smp: Bringing up secondary CPUs ...
[    0.120758] smp: Brought up 1 node, 1 CPU
[    0.120780] SMP: Total of 1 processors activated (48.00 BogoMIPS).
[    0.120796] CPU: All CPU(s) started in SVC mode.
[... 410 lines cut ...]
I (606) NETWORK_ADAPTER: *********************************************************************
I (616) NETWORK_ADAPTER: Supported features are:
I (626) NETWORK_ADAPTER: - WLAN over SDIO
I (626) NETWORK_ADAPTER: - BT/BLE
I (636) NETWORK_ADAPTER:    - HCI Over UART
I (636) NETWORK_ADAPTER:    - BT/BLE dual mode
I (646) BTDM_INIT: BT controller compile version [4bc0a8e]
I (646) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (756) phy: phy_version: 4390, 6b3c1f2, Sep 10 2020, 15:09:07, 0, 0
I (1226) SDIO_SLAVE: Using SDIO interface
I (1236) wifi:wifi driver task: 3ffde798, prio:23, stack:6656, core=0
I (1236) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1236) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1246) wifi:wifi firmware version: 6e1f9e4
I (1246) wifi:config NVS flash: disabled
I (1256) wifi:config nano formating: disabled
I (1256) wifi:Init data frame dynamic rx buffer num: 32
I (1266) wifi:Init management frame dynamic rx buffer num: 32
I (1266) wifi:Init management short buffer num: 32
I (1276) wifi:Init dynamic tx buffer num: 32
I (1276) wifi:Init static rx buffer size: 1600
I (1276) wifi:Init static rx buffer num: 10
I (1286) wifi:Init dynamic rx buffer num: 32
I (1296) wifi:mode : null
I (1296) NETWORK_ADAPTER: Initial set up done

shihui2014, 2023-03-13:
[image]

mantriyogesh, 2023-03-13:
So from your log, I still suspect if the SDIO is detected or not.
I would suggest to verify your device tree.

If the connections are correct and ESP is flashed with SDIO firmware, Host should detect the ESP SDIO first.

I am attaching working reference NXP imx 8mm (imx8mm-lpddr4-evk) dts for your reference.
We are not expert of device tree, but we believe below files may point you in better direction.
<details>
  <summary>imx8mm-evk.dts</summary>
  
 ```
u64@ubuntu:~/imx8_dev_board/imx-yocto-bsp/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-linux-libc-headers/5.16-r0/linux-5.16/arch/arm64/boot/dts/freescale$ cat imx8mm-evk.dts
// SPDX-License-Identifier: (GPL-2.0+ OR MIT)
/*
 * Copyright 2019-2020 NXP
 */

/dts-v1/;

#include <dt-bindings/usb/pd.h>
#include "imx8mm-evk.dtsi"

/ {
	model = "FSL i.MX8MM EVK board";
	compatible = "fsl,imx8mm-evk", "fsl,imx8mm";

	aliases {
		spi0 = &flexspi;
	};
};

&ddrc {
	operating-points-v2 = <&ddrc_opp_table>;

	ddrc_opp_table: opp-table {
		compatible = "operating-points-v2";

		opp-25M {
			opp-hz = /bits/ 64 <25000000>;
		};

		opp-100M {
			opp-hz = /bits/ 64 <100000000>;
		};

		opp-750M {
			opp-hz = /bits/ 64 <750000000>;
		};
	};
};

&flexspi {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_flexspi>;
	status = "okay";

	flash@0 {
		reg = <0>;
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "jedec,spi-nor";
		spi-max-frequency = <80000000>;
		spi-tx-bus-width = <1>;
		spi-rx-bus-width = <4>;
	};
};

&usdhc3 {
	assigned-clocks = <&clk IMX8MM_CLK_USDHC3_ROOT>;
	assigned-clock-rates = <400000000>;
[... 546 lines cut ...]
			MX8MM_IOMUXC_GPIO1_IO04_USDHC2_VSELECT	0x1d0
		>;
	};

	pinctrl_usdhc2_200mhz: usdhc2-200mhzgrp {
		fsl,pins = <
			MX8MM_IOMUXC_SD2_CLK_USDHC2_CLK		0x196
			MX8MM_IOMUXC_SD2_CMD_USDHC2_CMD		0x1d6
			MX8MM_IOMUXC_SD2_DATA0_USDHC2_DATA0	0x1d6
			MX8MM_IOMUXC_SD2_DATA1_USDHC2_DATA1	0x1d6
			MX8MM_IOMUXC_SD2_DATA2_USDHC2_DATA2	0x1d6
			MX8MM_IOMUXC_SD2_DATA3_USDHC2_DATA3	0x1d6
			MX8MM_IOMUXC_GPIO1_IO04_USDHC2_VSELECT	0x1d0
		>;
	};

	pinctrl_wdog: wdoggrp {
		fsl,pins = <
			MX8MM_IOMUXC_GPIO1_IO02_WDOG1_WDOG_B	0x166
		>;
	};
};
u64@ubuntu:~/imx8_dev_board/imx-yocto-bsp/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-linux-libc-headers/5.16-r0/linux-5.16/arch/arm64/boot/dts/freescale$
```
</details>

mantriyogesh, 2023-03-13:
To add further,
After doing:
1. ESP <== SDIO ==> Host connections made
2. Device tree mmc instance enabled (with correct/default pinctrl)
3. ESP flashed with SDIO binary

Your expected log (without even loading ESP32 host kernel module):

```
[   77.649578] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[   77.657019] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[   77.660243] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   77.662448] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   77.662545] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   77.662643] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   77.664445] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   77.664541] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   77.664637] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   77.664832] mmc1: new SDIO card at address 0001
```

Once you have detected the SDIO, then, you would want to port rpi_init.sh script as per your host (porting guide mentions this). This should be simple enough.
then you can load the esp32_sdio.ko driver with 'resetpin' configured using ported rpi_init.sh.

Once the kernel module loaded, you can see ESP has reset (using resetpin). Host will show log similar to:
```
[   77.665287] esp_probe: ESP network device detected
[   77.877892] Features supported are:
[   77.877901] 	 * WLAN
[   77.877906] 	 * BT/BLE
[   77.877911] 	   - HCI over SDIO
[   77.877916] 	   - BT/BLE dual mode
[   78.096179] esp_sdio: probe of mmc1:0001:2 failed with error -22
```
mmc instance will vary.

mantriyogesh, 2023-03-15:
Hello @shihui2014 

Were you manage to get CIS & first event like above?

shihui2014, 2023-03-15:
hello @mantriyogesh 
I haven't got this infomation。

mantriyogesh, 2023-03-15:
I think you can cross check your Linux SoC MMC config (Device Tree). We cannot assess your Device Tree to be correct or not (It varies from SoC to SoC)

If that is correct, ESP is connected correctly with expected pull-ups & ESP flashed with correct(matching) firmware, then you can expect the CIS prints.

You can also verify if (1) CIS is originated from ESP or not by shut the power to ESP -> No CIS for mmcX. (2) Power ESP -> CIS for mmcX printed in host log.

Make sure you reboot host to check CIS prints in both above tests (If device tree is not correctly configured/supported to poll the SD MMC hardware detection)

host log could be one from `/var/log/messages` or `/var/log/kern/log` or `dmesg` command.

mantriyogesh, 2023-03-15:
Okay, that will be easy to integrate actually.
You can embed your expected calls right from application.

But even in that case, the underlying sdio hardware still needs to be configured (https://github.com/espressif/esp-hosted/issues/212#issuecomment-1469472003 is still applicable)

shihui2014, 2023-03-15:
hi，I have obtained the following information from dmesg!
[  142.959567] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  142.979684] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  142.990920] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  142.999253] drivers/net/wireless/esp32/esp_sdio.c:736 Enter
[  143.004870] drivers/net/wireless/esp32/esp_sdio.c:737 Class=0
[  143.014081] drivers/net/wireless/esp32/esp_sdio.c:738 sdio vendor ID: 0x6666
[  143.023016] drivers/net/wireless/esp32/esp_sdio.c:739 sdio device ID: 0x2222
[  143.031704] drivers/net/wireless/esp32/esp_sdio.c:740 Function#: 1
[  143.039495] esp_probe: ESP network device detected
[  143.045161]
[  143.045161] Received INIT event from ESP32 peripheral
[  143.045174] EVENT: 0
[  143.059163] ESP peripheral capabilities: 0x1b
[  143.063550] Features supported are:
[  143.096948] drivers/net/wireless/esp32/esp_sdio.c:736 Enter
[  143.126903]   * WLAN
[  143.129122]   * BT/BLE
[  143.131488]     - HCI over UART
[  143.134634]     - BT/BLE dual mode
[  143.143276] drivers/net/wireless/esp32/esp_sdio.c:737 Class=2
[  143.169340] drivers/net/wireless/esp32/esp_sdio.c:738 sdio vendor ID: 0x6666
[  143.176421] drivers/net/wireless/esp32/esp_sdio.c:739 sdio device ID: 0x3333
[  143.196884] EVENT: 3
[  143.223037] drivers/net/wireless/esp32/esp_sdio.c:740 Function#: 2
[  143.242039] esp_sdio: probe of mmc1:0001:2 failed with error -22
[  143.271008] drivers/net/wireless/esp32/esp_sdio.c:805 esp32 sdio_register_driver ret:0
root@myd-y6ull14x14:~# [  144.395119] process_rx_packet, Dropping unsupported serial frame

shihui2014, 2023-03-15:
hi,@mantriyogesh ,thank you for your support,
When see this information,"esp_sdio: probe of mmc1:0001:2 failed with error -22 ",
What should I do next?

mantriyogesh, 2023-03-15:
esp_sdio: probe of mmc1:0001:2 failed with error -22
can be ignored. we don't use function 2 so returns.

What is concerning,
process_rx_packet, Dropping unsupported serial frame

mantriyogesh, 2023-03-15:
What is output of 

```
$ git log --oneline | head -2 
```
It should be https://github.com/espressif/esp-hosted/commit/d30942e7d863a78c1ea2d13c0f964529ff542fef

Can you use current master at both sides, ESP and host side?
https://github.com/espressif/esp-hosted/blob/81778f63a9d84fa46f8a907078f9e2681b391bf5/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L183-L202
You can tune timing there.

mantriyogesh, 2023-03-15:
Are you able to check ahead if you are able connect to AP (using demo app)?
Capture logs at both, ESP and Host.

- ESP
  - idf.py monitor OR minicom log
- Host: 
  - dmesg OR /var/log/messages OR /var/log/kern.log
  - Commands run and their output

shihui2014, 2023-03-15:
I visit the website" https://github.com/espressif/esp-hosted/releases " and download ESP-Hosted-FG-release_0.0.5.tgz  ,esp-hosted-release-fg-v0.0.5.zip

shihui2014, 2023-03-16:
yes,
esp32's log
I (69918) wifi:wifi driver task: 3ffdb308, prio:23, stack:6656, core=1
I (69918) wifi:wifi firmware version: 03528e0
I (69920) wifi:wifi certification version: v7.0
I (69920) wifi:config NVS flash: disabled
I (69924) wifi:config nano formating: disabled
I (69926) wifi:Init data frame dynamic rx buffer num: 32
I (69930) wifi:Init management frame dynamic rx buffer num: 32
I (69936) wifi:Init management short buffer num: 32
I (69938) wifi:Init dynamic tx buffer num: 64
I (69942) wifi:Init static I (69944) NETWORK_ADAPTER: Start Data Path
rx buffer size: 1600
I (69950) wifi:Init static rx buffer num: 16
I (69952) wifi:Init dynamic rx buffer num: 32
I (69956) wifi_init: rx ba win: 6
I (69958) wifi_init: tcpip mbox: 32
I (69962) wifi_init: udp mbox: 6
I (69966) wifi_init: tcp mbox: 6
I (69968) wifi_init: tcp tx win: 5744
I (69972) wifi_init: tcp rx win: 5744
I (69976) wifi_init: tcp mss: 1440
I (69978) wifi_init: WiFi IRAM OP enabled
I (69982) wifi_init: WiFi RX IRAM OP enabled
I (69988) wifi:mode : null
I (69990) NETWORK_ADAPTER: Initial set up done
I (70992) slave_ctrl: event ESPInit
kernel'e log
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0x600000, size 0xa00000
 10485760 bytes read: OK

NAND read: device 0 offset 0x1000000, size 0x100000
 1048576 bytes read: OK
Kernel image @ 0x80800000 [ 0x000000 - 0x83f710 ]
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
   Using Device Tree in place at 83000000, end 8300bcd4
Modify /soc/aips-bus@2200000/epdc@228c000:status disabled
ft_system_setup for mx6

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.3-gc724a73-dirty (root@ubuntu) (gcc version 9.2.0 (GCC)) #2 SMP PREEMPT Wed Mar 15 17:12:38 +08 2023
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Freescale i.MX6 ULL 14x14 MYiR Board
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x88000000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] percpu: Embedded 15 pages/cpu s32076 r8192 d21172 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: console=ttymxc0,115200 ubi.mtd=4 root=ubi0:rootfs rootfstype=ubifs mtdparts=gpmi-nand:5m(boot),1m(env),10m(kernel),1m(dtb),-(rootfs)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 109784K/262144K available (11264K kernel code, 728K rwdata, 4060K rodata, 1024K init, 430K bss, 21288K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=1.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2c8/0x468 with crng_init=0
[    0.000000] Switching to timer-based delay loop, resolution 41ns
[    0.000015] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[... 414 lines cut ...]
[   76.404555] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   76.415528] drivers/net/wireless/esp32/esp_sdio.c:736 Enter
[   76.432062] drivers/net/wireless/esp32/esp_sdio.c:737 Class=0
[   76.440496] drivers/net/wireless/esp32/esp_sdio.c:738 sdio vendor ID: 0x6666
[   76.449256] drivers/net/wireless/esp32/esp_sdio.c:739 sdio device ID: 0x2222
[   76.456333] drivers/net/wireless/esp32/esp_sdio.c:740 Function#: 1
[   76.465053] esp_probe: ESP network device detected
[   76.472193]
[   76.472193] Received INIT event from ESP32 peripheral
[   76.472210] EVENT: 0
[   76.484663] ESP peripheral capabilities: 0x1b
[   76.517068] drivers/net/wireless/esp32/esp_sdio.c:736 Enter
[   76.522680] drivers/net/wireless/esp32/esp_sdio.c:737 Class=2
[   76.538922] Features supported are:
[   76.542447]   * WLAN
[   76.544639]   * BT/BLE
[   76.576919]     - HCI over UART
[   76.580094]     - BT/BLE dual mode
[   76.583504] EVENT: 3
[   76.589477] drivers/net/wireless/esp32/esp_sdio.c:738 sdio vendor ID: 0x6666
[   76.596557] drivers/net/wireless/esp32/esp_sdio.c:739 sdio device ID: 0x3333
[   76.644527] drivers/net/wireless/esp32/esp_sdio.c:740 Function#: 2
[   76.664339] esp_sdio: probe of mmc1:0001:2 failed with error -22
[   76.698803] drivers/net/wireless/esp32/esp_sdio.c:805 esp32 sdio_register_driver ret:0
root@myd-y6ull14x14:~# [   77.820534] process_rx_packet, Dropping unsupported serial frame

shihui2014, 2023-03-16:
hi @mantriyogesh you say “connect to AP (using demo app)?”
the demo app is？

shihui2014, 2023-03-16:
connect to ap use by “esp-hosted-release-fg-v0.0.5\esp_hosted_fg\host\linux\host_control\ c_support ”？
may i use “c_support” only？Because my board has no python environment。

mantriyogesh, 2023-03-16:
I don't see your ESP full log.
Can you please help to get ESP log from flashing till end?
I suspect you are still using old release.

mantriyogesh, 2023-03-16:
Yes you can use C demo app from esp_hosted_fg\host\linux\host_control\c_support.

shihui2014, 2023-03-16:
An error occurred while compiling c_support
fatal error: protobuf-c/protobuf-c.h: No such file or directory
file protobuf-c/protobuf-c/protobuf-c.c  is not exist。

and esp log is :
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x2b (HSPI_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7012
load:0x40078000,len:15384
load:0x40080400,len:3816
entry 0x40080648
I (28) boot: ESP-IDF v5.0-beta1-742-g7bd5af7f1e 2nd stage bootloader
I (28) boot: compile time 09:35:03
I (28) boot: chip revision: V300
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 factory          factory app      00 00 00010000 00100000
I (99) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (107) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (114) boot: End of partition table
I (119) boot: Defaulting to factory image
I (123) boot_comm: chip revision: 3, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=288ach (166060) map
I (199) esp_image: segment 1: paddr=000388d4 vaddr=3ffbdb60 size=05864h ( 22628) load
I (208) esp_image: segment 2: paddr=0003e140 vaddr=40080000 size=01ed8h (  7896) load
I (212) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9d014h (643092) map
I (448) esp_image: segment 4: paddr=000dd03c vaddr=40081ed8 size=1c044h (114756) load
I (495) esp_image: segment 5: paddr=000f9088 vaddr=50000000 size=00010h (    16) load
I (511) boot: Loaded app from partition at offset 0x10000
I (511) boot: Disabling RNG early entropy source...
I (522) cpu_start: Pro cpu up.
I (523) cpu_start: Starting app cpu, entry point is 0x400813dc
I (0) cpu_start: App cpu up.
I (539) cpu_start: Pro cpu start user code
I (539) cpu_start: cpu freq: 240000000 Hz
I (539) cpu_start: Application information:
I (544) cpu_start: Project name:     network_adapter
I (549) cpu_start: App version:      e0bb23b
I (554) cpu_start: Compile time:     Oct 21 2022 09:34:49
I (560) cpu_start: ELF file SHA256:  985ab6d0449645a0...
I (566) cpu_start: ESP-IDF:          v5.0-beta1-742-g7bd5af7f1e
I (573) heap_init: Initializing. RAM available for dynamic allocation:
I (580) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (586) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (592) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (598) heap_init: At 3FFCCA28 len 000135D8 (77 KiB): DRAM
I (605) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (611) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (617) heap_init: At 4009DF1C len 000020E4 (8 KiB): IRAM
I (625) spi_flash: detected chip: gd
I (628) spi_flash: flash io: dio
W (632) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (646) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (660) NETWORK_ADAPTER: *********************************************************************
I (666) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5
I (674) NETWORK_ADAPTER:                 Transport used :: SDIO + UART
I (682) NETWORK_ADAPTER: *********************************************************************
I (690) NETWORK_ADAPTER: Supported features are:
I (694) NETWORK_ADAPTER: - WLAN over SDIO
I (698) ESP_BT: - BT/BLE
I (700) ESP_BT:    - HCI Over UART
I (702) ESP_BT:    - BT/BLE dual mode
I (706) NETWORK_ADAPTER: capabilities: 0x1b
I (718) BTDM_INIT: BT controller compile version [8020d24]
I (720) system_api: Base MAC address is not set
I (722) system_api: read default base MAC address from EFUSE
I (728) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1054) NETWORK_ADAPTER: ESP Bluetooth MAC addr: b8:d6:1a:6c:69:6a
I (1056) SDIO_SLAVE: Using SDIO interface

mantriyogesh, 2023-03-16:
> fatal error: protobuf-c/protobuf-c.h: No such file or directory
> file protobuf-c/protobuf-c/protobuf-c.c is not exist。

Missed this documentation step:
[image]

I will update the code likewise, but CONFIG_SUPPORT_ESP_SERIAL is mandatory.
https://github.com/espressif/esp-hosted/blob/81778f63a9d84fa46f8a907078f9e2681b391bf5/esp_hosted_fg/host/linux/host_driver/esp32/Makefile#L1

shihui2014, 2023-03-16:
I have already set CONFIG_SUPPORT_ESP_SERIAL = y

and esp32 log is :
root@myd-y6ull14x14:~# insmod esp32_sdio.ko
[   34.381029] esp_reset, ESP32: Resetpin of Host is 68
[   34.455773] esp_sdio: probe of mmc1:0001:1 failed with error -16
[   34.462340] esp_sdio: probe of mmc1:0001:2 failed with error -16
[   34.479966] drivers/net/wireless/esp32/esp_sdio.c:806 esp32 sdio_register_driver ret:0
root@myd-y6ull14x14:~# [   35.047012] can-3v3: disabling

kernel log is:
.
I (659) NETWORK_ADAPTER: *********************************************************************
I (665) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5
I (673) NETWORK_ADAPTER:                 Transport used :: SDIO + UART
I (681) NETWORK_ADAPTER: *********************************************************************
I (689) NETWORK_ADAPTER: Supported features are:
I (693) NETWORK_ADAPTER: - WLAN over SDIO
I (697) ESP_BT: - BT/BLE
I (699) ESP_BT:    - HCI Over UART
I (701) ESP_BT:    - BT/BLE dual mode
I (705) NETWORK_ADAPTER: capabilities: 0x1b
I (717) BTDM_INIT: BT controller compile version [8020d24]
I (719) system_api: Base MAC address is not set
I (721) system_api: read default base MAC address from EFUSE
I (727) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1055) NETWORK_ADAPTER: ESP Bluetooth MAC addr: b8:d6:1a:6c:69:6a
I (1057) SDIO_SLAVE: Using SDIO interface
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7012
load:0x40078000,len:15384
load:0x40080400,len:3816
entry 0x40080648
I (27) boot: ESP-IDF v5.0-beta1-742-g7bd5af7f1e 2nd stage bootloader
I (27) boot: compile time 09:35:03
I (28) boot: chip revision: V300
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 factory          factory app      00 00 00010000 00100000
I (99) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (114) boot: End of partition table
I (118) boot: Defaulting to factory image
I (123) boot_comm: chip revision: 3, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=288ach (166060) map
I (199) esp_image: segment 1: paddr=000388d4 vaddr=3ffbdb60 size=05864h ( 22628) load
I (208) esp_image: segment 2: paddr=0003e140 vaddr=40080000 size=01ed8h (  7896) load
I (211) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9d014h (643092) map
I (447) esp_image: segment 4: paddr=000dd03c vaddr=40081ed8 size=1c044h (114756) load
I (495) esp_image: segment 5: paddr=000f9088 vaddr=50000000 size=00010h (    16) load
I (510) boot: Loaded app from partition at offset 0x10000
I (510) boot: Disabling RNG early entropy source...
I (522) cpu_start: Pro cpu up.
I (522) cpu_start: Starting app cpu, entry point is 0x400813dc
I (0) cpu_start: App cpu up.
I (538) cpu_start: Pro cpu start user code
I (538) cpu_start: cpu freq: 240000000 Hz
I (539) cpu_start: Application information:
I (543) cpu_start: Project name:     network_adapter
I (549) cpu_start: App version:      e0bb23b
I (554) cpu_start: Compile time:     Oct 21 2022 09:34:49
I (560) cpu_start: ELF file SHA256:  985ab6d0449645a0...
I (566) cpu_start: ESP-IDF:          v5.0-beta1-742-g7bd5af7f1e
I (573) heap_init: Initializing. RAM available for dynamic allocation:
I (580) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (586) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (592) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (598) heap_init: At 3FFCCA28 len 000135D8 (77 KiB): DRAM
I (604) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (611) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (617) heap_init: At 4009DF1C len 000020E4 (8 KiB): IRAM
I (624) spi_flash: detected chip: gd
I (627) spi_flash: flash io: dio
W (631) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (646) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (659) NETWORK_ADAPTER: *********************************************************************
I (665) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5
I (673) NETWORK_ADAPTER:                 Transport used :: SDIO + UART
I (681) NETWORK_ADAPTER: *********************************************************************
I (689) NETWORK_ADAPTER: Supported features are:
I (693) NETWORK_ADAPTER: - WLAN over SDIO
I (697) ESP_BT: - BT/BLE
I (699) ESP_BT:    - HCI Over UART
I (701) ESP_BT:    - BT/BLE dual mode
I (705) NETWORK_ADAPTER: capabilities: 0x1b
I (717) BTDM_INIT: BT controller compile version [8020d24]
I (719) system_api: Base MAC address is not set
I (721) system_api: read default base MAC address from EFUSE
I (727) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1061) NETWORK_ADAPTER: ESP Bluetooth MAC addr: b8:d6:1a:6c:69:6a
I (1061) SDIO_SLAVE: Using SDIO interface

shihui2014, 2023-03-16:
Hello @mantriyogesh Why do the following errors occur？
insmod esp32_sdio.ko
[ 1104.401391] esp_sdio: probe of mmc1:0001:1 failed with error -16
[ 1104.407978] esp_sdio: probe of mmc1:0001:2 failed with error -16

mantriyogesh, 2023-03-16:
Log at host,
`Dropping unsupported serial frame` can only comes at line 423:

https://github.com/espressif/esp-hosted/blob/81778f63a9d84fa46f8a907078f9e2681b391bf5/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L409-L424

Have you checked the file mentioned in  https://github.com/espressif/esp-hosted/issues/212#issuecomment-1471255862

You might want to re-run rpi_init.sh (unload and cleanup the kernel driver and rebuild and load it back after above changes)
Revert any timing related changes if you have done at ESP side.

shihui2014, 2023-03-16:
i have define  CONFIG_SUPPORT_ESP_SERIAL =1 in esp.h
so line 423,will not appear,
Now the question is 
[ 1104.401391] esp_sdio: probe of mmc1:0001:1 failed with error -16
[ 1104.407978] esp_sdio: probe of mmc1:0001:2 failed with error -16

shihui2014, 2023-03-16:
my makefile is ，
obj-$(CONFIG_WLAN_VENDOR_ESP32)	+= esp32_sdio.o
esp32_sdio-objs:=esp_sdio.o esp_bt.o main.o esp_sdio_api.o esp_stats.o esp_serial.o esp_rb.o
and i have define CONFIG_SUPPORT_ESP_SERIAL =1 in esp.h ，This works the same way as CONFIG_SUPPORT_ESP_SERIAL = y

mantriyogesh, 2023-03-16:
Current code changes please?
$ git status 
$ git diff

Both ESP and Host side?

Also are you getting first event at Host (host log) will tell.
It will be important to understand if the esp_probe is being called or not and where it is failing.
similar issue and some explanation with logs addition:
https://github.com/espressif/esp-hosted/issues/198#issuecomment-1463210942

shihui2014, 2023-03-16:
okey ，
I have used the latest code on esp side for esp32_sdio.ko，but the esp_bins file for host where can i download？

shihui2014, 2023-03-16:
esp_bins

shihui2014, 2023-03-16:
I have used the latest code for esp side ,but where can i download esp bins file for host side?

mantriyogesh, 2023-03-16:
> esp bins file for host side

0.0.5 work with latest current master [81778f6](https://github.com/espressif/esp-hosted/commit/81778f63a9d84fa46f8a907078f9e2681b391bf5)

But as you need might want to change the timing, You will need to flash from source.
It is easy to flash using source (and much usable as well).

If you need binaries only, Let me know.
I will get you FG 0.0.5 based binaries with timing adjusted (as above mentioned in https://github.com/espressif/esp-hosted/issues/212#issuecomment-1469789618)

mantriyogesh, 2023-03-16:
Regarding https://github.com/espressif/esp-hosted/issues/212#issuecomment-1471382604, Can you please localise by adding prints in esp_sdio.c, which call fails?
Is the esp_probe() even entered once after loading module?

shihui2014, 2023-03-16:
for the err esp_sdio: probe of mmc1:0001:1 failed with error -16 
Refer to this address https://github.com/espressif/esp-hosted/issues/198#issuecomment-1463210942 ihave add log 
kernel log is :
[   47.200294] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[   47.222244] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[   47.234302] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   47.244269] esp_probe:741
[   47.248749] esp_probe: ESP network device detected
[   47.255570] init_sdio_func:659 sdio_enable_func:ret:0
[   47.264732] init_sdio_func:666 sdio_claim_irq:ret:0
[   47.271818]
[   47.271818] Received INIT event from ESP32 peripheral
[   47.271834] EVENT: 0
[   47.284631] esp_probe:749 init_sdio_func ret:2130723472
[   47.290082] ESP peripheral capabilities: 0x1b
[   47.294453] Features supported are:
[   47.301294] init_context:317
[   47.304636] esp_probe:756 init_context ret:0
[   47.311408]   * WLAN
[   47.313625]   * BT/BLE
[   47.315992]     - HCI over UART
[   47.357081] esp_probe:741
[   47.359777] esp_sdio: probe of mmc1:0001:2 failed with error -22
[   47.381836]     - BT/BLE dual mode
[   47.385276] EVENT: 3
root@myd-y6ull14x14:~#
esp side log is 

I (627) spi_flash: flash io: dio
W (631) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (646) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (659) NETWORK_ADAPTER: *********************************************************************
I (665) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5
I (673) NETWORK_ADAPTER:                 Transport used :: SDIO + UART
I (681) NETWORK_ADAPTER: *********************************************************************
I (689) NETWORK_ADAPTER: Supported features are:
I (693) NETWORK_ADAPTER: - WLAN over SDIO
I (697) ESP_BT: - BT/BLE
I (699) ESP_BT:    - HCI Over UART
I (701) ESP_BT:    - BT/BLE dual mode
I (705) NETWORK_ADAPTER: capabilities: 0x1b
I (717) BTDM_INIT: BT controller compile version [8020d24]
I (719) system_api: Base MAC address is not set
I (721) system_api: read default base MAC address from EFUSE
I (727) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1053) NETWORK_ADAPTER: ESP Bluetooth MAC addr: b8:d6:1a:6c:69:6a
I (1053) SDIO_SLAVE: Using SDIO interface
I (3688187) wifi:wifi driver task: 3ffdb308, prio:23, stack:6656, core=1
I (3688187) wifi:wifi firmware version: 03528e0
I (3688189) wifi:wifi certification version: v7.0
I (3688189) wifi:config NVS flash: disabled
I (3688193) wifi:config nano formating: disabled
I (3688197) wifi:Init data frame dynamic rx buffer num: 32
I (3688201) wifi:Init management frame dynamic rx buffer num: 32
I (3688205) wifi:Init management short buffer num: 32
I (3688209) wifi:Init dynamic tx buffer num: 64
I (3688213) wifi:Init static rx buffer size: 1600
I (3688217) wifi:Init static rx buffer num: 16
I (3688219) wifi:Init dynamic rx buffer num: 32
I (3688223) wifi_init: rx ba win: 6
I (3688225) wifi_init: tcpip mbox: 32
I (3688229) wifi_init: udp mbox: 6
I (3688233) wifi_init: tcp mbox: 6
I (3688235) wifi_init: tcp tx win: 5744
I (3688239) wifi_init: tcp rx win: 5744
I (3688243) wifi_init: tcp mss: 1440I (3688247) NETWORK_ADAPTER: Start Data Path
m
I (3688251) wifi_init: WiFi IRAM OP enabled
I (3688255) wifi_init: WiFi RX IRAM OP enabled
I (3688261) wifi:mode : null
I (3688261) NETWORK_ADAPTER: Initial set up done
I (3689265) slave_ctrl: event ESPInit

mantriyogesh, 2023-03-16:
great! everything looks great. proceed ahead with c demo app.

shihui2014, 2023-03-16:
But it doesn't seem to be running  
printk(KERN_INFO "%s: ESP network device detected\n", __func__);

	context = init_sdio_func(func);
	printk(KERN_INFO "%s:%u init_sdio_func ret:%d\n",__func__,__LINE__,context);
	if (!context) {
		return -ENOMEM;
	}

mantriyogesh, 2023-03-16:
```
[ 47.304636] esp_probe:756 init_context ret:0
[ 47.311408] * WLAN
[ 47.313625] * BT/BLE
[ 47.315992] - HCI over UART
[ 47.357081] esp_probe:741
[ 47.359777] esp_sdio: probe of mmc1:0001:2 failed with error -22
[ 47.381836] - BT/BLE dual mode
[ 47.385276] EVENT: 3
```
shows your sdio is fine.
It has received the init event as well.

Have you ported the rpi_init.sh to your platform as per stated in porting)guide?

shihui2014, 2023-03-16:
esp_hosted_fg\host\linux\host_control\c_support  
Makefile 
SRC += $(DIR_COMMON)/protobuf-c/protobuf-c/protobuf-c.c
the file protobuf-c.c dos not exist,so i can't compile c_support

mantriyogesh, 2023-03-16:
Have you checked this?
https://user-images.githubusercontent.com/67477199/225508224-c7b22446-abde-4c9d-92aa-ce4968e9ac39.png

mantriyogesh, 2023-03-16:
I know that you are not using Raspberry Pi.
But to build the code, you will still need full esp-hosted repo.
That includes submodules.

Where you have ESP-Hosted cloned, please update submodules as shown command in picture.
You should be able to find protobuf-c.c file after updating submodule.

rpi_init,sh is of course Raspberry Pi specific. But when you still need to port it as per your platform.
Basically, remove the things that are raspberry Pi specific but keep generic linux.
As I see you are interested in UART, still I suggest you to first get SDIO only working.

For UART, you may need to port rpi_init.sh to make you SoC UART GPIOs in correct alternate function.
Please check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#122-rpi_initsh

shihui2014, 2023-03-16:
root@myd-y6ull14x14:~# ./test.out
sudo ./test.out
[
 set_sta_mac_addr               ||
 get_sta_mac_addr               ||
 set_softap_mac_addr            ||
 get_softap_mac_addr            ||
 get_ap_scan_list               ||
 sta_connect                    ||
 get_sta_config                 ||
 sta_disconnect                 ||
 set_wifi_mode                  ||
 get_wifi_mode                  ||
 reset_softap_vendor_ie         ||
 set_softap_vendor_ie           ||
 softap_start                   ||
 get_softap_config              ||
 softap_connected_sta_list      ||
 softap_stop                    ||
 set_wifi_powersave_mode        ||
 get_wifi_powersave_mode        ||
 set_wifi_max_tx_power          ||
 get_wifi_curr_tx_power         ||
 ota <ESP 'network_adapter.bin' path>
]

For example,
sudo ./test.out set_sta_mac_addr
root@myd-y6ull14x14:~# ./test.out get_wifi_mode
Enable heartbeat with duration 20
Heartbeat operation successful
wifi mode is : none

Requested operation complete
Sleeping for some time just to showcase heartbeat
2021-12-07 11:33:25 >  App EVENT: Heartbeat event [1]

^C
Clean-up and exit
Disable Heartbeat
Heartbeat operation successful
root@myd-y6ull14x14:~# ./test.out get_ap_scan_list
Enable heartbeat with duration 20
Heartbeat operation successful
Number of available APs is 1
0) ssid "zhoudaye" bssid "f0:b4:29:6e:bc:f6" rssi "-91" channel "2" auth mode "4"

Requested operation complete
Sleeping for some time just to showcase heartbeat
2021-12-07 11:34:24 >  App EVENT: Heartbeat event [1]
2021-12-07 11:34:49 >  App EVENT: Heartbeat event [1]
Disable Heartbeat

mantriyogesh, 2023-03-16:
Great. You can now integrate the APIs into your application and control the ESP-Hosted right through your application.

Please do check ctrl_apis.md and serial_apis.md for all APIs from
https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg/docs/common

shihui2014, 2023-03-16:
hi,@mantriyogesh  I scanned the wifi ssid using c_support！Thank you very much!

shihui2014, 2023-03-16:
okey,
Is ap hotspot supported?

shihui2014, 2023-03-16:
Does wpa_supplicant or iw be supported?

mantriyogesh, 2023-03-16:
You can use ESP-Hosted-NG flavour for that.

shihui2014, 2023-03-16:
okey,thanks a lot!

shihui2014, 2023-03-16:
Can you tell me how to connect wifi using c demo?the cmd is？c_demo.md It's not clear！

shihui2014, 2023-03-17:
How can a client obtain an ip address in AP mode?

shihui2014, 2023-04-07:
hello @mantriyogesh 
We are ready for mass production，
But found a problem, can not enter the esp download mode,
Does it have anything to do with the external pull-up resistance?

mantriyogesh, 2023-04-10:
Download mode selection depends on GPIO0 pin, EN pin. Please check some details and known ways to handle this behaviour.
https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html

Also please have a look at 'strapping pins' in esp32 datasheet.

mantriyogesh, 2024-01-19:
Hello @shihui2014 is the problem solved?

Strapping pins if need to be used, need to pull them up. Check specific Technical Reference manual for more info of pin.
Also make sure that you use pin correctly, for using boot mode and download mode.

These limitations will not happen if strapping pins are avoided.

closed 2026-09-28 by mantriyogesh