espressif/esp-hosted #213: ESP32-C6 support

timfrankentwf, 2023-03-13:
Hi all,

For a proof-of-concept project, the ESP32-C6 will be used to add Wi-Fi 6, Bluetooth 5.x and maybe even Thread support to Linux Platforms via SDIO (IMX8-based).
I can see that the ESP-C6 is currently not mentioned in any of the documents (understandable since engineering samples are currently hard to get).
Still, I want to see if I can make it work with the limited support this chip has at the moment.
Because the source for the Next Generation version is not available, I'll be using the First Generation to port for the ESP32-C6.

Is there any work done yet to support the ESP32-C6?

Thanks!

mantriyogesh, 2023-03-13:
Hello @timfrankentwf 

ESP-Hosted-FG support will be added in this week for ESP32-C6 'SPI only' transport.

mantriyogesh, 2023-03-13:
ESP32-C6 SDIO support might take some time to add.

'SPI only' platform will have all features like WiFi 6 & BT 5.x.
Thread support on top of ESP-Hosted is in discussion phase (very initial phase).

timfrankentwf, 2023-03-13:
Hi @mantriyogesh,

Thanks for your responds and good to hear that `SPI-only` support will be added this week.
Since SDIO and UART are only specified as transport interfaces it will at least be a good starting point.

I will investigate further in the feasibility of developing the SDIO interface internally.

mantriyogesh, 2023-03-13:
Hello @timfrankentwf 

Sorry, I could not understand by
`Since SDIO and UART are only specified as transport interfaces`. [ESP32-C6](https://www.espressif.com/en/products/socs/esp32-c6) supports SPI as well.
Any reference?

timfrankentwf, 2023-03-13:
Hi @mantriyogesh ,

Sorry, I didn't format that statement correctly.
The requirements of the project I work on has defined SDIO and UART only for Wi-Fi and Bluetooth.
SPI would be used for testing/development, though SDIO is originally defined as interface towards the Linux Host.

mantriyogesh, 2023-03-13:
ohh okay.

mantriyogesh, 2023-03-15:
Hello @timfrankentwf 

ESP32-C6 is now available with SPI only support in the ESP-Hosted-FG.
Currently Wi-Fi only is supported for ESP32-C3.

Other transports, like SPI+UART, SDIO, SDIO+UART & Bluetooth will be supported step wise manner.

timfrankentwf, 2023-03-15:
Hi @mantriyogesh 

Thank you! I noticed the merge yesterday and already looked into it.
I'm setting up the necessary build-environment within a Yocto BSP project using previous issues listed with info with tips.
Currently testing how far I can go with SDIO by including the ESP32-C6, the Linux Platform detects the module on the mmc-interface as an high-speed SDIO card.

I might switch to SPI for testing and make sure it all works before continuing with SDIO.

timfrankentwf, 2023-06-14:
Hi @mantriyogesh,

Do you have any update on the SDIO support for the C6?
I read that there were some timing/DMA issues, any progress on this?

Thank you!

mantriyogesh, 2023-06-14:
Hello @timfrankentwf 
No, not yet.
We will be able to get some progress next week.

mantriyogesh, 2023-07-07:
Hello @timfrankentwf  ,

Apologies for delay..
ESP32-C6 support for SDIO is added for Linux in master under 391cb8efe593812cf1e7dc8c4e1c9302d40cc18f