espressif/esp-hosted #216: ESP hosted NG driver not loading on kernel 6.1.19+

tech9492, 2023-03-19:
Hi,

I have been using ESP Hosted NG with RPi OS kernel 5.15.61+ for a while now and have not had any issues. 

However, last night I updated the raspberry pi and a new kernel was installed, version  6.1.19+. Since then I cannot load the ESP hosted driver. Please find below a screenshot of the errors I ran into while loading the driver.

Thank you.

[image]

mantriyogesh, 2023-03-20:
Hello @tech9492 

Thanks to [Mr-Bossman](https://github.com/Mr-Bossman), this issue could be fixed.
Can you please check https://github.com/espressif/esp-hosted/pull/209 ?

Mr-Bossman, 2023-03-20:
Speaking of this @mantriyogesh is there an eta on giulio and my commits

mantriyogesh, 2023-03-20:
Hello @Mr-Bossman, we will test and get this in asap.
@tech9492 did this work for you?

tech9492, 2023-03-20:
Hi @mantriyogesh, I had a go at it and it now loads through `main.c` upto `esp_cfg80211.c` where there are 3 errors. Please find below a screenshot.

FYI, I carried out the modifications from the following 3 commits; [855a2b6](https://github.com/espressif/esp-hosted/pull/209/commits/855a2b6f638e332abaabec6e98f4b7a01b40915a), [c629345](https://github.com/espressif/esp-hosted/pull/209/commits/c62934582a7bb5ed2e28a2744ed4814ea76ec55d) & [b11ea0f](https://github.com/espressif/esp-hosted/pull/209/commits/b11ea0f2d9eb389c45df3576dc558276aea29ddf) . For commit [b11ea0f](https://github.com/espressif/esp-hosted/pull/209/commits/b11ea0f2d9eb389c45df3576dc558276aea29ddf), I skipped `esp_sdio.c` since I am using SPI protocol. 

There are 2 commits for `esp_sdio.c`; [094798f](https://github.com/espressif/esp-hosted/pull/209/commits/094798f215cddc4bdc1f5318b0cf8a8c684715ea) & [b11ea0f](https://github.com/espressif/esp-hosted/pull/209/commits/b11ea0f2d9eb389c45df3576dc558276aea29ddf). I do not know if similar changes are required for `esp_spi.c and how to proceed.

Thank you

[image]

Mr-Bossman, 2023-03-20:
@tech9492 so you didn't include [b11ea0f](https://github.com/espressif/esp-hosted/pull/209/commits/b11ea0f2d9eb389c45df3576dc558276aea29ddf) please include it

Mr-Bossman, 2023-03-20:
ok try now works on linux-next

tech9492, 2023-03-23:
Hi @mantriyogesh, is there any solutions to the above errors encountered in `esp_spi.c` ?

Thank you

mantriyogesh, 2023-03-23:
Hello @tech9492 

Kernel stopped supporting this function, will check alternative way.

mantriyogesh, 2023-03-23:
Keep you posted on this

tech9492, 2023-03-23:
Alright sure. Thank you

Mr-Bossman, 2023-03-24:
Yeah so there isn't something to replace the spi_bus_num function. And it shouldn't have been used to begin with. Unfortunately this means that we would have to work around this. Personally I would do something like i did in branch dev/new-useDT in my fork of this.

Mr-Bossman, 2023-03-25:
try this. i havent tested it. https://github.com/Mr-Bossman/esp-hosted/commit/4b0fc803a6e6f40e634b25ccc7800a92e8ea5699 you will need to load the esp_hosted_ng/host/esp_spi.dts instead of the original dts

tech9492, 2023-03-27:
Hi @Mr-Bossman, thank you very much for that new commit [`4b0fc80`](https://github.com/Mr-Bossman/esp-hosted/commit/4b0fc803a6e6f40e634b25ccc7800a92e8ea5699). I actually replaced the content of `spidev_disabler.dts` with `esp_spi.dts`. 
It loads the esp module successfully without any issues. However when I run `ifconfig` and `hciconfig`,espsta0 and hci0 are not present.

Please find below the output of `dmesg`, if that might help to find the issue. 

Thank you.
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.19+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1637 Tue Mar 14 11:01:56 GMT 2023
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Plus Rev 1.2
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x0bc00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=Composite-1:720x480@60i vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=41f0d03e-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 177016K/458752K available (8984K kernel code, 1450K rwdata, 2772K rodata, 432K init, 589K bss, 19592K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 33369 entries in 98 pages
[    0.000000] ftrace: allocated 98 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000006] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000080] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000179] bcm2835: system timer (irq = 27)
[    0.001060] Console: colour dummy device 80x30
[    0.001647] printk: console [tty1] enabled
[    0.001736] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060338] pid_max: default: 32768 minimum: 301
[    0.060573] LSM: Security Framework initializing
[    0.060916] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.061003] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.062802] cgroup: Disabling memory control group subsystem
[    0.063275] CPU: Testing write buffer coherency: ok
[    0.065159] cblist_init_generic: Setting adjustable number of callback queues.
[    0.065245] cblist_init_generic: Setting shift to 0 and lim to 1.
[    0.065533] cblist_init_generic: Setting shift to 0 and lim to 1.
[    0.065890] Setting up static identity map for 0x8220 - 0x8258
[    0.067378] devtmpfs: initialized
[    0.080043] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.080575] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.080699] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.122474] pinctrl core: initialized pinctrl subsystem
[    0.124653] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.127970] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.134131] audit: initializing netlink subsys (disabled)
[    0.135832] thermal_sys: Registered thermal governor 'step_wise'
[    0.136759] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.136861] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.137138] Serial: AMBA PL011 UART driver
[    0.141569] audit: type=2000 audit(0.130:1): state=initialized audit_enabled=0 res=1
[    0.145823] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.170886] raspberrypi-firmware soc:firmware: Attached to firmware from 2023-03-17T10:52:42, variant start
[    0.180916] raspberrypi-firmware soc:firmware: Firmware hash is 82f3750a65fadae9a38077e3c2e217ad158c8d54
[    0.232246] kprobes: kprobe jump-optimization is enabled. All kprobes are optimized if possible.
[    2.430877] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
[    2.433773] SCSI subsystem initialized
[... 1180 lines cut ...]
[  307.759475] EVENT: 0
[  307.759486] EVENT: 0
[  307.759496] EVENT: 0
[  307.759506] EVENT: 0
[  307.759516] EVENT: 0
[  307.759527] EVENT: 0
[... cut ...]
[  307.759599] EVENT: 0
[  307.759609] EVENT: 0
[  307.759620] EVENT: 0
[  307.759630] EVENT: 0
[  307.759640] EVENT: 0
[  307.759651] ESP chipset not recognized, ignoring [255]
[  312.810949] esp32: Command[1] timed out
[  312.811009] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[  312.811074] esp32: add_network_iface() failure, ret: -1
[  312.811097] network iterface init failed
[  312.811109] ESP peripheral capabilities: 0x0
[  312.811123] Capabilities: 0x0. Features supported are:
```

tech9492, 2023-03-27:
> Ah yeah this issue actually has nothing to do with my patches.

My bad, I thought it was since you mentioned you haven't tested it.

> This happened on an unmodified esp_hosted_ng on my RPI a while ago and I just rolled back and it fixed the issue.

You mean you rolled back the kernel?

Mr-Bossman, 2023-03-27:
No I rolled back this repository to the above commit

Mr-Bossman, 2023-03-27:
> > Ah yeah this issue actually has nothing to do with my patches.
> 
> My bad, I thought it was since you mentioned you haven't tested it.
> 
I didn't test it, but you did, and the changes I made for the spi seemed to do what I expected.
> > This happened on an unmodified esp_hosted_ng on my RPI a while ago and I just rolled back and it fixed the issue.
> 
> You mean you rolled back the kernel?

tech9492, 2023-03-27:
> No I rolled back this repository to the above commit

Oh alright. 

Just wondering are you running esp firmware NG 1.0.1 or 1.0.2? 

Thank you again for the patches.

Mr-Bossman, 2023-03-27:
> > No I rolled back this repository to the above commit
> 
> Oh alright. 
> 
> Just wondering are you running esp firmware NG 1.0.1 or 1.0.2?

I don't remember probably tested with both. I'll check when I get home(if I remember).

> 
> Thank you again for the patches.

mantriyogesh, 2023-05-02:
Hello @tech9492 ,

are you still blocked?

kapilkedawat, 2023-05-05:
@tech9492 this is fixed, please try with latest driver.

tech9492, 2023-05-25:
Hello @mantriyogesh , @kapilkedawat 

Thank you for the update. Flashed firmware 1.0.2 with the modifications of @Mr-Bossman solved the issue.

Thank you again.