espressif/esp-hosted #217: ESP32-C6 sdio fg support

MrHaiCoder, 2023-03-22:
esp_host_fg supports esp32C6 and uses sdio to transfer network data. linux_hosted will support the current version. Will the main adaptation work in esp32c6 mcu?

mantriyogesh, 2023-03-22:
Yes, Full fledge support for ESP32-C6 is ongoing.
Linux support is added with SPI. Will be adding that first in MCU.
Other transports combinations also will come, once they stable supported in ESP-IDF.

Linux and MCU will go hand in hand.

MrHaiCoder, 2023-03-23:
> Yes, Full fledge support for ESP32-C6 is ongoing. Linux support is added with SPI. Will be adding that first in MCU. Other transports combinations also will come, once they stable supported in ESP-IDF.
> 
> Linux and MCU will go hand in hand.

Hello, may I know when the support for ESP32 c6 fg sdio can be developed? Currently, our project uses this solution, and the pcb has been laid out. Knowing the approximate time is convenient for subsequent project arrangements

mantriyogesh, 2023-03-23:
Latest ESP-IDF master sdio, just works on ESP32, but unstable on ESP32-C6.
So we are blocked on this.
it may take couple of weeks.

MrHaiCoder, 2023-04-01:
Now esp32 c6 has idf branch can support sdio up ?
Our side of the pcb has been laid out, I hope to be able to do basic testing soon

mantriyogesh, 2023-04-03:
Actually, I face timing issue with Raspberry Pi for ESP32-C6, so it is blocked for me.
But recently I leant that people are not facing such issues for other hosts.

If you are okay, I can give untested ESP32-C6 (because of above limitation) FG code patch to use and try. Please confirm.

If yes, GitHub IDF master is expected to be used for this. You can check [repo documentation](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg) ESP32-C6 'setup for SPI' for setting up IDF. Once IDF is setup you can directly build default transport SDIO for ESP32-C6 from patch using `idf.py flash monitor`

MrHaiCoder, 2023-04-26:
Our company has obtained this patch and is currently integrating it

MrHaiCoder, 2023-04-27:
We used this patch. The host sdio is configured in 4-wire mode with 25M transmission rate. Both sdio data and cmd have 47k pull-up resistor.
In ESP_ The init function includes the SDIO detection interface provided by our active chip;
The device can be identified, and the log is as follows;
Question: esp_read_reg  fail

[root@Zeratul:mnt]# insmod esp32_sdio.ko 
[    9.664075] ingenic,sdhci 13070000.msc: card insert manually
[    9.670113] ------------------------------------------[mmc_rescan][2573]
[root@Zeratul:mnt]# [    9.758720] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[    9.767598] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[    9.774642] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[    9.782733] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    9.788628] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[    9.794380] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[    9.801501] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    9.807269] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[    9.812998] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[    9.819285] mmc1: queuing unknown CIS tuple 0x91 (3 bytes)
[    9.825047] mmc1: new SDIO card at address 0001
[    9.830741] *** esp_probe: ESP network device detected ***
[   11.236103]  [init_context][333] 
[   11.239759] esp_sdio: probe of mmc1:0001:1 failed with error -145
[   11.246626] esp_sdio: probe of mmc1:0001:2 failed with error -22

[image]

mantriyogesh, 2023-04-27:
that is timed out.
Can you also use resetpin?

mantriyogesh, 2023-04-27:
so may I know you are using ESP32 or ESP32-C6 now?

MrHaiCoder, 2023-04-27:
esp32 c6，We have received the patch provided by your company
ESP32-C6_SDIO_FG_support.patch

mantriyogesh, 2023-04-27:
I suppose you are referring to 
[2023.04.03_ESP32-C6_FG_support.tgz](https://github.com/espressif/esp-hosted/files/11343171/2023.04.03_ESP32-C6_FG_support.tgz)

As the readme says while sending the patch, the code is not tested.
However later it had failed on Raspberry Pi (but there was possibility of issues as we were using wired setup).

We are still working to get it working with proper PCB with Raspberry Pi.

If this doesn't work, it will need development attention and might consume lot time.
I would suggest to wait till it is fully supported

mantriyogesh, 2023-04-27:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_sdio_slave_protocol.html#sending-fifo

PKT_LEN is getting failed , that hints on of:
1. Packet (ESP->Host) from slave is not correctly queued to sdmmc hardware driver
2. SDIO timing is not correct

I strongly suggest to wait for further integration of sdio into master for c6.

mantriyogesh, 2023-04-27:
https://github.com/espressif/esp-hosted/issues/217#issuecomment-1493551506 this is unfortunately failed.

we are checking in sdio driver for possible issues, but may take some time to resolve.

MrHaiCoder, 2023-05-04:
Do you currently have a device that can verify the relevant code?
When can you possibly integrate SDIO for ESP c6?

mantriyogesh, 2023-07-07:
Hello @MrHaiCoder ,

JFYI, ESP32-C6 support for SDIO is added for Linux in master under 391cb8efe593812cf1e7dc8c4e1c9302d40cc18f