espressif/esp-hosted #220: kernel 4.14 and version 1.0.2, sdio problem

yihaili, 2023-04-17:
The kernel sdio printing information is as follows：
```
sun8iw20-pinctrl 2000000.pinctrl: 2000000.pinctrl supply vcc-pc not found, using dummy regulator
sunxi-mmc 4021000.sdmmc: SD/MMC/SDIO Host Controller Driver(v4.20 2021-05-12 11:02)
sunxi-mmc 4021000.sdmmc:  ctl-spec-caps ：8
sunxi-mmc 4021000.sdmmc: No vmmc regulator found
sunxi-mmc 4021000.sdmmc: No vqmmc regulator found
sunxi-mmc 4021000.sdmmc: No vdmmc regulator found
sunxi-mmc 4021000.sdmmc: No vd33sw regulator found
sunxi-mmc 4021000.sdmmc: No vd18sw regulator found
sunxi-mmc 4021000.sdmmc: No vq33sw regulator found
sunxi-mmc 4021000.sdmmc: No vq18sw regulator found
sunxi-mmc 4021000.sdmmc: Cann't get pin bias hs pinstate,check if needed
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 0Hz bm PP pm UP vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: no vqmmc,Check if there is regulator
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: detmode:gpio polling
ashmem: initialized
sunxi-mmc 4021000.sdmmc: smc 1 p1 err, cmd 52, RTO !!
exFAT: Version 1.3.0
sunxi-mmc 4021000.sdmmc: smc 1 p1 err, cmd 52, RTO !!
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 0Hz bm PP pm OFF vdd 0 width 1 timing LEGACY(SDR12) dt B
esp_reset, ESP32: Triggering ESP reset.
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 0Hz bm PP pm UP vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: no vqmmc,Check if there is regulator
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: smc 1 p1 err, cmd 52, RTO !!
sunxi-mmc 4021000.sdmmc: smc 1 p1 err, cmd 52, RTO !!
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing SD-HS(SDR25) dt B
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 50000000Hz bm PP pm ON vdd 21 width 1 timing SD-HS(SDR25) dt B
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 50000000Hz bm PP pm ON vdd 21 width 4 timing SD-HS(SDR25) dt B
mmc1: queuing unknown CIS tuple 0x80 (1 bytes)file or directory
mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
mmc1: new high speed SDIO card at address 0001
esp_probe: ESP network device detected
esp_sdio: probe of mmc1:0001:2 failed with error -22
```

After the driver loading is completed, many events are received:
```
Received ESP bootup event
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x1
Capabilities: 0x1. Features supported are:
	 * WLAN on SDIO
EVENT: 1
Length not matching to firmware data size
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
[... 128 lines cut ...]
I (706) wifi_init: tcp mbox: 6
I (708) wifi_init: tcp tx win: 5744
I (712) wifi_init: tcp rx win: 5744
I (714) wifi_init: tcp mss: 1440
I (718) wifi_init: WiFi IRAM OP enabled
I (722) wifi_init: WiFi RX IRAM OP enabled
I (726) BTDM_INIT: BT controller compile version [2c56073]
I (734) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1046) FW_MAIN: ESP Bluetooth MAC addr: b4-8a- a-2f-ba-36

I (1046) FW_SDIO_SLAVE: Using SDIO interface
E (1046) FW_SDIO_SLAVE: SDIO timing: 1

I (1052) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1296) FW_MAIN: Start Data Path
I (2058) FW_SDIO_SLAVE: last reset cause: 1x
I (2058) FW_MAIN: Initial set up done
I (43490) FW_MAIN: INIT Interface command

I (43494) wifi:mode : sta (b4:8a:0a:2f:ba:34)
I (43496) wifi:enable tsf
I (43496) FW_CMD: Wifi Sta mode set

I (43538) FW_MAIN: Get MAC command
```

mantriyogesh, 2023-04-17:
Hello @yihaili 

This issue is known but I will be able to build the binaries tomorrow for 1.0.2.

mantriyogesh, 2023-04-18:
[2023.04.18_NG_FW2_latest.tgz](https://github.com/espressif/esp-hosted/files/11258158/2023.04.18_NG_FW2_latest.tgz)

mantriyogesh, 2023-04-18:
[2023.04.18_NG_FW2_latest.tgz](https://github.com/espressif/esp-hosted/files/11258196/2023.04.18_NG_FW2_latest.tgz)

mantriyogesh, 2023-04-18:
Please test on the binaries provided if the issue still observed.
As reference from your earlier log, you can use firmware binaries for ESP SDIO timing adjusted binaries: 2023.04.18_NG_FW2_latest/esp_bins/Timing_change_N_P/

yihaili, 2023-04-18:
Hello @mantriyogesh
I have tried both firmwares and found that the previous issue still persists.
After the driver was loaded, I kept receiving these events continuously for a while before they stopped.Then, using the ifconfig -a command, I was able to see the espsta0 device. After using wpa_supplicant -D nl80211 -i espsta0 -c /root/wifi.conf, the ESP device keeps printing "FW_MAIN: Scan request".

### Logs for Linux:
```
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 0Hz bm PP pm UP vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: no vqmmc,Check if there is regulator
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: smc 1 p1 err, cmd 52, RTO !!
sunxi-mmc 4021000.sdmmc: smc 1 p1 err, cmd 52, RTO !!
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 25000000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
sunxi-mmc 4021000.sdmmc: sdc set ios:clk 25000000Hz bm PP pm ON vdd 21 width 4 timing LEGACY(SDR12) dt B
mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
------run rc.modules file-----mmc1: queuing unknown CIS tuple 0x82 (1 bytes)

mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
mmc1: new SDIO card at address 0001
esp_probe: ESP network device detected
esp_sdio: probe of mmc1:0001:2 failed with error -22
------run rc.final file-----
/etc/rc.common: line 128: /usr/bin/sf_4p_software: not found
/etc/rc.common: line 128: /etc/init.d/sf_check_app_alive.sh: not found
udhcpc: started, v1.27.2
libphy: 4500000.eth: probed
sunxi-gmac 4500000.eth eth0: eth0: Type(7) PHY ID 02430c54 at 0 IRQ poll (4500000.eth-0:00)
udhcpc: sending discover

Received ESP bootup event
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x1
Capabilities: 0x1. Features supported are:
         * WLAN on SDIO
EVENT: 1
Length not matching to firmware data size
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
Capabilities: 0x0. Features supported are:
EVENT: 0
ESP peripheral capabilities: 0x0
[... 407 lines cut ...]
I (222327) FW_MAIN: Scan request

I (224511) FW_MAIN: Scan request

I (226695) FW_MAIN: Scan request

I (228879) FW_MAIN: Scan request

I (231063) FW_MAIN: Scan request

I (231999) FW_MAIN: Scan request

I (234183) FW_MAIN: Scan request

I (235119) FW_MAIN: Scan request

I (237303) FW_MAIN: Scan request

I (239487) FW_MAIN: Scan request

I (241671) FW_MAIN: Scan request

I (243855) FW_MAIN: Scan request
```
I am not sure where the problem still exists.

mantriyogesh, 2023-04-18:
What is the host side git commit?

yihaili, 2023-04-18:
81778f63a9d84fa46f8a907078f9e2681b391bf5

mantriyogesh, 2023-04-18:
For 'FW_MAIN: Scan request'

$ sudo killall wpa_supplicant
$ sudo killall -9 wpa_supplicant
Re-run wpa_supplicant with additional debug logs to understand what is the problem.
You can add `-d`  (may be more ds like -dd -ddd) for additional debug logs.

Most possible reasons are the SSID you had configured in /root/wifi.conf is not visible or incorrect password.
Anyway wpa supplicant logs should show you the exact issue.

yihaili, 2023-04-18:
Is it possible that it's a problem with the antenna?

yihaili, 2023-04-21:
Good morning @mantriyogesh
I tried changing a board for testing and was surprised to find that it worked properly. I guess there must be a difference between the two boards, or there may be some issues with the boards.
```
Linux version 5.4.61 (lyh@liyihai) (riscv64-unknown-linux-gnu-gcc (C-SKY RISCV Tools V1.8.4 B20200702) 8.1.0, GNU ld (GNU Binutils) 2.32) #180 PREEMPT Thu Apr 20 07:08:28 UTC 2023
Zone ranges:
  DMA32    [mem 0x0000000040000000-0x0000000043ffffff]
  Normal   empty
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000040000000-0x0000000043ffffff]
Initmem setup node 0 [mem 0x0000000040000000-0x0000000043ffffff]
On node 0 totalpages: 16384
  DMA32 zone: 224 pages used for memmap
  DMA32 zone: 0 pages reserved
  DMA32 zone: 16384 pages, LIFO batch:3
elf_hwcap is 0x20112d
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0 
Built 1 zonelists, mobility grouping on.  Total pages: 16160
Kernel command line: earlyprintk=sunxi-uart,0x02500000 clk_ignore_unused initcall_debug=0 console=ttyS0,115200 loglevel=8 root=/dev/mmcblk0p5 init=/pseudo_init partitions=boot-resource@mmcblk0p1:env@mmcblk0p2:env-redund@mmcblk0p3:boot@mmcblk0p4:rootfs@mmcblk0p5:recovery@mmcblk0p6:rootfs_data@mmcblk0p7:UDISK@mmcblk0p8 cma=0M snum= mac_addr= wifi_mac= bt_mac= specialstr= gpt=1 androidboot.mode=normal androidboot.hardware=sun20iw1p1 boot_type=2 androidboot.boot_type=2 gpt=1 uboot_message=2018.05-ga81c5b4(11/12/2021-03:30:42) mbr_
Dentry cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
Inode-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
Sorting __ex_table...
mem auto-init: stack:off, heap alloc:off, heap free:off
Memory: 56284K/65536K available (4398K kernel code, 379K rwdata, 1092K rodata, 140K init, 218K bss, 9252K reserved, 0K cma-reserved)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
rcu: Preemptible hierarchical RCU implementation.
	Tasks RCU enabled.
rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
plic: mapped 200 interrupts with 1 handlers for 2 contexts.
riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
sched_clock: 64 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
riscv_timer_clockevent depends on broadcast, but no broadcast function available
clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
ASID allocator initialised with 65536 entries
rcu: Hierarchical SRCU implementation.
devtmpfs: initialized
random: get_random_u32 called from 0xffffffe00019dc7e with crng_init=0
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: 0, 6144 bytes, linear)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic allocations
cpuidle: using governor menu
rtc_ccu: sunxi ccu init OK
clock: sunxi ccu init OK
clock: sunxi ccu init OK
iommu: Default domain type: Translated 
sunxi iommu: irq = 4
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Bluetooth: Core ver 2.22
NET: Registered protocol family 31
Bluetooth: HCI device and connection manager initialized
Bluetooth: HCI socket layer initialized
Bluetooth: L2CAP socket layer initialized
pwm module init!
clocksource: Switched to clocksource riscv_clocksource
sun8iw20-pinctrl 2000000.pinctrl: 2000000.pinctrl supply vcc-pd not found, using dummy regulator
sun8iw20-pinctrl 2000000.pinctrl: initialized sunXi PIO driver
NET: Registered protocol family 2
tcp_listen_portaddr_hash hash table entries: 256 (order: 0, 4096 bytes, linear)
[... 230 lines cut ...]
root@TinaLinux:/# esp_cfg80211_scan
esp_cfg80211_authenticate
Authentication request: 8c:53:c3:c3:50:8a 1 0 0 0
esp_cfg80211_associate
Association request: 8c:53:c3:c3:50:8a 1 33
Connection status: 0
process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
NOHZ: local_softirq_pending 08
process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
NOHZ: local_softirq_pending 08
esp_cfg80211_add_key
esp_cfg80211_add_key

root@TinaLinux:/# 
root@TinaLinux:/# 
root@TinaLinux:/# udhcpc -i espsta0
udhcpc: started, v1.27.2
udhcpc: sending discover
udhcpc: sending select for 192.168.16.159
udhcpc: lease of 192.168.16.159 obtained, lease time 43200
udhcpc: ifconfig espsta0 192.168.16.159 netmask 255.255.255.0 broadcast 192.168.16.255
esp_inetaddr_event: NETDEV_UP interface espsta0 ip changed to  192.168.016.159 
udhcpc: setting default routers: 192.168.16.1

```

mantriyogesh, 2023-04-21:
That's great.
Is the new board same as your expected target board?

mantriyogesh, 2023-04-24:
i hope if everything working as expectation, we can close this issue?

yihaili, 2023-04-25:
Yes, thank you

closed 2023-04-25 by yihaili