espressif/esp-hosted #223: SPI throughput performance

gdaneels, 2023-04-24:
Hi,

We managed to get esp-hosted-**first-generation** working with a Linux host (`iMX6ULL, kernel 5.4.210+gf2663e3, yocto build (dunfell)`), both with the ESP32-S3 and the ESP32-C3. However, the wireless performance is not great: when I use the Linux host as (Soft)AP or station and I make a connection with another laptop (either as station or AP), I get an iperf3 throughput only up to 6 or 7 mbps (both for TCP and UDP). `SPI_CLK_MHZ` is set to 28 MHz (at 30 MHz, communication did not work anymore), SPI mode is set to 2. 

I also did the raw throughput SPI test, that is included in the first generation source code, and with that I get similar SPI throughput performance of 7 - 8 mbps.

Measured at host (sending from ESP32 to host):
```
[   80.405399] 13-14 sec       7927 kbits/sec
[   80.405399] 
[   81.445400] 14-15 sec       7893 kbits/sec
[   81.445400] 
[   82.485402] 15-16 sec       7927 kbits/sec
[   82.485402] 
[   83.525397] 16-17 sec       7938 kbits/sec
[   83.525397] 
[   84.565401] 17-18 sec       7870 kbits/sec
```

Looking at the figures in the documentation, I was expecting a wireless throughput performance (and for sure an SPI throughput) of around 13 up to 16 mbps. SPI and wireless throughput below 8 mbps is below expectation, I guess?

My IDF version is `ESP-IDF v4.4.4-148-g4c2afac355`, working on the latest esp-hosted master branch.

I have included my .dbt device tree: [imx6ull-14x14-evk.txt](https://github.com/espressif/esp-hosted/files/11311275/imx6ull-14x14-evk.txt).

Any ideas how to speed up the connection or at least debug it to know the exact problem?

Thanks in advance.

mantriyogesh, 2023-04-24:
Hello @gdaneels 

That is so great that you have tested raw throughput test. This is very clear idea that the transport is max performance. Thanks a lot, for reading the open issues and getting hands dirty before opening issue.

Is it possible that you have logic analyzer with you?

Anyway before jumping to conclusions, I think it is better to get something as base.

No worries even if you don't have analyzer, then we can try manual technique to understand better.

gdaneels, 2023-04-24:
I don't have a logic analyzer at my disposal right now, but I will try to make work of it in the coming days. In the meanwhile, is it possible to already continue with the manual technique to understand better?

mantriyogesh, 2023-04-24:
Actually, we could have checked the clock (Freq and distribution) and cs pattern which generally give very clear picture.

Now, I think huge bunch of tests -
1. How raw output other way round?
2. Any difference when  spi mode , specially spi mode 1 and 3 both sides, also when mode mismatched
3. Do you see same behaviour if cs changed or altogether other spi instance used on host
4. Input power is exact what is expected to be? Use exact power adaptor and wore which SoC came with or sold with? Less than perfect power hurts peripheral performance first.
5. Worth to see exactly similar issue of clock on raspberry Pi we faced -> 
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#12-raspberry-pi-software-setup
This is RaspberryPi specific https://github.com/raspberrypi/linux/issues/2286.

gdaneels, 2023-04-25:
Hi @mantriyogesh,

I did manage to get my hands on a (low-cost, 24MHz) logic analyzer. I've put the `SPI_CLK_MHZ` to 12 MHz because of the limited sample rate of 24 MHz of my logic analyzer. I've analyzed two scenario's:

- from ESP32 to host: [image]
- from host to ESP32: [image]

Complete sigrok Pulseview logs: [logic-analyzer.zip](https://github.com/espressif/esp-hosted/files/11323053/logic-analyzer.zip). Can you view these?

In both scenario's, the SPI throughput is around 6 mbps. Do you have any reference throughput you get at 12 MHz SPI clock speed?

You have specific setups in mind I should test with the logic analyzer?

I will try the other things you suggested further today or tomorrow.

mantriyogesh, 2023-04-25:
Okay.. I'll check these tomorrow morning and get back to you

gdaneels, 2023-04-25:
> Okay.. I'll check these tomorrow morning and get back to you

Thanks. BTW, these are raw (short) output snippets: [esp32-to-host.txt](https://github.com/espressif/esp-hosted/files/11323176/esp32-to-host.txt), [host-to-esp32.txt](https://github.com/espressif/esp-hosted/files/11323177/host-to-esp32.txt)

Isn't it also strange that in raw output logs from ESP32 to host, the kbit/sec is always higher than what is received at the host side? I see that behavior consistently when sending data from the ESP32 to iMX6.

mantriyogesh, 2023-04-26:
I did not example understand. Can you please help point me to lines in log above logs?

mantriyogesh, 2023-04-26:
why is your clock discontinuous?
It wastes time domain and effective frequency will be smaller.
[image]

gdaneels, 2023-04-26:
Hi @mantriyogesh, thanks for your answer.

> I did not exactly understand. Can you please help point me to lines in log above logs?

So I did 2 raw SPI throughput tests at `SPI_CLK_MHZ` to 12 MHz: 

- from ESP32 to host, by setting `TEST_RAW_TP__ESP_TO_HOST` to 1. These logs are the (part) of the output of that test: [esp32-to-host.txt](https://github.com/espressif/esp-hosted/files/11323176/esp32-to-host.txt). Similarly, the _first_ image in my reaction above belongs to this test. In the logic-analyzer.zip file, the _esp-to-host-12mhz-24mhz-10.sr_ log belongs to this test.
- from host to ESP32 to host, by setting `TEST_RAW_TP__ESP_TO_HOST` to 0. These logs are (part) of the output of that test: [host-to-esp32.txt](https://github.com/espressif/esp-hosted/files/11323177/host-to-esp32.txt). Similarly, the _second_ image in my reaction above belongs to this test. In the logic-analyzer.zip file, the _host-to-esp-12mhz-24mhz-10.sr_ log belongs to this test.

> why is your clock discontinuous?
It wastes time domain and effective frequency will be smaller.

Mmh, good question. The expected behavior would be that the clock is continuous from the moment the CS line goes up, right? Would this be a problem caused by the host?

Do you perhaps have a reference logic analyzer trace of a raw SPI throughput test to compare to?

mantriyogesh, 2023-04-26:
issue 224 also has logs of clock.
But anyway base understanding is that clock will not be idle unless the CS transitions happen.
CS should toggle only after said buffer size (1600 bytes) transfer is over.

yes, you might have to check why the host spi driver is doing this.
There are cases in host, sometimes the configurations include dummy symbols or discontinuous clocks added.

Apart from discontinuous clock I don't thing there are any issues in your graph.

We had introduced cs_change variable in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#124-peripheral-configurations
for ESP32 as we had observed timing issues while working with Raspberry Pi (only for ESP32 chipset no other ESP chipsets like C2/C3 etc).
worth to disable it in general. But anyway clock spacing looks a lot more than that cs_change should introduce.

Just to let you know, in your case cs_change is **not in effect** as you are **not** using ESP32. this is just to let you know how configs could affect. Also, ESP32 even with cs_change, performed 7-8Mbps on 10MHz clock on Raspberry Pi.

mantriyogesh, 2023-05-02:
Hello @gdaneels ,

Any further findings on this?

gdaneels, 2023-05-02:
Hi @mantriyogesh, in the meanwhile we have investigated this discontinuous clock in more detail. It seems to be an issue of the ECSPI controller on the iMX6 and its DMA implementation, introducing these gaps after 8 bits. After finding this out, I also disabled the DMA usage of the ECSPI controller on the iMX6 (after which the clock was more continuous) and that increased the SPI throughput test to 9 Mbits/s (at 28 MHz clock): still a lot room for improvement I would guess. Still further investigating the issue.

closed 2023-10-11 by gdaneels