espressif/esp-hosted #224: Command[1] timed out - Jetson Nano porting

wizath, 2023-04-25:
Simillar issue to https://github.com/espressif/esp-hosted/issues/191

Hi,

I'm trying to port ESP Hosted NG on Jetson Nano, ATM using esp32 via SPI.
Here are my modifications: https://github.com/wizath/esp-hosted

Brief:
```
[   25.715117] esp32: Command[3] timed out
[   25.719735] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   25.727674] esp32: add_network_iface() failure, ret: -1
[   25.733107] network iterface init failed
```
It switches around command[0] and command[3], sometimes it can boot UP and go till the "Scan Request"

=== Success logs ===
Kernel
```
[   41.512324] esp32_spi: loading out-of-tree module taints kernel.
[   41.528911] esp_reset, ESP32: Triggering ESP reset.
[   42.549123] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[   43.972063] spi drdy handler
[   43.972191] spi interrupt handler
[   43.972205] starting rx
[   43.973882] stopping rx
[   43.973888] next rx pkt->
[   43.973896] spi_rx: ffffffc0e01b5240: 03 00 00 00 1d 00 0c 00 73 01 00 00 01 00 16 00  ........s.......
[   43.983290] spi interrupt handler
[   43.984336] spi_rx: ffffffc0e01b5250: 15 00 00 00 03 01 00 02 01 0a 00 01 f8 04 01 00  ................
[   43.993873] spi_rx: ffffffc0e01b5260: 01 07 01 00 02 01 00 00 00                       .........
[   44.002692] 
               Received ESP bootup event
[   44.002697] EVENT: 3
[   44.002698] EVENT: 2
[   44.002700] EVENT: 0
[   44.002702] EVENT: 4
[   44.002704] EVENT: 1
[   44.002707] esp32: process_fw_data ESP chipset's last reset cause: 
[   44.002710] POWERON_RESET
[   44.002713] esp32: ESP Firmware version: 1.0.2
[   44.002716] ESP chipset detected [esp32]
[   44.003010] sending init iface
[   44.018504] esp32: Processing Command [0x1]
[   44.018519] Next TX pkt
[   44.018524] spi_tx: ffffffc0e01b4880: 00 00 01 00 08 00 0c 00 16 00 00 00 01 00 00 00  ................
[   44.028684] spi_tx: ffffffc0e01b4890: 00 00 00 00                                      ....
[   44.038507] spi interrupt handler
[   44.038582] process_rx_buf:325--header->if_type err:15
[   44.038588] stopping rx
[   44.038705] spi interrupt handler
[   44.046888] spi drdy handler
[   44.046912] starting rx
[   44.048297] process_rx_buf:325--header->if_type err:15
[   44.048303] stopping rx
[   44.048328] spi interrupt handler
[   44.048345] starting rx
[   44.049723] spi interrupt handler
[   44.049738] stopping rx
[   44.049744] next rx pkt->
[   44.049750] spi_rx: ffffffc0e01b2080: 00 00 02 00 08 00 0c 00 00 00 00 00 01 02 00 00  ................
[   44.059254] spi_rx: ffffffc0e01b2090: 00 00 00 00                                      ....
[   44.067795] Resp for command [1]
[   44.086524] esp32: Processing Command [0x3]
[   44.086552] Next TX pkt
[   44.086557] spi_tx: ffffffc0e01b16c0: 00 00 01 00 08 00 0c 00 18 00 00 00 03 00 00 00  ................
[   44.096402] spi_tx: ffffffc0e01b16d0: 00 00 00 00                                      ....
[   44.108456] process_rx_buf:325--header->if_type err:15
[   44.108461] stopping rx
[   44.108470] spi interrupt handler
[... 479 lines cut ...]
I (705) wifi_init: tcp rx win: 5744
I (707) wifi_init: tcp mss: 1440
I (711) wifi_init: WiFi IRAM OP enabled
I (715) wifi_init: WiFi RX IRAM OP enabled
I (719) BTDM_INIT: BT controller compile version [425d342]
I (725) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1051) FW_MAIN: ESP Bluetooth MAC addr: 3c-71-bf-59-2b-36

I (1051) FW_SPI: Using SPI interface
I (1051) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1059) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2067) FW_MAIN: Initial set up done
I (2141) FW_MAIN: INIT Interface command

I (2143) wifi:mode : sta (3c:71:bf:59:2b:34)
I (2143) wifi:enable tsf
I (2143) FW_CMD: Wifi Sta mode set

I (2193) FW_MAIN: Get MAC command
```

Logic analyzer:
[10mhz_spi2_error3.vcd.zip](https://github.com/espressif/esp-hosted/files/11320812/10mhz_spi2_error3.vcd.zip)

Any ideas?

mantriyogesh, 2023-04-25:
how to open vcd file?
Also, can you please add reset pin in your graphs (some times CS/SPI transactions show down suspicious when reset pin is actually applied).

Can you try to use SPI 1 or 3 both sides? 

You can actually check raw throughput performance using https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md . this way will get to know if base transport is correctly working or not.

wizath, 2023-04-25:
I've tried all SPI modes 0-3, no visible change in any mode.
I'll do the speed test in a second.

VCD file can be opened by https://sigrok.org/wiki/PulseView. If that helps I can export in csv

wizath, 2023-04-25:
Testing RAW TP, there is simillar issue. 
```
sudo c_support/test.out get_sta_mac_addr
Enable heartbeat with duration 20
Control response timed out after 30 sec
Response not received
```

ESP confirm that command was sent, but the response never arrives, thus there is timeout
```
20-21 sec       0.00 kbits/sec
I (18249) slave_ctrl: HB timer started for 20 sec
```

This is 
Can this be caused by this weird effect when DR (CH4) goes down, then the channels 0-1-3 gets this little transition???
[image]

wizath, 2023-04-25:
I was wondering if that 0xFF at the start of the transmission is not throwing the whole packet out due to
header error

```
[ 1995.412362] śro, 26 kwi 2023, 00:04:00 CEST
[ 2004.089900] Next TX pkt
[ 2004.089910] spi_tx: ffffffc0f47a9e40: 02 00 17 00 0c 00 14 05 01 00 00 00 01 08 00 63  ...............c
[ 2004.099481] spi_tx: ffffffc0f47a9e50: 74 72 6c 52 65 73 70 02 09 00 10 79 ca 07 04 08  trlResp....y....
[ 2004.109616] spi_tx: ffffffc0f47a9e60: 01 10 14                                         ...
[ 2004.132065] spi interrupt handler
[ 2004.133807] spi interrupt handler
[ 2004.133834] spi_rx: ffffffc098e14840: ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.143641] spi_rx: ffffffc098e14850: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.153945] spi_rx: ffffffc098e14860: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.163887] spi_rx: ffffffc098e14870: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.173690] spi_rx: ffffffc098e14880: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.183460] spi_rx: ffffffc098e14890: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.193027] spi_rx: ffffffc098e148a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.202508] spi_rx: ffffffc098e148b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.211970] spi_rx: ffffffc098e148c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.221442] spi_rx: ffffffc098e148d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 2004.230898] spi_rx: ffffffc098e148e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

wizath, 2023-04-25:
Once-in-a-lifetime proper startup event
```
[kwi26 00:24] esp_reset, ESP32: Resetpin of Host is 200
[  +0,000223] esp_reset, ESP32: Triggering ESP reset.
[  +1,021416] ESP: SPI host config: GPIOs: Handshake[194] DataReady[14]
[  +0,001496] ESP host driver claiming SPI bus [0],chip select [0] with init SPI Clock [1]
[  +0,257629] IPv6: ADDRCONF(NETDEV_UP): ethsta0: link is not ready
[  +0,010163] IPv6: ADDRCONF(NETDEV_UP): ethap0: link is not ready
[ +35,578639] esp_reset, ESP32: Resetpin of Host is 200
[  +0,000223] esp_reset, ESP32: Triggering ESP reset.
[  +1,021902] ESP: SPI host config: GPIOs: Handshake[194] DataReady[14]
[  +0,000847] ESP host driver claiming SPI bus [0],chip select [0] with init SPI Clock [1]
[  +0,170675] spi drdy handler
[  +0,000281] spi interrupt handler
[  +0,000172] starting rx
[  +0,013692] spi interrupt handler
[  +0,001818] spi_rx: ffffffc0de23f840: 04 00 0e 00 0c 00 2e 01 00 00 00 00 00 0c 02 01  ................
[  +0,000046] spi interrupt handler
[  +0,010503] spi_rx: ffffffc0de23f850: 00 01                                            ..
[  +0,009318] skb len 1600 offset 12 struct 12
[  +0,000200] stopping rx
[  +0,000030] 
              Received INIT event from ESP32 peripheral
[  +0,000018] EVENT: 2
[  +0,000013] EVENT: 1
[  +0,000043] EVENT: 0
[  +0,000015] ESP peripheral capabilities: 0xf8
[  +0,047808] IPv6: ADDRCONF(NETDEV_UP): ethsta0: link is not ready
[  +0,006594] IPv6: ADDRCONF(NETDEV_UP): ethap0: link is not ready
[  +0,012204] spi drdy handler
[  +0,000024] starting rx
[  +0,013167] spi interrupt handler
[  +0,000068] spi_rx: ffffffc0de23e840: ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  +0,009382] spi interrupt handler
[  +0,000094] spi_rx: ffffffc0de23e850: 00 00                                            ..
[  +0,008358] fixing 0xff
[  +0,000005] skb len 1600 offset 0 struct 12
[  +0,000003] process_rx_buf:247--offset err:0
[  +0,000004] stopping rx
[  +0,114145] EVENT: 3
[  +0,000005] ESP peripheral capabilities: 0x1
[  +0,000002] esp32: start testing of Host->ESP raw throughput
```

There is clearly some corruption in received frames, since
```
[kwi26 00:27] esp_reset, ESP32: Resetpin of Host is 200
[  +0,000219] esp_reset, ESP32: Triggering ESP reset.
[  +1,030469] ESP: SPI host config: GPIOs: Handshake[194] DataReady[14]
[  +0,000885] ESP host driver claiming SPI bus [0],chip select [0] with init SPI Clock [1]
[  +0,147340] spi drdy handler
[  +0,000283] spi interrupt handler
[  +0,000178] starting rx
[  +0,013964] spi interrupt handler
[  +0,001025] spi interrupt handler
[  +0,000084] spi_rx: ffffffc0de23b840: 04 00 0e 00 0c 00 2e 01 00 00 00 00 00 0c 02 01  ................
[  +0,010356] spi_rx: ffffffc0de23b850: 00 01                                            ..
[  +0,009084] skb len 1600 offset 12 struct 12
[  +0,000065] process_rx_buf:269--data_path err
```

INIT EVENT is corrupted

wizath, 2023-04-25:
Okay, got some progress

By dividing
```
...
data_path = 1;

status = request_irq(SPI_IRQ, spi_interrupt_handler,
                IRQF_SHARED | IRQF_TRIGGER_RISING,
                "ESP_SPI", spi_context.esp_spi_dev);
....
        atomic_set(&tx_pending, 0);
        msleep(200);

        return 0;
```

Software now consistently boots and starts throughput test regardless of SPI speed

wizath, 2023-04-25:
Kernel:
```
[kwi26 00:53] 29-30 sec       923 kbits/sec
              
[  +1,023968] 30-31 sec       923 kbits/sec         
[  +1,023971] 31-32 sec       923 kbits/sec         
[  +1,023969] 32-33 sec       923 kbits/sec
```

ESP
```
30-31 sec       844.06 kbits/sec
31-32 sec       844.06 kbits/sec
32-33 sec       832.66 kbits/sec
33-34 sec       809.84 kbits/sec
```

wizath, 2023-04-25:
ESP->HOST TP TEST

Kernel - there are some offset errors happening, but the transfer is still going
```
              
[  +1,023622] 65-66 sec       832 kbits/sec
              
[  +1,023992] 66-67 sec       878 kbits/sec
              
[  +0,800080] process_rx_buf:247--offset err:32929
[  +0,227931] 67-68 sec       866 kbits/sec
              
[  +0,766361] process_rx_buf:247--offset err:49232
[  +0,253628] 68-69 sec       866 kbits/sec
              
[kwi26 01:40] 69-70 sec       878 kbits/sec
              
[  +1,020003] 70-71 sec       855 kbits/sec
              
[  +0,398465] process_rx_buf:247--offset err:49232
[  +0,625523] 71-72 sec       855 kbits/sec
              
[  +0,677928] process_rx_buf:247--offset err:49232
[  +0,346072] 72-73 sec       866 kbits/sec
              
[  +1,024009] 73-74 sec       855 kbits/sec
              
[  +1,023992] 74-75 sec       878 kbits/sec
         
```

ESP
```
65-66 sec       1564.00 kbits/sec
66-67 sec       1564.00 kbits/sec
67-68 sec       1610.00 kbits/sec
68-69 sec       1725.00 kbits/sec
69-70 sec       1725.00 kbits/sec
70-71 sec       1713.50 kbits/sec
71-72 sec       1713.50 kbits/sec
72-73 sec       1748.00 kbits/sec
73-74 sec       1690.50 kbits/sec
74-75 sec       1713.50 kbits/sec
75-76 sec       1713.50 kbits/sec
```

mantriyogesh, 2023-04-26:
https://github.com/espressif/esp-hosted/issues/224#issuecomment-1522482013
in this case I don't see [next rx pkt->](https://github.com/espressif/esp-hosted/compare/master...wizath:esp-hosted:master#diff-376c997542ed8caae30ab0a0d30cbff66a894804f1e92e8dcee9b1cc6aa3fcfdR613) trace. I am surprised why this was not printed.

mantriyogesh, 2023-04-26:
yes data_path variable actually doesn't make much sense in Rx path. Rx path should always be open.
I will have to review this part of code, but anyway data_path being set before ISRs are registered, will unblock any Rx side check for data_path.

mantriyogesh, 2023-04-26:
https://sigrok.org/wiki/Downloads#Mac_OS_X
dmg is not working so I am forced to build from source (which failing step by step!).
I will try to get it install to see the graphs

mantriyogesh, 2023-04-26:
how did you manage to change the SPI mode at ESP32?

You do not have any binary with changed SPI mode for NG.. Am I correct?

Possible to test FG with SPI mode changed at both ESP and Host side?

mantriyogesh, 2023-04-26:
https://github.com/espressif/esp-hosted/issues/224#issuecomment-1522500580
Position of printing log is incorrect.
actually, 0xff will be ignored at [if (header->if_type >= ESP_MAX_IF) {](https://github.com/espressif/esp-hosted/compare/master...wizath:esp-hosted:master#diff-3e2fe38f9ab60cd9ad7c5aeaa8c876a2e40c08d160c0913be0dc4754c3463440L227)

mantriyogesh, 2023-04-26:
https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L517 -> can add "esp_spi_tx" log or better here: https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L368

---

For Rx (after filtering) : https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L393

---
Before rx filtering at ESP: https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L407

---

In **FG**

SPI mode change:
ESP: https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L563
Host: https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L368

---

To comment readjust of freq: comment: https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L190

---

Master driver Freq: 

Change here https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L33 with your desired freq

basically, https://github.com/espressif/esp-hosted/blob/c926fc3fb4218b54e06508bd9f0c93848a0e1c5f/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L293-L295
will have no effect and you can control freq only through master (to try with lower freq)
----

Basically intention is to compare ESP-Tx and Linux-Rx if there is any bit shifting found.
Vice versa, Host-Tx and ESP-Rx. to be compared. These comparison to be done before filtering packet (as any such errors will lead to packet filtered and dropped)

mantriyogesh, 2023-04-26:
In your graphs, I could notice MISO and MOSI have different time transitions. some times they are in sync with clk and some time they shift.

Worth to try disabling data ready if any difference observed. check this comment: https://github.com/espressif/esp-hosted/issues/191#issuecomment-1517729636

wizath, 2023-04-26:
> https://sigrok.org/wiki/Downloads#Mac_OS_X dmg is not working so I am forced to build from source (which failing step by step!). I will try to get it install to see the graphs

You can also try
https://github.com/DreamSourceLab/DSView
this works for me on MacOSX

mantriyogesh, 2023-04-26:
> You can also try
> https://github.com/DreamSourceLab/DSView
> this works for me on MacOSX
I had checked your logs already: https://github.com/espressif/esp-hosted/issues/224#issuecomment-1522739232

mantriyogesh, 2023-04-26:
https://github.com/espressif/esp-hosted/issues/224#issuecomment-1522995497

yes yes. that is great approach to check raw data. Please check other comments above.

wizath, 2023-04-26:
OK, after a little cleaning from my desk (or several power-cycles of the board??) the ESP->HOST TP test run flawlessly, reglardless of multiple resets, without any errors, SPI_2 mode. Proceeding to debug the NG now

mantriyogesh, 2023-05-02:
Thats great. thanks for the update..

mantriyogesh, 2023-07-19:
Can you please close this issue?