espressif/esp-hosted #226: Raspberry PI 4B using Esp32_Wroom_32e as WLan card ， failed

HanJiec, 2023-05-05:
Pi OS: ubuntu 22 .kernel is 5.15
using SDIO only firmware 
SDIO pins  and reset pin are connected between Pi and Esp32.
CMD and DATA0~3 are pull-up by Rpi  gpio setting
 After modprobe bluetooth and cfg80211 , insmod esp32_sdio.ko resetpin=6.
But espsta0  not comes.
The Log is as the following:
[23133.927844] Enter esp_init
[23133.927876] gpio_request(6) ret 0
[23133.927884] gpio_direction_output ret 0
[23134.127982] esp_reset, ESP32: Triggering ESP reset.
[23134.335555] esp_sdio: probe of mmc1:0001:1 failed with error -110
[23134.776229] mmc1: card 0001 removed
[23136.074618] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[23136.082639] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[23136.085975] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[23136.091816] mmc1: new SDIO card at address 0001
[23136.093535] esp_probe: ESP network device detected
[23136.093775] Rx Pre ====== 0
[23136.093785] Rx Pos ======  0
[23136.093842] Tx Pre ======  0
[23136.093850] Tx Pos ======  10
[23136.544299]
               Received ESP bootup event
[23136.544325] EVENT: 3
[23136.544338] ESP chipset detected [esp32]
[23136.544349] EVENT: 0
[23136.544359] ESP peripheral capabilities: 0x1d
[23136.751267] ESP Bluetooth init
[23136.751783] Capabilities: 0x1d. Features supported are:
[23136.751800]   * WLAN on SDIO
[23136.751809]   * BT/BLE
[23136.751818]     - HCI over SDIO
[23136.751826]     - BT/BLE dual mode
[23136.751836] EVENT: 4
[23136.751846] EVENT: 1
[23136.751856] esp32: process_fw_data ESP chipset's last reset cause:
[23136.751866] POWERON_RESET
[23136.751875] esp32: ESP Firmware version: 1.0.2
[23141.847330] esp32: Command[1] timed out
[23141.847381] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[23141.847442] esp32: add_network_iface() failure, ret: -1
[23141.847462] network iterface init failed
ubuntu@ubuntu-desktop:~/esp32/esp-hosted/esp_hosted_ng/host$ sudo ifconfig espsta0 up
espsta0: ERROR while getting interface flags: No such device

HanJiec, 2023-05-05:
Pi OS: ubuntu 22 .kernel is 5.15
using SDIO only firmware
SDIO pins and reset pin are connected between Pi and Esp32.
CMD and DATA03 are pull-up by Rpi gpio setting
After modprobe bluetooth and cfg80211 , insmod esp32_sdio.ko resetpin=6.
But espsta0 not comes.
The Log is as the following:
[23133.927844] Enter esp_init
[23133.927876] gpio_request(6) ret 0
[23133.927884] gpio_direction_output ret 0
[23134.127982] esp_reset, ESP32: Triggering ESP reset.
[23134.335555] esp_sdio: probe of mmc1:0001:1 failed with error -110
[23134.776229] mmc1: card 0001 removed
[23136.074618] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[23136.082639] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[23136.085975] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[23136.091816] mmc1: new SDIO card at address 0001
[23136.093535] esp_probe: ESP network device detected
[23136.093775] Rx Pre ====== 0
[23136.093785] Rx Pos ======  0
[23136.093842] Tx Pre ======  0
[23136.093850] Tx Pos ======  10
[23136.544299]
               Received ESP bootup event
[23136.544325] EVENT: 3
[23136.544338] ESP chipset detected [esp32]
[23136.544349] EVENT: 0
[23136.544359] ESP peripheral capabilities: 0x1d
[23136.751267] ESP Bluetooth init
[23136.751783] Capabilities: 0x1d. Features supported are:
[23136.751800]   * WLAN on SDIO
[23136.751809]   * BT/BLE
[23136.751818]     - HCI over SDIO
[23136.751826]     - BT/BLE dual mode
[23136.751836] EVENT: 4
[23136.751846] EVENT: 1
[23136.751856] esp32: process_fw_data ESP chipset's last reset cause:
[23136.751866] POWERON_RESET
[23136.751875] esp32: ESP Firmware version: 1.0.2
[23141.847330] esp32: Command[1] timed out
[23141.847381] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[23141.847442] esp32: add_network_iface() failure, ret: -1
[23141.847462] network iterface init failed
ubuntu@ubuntu-desktop:~/esp32/esp-hosted/esp_hosted_ng/host$ sudo ifconfig espsta0 up
espsta0: ERROR while getting interface flags: No such device

mantriyogesh, 2023-05-09:
Hello @HanJiec 

We have tested Raspberry Pi With Raspberry Pi O, as it give full control and exposes /boot/config.txt to alter config.
Raspberry Pi OS is also based on debian.

Have you attached the [external pull-ups required](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#sd-pull-up-requirements)?

HanJiec, 2023-05-09:
Using Rpi internal pulls-ups  can works ? I didn't attach extern pull-ups

HanJiec, 2023-05-12:
Unfortunatively ， I cannot success。
If I add 5 pullups to CMD and DAT0~DAT3 each,the card cannot attached. the following log stops at  "card 0001 removed" 
[  665.475170] esp_init
[  665.475406] esp_reset, ESP32: Triggering ESP reset.
[  665.682735] esp_sdio: probe of mmc1:0001:1 failed with error -110
[  665.711024] mmc1: card 0001 removed
Then if  I remove the pullup of DAT2 the following log  voluntarily out :
[  688.226064] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[  688.234102] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[  688.237483] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[  688.243071] mmc1: new SDIO card at address 0001
[  688.243522] esp_probe: ESP network device detected
[  688.243778] Rx Pre ====== 0
[  688.243789] Rx Pos ======  0
[  688.243851] Tx Pre ======  0
[  688.243860] Tx Pos ======  10
[  689.124507]
               Received ESP bootup event
[  689.124533] EVENT: 3
[  689.124546] ESP chipset detected [esp32]
[  689.124557] EVENT: 0
[  689.124566] ESP peripheral capabilities: 0x1d
[  689.330724] ESP Bluetooth init
[  689.331216] Capabilities: 0x1d. Features supported are:
[  689.331232]   * WLAN on SDIO
[  689.331240]   * BT/BLE
[  689.331249]     - HCI over SDIO
[  689.331258]     - BT/BLE dual mode
[  689.331266] EVENT: 4
[  689.331276] EVENT: 1
[  689.331286] esp32: process_fw_data ESP chipset's last reset cause:
[  689.331296] RTCWDT_RTC_RESET
[  689.331305] esp32: ESP Firmware version: 1.0.2
[  694.510805] esp32: Command[1] timed out
[  694.510855] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[  694.510902] esp32: add_network_iface() failure, ret: -1
[  694.510920] network iterface init failed
Now DAT2 pullup is removed , I rmmod esp32_sdio: ubuntu@ubuntu-desktop:~/Esp32_20230509$ sudo rmmod esp32_sdio.ko
the follwing log out:
[  699.400545] mmc1: card 0001 removed
[  699.451844] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[  699.460103] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[  699.463456] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[  699.468754] mmc1: new SDIO card at address 0001
[  699.469223] esp_probe: ESP network device detected
[  699.469489] Rx Pre ====== 0
[  699.469499] Rx Pos ======  38
[  699.469696] Tx Pre ======  0
[  699.469706] Tx Pos ======  10
[  949.059340] esp_exit
and at state DAT2 pullup removed and re-insmod esp32_sdio,ko the following log out ：
[  986.664244] esp_init
[  986.664482] esp_reset, ESP32: Triggering ESP reset.
[  986.871756] esp_sdio: probe of mmc1:0001:1 failed with error -110
[  987.220584] mmc1: card 0001 removed
[  988.518921] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[  988.526680] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[  988.530047] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[  988.536342] mmc1: new SDIO card at address 0001
[  988.536903] esp_probe: ESP network device detected
[  988.537191] Rx Pre ====== 0
[  988.537202] Rx Pos ======  0
[  988.537270] Tx Pre ======  0
[  988.537280] Tx Pos ======  10
[  989.085598]
               Received ESP bootup event
[  989.085613] EVENT: 3
[  989.085620] ESP chipset detected [esp32]
[  989.085626] EVENT: 0
[  989.085631] ESP peripheral capabilities: 0x1d
[  989.291500] ESP Bluetooth init
[  989.291999] Capabilities: 0x1d. Features supported are:
[  989.292015]   * WLAN on SDIO
[  989.292024]   * BT/BLE
[  989.292033]     - HCI over SDIO
[  989.292041]     - BT/BLE dual mode
[  989.292050] EVENT: 4
[  989.292061] EVENT: 1
[  989.292070] esp32: process_fw_data ESP chipset's last reset cause:
[  989.292081] POWERON_RESET
[  989.292091] esp32: ESP Firmware version: 1.0.2
[  994.547594] esp32: Command[1] timed out
[  994.547640] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[  994.547682] esp32: add_network_iface() failure, ret: -1
[  994.547698] network iterface init failed
ubuntu@ubuntu-desktop:~/Esp32_20230509$

I use 22K pullup .
From log , Rpi have get Firmware version (1.0.2) from esp32 ,so I guess sdio link is built,   Capabilities(0x1d) is transfered , is it right ?

HanJiec, 2023-05-15:
Any comments are waiting

mantriyogesh, 2023-05-15:
Yes the first event is transferred. but transfer ahead not successful.

A. Can you please send your setup image/picture?
B. Please send below:
1. esp log
2. git commit at both, esp and host
3. host dmesg

C. Can you please try changing timing at 
https://github.com/espressif/esp-hosted/blob/3500b252d904b6a5b65ae50eba04f3eac0cf2a33/esp_hosted_ng/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L104

similar to https://github.com/espressif/esp-hosted/blob/3500b252d904b6a5b65ae50eba04f3eac0cf2a33/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L201

HanJiec, 2023-05-15:
A The setup Image ：
[image]

HanJiec, 2023-05-15:
A  The setup image is:
[image]

mantriyogesh, 2023-05-15:
Lines that need pull-up: CMD and DATA (DAT0 - DAT3) i.e. total 5.

What is wire lengths? try make as short as possible, with all in equal size which serve SDIO peripheral.
Try to connect all grounds (to reduce the noise and increase signal integrity).

HanJiec, 2023-05-15:
You  can see the image ? I cannot see it in my web brower

HanJiec, 2023-05-15:
The the wire for SDIO ,length is 12cm , are same size. Only 4 pull-ups in image , because all 5 pull-ups are connected , It cannot run , dmesg log stops at "[ 665.475406] esp_reset, ESP32: Triggering ESP reset."

mantriyogesh, 2023-05-15:
> because all 5 pull-ups are connected , It cannot run

Actually 5 pull-ups is requirement. you can lower length **as much as possible**

Wire lengths need to be <6cm. SDIO protocol face signal integrity issues if longer wires used.
We generally recommend to use PCB board or good wires with much smaller possible length.

With 5 pull-ups, try to change the timing as mentioned above.

HanJiec, 2023-05-15:
> Actually 5 pull-ups is requirement. you can lower length **as much as possible**
>
> Wire lengths need to be <6cm. SDIO protocol face signal integrity issues if longer wires used. We generally [...]

I Using prebuilt firmware 1.0.1  (https://github.com/espressif/esp-hosted/releases) , Your suggestion about timing should compile the firmware (stored in Esp32) by myself? now I didn't know how to compile it .
Host driver source commit is 
$ git log -1
commit cf871bbaef0906b34812a1e920c745fe101d855e
Merge: 81778f6 1294f8a
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Wed Apr 19 12:55:03 2023 +0800

    Merge branch 'bugfix/compatibility_with_higher_kernels' into 'master'

    Add support for Linux version 6.1.21

    See merge request app-frameworks/esp_hosted!283

mantriyogesh, 2023-05-15:
Please use ESP-Hosted master for NG, where you can build and flash esp using ESP-IDF.

By the way, latest release 1.0.2 is not backward compatible with 1.0.1.
Better to use latest master for current debugging.

HanJiec, 2023-05-15:
> Please use ESP-Hosted master for NG, where you can build and flash esp using ESP-IDF.
> 
> By the way, latest release 1.0.2 is not backward compatible with 1.0.1. Better to use latest master for current debugging.

I did use 1.0.2 , 1.0.1 is mis-writing:
[ 989.292091] esp32: ESP Firmware version: 1.0.2

HanJiec, 2023-05-15:
I want try it using SPI first , does wire legth is  critical?

mantriyogesh, 2023-05-15:
SPI doesn't face signal integrity issues. can use your wires (~10-12 cm)

HanJiec, 2023-05-16:
For SDIO driver , in esp_sdio.c, 
static const struct sdio_device_id esp_devices[] = {
	{ SDIO_DEVICE(ESP_VENDOR_ID, ESP_DEVICE_ID_1) },
	{ SDIO_DEVICE(ESP_VENDOR_ID, ESP_DEVICE_ID_2) },
	{}
};
The above id is read from ESP device by SDIO signal line  and compare ?

HanJiec, 2023-05-16:
using https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng/host  as host driver code +  ESP Firmware version: 1.0.2 as SPI solution , can it works ?  
 host code ,commit log-l : 
commit 3500b252d904b6a5b65ae50eba04f3eac0cf2a33 (HEAD -> master, origin/master, origin/HEAD)
Merge: d515213 c13c46a
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Thu May 11 16:49:42 2023 +0800

    Merge branch 'bugfix/rssi_correction' into 'master'

    esp_hosted_ng: Correct signal strength in scan results

    See merge request app-frameworks/esp_hosted!287

mantriyogesh, 2023-05-16:
It would work.
However, for 1.0.2 firmware better ton use source code from same git commit where the binaries were formed.

For 1.0.2, host code https://github.com/espressif/esp-hosted/tree/release/ng-v1.0.2 could be referred.

This is well tested. However, if that doesn't meet your expectations and need something from master, you can use current master which is currently compatible with 1.0.2.

mantriyogesh, 2023-05-16:
So, were you able to test https://github.com/espressif/esp-hosted/issues/226#issuecomment-1547423835
With spi?

HanJiec, 2023-05-16:
> So, were you able to test [#226 (comment)](https://github.com/espressif/esp-hosted/issues/226#issuecomment-1547423835) With spi?

Yes

HanJiec, 2023-05-17:
Via SPI , wifi station espsta0 comes . Using wpa_supplicant  configure , Raspberry Pi  can link  to wifi AP . Ping www.baidu.com successfully.
I don't know what code  set MISO/MOSI/SCLK pin to 21/19/23 . Would you please give me some guidance?

HanJiec, 2023-05-17:
[image]

mantriyogesh, 2023-05-17:
As I understand you are asking about Raspberry Pi Pins.
Please check https://pinout.xyz/pinout/spi
(Raspberry Pi datasheet will be better reference than this obviously)

So we are using SPI0 and not SPI1.
I am not pretty sure if these pins are manually configurable by RaspberryPi, if you intent to change them.
May be Raspberry Pi data sheet will help.
In case you wish to use SPI1, you might have to change spi_disabler.dts accordingly from repo.

On the contrary, ESP side GPIOs are all configurable, but then they will take standard GPIO_MUX path than IO_MUX path. (check ESP datasheet if need to go in detail) with negligible lower performance but with a needed good flexibility of using any GPIO.

mantriyogesh, 2023-05-17:
yes, in general, your expected linux should be configured with **Device Tree** (DT)
1. Enable your SPI Device instance in DT
2. Do not load hosted module yet. first test with spidev Tx and Rx is fine
3. change the expected pins, module loader script etc used for SPI as mentioned in per esp_hosted_fg porting_guide.md
4. test the GPIOs like resetpin actually restting ESP, Handshake ISR & Data ready ISR
5. load the module and check if first instance is received.

mantriyogesh, 2023-05-17:
You might have to see if there is any timing mismatch in between your new Linux and ESP.
In general, people have issues setting up the transport (but it has to be done!). Once the SPI timings are fine (or raw throughput is fine), you can add your own demo app and remove c demo app with your own logic.

HanJiec, 2023-05-17:
> 

Sorry , "remove c demo app" , c means what ?

mantriyogesh, 2023-05-17:
https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg/host/linux/host_control/c_support -> test.c

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/c_demo.md

Is user space demo application to showcase how the API to be used.
your application might re-use this app or you might develop referring to this c demo application.

HanJiec, 2023-05-17:
I always using esp_hosted_ng. Using nothing from esp_hosted_fg. 
The process is as the following :
make esp32_spi.ko 
make  dtbo：
      dtc spidev_disabler.dts -O dtb > spidev_disabler.dtbo
      cp  spidev_disabler.dtbo /boot/firmware/overlays
modprobe bluetooth ;modprobe cfg80211;insmod esp32_spi.ko resetpin=6

And  then using wpa_supplicant to bring up esta0 and link to wifi AP

mantriyogesh, 2023-05-17:
okay

HanJiec, 2023-05-26:
Using  SDIO， in esp_sdio.c   esp_init_interface_layer  include  sdio_register_driver to register a slave driver (for esp32).
Using Spi , in esp_spi.c , no calling  spi_register_driver, this is confusing me ?
Could you please give me some help ?

mantriyogesh, 2023-05-26:
It is a way to use underlying Linux bus driver for your SoC. APIs provided by Linux for different buses could have different way to use. Not necessarily would resemble names.

For sdio,
sdio_register_driver used. See:
https://elixir.bootlin.com/linux/latest/source/drivers/mmc/core/sdio_bus.c#L271

Whereas, In spi case we use 

https://github.com/espressif/esp-hosted/blob/2755b0db0280510757ed9c3e221ed9a1630be981/esp_hosted_ng/host/spi/esp_spi.c#L514

See -
https://elixir.bootlin.com/linux/latest/source/drivers/spi/spi.c#L744

HanJiec, 2023-05-26:
spi_new_device  creates a device (Esp32) .  I think there should be  a  Spi Slave driver to work with this device, in linux interior ? Could you help tell which driver which code create such a driver ?

mantriyogesh, 2023-05-26:
Now the question are going out of scope for ESP-Hosted.

Nevermind, I had came across one interesting post at https://www.linkedin.com/posts/activity-7067402712312651776-vi_v . This is general (not related to Esp-Hosted) post and credits to author.

HanJiec, 2023-06-08:
In real project  PCB , the reset Pin ( Esp32 EN) is connected to other CPU , the pin can be always set to high level, but canot control it to send a reset signal .  Can it works ?
I try " insmod esp32-spi.ko resetpin=-1 " , dmesg shows the the driver don't go forward.
Please give some advice about this issue.

HanJiec, 2023-06-08:
And in PCB， all VDD pin is controled by the same source : VCC_WIFI
[image]
Please comment this issue. Thanks.

mantriyogesh, 2023-06-08:
Reset pin is mandatory.
The High->Low->High (with small delays in between transitions) when applied on EN pin of ESP32, will only reset the esp32.

Intention is whenever the kernel module is reloaded, the esp32 should be reset so as the sdio is detected.
make sure that you have https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SDIO_setup.md#12-raspberry-pi-software-setup
done

```
dtoverlay=sdio,poll_once=off
```

will allow to poll the esp32 sdio slave (kind of hot pluggable sdio identification on esp reset)

```
dtoverlay=disable-bt
```

will stop onboard bt and only use from esp.

mantriyogesh, 2023-06-08:
Above is expectation, however config/logs will help to assess.

ESP -> `idf.py flash monitor` logs  OR minicom over ESP's USB port
Host -> Device Tree or /boot/config.txt, dmesg logs

Just to keep common understanding, getting the platform up (with transport) is requirement of ESP-Hosted project.
Any issues introduced because of incorrect configuration, we will try to address those with good intention. But doesn't fall into ESP-Hosted issues.

HanJiec, 2023-06-20:
By Spi connection , after insmod esp32_spi.ko resetpin=6, the wlan card espsta0 appear. This means the esp32 wifi card is OK. But after rmmod esp32_spi.ko and the insmod again , it fails.

[  513.354223] spi_exit
[  563.071489] esp_reset, ESP32: Triggering ESP reset.
[  563.278425] sysfs: cannot create duplicate filename '/bus/spi/devices/spi0.0'
[  563.278442] CPU: 3 PID: 3697 Comm: insmod Tainted: G         C OE     5.15.0-1032-raspi #35-Ubuntu
[  563.278449] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[  563.278453] Call trace:
[  563.278456]  dump_backtrace+0x0/0x200
[  563.278467]  show_stack+0x20/0x30
[  563.278471]  dump_stack_lvl+0x8c/0xb8
[  563.278478]  dump_stack+0x18/0x34
[  563.278482]  sysfs_warn_dup+0x6c/0x90
[  563.278489]  sysfs_do_create_link_sd+0xf8/0x100
[  563.278495]  sysfs_create_link+0x28/0x50
[  563.278500]  bus_add_device+0x74/0x150
[  563.278510]  device_add+0x370/0x780
[  563.278514]  __spi_add_device+0x80/0x174
[  563.278519]  spi_add_device+0x68/0xa0
[  563.278523]  spi_new_device+0xd0/0x15c
[  563.278527]  esp_init_interface_layer+0x164/0x590 [esp32_spi]
[  563.278550]  esp_init+0x1a4/0x1000 [esp32_spi]
[  563.278564]  do_one_initcall+0x4c/0x2c0
[  563.278570]  do_init_module+0x50/0x260
[  563.278575]  load_module+0xa30/0xbd0
[  563.278579]  __do_sys_finit_module+0xa8/0x114
[  563.278583]  __arm64_sys_finit_module+0x28/0x3c
[  563.278587]  invoke_syscall+0x50/0x120
[  563.278594]  el0_svc_common.constprop.0+0x6c/0x1a0
[  563.278599]  do_el0_svc+0x30/0xb0
[  563.278605]  el0_svc+0x4c/0x170
[  563.278609]  el0t_64_sync_handler+0xa4/0x12c
[  563.278614]  el0t_64_sync+0x1a4/0x1a8
[  563.278717] spi-bcm2835 fe204000.spi: can't add spi0.0, status -17
[  563.278737] Failed to add new SPI device
[  563.278744] spi_exit
[  563.694146] spi_exit:spi_context.esp_spi_dev=0
[  563.694165] Failed Init SPI device

 Plese help me on this issue.

HanJiec, 2023-06-21:
Please see my comments:
 
> https://github.com/espressif/esp-hosted/blob/8dc9280339e1539ba4a5bae65a77b3fe99c9ce14/esp_hosted_ng/host/spi/esp_spi.c#L614
> 
> 
> 'Failed Init SPI device' means your spi instance is not claimable by the spi driver.
> After unloading the driver, can you please check:
> 
> 1. ls /dev/spidev*
> 2. lsmod
> 3. ls /bus/spi/devices/spi*
> 
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ modprobe bluetooth
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ modprobe cfg80211
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ sudo insmod esp32_spi.ko resetpin=6
[sudo] password for ubuntu:
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ sudo rmmod esp32_spi
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ ls /dev/spidev*
/dev/spidev0.1
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ lsmod
Module                  Size  Used by
bnep                   32768  2
binfmt_misc            24576  1
btsdio                 20480  0
bluetooth             712704  8 btsdio,bnep
vc4                   339968  10
ecdh_generic           16384  1 bluetooth
ecc                    36864  1 ecdh_generic
snd_soc_hdmi_codec     24576  2
snd_soc_core          311296  2 vc4,snd_soc_hdmi_codec
brcmfmac              417792  0
bcm2835_codec          53248  0
snd_compress           28672  1 snd_soc_core
ac97_bus               16384  1 snd_soc_core
snd_bcm2835            36864  1
bcm2835_v4l2           49152  0
bcm2835_isp            36864  0
brcmutil               28672  1 brcmfmac
v4l2_mem2mem           45056  1 bcm2835_codec
bcm2835_mmal_vchiq     40960  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
videobuf2_vmalloc      20480  1 bcm2835_v4l2
videobuf2_dma_contig    24576  2 bcm2835_codec,bcm2835_isp
snd_pcm_dmaengine      20480  1 snd_soc_core
videobuf2_memops       20480  2 videobuf2_vmalloc,videobuf2_dma_contig
cfg80211              966656  1 brcmfmac
videobuf2_v4l2         32768  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
snd_seq_midi           20480  0
snd_seq_midi_event     16384  1 snd_seq_midi
videobuf2_common       81920  8 bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
snd_pcm               163840  5 snd_bcm2835,snd_soc_hdmi_codec,snd_compress,snd_soc_core,snd_pcm_dmaengine
v3d                    94208  6
snd_rawmidi            49152  1 snd_seq_midi
drm_kms_helper        344064  2 vc4
snd_seq                90112  2 snd_seq_midi,snd_seq_midi_event
videodev              282624  6 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
crct10dif_ce           20480  1
gpu_sched              49152  1 v3d
raspberrypi_hwmon      16384  0
cec                    81920  2 drm_kms_helper,vc4
snd_seq_device         20480  3 snd_seq,snd_seq_midi,snd_rawmidi
mc                     73728  6 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
fb_sys_fops            20480  1 drm_kms_helper
snd_timer              45056  2 snd_seq,snd_pcm
syscopyarea            16384  1 drm_kms_helper
vc_sm_cma              40960  2 bcm2835_mmal_vchiq,bcm2835_isp
sysfillrect            20480  1 drm_kms_helper
snd                   126976  11 snd_seq,snd_seq_device,snd_bcm2835,snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi
sysimgblt              16384  1 drm_kms_helper
bcm2835_gpiomem        16384  0
rpivid_mem             16384  0
[... 40 lines cut ...]
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ cd /sys/bus/spi/devices/spi0.0
-bash: cd: /sys/bus/spi/devices/spi0.0: No such file or directory
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ cd /sys/bus/spi/devices/spi0.1
ubuntu@ubuntu-desktop:/sys/bus/spi/devices/spi0.1$ ls
driver  driver_override  modalias  of_node  power  spidev  statistics  subsystem  uevent
ubuntu@ubuntu-desktop:/sys/bus/spi/devices/spi0.1$

> Also, can you please state how are you loading or unloading **spidev**?
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ cat  /boot/firmware/config.txt | grep spidev
dtoverlay=spidev_disabler
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ cat spidev_disabler.dts
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2708";
    fragment@0 {
        target = <&spidev0>;
        __overlay__ {
            status = "disabled";
        };
    };
};
ubuntu@ubuntu-desktop:~/Esp32_20230516/host$ ls  /boot/firmware/overlays | grep spidev    
spidev_disabler.dtbo

HanJiec, 2023-06-26:
Hi, mantriyogesh ,
Above answer is enough for positioning my question ？

mantriyogesh, 2023-06-26:
Hello @HanJiec ,

The answer is fine.
We will need to try at our place ( till now we didn't face such issue)

Can you please send the diff from the repo, be it code or rpi_init.sh script etc.?

HanJiec, 2023-06-27:
I didn't use  rpi_init.sh.  I modified  Makefile .  In  esp_hosted_ng/host dir , using make to produce ko.  And  load module according the bottom lines of rpi_init.sh: sudo modprobe bluetooth; sudo modprobe cfg80211; sudo insmod ./esp32_sdio.ko resetpin=6;
The following is Makefile modified:
#2023/3/31 Created by HanJc for using in prj X94M 

MODULE_NAME=esp32_spi
obj-m := $(MODULE_NAME).o
$(MODULE_NAME)-y := spi/esp_spi.o 
$(MODULE_NAME)-y += esp_bt.o main.o esp_cmd.o esp_wpa_utils.o esp_cfg80211.o esp_stats.o 

KERNELDIR ?= /lib/modules/$(shell uname -r)/build

EXTRA_CFLAGS += -I$(shell pwd)/include -I$(shell pwd)/spi

all default: modules
install: modules_install

modules modules_install help clean:
	$(MAKE) -C $(KERNELDIR) M=$(shell pwd) $@	

Please find  diff in the following:
diff --git a/esp_hosted_ng/host/Makefile b/esp_hosted_ng/host/Makefile
index 4bf3da6..4d8c58e 100644
--- a/esp_hosted_ng/host/Makefile
+++ b/esp_hosted_ng/host/Makefile
@@ -1,53 +1,19 @@
-CONFIG_TEST_RAW_TP := n
-CONFIG_ENABLE_MONITOR_PROCESS = n
+#2023/3/31 Created by HanJc for using in prj X94M 
 
-# Toolchain Path
-CROSS_COMPILE := /usr/bin/arm-linux-gnueabihf-
-# Linux Kernel header
-KERNEL := /lib/modules/$(shell uname -r)/build
-# Architecture
-ARCH := arm
-
-#Default interface is sdio
-MODULE_NAME=esp32_sdio
-
-#Targets passed overrrides default value
-ifeq ($(target), sdio)
-	MODULE_NAME=esp32_sdio
-endif
-
-ifeq ($(target), spi)
-	MODULE_NAME=esp32_spi
-endif
-
-ifeq ($(CONFIG_ENABLE_MONITOR_PROCESS), y)
-	EXTRA_CFLAGS += -DCONFIG_ENABLE_MONITOR_PROCESS
-endif
-
-ifeq ($(CONFIG_TEST_RAW_TP), y)
-	EXTRA_CFLAGS += -DCONFIG_TEST_RAW_TP
-endif
-
-EXTRA_CFLAGS += -I$(PWD)/include -I$(PWD)
+MODULE_NAME=esp32_spi
+obj-m := $(MODULE_NAME).o
+$(MODULE_NAME)-y := spi/esp_spi.o 
+$(MODULE_NAME)-y += esp_bt.o main.o esp_cmd.o esp_wpa_utils.o esp_cfg80211.o esp_stats.o 
 
-ifeq ($(MODULE_NAME), esp32_sdio)
-	EXTRA_CFLAGS += -I$(PWD)/sdio
-	module_objects += sdio/esp_sdio.o sdio/esp_sdio_api.o
-endif
+KERNELDIR ?= /lib/modules/$(shell uname -r)/build
 
-ifeq ($(MODULE_NAME), esp32_spi)
-	EXTRA_CFLAGS += -I$(PWD)/spi
[... 324 lines cut ...]
+    printk("%s", __func__ ) ;
 	disable_irq(SPI_IRQ);
 	disable_irq(SPI_DATA_READY_IRQ);
 	close_data_path();
@@ -669,6 +711,7 @@ static void spi_exit(void)
 		gpio_free(SPI_DATA_READY_PIN);
 	}
 
+    printk("%s:spi_context.esp_spi_dev=%x", __func__ , spi_context.esp_spi_dev  ) ;
 	if (spi_context.esp_spi_dev)
 		spi_unregister_device(spi_context.esp_spi_dev);
 
diff --git a/esp_hosted_ng/host/spi/esp_spi.h b/esp_hosted_ng/host/spi/esp_spi.h
index 577c562..db931f3 100644
--- a/esp_hosted_ng/host/spi/esp_spi.h
+++ b/esp_hosted_ng/host/spi/esp_spi.h
@@ -17,7 +17,7 @@
 #ifndef _ESP_SPI_H_
 #define _ESP_SPI_H_
 
-#include "esp.h"
+#include "../include/esp.h"
 
 #define HANDSHAKE_PIN           22
 #define SPI_IRQ                 gpio_to_irq(HANDSHAKE_PIN)

HanJiec, 2023-06-27:
Hi  mantriyogesh ，

I sloved it, It 's my fault.

HanJiec, 2023-06-27:
In order to study , I  kmalloc  some memory for spi_context.esp_spi_dev->dev.init_name, and strcpy "esp32spidev" to this item in esp_spi.c .  It fails the rmmod esp32_spi.ko.
Failing rmmod is a sub-issue. For my project , I need  your more support and  want continue  this issue.

mantriyogesh, 2023-07-21:
Please follow https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng

Instead of using 1.0.2 in your case, please use master.
Build the host Linux driver & esp firmware using same code base and same git commit.

While following the documentation, you will come across [hardware and software setup](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md)
with spi only to flash the esp firmware **from sources**

HanJiec, 2023-07-25:
In  https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng/esp/esp_driver , It is  [esp-idf @ 91dc99a](https://github.com/espressif/esp-idf/tree/91dc99a3ce09b66c51b8be884e2ed6b6e1558747)  , do we using esp-idf @ 91dc99a to compile firmware?

mantriyogesh, 2023-07-25:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md
will setup the ESP-IDF for you.
Please make sure that all local changes saved outside the repo as setup.sh will clean the repo.

HanJiec, 2023-07-25:
Should it be built  in Linux host system？

mantriyogesh, 2023-07-25:
which machine is used as development environment?

Your development environment should flash the ESP32. It would be typically your laptop/desktop. Any Linux/Windows/Mac should work.

For changing the number of cores, 
1. 
```bash
$ cd esp_hosted_fg/esp/esp_driver/network_adapter
```
2. Change `sdkconfig.defaults.esp32`
https://github.com/espressif/esp-hosted/blob/644eb875dc1997916cfc5c0f66b5b9964e2ca55d/esp_hosted_fg/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32#L22-L27

to 

> CONFIG_FREERTOS_UNICORE=**y**
> CONFIG_FREERTOS_HZ=400
> **#** CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1=y
> **#** CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
> **#** CONFIG_BTDM_CTRL_PINNED_TO_CORE=0
> **#** CONFIG_BTDM_CTRL_PINNED_TO_CORE_0=y

```bash
$ cd esp_hosted_fg/esp/esp_driver/network_adapter
$ rm -rf sdkconfig build
```
Follow normal flashing ESP from sources using documentation for your transport.

mantriyogesh, 2023-07-25:
please note that `rm -rf sdkconfig` is important to take effect of changed `sdkconfig.defaults.esp32`

HanJiec, 2023-07-31:
The firmware bin which are download github is divided into  a few kinds. SPI-Only/SDIO-Only/... 
How to config these ?
[image]

mantriyogesh, 2023-07-31:
These are alternative transport ESP-Hosted provides.
Have you checked the documentation?

You have to choose the transport and use the correct one.

HanJiec, 2023-07-31:
Do you mean the doc which is pointed by url : https://github.com/espressif/esp-hosted/tree/644eb875dc1997916cfc5c0f66b5b9964e2ca55d/esp_hosted_ng#22-transport-layer-protocol ?
In https://github.com/espressif/esp-hosted/blob/644eb875dc1997916cfc5c0f66b5b9964e2ca55d/esp_hosted_ng/docs/sdio_protocol.md ,  in the line “0x3FF5508C: Interrupt vector used by host to interrupt ESP peripheral” ，0x3FF5508C is a register in ESP32 ？

mantriyogesh, 2023-07-31:
https://github.com/espressif/esp-hosted/tree/644eb875dc1997916cfc5c0f66b5b9964e2ca55d/esp_hosted_ng#31-setup -> [Hardware and Software Setup](https://github.com/espressif/esp-hosted/blob/644eb875dc1997916cfc5c0f66b5b9964e2ca55d/esp_hosted_ng/docs/setup.md).

HanJiec, 2023-08-01:
I ever read this doc and study it again. It doesn't describe how to select transport Layer when compile a firmware. It provide the compiled firmware bin and guide us how to flash the firmware into esp board.
I compile the sdio firmware bin directory  with spi. I found  sdkconfig is different. In ESP-Hosted-NG_release_v1.0.2\esp32\sdio_only\sdkconfig , there are line : CONFIG_ESP_SDIO_HOST_INTERFACE=y ; while in  ESP-Hosted-NG_release_v1.0.2\esp32\spi_only , there is line : CONFIG_ESP_SPI_HOST_INTERFACE=y.  Can I write  these config lines  into  esp-hosted/esp_hosted_fg/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32 , are there any other configs  for SDIO or SPI.

HanJiec, 2023-11-03:
Hi, 
wpa_supplicant -B -i espsta0 -c /etc/wpa_supplicant/wpa_supplicant.conf                                            <
x9hp_ms:/etc/wpa_supplicant # ifconfig
espsta0   Link encap:Ethernet  HWaddr 3c:e9:0e:1a:47:9c  Driver esp_sdio
          inet6 addr: fe80::3ee9:eff:fe1a:479c/64 Scope: Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:57 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 TX bytes:0
why the wifi station is of inet6 mode ？how to change inet4 ?

mantriyogesh, 2023-11-03:
Is the DHCP client running on espsta0?
You can run
sudo dhclient -v espsta0

Again it depends which DHCP client software you have.
ESP-Hosted will just expose the espsta0. IP assignment is left open to configure.

HanJiec, 2023-11-06:
> Is the DHCP client running on espsta0? You can run sudo dhclient -v espsta0
> 
> Again it depends which DHCP client software you have. ESP-Hosted will just expose the espsta0. IP assignment is left open to configure.

Android  have no dhclient ， try cp some linux dhclient in adb device， always fail  to execute .And DL a open source , It seems needing lots of time to compile it .  Is  some other method to configure IP ？

[6 more replies]

closed 2023-11-15 by HanJiec