espressif/esp-hosted #231: Correct way to increase RX buffer size on Linux host

NanokodasKarolis, 2023-05-15:
Hi, we are working on porting the FG drivers to a proprietary SoC running Linux host using the SDIO interface with ESP32. Everything seems to be smooth on the SDIO lines. Though the host side seems to be dropping packets when we run some more bandwidth heavy applications. The loss happens usually in bursts since our application is pretty volatile when it comes to incoming traffic.
```
# ifconfig ethsta0
ethsta0   Link encap:Ethernet  HWaddr E0:E2:E6:37:BC:78  
          inet addr:169.254.169.2  Bcast:169.254.169.3  Mask:255.255.255.252
          UP BROADCAST RUNNING MULTICAST  MTU:1280  Metric:1
          RX packets:7657 errors:0 dropped:228 overruns:0 frame:0
          TX packets:8679 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:465089 (454.1 KiB)  TX bytes:5932520 (5.6 MiB)
```
We assume that we have enough processing power to handle higher traffic as long as we could buffer up those packets for a while.  I am not very familiar with Linux network settings so apologies in advance if this question is not related to this specific library, but rather a generic Linux question instead.

What would be the correct way to increase the RX packet buffer count/size?

mantriyogesh, 2023-05-16:
Actually Linux is a beast when comes to customisation.

As I can understand Rx at Linux is being dropped, some of the ways is to shape the traffic.

One alternate is using tc.
https://www.slideshare.net/suselab/linux-linux-traffic-control

It is complicated and needs study to apply.

Another one is using RSS or RPS. https://www.kernel.org/doc/Documentation/networking/scaling.txt

I am not sure if the queue control is supported out of the box for ethsta0 (being virtual interface), you may have to add multiple queues at interface and queues will be processed by different processors to achieve parallel processing. Again everything has advantage and disadvantage, so you might need to check which traffic shaping is fine for you.

mantriyogesh, 2023-05-16:
You can also optimise only features you need in hosted and disable others to free up space and use that space for bigger queues in hosted software.

For example, you don't need Bluetooth, disable it etc.

NanokodasKarolis, 2023-05-20:
Thanks for the replies @mantriyogesh. It appears that we were getting corrupted RX packets and the host side was dropping them because of this. Fixing this no longer shows the issue.

closed 2023-05-20 by NanokodasKarolis