espressif/esp-hosted #232: Unable to compile NG firmware for esp32s3

sergey-suloev, 2023-05-16:
/home/sergey/Projects2/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c: In function 'queue_next_transaction':
/home/sergey/Projects2/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c:401:30: error: 'ESP_SPI_CONTROLLER' undeclared (first use in this function)
  ret = spi_slave_queue_trans(ESP_SPI_CONTROLLER, spi_trans, portMAX_DELAY);

sergey-suloev, 2023-05-16:
> Yes, That should be it. But other chipsets will be added in some time.

so, in case if I add definitions for S3 the host driver itself should work ?

sergey-suloev, 2023-05-16:
> Untill ng code is not added, you can use FG is reference to add.

ok, thanks

I am not sure why you used 
#define ESP_SPI_CONTROLLER     1

According to spec SPI1 is used by some internal peripheral

sergey-suloev, 2023-05-17:
Received ESP bootup event
[    9.428526] EVENT: 3
[    9.430757] EVENT: 2
[    9.432953] EVENT: 0
[    9.435188] EVENT: 4
[    9.437395] EVENT: 1
[    9.439620] esp32: process_fw_data ESP chipset's last reset cause: 
[    9.439630] POWERON_RESET
[    9.448585] esp32: ESP Firmware version: 1.0.2
[    9.453054] ESP chipset detected [esp32-s3]
[   10.846843] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   14.557974] esp32: Command[3] timed out
[   14.561853] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   14.569059] esp32: add_network_iface() failure, ret: -1
[   14.569430] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.574350] network iterface init failed
[   14.574363] ESP peripheral capabilities: 0xa0
[   14.589022] Capabilities: 0xa0. Features supported are:
[   14.594290] 	 * WLAN on SPI

mantriyogesh, 2023-05-19:
I can see the first event is received.
Can you send ESP and Host logs?

Can you perform raw throughput test to test transport?
I am suspecting something wrong at transport level like timing etc mismatch in between host and ESP

mantriyogesh, 2023-12-07:
Can close this now?