espressif/esp-hosted #233: The ESP32 uses the esp-hosted SDK as the WiFi NIC for the SPI communication. In addition to the function of SPI wifi network card, does it support Bluetooth?

JS-Ye, 2023-05-18:
### ESP chip/Product Platform:

- ESP chip:  ESP32-WROOM-32E
- Product Platform:  https://github.com/espressif/esp-hosted

### The ESP32 uses the esp-hosted SDK as the WiFi NIC for the SPI communication. In addition to the function of SPI wifi network card, does it support Bluetooth? If Bluetooth is supported, how should I add Bluetooth?

Bluetooth needs to achieve the following two functions:

1. Customize the 128bit service UUID, receive feature UUID, and send UUID.

2. After entering the data transmission mode, ESP32 acts as the Bluetooth slave, and APP acts as the host to send data to ESP32.

mantriyogesh, 2023-05-19:
Hello @JS-Ye 

Can you please let us know the host used?
Linux /MCU, kernel version?

zhlftest, 2023-05-19:
hello @mantriyogesh the host is linux ,But they didn't use the device tree

mantriyogesh, 2023-05-19:
Yes Bluetooth and Wi-Fi are both supported using ESP-Hosted (By default this is enabled, if needed, you can configure only bluetooth or only Wi-Fi ).

Follow the documentation and video at https://youtu.be/g14aEjnjRLw for bluetooth related questions.

mantriyogesh, 2023-05-19:
Hello @zhlftest 

Is the Bluetooth LE or BR/EDR?

> Customize the 128bit service UUID, receive feature UUID, and send UUID. 

Are you talking about GATT service?

JS-Ye, 2023-05-19:
Hello @mantriyogesh  
   
I wonder if you can change 128bitUUID by adding an AT instruction to esp-hosted.  
>Customize the 128bit service UUID, receive feature UUID, and send UUID.

mantriyogesh, 2023-05-19:
ESP-Hosted doesn't understand AT commands as such, unless you have something on top as your own AT interpreter. I am still trying to understand, how AT is relevant.

mantriyogesh, 2023-05-19:
just to clarify .. BT controller is not the one that supports AT as such.
AT commands usually work on top of RFCOMM and is a data for controller i.e. controller doesn't have any special or separate handling for AT commands

RFCOMM layer will reach till stack which is still at host.
From host to controller HCI communication till bluetooth controller.
How do you communicate tith BlueZ stack, is upto you.

mantriyogesh, 2023-06-27:
i hope the answers have clarified your question.
Can we close this thread?

JS-Ye, 2023-06-29:
Ok, thank you for your support, this problem can be turned off.

closed 2023-06-29 by JS-Ye