espressif/esp-hosted #235: "espsta0 " unable to set tx power,esp32c3

renyeguo, 2023-05-26:
"iwconfig espsta0 txpower 30”
Error for wireless request "Set Tx Power" (8B26) :
SET failed on device espsta0 ; Operation not supported.

kapilkedawat, 2023-06-04:
Support for setting transmit power is available now.
Please use `iw dev espsta0 set txpower limit 2000`  in with latest code.

renyeguo, 2023-06-06:
well, I pulled the latest esp_host_ng code, However, I still use "esp-hosted -NG Release 1.0.2" for the esp side firmware. Now when I run "iw dev espsta0 set txpower limit 2000", the following error occurs:

[789.802811] [0] esp32_spi_ng:wait_and_decode_cmd_resp: Command[16] timed out

[789.810280] [0] esp32_spi_ng:cmd_set_tx_power: esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22

command failed: Invalid argument (-22)

Check whether the esp side firmware needs to be updated and whether a version newer than ESP-Hosted-NG Release 1.0.2 is available，thanks

renyeguo, 2023-06-06:
@kapilkedawat

kapilkedawat, 2023-06-06:
@renyeguo please compile the new firmware. Changes are both in host driver and firmware.

renyeguo, 2023-06-06:
@kapilkedawat Ok, I got it, now you are the one to open source the source code of ESP-hosted-ng. Now my network is not in good condition, and it is very slow for me to build a new ESP-IDF project. Could you please provide the compiled firmware of your esp side, I want to verify this problem quickly

renyeguo, 2023-06-06:
[image]

kapilkedawat, 2023-06-07:
Please try with attached firmware.
[firmware.zip](https://github.com/espressif/esp-hosted/files/11673109/firmware.zip)

renyeguo, 2023-06-07:
thank u

kapilkedawat, 2023-06-12:
@mantriyogesh this support is available now, please close this issue.

closed 2023-06-12 by mantriyogesh