espressif/esp-hosted #237: The wifi will not auto reconnect after event espsta0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD

bert-qin, 2023-06-05:
Hello,

I'm using esp32c3 in my product, using the hosted-ng solution.
I've encountered a problem and looking for some help here.

The customer uses our product in Germany. When the esp32 is connected to the wifi router, the connection is disconnected after about 30 seconds, and it will not automatically reconnect again.

Here is some version and configuration information:

> [   16.624606] esp32: ESP Firmware version: 1.0.2
[   16.629538] ESP chipset detected [esp32-c3]
[   16.735884] ESP peripheral capabilities: 0xe8
[   16.967285] ESP Bluetooth init
[   16.982707] Capabilities: 0xe8. Features supported are:
[   16.988136]   * WLAN on SPI
[   16.990960]   * BT/BLE
[   16.993341]     - HCI over SPI
[   16.996418]     - BLE only

> root@ocpp510:# uname -a
Linux ocpp510 5.10.9-1.0.0+g49ec67ad2282 #1 SMP PREEMPT Tue Mar 14 05:09:26 UTC 2023 armv7l GNU/Linux
root@ocpp510:~#

> root@ocpp510:# wpa_supplicant -v
wpa_supplicant v2.9
Copyright (c) 2003-2019, Jouni Malinen <j@w1.fi> and contributors
root@ocpp510:~#

> 278 root      6916 S    /usr/sbin/wpa_supplicant -D nl80211 -i espsta0 -c /etc/wpa_supplicant.conf -g espsta0

> root@ocpp510:~# cat /etc/wpa_supplicant.conf
country=EU
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
update_config=1
network={
        ssid="Lab"
        psk=3e6b54d412e166b161ede254b91c8e98a2797d24933028e650eab33259afa665
}

This is the log of wpa_supplicant when the network is disconnected.

> wpa_supplicant.service - WPA supplicant
     Loaded: loaded (/etc/systemd/system/wpa_supplicant.service; enabled; vendor preset: disabled)
     Active: active (running) since Fri 2023-06-02 11:28:16 UTC; 2min 10s ago
   Main PID: 272 (wpa_supplicant)
     CGroup: /system.slice/wpa_supplicant.service
             `-272 /usr/sbin/wpa_supplicant -D nl80211 -i espsta0 -c /etc/wpa_supplicant.conf -g espsta0
Jun 02 11:28:16 ocpp510 systemd[1]: Started WPA supplicant.
Jun 02 11:28:16 ocpp510 wpa_supplicant[272]: Successfully initialized wpa_supplicant
Jun 02 11:28:16 ocpp510 wpa_supplicant[272]: rfkill: Cannot open RFKILL control device
Jun 02 11:28:18 ocpp510 wpa_supplicant[272]: espsta0: SME: Trying to authenticate with d2:21:f9:1d:17:b6 (SSID='Lab' freq=2412 MHz)
Jun 02 11:28:18 ocpp510 wpa_supplicant[272]: espsta0: Trying to associate with d2:21:f9:1d:17:b6 (SSID='Lab' freq=2412 MHz)
Jun 02 11:28:18 ocpp510 wpa_supplicant[272]: espsta0: Associated with d2:21:f9:1d:17:b6
Jun 02 11:28:18 ocpp510 wpa_supplicant[272]: espsta0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jun 02 11:28:20 ocpp510 wpa_supplicant[272]: espsta0: WPA: Key negotiation completed with d2:21:f9:1d:17:b6 [PTK=CCMP GTK=CCMP]
Jun 02 11:28:20 ocpp510 wpa_supplicant[272]: espsta0: CTRL-EVENT-CONNECTED - Connection to d2:21:f9:1d:17:b6 completed [id=0 id_str=]
Jun 02 11:28:53 ocpp510 wpa_supplicant[272]: espsta0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD

I did some experiments and found when esp32 received CTRL-EVENT-REGDOM-CHANGE, the network disconnects and does not automatically reconnect. If I turn off the router, change the channel of the wifi router, or ban esp32 on the router and then allow it, I get the event right away. 

I use the command `usr/sbin/wpa_supplicant -D nl80211 -i espsta0 -c /etc/wpa_supplicant.conf -g espsta0 -ddd &`
to turn on the debugging mode of wap_supplicant, and when I ban esp32 on router, I found the log as follows:

> root@ocpp510:~# [ 1837.938245] Disconnect event for ssid ocpp510 [reason:2]
[ 1837.943792] esp_cfg80211_del_key
nl80211: Event message available
nl80211: Drv Event 48 (NL80211_CMD_DISCONNECT) received for espsta0
nl80211: Ignore disconnect event when using userspace SME
[ 1837.998234] esp_cfg80211_del_key
[ 1838.038136] esp_cfg80211_del_key
[ 1838.057584] esp_cfg80211_del_key
[ 1838.087616] esp_cfg80211_del_key
[ 1838.118124] esp_cfg80211_del_key
RTM_NEWLINK: ifi_index=6 ifname=espsta0 wext ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
nl80211: Event message available
nl80211: Drv Event 36 (NL80211_CMD_REG_CHANGE) received for espsta0
nl80211: Regulatory domain change
initiator=0
type=1
espsta0: Event CHANNEL_LIST_CHANGED (27) received
espsta0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
espsta0: Updating hw mode
nl80211: Regulatory information - country=00
nl80211: 2402-2472 @ 40 MHz 20 mBm
nl80211: 2457-2482 @ 20 MHz 20 mBm (no IR)
nl80211: 2474-2494 @ 20 MHz 20 mBm (no OFDM) (no IR)
nl80211: 5170-5250 @ 80 MHz 20 mBm (no IR)
nl80211: 5250-5330 @ 80 MHz 20 mBm (DFS) (no IR)
nl80211: 5490-5730 @ 160 MHz 20 mBm (DFS) (no IR)
nl80211: 5735-5835 @ 80 MHz 20 mBm (no IR)
nl80211: 57240-63720 @ 2160 MHz 0 mBm
nl80211: Added 802.11b mode based on 802.11g information
nl80211: Mode IEEE 802.11g: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472[NO_IR] 2484[NO_IR]
nl80211: Mode IEEE 802.11b: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472[NO_IR] 2484[NO_IR]

I found a related message
> nl80211: Ignore disconnect event when using userspace SME.

When I query the status of the wifi connection, the response I get is

> root@ocpp510:# wpa_cli status
Selected interface 'espsta0'
espsta0: Control interface command 'STATUS'
EAPOL: EAP Session-Id not available
CTRL-DEBUG: ctrl_sock-sendto: sock=14 sndbuf=180224 outq=0 send_len=233
bssid=a2:fb:7e:57:40:56
freq=2412
ssid=ocpp510
id=0
mode=station
pairwise_cipher=CCMP
group_cipher=CCMP
key_mgmt=WPA2-PSK
wpa_state=COMPLETED
ip_address=192.168.9.7
address=60:55:f9:83:9d:6c
uuid=57e8a86b-a897-5067-90e8-f824e4d33d82
root@ocpp510:~#

I found wpa_supplicant doesn't know that wifi has been disconnected.
 Is this why it won't reconnect after disconnecting? 
wpa_supplicant is a standard software, I tried it on a laptop, and the same configuration can automatically reconnect. 
How can I solve this problem? 
I need help.

kapilkedawat, 2023-06-05:
Hi @bert-qin, 

We have tried to change reg domain using iw, changing the reg doesn't cause disconnect. Can you please share firmware logs in debug mode.

bert-qin, 2023-06-05:
hi @kapilkedawat ,
Thanks for your reply.
From the log above, we can see that the 

> nl80211: Drv Event 48 (NL80211_CMD_DISCONNECT) received for espsta0 

event was received first and then 

> espsta0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD 

was received. So I have no doubts with reg dom change will not cause disconnection.

My biggest concern now is that it cannot automatically reconnect after disconnection. 
For example, power off and on the router.
Can you test if the esp32 can reconnect automatically after restarting the router?

> Can you please share firmware logs in debug mode for further debugging.

Do you mean the debug log of esp32? Sorry, I am directly using the firmware compiled by you, and I don't know how to open it. I can study if necessary.

kapilkedawat, 2023-06-05:
@bert-qin Can you please try with attached patch and see if that helps.

[cfg_deauth.patch](https://github.com/espressif/esp-hosted/files/11653710/cfg_deauth.patch)

bert-qin, 2023-06-06:
@kapilkedawat Perfect, this patch solves the problem of not being able to automatically reconnect after being disconnected from the router. Thank you.

closed 2023-06-06 by bert-qin