espressif/esp-hosted #238: how to configure voice data which from bt to PCM format ?

zhou88mang, 2023-06-05:
I successfully implemented the Bluetooth function using btstack + esp-hosted, but there was one problem, i set BR/EDR Sync(SCO/eSCO) default datapath PCM .By measuring the output pin of the pcm, I found that the format of these signals is i2s format , but the signal I really want is PCM format, is there any way to configure the format of these output signals?

mantriyogesh, 2023-06-05:
Thank you @zhou88mang for the details.

Which host do you use? (Linux/MCU)? Which one precisely? Which stack do you use for Bluetooth?

Can you please provide the sdkconfig , esp log and host side bluetooth logs like btmon, Bluetooth TL logs?

zhou88mang, 2023-06-05:
> Thank you @zhou88mang for the details.
> 
> Which host do you use? (Linux/MCU)? Which one precisely? Which stack do you use for Bluetooth?
> 
> Can you please provide the sdkconfig , esp log and host side bluetooth logs like btmon, Bluetooth TL logs?
my main host chip  is x2000 and platform is freertos , Bluetooth stack is  btstack 
this is sdkconfig:

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=3
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_RTC_SLOW_CLOCK_SUPPORT_8MD256=y
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_MMU_LINEAR_ADDRESS_REGION_NUM=5
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_SIZE=64
[... 2515 lines cut ...]

[834.502222] [00:13:54.502] ACL => 80 00 0F 00 0B 00 41 00 09 EF 0C 00 0D 0A 4F 4B 0D 0A 40
[834.502544] [00:13:54.502] LOG -- hci_transport_h4.c.190: hci_transport_h4_trigger_next_read: 1 bytes
[834.502898] [00:13:54.502] EVT <= 6E 00
[834.503043] [00:13:54.503] EVT <= 79 02 44 00
[834.503208] [00:13:54.503] EVT <= 89 02 02 00
[834.503373] [00:13:54.503] EVT <= 79 02 44 00
[834.512729] [00:13:54.512] LOG -- hci_transport_h4.c.190: hci_transport_h4_trigger_next_read: 2 bytes
[834.513088] [00:13:54.513] LOG -- hci_transport_h4.c.190: hci_transport_h4_trigger_next_read: 5 bytes
[834.513442] [00:13:54.513] EVT <= 13 05 01 80 00 01 00
[834.513644] [00:13:54.513] LOG -- hci_transport_h4.c.190: hci_transport_h4_trigger_next_read: 1 bytes
[834.544615] [00:13:54.544] LOG -- hci_transport_h4.c.190: hci_transport_h4_trigger_next_read: 4 bytes
[834.544973] [00:13:54.544] LOG -- hci_transport_h4.c.190: hci_transport_h4_trigger_next_read: 20 bytes
[834.545332] [00:13:54.545] ACL <= 80 20 14 00 10 00 44 00 0B FF 14 00 00 41 54 2B 56 47 4D 3D 31 34 0D 86
[834.545701] [00:13:54.545] LOG -- rfcomm.c.1447: RFCOMM data UIH_PF, new credits channel 0x02: 0, now 2
[834.546060] [00:13:54.546] LOG -- hfp.c.2109: HFP_AG_RX: 'AT+VGM=14\r'
[834.546310] [00:13:54.546] LOG -- hfp.c.1352: command string 'AT+VGM=', handsfree 0 -> cmd id 42
[834.546644] [00:13:54.546] LOG -- hfp.c.1614: hfp parse HFP_CMD_SET_MICROPHONE_GAIN 14

[834.546949] [00:13:54.546] LOG -- hfp_ag.c.2521: HF microphone gain = 14
[834.547199] Microphone volume: gain 14
[834.547340] [00:13:54.547] LOG -- hfp.c.241: HFP_TX
OK

my problem is that i found data from pcm interface of esp32 is i2s format by measuring , but the pcm interface of my host chip can only support pcm protocol(mode A , mode B), so i want know  whether  esp32 can set pcm signal  to pcm format instead of i2s format

mantriyogesh, 2023-06-06:
Hello @zhou88mang ,

Thank you for the logs.

Can you please attach the sdkconfig at your side?

zhou88mang, 2023-06-06:
> Hello @zhou88mang ,
> 
> Thank you for the logs.
> 
> Can you please attach the sdkconfig at your side?

this is sdkconfig :  
[sdkconfig.txt](https://github.com/espressif/esp-hosted/files/11660372/sdkconfig.txt)

zhou88mang, 2023-06-06:
> 
Thank for your reply !
I'll take a closer look at the protocol stack I'm using

zhou88mang, 2023-06-06:
> I hope you have not changed the code apart from enabling uart and using max SCO connections.
> 
> Crux here is that ESP-Hosted uses HCI interface (stack over host, controller over esp) whereas, ESP-IDF example hfp_ag uses Bluedroid directly in conjunction with controller locally present.

but, if i just use HFP-AG demo , i did not found any ways to configure pcm format , it just provided  the function "app_gpio_pcm_io_cfg", I don't know how it generated these pcm signals inside,  I suspect this was done inside the controller,
maybe it has been done in function "esp_bt_controller_init" ,  i dont know.

mantriyogesh, 2023-06-06:
The Bluetooth team was looking in this matter till you had closed https://github.com/espressif/esp-idf/issues/11589

mantriyogesh, 2023-06-27:
any progress ahead?