espressif/esp-hosted #241: (Retrieving CTRL Sync Message) ctrl_wait_and_parse_sync_resp fails due to rx_buf_len always being 0

lawrencerb, 2023-07-06:
Hello

I believe I've found an issue in the function ctrl_wait_and_parse_sync_resp. (esp_hosted_fg > control_lib > ctrl_core.c)

In this function, rx_buf_len is always set to 0 and never modified before it's checked:
```
ctrl_cmd_t * ctrl_wait_and_parse_sync_resp(ctrl_cmd_t *app_req)
{
    ctrl_cmd_t * rx_buf = NULL;
    int rx_buf_len = 0;

    rx_buf = get_response(&rx_buf_len, app_req->cmd_timeout_sec);
    if (!rx_buf || !rx_buf_len) {
        printf("Response not received\n");
        if (rx_buf) {
            mem_free(rx_buf);
        }
    }
    return rx_buf;
}
```
Because rx_buf_len is always 0, the if (!rx_buf || !rx_buf_len) check will always pass the !rx_buf_len condition. This causes the function to print "Response not received" and free rx_buf every time, regardless of whether a valid response was received.

mantriyogesh, 2023-07-06:
Hello @lawrencerb 
Can you please help send 
1. Esp log
2. Host log - dmesg, commands fired and their output?

Is the ESP-Hosted ported correctly?

lawrencerb, 2023-07-06:
Yes, I'm sorry. I'm porting the Host solution to ThreadX and forgot to pass on the pointer for the length. My Bad
`static ctrl_cmd_t * get_response(int *read_len, int timeout_sec)
{
	uint8_t * buf = NULL;
	esp_queue_elem_t *elem = NULL;
	int ret = 0;
	timeout_sec = timeout_sec * (UINT)TX_TIMER_TICKS_PER_SECOND;

	/* 1. Any problems in response, return NULL */
	if (!read_len) {
		printf("Invalid input parameter\n");
		return NULL;
	}

	/* 2. If timeout not specified, use default */
	if (!timeout_sec)
		timeout_sec = (UINT)DEFAULT_CTRL_RESP_TIMEOUT * (UINT)TX_TIMER_TICKS_PER_SECOND;

	/* 3. Wait for response */
	ret = hosted_get_semaphore((UINT)READ_SEMAPHORE, timeout_sec);
	if (ret) {
		if (errno == ETIMEDOUT)
			printf("Control response timed out after %u sec\n", timeout_sec);
		else
			printf("ctrl lib error[%u] in sem of timeout[%u]\n", errno, timeout_sec);
		/* Unlock semaphore in negative case */
		hosted_post_semaphore((UINT)CTRL_SEMAPHORE);
		return NULL;
	}

	/* 4. Fetch response from `ctrl_msg_queue` */
	ret = tx_queue_receive(&ctrl_msg_queue, &elem, TX_NO_WAIT);
	if (ret == TX_SUCCESS){

		if (!(elem->buf_len)) {
			mem_free(elem->buf);
			mem_free(elem);
			return NULL;
		}
		buf = elem->buf;
		mem_free(elem);
		/* Free queue element and return the rx data */
		return (ctrl_cmd_t*)buf;

	} else {
		printf("Failed to receive pointer from ctrl_msg_queue\r\n");
		assert(ret);
		return NULL;
	}
	return NULL;
}`

closed 2023-07-06 by lawrencerb