espressif/esp-hosted #242: esp hosted fg error with -84

gauthamikosanam, 2023-07-13:
Hi 
I am using arm cortex a5d2x microprocessor as host and esp32 module as slave. In between this communication I am getting error.
I have used fg 0.0.5 version and compiled in kernel and flashed the respective binaries in esp 32 module.
Attaching the logs for both host and esp ,please let me know the solution.
[host_fg.txt](https://github.com/espressif/esp-hosted/files/12036708/host_fg.txt)
[esp_fg.txt](https://github.com/espressif/esp-hosted/files/12036711/esp_fg.txt)
esp_sdio: probe of mmc0:0001:1 failed with error -84

mantriyogesh, 2023-07-13:
Send us setup image.

Please go through https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md to understand things required while porting sdio for FG software.

Is the NG worked earlier and FG only is not working?

gauthamikosanam, 2023-07-13:
Yes NG worked earlier now I want softap mode so I moved to FG.

mantriyogesh, 2023-07-13:
did you try to tune dio timing https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#242-sdio?

gauthamikosanam, 2023-07-13:
Can you please send the patch for esp hosted fg sdio timing -SDIO_SLAVE_TIMING_NSEND_PSAMPLE.
Beacuse I am only flashing the bin files to board.Not having the idea of the source code, please let me know that process.

mantriyogesh, 2023-07-13:
Hello @gauthamikosanam 

We used to provide the binaries as ESP side code was not open sourced.
Now you should not face issue doing that.
We have added simple scripts to setup environment & make the flashing as easy as possible.

SDIO setup document is at https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SDIO_setup.md

Anyway, Please check the full FG documentation at https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg with keeping SDIO_only mode in mind.
Some of the earlier reported GitHub issues would have automatically resolved if the documentation was followed step wise and correctly. 

We try to keep documentation as small as possible and up to date.

gauthamikosanam, 2023-07-13:
I have flashed the sdio only binary files to esp32 module and can you set the sdio timing to SDIO_SLAVE_TIMING_NSEND_PSAMPLE and send the patch,because previous also for NG you have send the patch for sdio timing.Then I can able to communicate between host and esp module.

mantriyogesh, 2023-07-17:
please help close this issue.

gauthamikosanam, 2023-07-17:
Hi Yogesh,
I could not able to compile the source code,getting errors.Could you please send the bin files with respect to sdio timing ,will flash to the esp32 target board.
Attaching the error log.
[CMakeOutput.log](https://github.com/espressif/esp-hosted/files/12066420/CMakeOutput.log)
[CMakeError.log](https://github.com/espressif/esp-hosted/files/12066422/CMakeError.log)
gauthami@gauthami-Latitude-3420:~/esp-hosted-release-fg-v0.0.5/esp_hosted_fg/esp/esp_driver/network_adapter$ cmake .
*************************************************************************************
                    Building ESP-Hosted-FG Firmware version :: 0.0.5 
*************************************************************************************
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
CMake Error at /home/gauthami/esp-idf/tools/cmake/build.cmake:365 (message):
  Failed to run Python dependency check.  Python: python, Error: No such file
  or directory
Call Stack (most recent call first):
  /home/gauthami/esp-idf/tools/cmake/build.cmake:500 (__build_check_python)
  /home/gauthami/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!

mantriyogesh, 2023-07-17:
As my earlier comment I had mentioned, please use base commit as https://github.com/espressif/esp-hosted/commit/11e90f185a5a9922343e73e44032816a0e3ddedb 

Can you show output of 
$ cd <esp-hosted-root-dir>
$ git branch
$ git log | head -5

Also, I see your current folder is <esp_hosted_fg/esp/esp_driver/network_adapter> which is not matched as per documentation. should have followed the exact steps.

[image]

mantriyogesh, 2023-07-17:
also make sure you have C compiler installed where you are trying to build.

gauthamikosanam, 2023-07-25:
Hi Yogesh,
I have taken the esp hosted fg 0.0.5 version firmware.
[image]
I have flashed the bin files to board-esp hosted fg 0.0.5.tgz
I have taken the source code.tar.gz for esp32 driver.
is this process correct? I am unable to  understand how to take same git base 11e90f1 for host and esp.

mantriyogesh, 2023-07-25:
ESP-Hosted-FG building procedure.
This is all well documented. I am repeating here.

1. Checkout ESP-Hosted in new directory, say slave_hosted_25jul23
2. git checkout https://github.com/espressif/esp-hosted/commit/11e90f185a5a9922343e73e44032816a0e3ddedb
3. git submodule update --init --recursive
4. scp this code base dir slave_hosted_25jul23 to host with name say host_hosted_25jul23
5. in slave_hosted_25jul23, follow documentation to flash the ESP firmware using source compilation in above image. Please spare some time to check the documentation, repeating the documentation as comment really does not make any sense (again and again)
6. in host_hosted_25jul23, check rpi_init.sh if port is needed. Porting guide is at https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md
Unless it is ported correctly, kernel module will not work.
7. build kernel module using rpi_init.sh (check documentation)

gauthamikosanam, 2023-07-27:
Hi yogesh,
I have taken the git checkout of 11e90f1 and tried to compile the driver of fg esp32 getting error.
/home/wifi/esp_slave_hosted/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c:457:22: error: passing argument 1 of 'sdio_release_host' from incompatible pointer type [-Werror=incompatible-pointer-types]
    RELEASE_SDIO_HOST(context);
                      ^
/home/wifi/esp_slave_hosted/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c:54:50: note: in definition of macro 'RELEASE_SDIO_HOST'
   #define RELEASE_SDIO_HOST(x) sdio_release_host(x)
                                                  ^
In file included from /home/wifi/esp_slave_hosted/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c:22:0:
/home/yocto/build/tmp/work-shared/rugged-board-a5d2x-sd1/kernel-source/include/linux/mmc/sdio_func.h:118:13: note: expected 'struct sdio_func *' but argument is of type 'struct esp_sdio_context *'
 extern void sdio_release_host(struct sdio_func *func);

Any suggestion for this error?

mantriyogesh, 2023-07-27:
This is fixed on master, please do `git pull`

gauthamikosanam, 2023-07-27:
Done with git pull,still I am getting the errors.
/home/wifi/esp_slave_hosted/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c:76:31: error: 'ESP_DEVICE_ID_2' undeclared here (not in a function); did you mean 'ESP_DEVICE_ID_1'?
  { SDIO_DEVICE(ESP_VENDOR_ID, ESP_DEVICE_ID_2) },
                               ^
/home/yocto/build/tmp/work-shared/rugged-board-a5d2x-sd1/kernel-source/include/linux/mmc/sdio_func.h:97:31: note: in definition of macro 'SDIO_DEVICE'
  .vendor = (vend), .device = (dev)
I included the header file esp_sdio_decl.h still getting error.

mantriyogesh, 2023-07-27:
Have you taken 
https://github.com/espressif/esp-hosted/blob/0702eda3ceca996e094641bfdf402e2f5453f3e3/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L248-L251
into consideration from rpi_init.sh?

gauthamikosanam, 2023-07-27:
No local changes , I am compiling the host_driver/esp32/Makefile
In Makefile I am using, 
CR_C := arm-linux-gnueabihf-
all: 
        make ARCH=arm CROSS_COMPILE=$(CROSS_COMPILE) -C /home/yocto/build/tmp/work/rugged_board_a5d2x_sd1-poky-linux-musleabi/linux-rba5d2x/4.9-r0/build M=$(PWD) modules

mantriyogesh, 2023-07-27:
please post your rpi_init.sh output

Use 
```bash
$ bash -x rpi_init.sh 
```

gauthamikosanam, 2023-07-27:
Getting ESP_VENDOR_ID,ESP_DEVICE_ID_1 and ESP_DEVICE_ID_2 errors
/home/wifi/esp_slave_hosted/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio_decl.h:91:2: error: #error "SDIO not supported for selected ESP device"
 #error "SDIO not supported for selected ESP device"
root@0d4c4bb56531:/home/wifi/esp_slave_hosted/esp_hosted_fg/host/linux/host_control# bash -x rpi_init.sh
+ RESETPIN=
+ BT_INIT_SET=0
+ TEST_RAW_TP=0
+ IF_TYPE=sdio
+ MODULE_NAME=esp32_sdio.ko
+ ESP_SLAVE_CHIPSET=
+ parse_arguments
+ '[' '' '!=' '' ']'
+ '[' sdio = '' ']'
+ '[' sdio == sdio ']'
+ select_esp_slave
+ case $ESP_SLAVE_CHIPSET in
+ echo '***** Err: Please set expected ESP slave chipset ****'
***** Err: Please set expected ESP slave chipset ****
+ exit 1

gauthamikosanam, 2023-07-27:
In rpi_init.sh file kept esp slave chipset to esp32 and run the command bash -x rpi_init.sh
Attaching the output.
[fg log.txt](https://github.com/espressif/esp-hosted/files/12181546/fg.log.txt)

gauthamikosanam, 2023-07-28:
Hi Yogesh,
Compiled the driver but not inserting the driver getting error.
esp_sdio: probe of mmc0:0001:1 failed with error -110
esp_sdio: probe of mmc0:0001:2 failed with error -110
Attaching the logs of both host and esp.
[esp fg.txt](https://github.com/espressif/esp-hosted/files/12192697/esp.fg.txt)
[host_log.txt](https://github.com/espressif/esp-hosted/files/12192705/host_log.txt)

mantriyogesh, 2023-07-28:
Check suggestions in old but and try again. Specially wet timings

gauthamikosanam, 2023-08-01:
How can I change the sdio timings?

mantriyogesh, 2023-08-01:
Have you checked this comment: https://github.com/espressif/esp-hosted/issues/242#issuecomment-1633738540? 
Let us know once followed porting guide. your answers should lie in there.

gauthamikosanam, 2023-08-02:
Hi Yogesh,
root@0d4c4bb56531:/home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/network_adapter/main# git branch
* (HEAD detached from 11e90f1)
I changed this line in main/sdio_slave_api.c file
-	ESP_LOGI(TAG, "%s: ESP32 SDIO timing: %u\n", __func__, config.timing);
+	ESP_LOGI(TAG, "%s: Base Hosted expected: 11e90f1\nESP32 SDIO timing: %u\n", __func__, config.timing);
After that cmake I did.
root@0d4c4bb56531:/home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver# cmake .
HEAD is now at 0702eda Merge branch 'bugfix/sdio_claim_release_macros_ym2_26jul' into 'master'
fatal: reference is not a tree: cbce221e88d52665523093b2b6dd0ebe3f1243f1
fatal: reference is not a tree: 91dc99a3ce09b66c51b8be884e2ed6b6e1558747
Unable to checkout 'cbce221e88d52665523093b2b6dd0ebe3f1243f1' in submodule path 'esp-idf'
Unable to checkout '91dc99a3ce09b66c51b8be884e2ed6b6e1558747' in submodule path '../../../esp_hosted_ng/esp/esp_driver/esp-idf'
fatal: reference is not a tree: d037ec89546fad14b5c4d5456c2e23a71e554966
fatal: reference is not a tree: ddd61e9a276926496437665dd217cba2850db6b9
fatal: reference is not a tree: b207a7da7a9758dbccea0dba1023fd36c47b3e65
fatal: reference is not a tree: 9478bc510fa9fc792d98efbc80dc547b97a13b99
fatal: reference is not a tree: eeecc49ce8af123cf8ad40efdb9673e37b56230f
fatal: reference is not a tree: 998e05636344949ba219e8ac56feb733e27bf5d2
fatal: reference is not a tree: da0306da9b882c4bbd4c6ed5ee497c5c0e9945c4
fatal: reference is not a tree: d348621ca93571343a56862df7de4ff3bc9b5667
fatal: reference is not a tree: 269a58f18597aedec31c439e34cdaea47229dda3
fatal: reference is not a tree: 65b3c0857c9f207dcb0b646f2b4ebd8e5edf83c1
fatal: reference is not a tree: dffabb067fb3c39f486033d2e47eb4b1416f0c82
fatal: reference is not a tree: 785e946222f6c6ab453d7d47c15c551d4a181f79
fatal: reference is not a tree: 5beae143700db54c6e9bd4b15a568abe2f305723
fatal: reference is not a tree: 0dbb3f71c5f6fae3747a9d935372773762baf852
fatal: reference is not a tree: 7d2bf62b7e6afaf38153041a9d53c21aeeca9a25
Unable to checkout 'd037ec89546fad14b5c4d5456c2e23a71e554966' in submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Unable to checkout 'ddd61e9a276926496437665dd217cba2850db6b9' in submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib'
Unable to checkout 'b207a7da7a9758dbccea0dba1023fd36c47b3e65' in submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib'
Unable to checkout '9478bc510fa9fc792d98efbc80dc547b97a13b99' in submodule path 'components/bt/host/nimble/nimble'
Unable to checkout 'eeecc49ce8af123cf8ad40efdb9673e37b56230f' in submodule path 'components/cmock/CMock'
Unable to checkout '998e05636344949ba219e8ac56feb733e27bf5d2' in submodule path 'components/esp_phy/lib'
Unable to checkout 'da0306da9b882c4bbd4c6ed5ee497c5c0e9945c4' in submodule path 'components/esp_wifi/lib'
Unable to checkout 'd348621ca93571343a56862df7de4ff3bc9b5667' in submodule path 'components/json/cJSON'
Unable to checkout '269a58f18597aedec31c439e34cdaea47229dda3' in submodule path 'components/lwip/lwip'
Unable to checkout '65b3c0857c9f207dcb0b646f2b4ebd8e5edf83c1' in submodule path 'components/mbedtls/mbedtls'
Unable to checkout 'dffabb067fb3c39f486033d2e47eb4b1416f0c82' in submodule path 'components/mqtt/esp-mqtt'
Unable to checkout '785e946222f6c6ab453d7d47c15c551d4a181f79' in submodule path 'components/openthread/lib'
Unable to checkout '5beae143700db54c6e9bd4b15a568abe2f305723' in submodule path 'components/openthread/openthread'
Unable to checkout '0dbb3f71c5f6fae3747a9d935372773762baf852' in submodule path 'components/spiffs/spiffs'
Unable to checkout '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25' in submodule path 'components/unity/unity'
Detecting the Python interpreter
Checking "python3" ...
Checking "python" ...
"python" has been detected
Checking Python compatibility
Traceback (most recent call last):
  File "/home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf/tools/python_version_checker.py", line 45, in <module>
    check()
  File "/home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf/tools/python_version_checker.py", line 39, in check
    _ver_to_str(OLDEST_PYTHON_SUPPORTED), _ver_to_str(sys.version_info[:3])
RuntimeError: ESP-IDF supports Python 3.7 or newer but you are using Python 2.7.12. Please upgrade your installation as described in the documentation.
###### Setup Done ######
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver
Then again whatever changes i made in main/sdio_slave_api.c file, that line is not there. 
After cmake how to flash the bin files of sdio only to board. Where is the path?

mantriyogesh, 2023-08-02:
1. Is your development machine internet connected?
3. Cmake cleans all changed file as it is said in the documentation. You can change and git add and  commit locally to save the changes.

Alternatively, if you plan to reuse existing IDF, you can change the sdio file with changed timing. And in existing IDF environment, `$ idf.py flash monitor`

We will take this input to improve to allow changed files in cmake.

gauthamikosanam, 2023-08-03:
Hi @mantriyogesh ,

    Is your development machine internet connected?
--> it is connected internet. not sure what is the dependency. 
    Cmake cleans all changed file as it is said in the documentation. You can change and git add and commit locally to save the changes.
--> i am simply given "Cmake", after that local changes not found.

Alternatively, if you plan to reuse existing IDF, you can change the sdio file with changed timing. And in existing IDF environment, $ idf.py flash monitor
Getting this error.
gauthami@gauthami-Latitude-3420:~/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf$ . ./export.sh 
Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Adding ESP-IDF tools to PATH...
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Checking if Python packages are up to date...
Constraint file: /home/gauthami/.espressif/espidf.constraints.v5.2.txt
Requirement files:
 - /home/gauthami/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/gauthami/.espressif/python_env/idf5.2_py3.8_env/bin/python
The following Python requirements are not satisfied:
Requirement 'esptool==4.7.dev1,~=4.5' was not met. Installed version: 4.7.dev1
To install the missing packages, please run "install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/gauthami/.espressif/python_env/idf5.2_py3.8_env
    Python interpreter used: /home/gauthami/.espressif/python_env/idf5.2_py3.8_env/bin/python

mantriyogesh, 2023-08-03:
This is not the earlier IDF. this is new one from Hosted repo. if you want to use this,
Please do
```sh
$ rm -rf ~/.espressif
$ cd ~/esp_slave_hosted
$ git clean -f -x && git clean -xffd && git submodule foreach --recursive git clean -xffd
$ git submodule update --init --recursive
$ git status # should show clean
$ cd ~/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf
$ git clean -f -x && git clean -xffd && git submodule foreach --recursive git clean -xffd
$ git submodule update --init --recursive
$ ./install.sh
$ . ./export.sh
$ cd ~/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/network_adapter
$ rm -rf sdkconfig build
<Do your changes if not already present for sdio timing>
$ idf.py flash monitor
```

gauthamikosanam, 2023-08-03:
Hi,
git status
root@0d4c4bb56531:/home/wifi/esp_slave_hosted# git status
HEAD detached from 11e90f1
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

	modified:   esp_hosted_fg/esp/esp_driver/esp-idf (new commits, modified content)
	modified:   esp_hosted_ng/esp/esp_driver/esp-idf (new commits)

no changes added to commit (use "git add" and/or "git commit -a")

root@0d4c4bb56531:/home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf# ./install.sh 
Detecting the Python interpreter
Checking "python3" ...
Checking "python" ...
"python" has been detected
Checking Python compatibility
Traceback (most recent call last):
  File "/home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf/tools/python_version_checker.py", line 45, in <module>
    check()
  File "/home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf/tools/python_version_checker.py", line 39, in check
    _ver_to_str(OLDEST_PYTHON_SUPPORTED), _ver_to_str(sys.version_info[:3])
RuntimeError: ESP-IDF supports Python 3.7 or newer but you are using Python 2.7.12. Please upgrade your installation as described in the documentation.

mantriyogesh, 2023-08-03:
install.sh would not be useful unless your 'modified goes away'

share output for all:
$ cd ~/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf
$ git clean -f -x && git clean -xffd && git submodule foreach --recursive git clean -xffd
$ git submodule update --init --recursive
$ cd ~/esp_slave_hosted/esp_hosted_ng/esp/esp_driver/esp-idf
$ git clean -f -x && git clean -xffd && git submodule foreach --recursive git clean -xffd
$ git submodule update --init --recursive

$ git status

mantriyogesh, 2023-08-03:
By the way, this falls under ESP-IDF.

If you face issue, last line is self explanatory:
RuntimeError: ESP-IDF supports Python 3.7 or newer but you are using Python 2.7.12. Please upgrade your installation as described in the documentation

gauthamikosanam, 2023-08-03:
Hi ,
Please find the log of each command.
[log.txt](https://github.com/espressif/esp-hosted/files/12250175/log.txt)

gauthamikosanam, 2023-08-03:
Hi,
Meanwhile I am testing with raspberry Pi 3 Model B+ as target.
Please find the logs.
[   77.877892] Features supported are:
[   77.877901] 	 * WLAN
[   77.877906] 	 * BT/BLE
[   77.877911] 	   - HCI over SDIO
[   77.877916] 	   - BT/BLE dual mode 
These are not getting.
[esp.txt](https://github.com/espressif/esp-hosted/files/12250200/esp.txt)
[rasp_host.txt](https://github.com/espressif/esp-hosted/files/12250201/rasp_host.txt)

mantriyogesh, 2023-08-03:
root@0d4c4bb56531:/home/wifi/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/esp-idf# git submodule update --init --recursive
fatal: reference is not a tree: d037ec89546fad14b5c4d5456c2e23a71e554966
fatal: reference is not a tree: ddd61e9a276926496437665dd217cba2850db6b9
fatal: reference is not a tree: b207a7da7a9758dbccea0dba1023fd36c47b3e65
fatal: reference is not a tree: 9478bc510fa9fc792d98efbc80dc547b97a13b99
fatal: reference is not a tree: eeecc49ce8af123cf8ad40efdb9673e37b56230f
fatal: reference is not a tree: 998e05636344949ba219e8ac56feb733e27bf5d2
fatal: reference is not a tree: da0306da9b882c4bbd4c6ed5ee497c5c0e9945c4
fatal: reference is not a tree: d348621ca93571343a56862df7de4ff3bc9b5667
fatal: reference is not a tree: 269a58f18597aedec31c439e34cdaea47229dda3
fatal: reference is not a tree: 65b3c0857c9f207dcb0b646f2b4ebd8e5edf83c1
fatal: reference is not a tree: dffabb067fb3c39f486033d2e47eb4b1416f0c82
fatal: reference is not a tree: 785e946222f6c6ab453d7d47c15c551d4a181f79
fatal: reference is not a tree: 5beae143700db54c6e9bd4b15a568abe2f305723
fatal: reference is not a tree: 0dbb3f71c5f6fae3747a9d935372773762baf852
fatal: reference is not a tree: 7d2bf62b7e6afaf38153041a9d53c21aeeca9a25
Unable to checkout 'd037ec89546fad14b5c4d5456c2e23a71e554966' in submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Unable to checkout 'ddd61e9a276926496437665dd217cba2850db6b9' in submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib'
Unable to checkout 'b207a7da7a9758dbccea0dba1023fd36c47b3e65' in submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib'
Unable to checkout '9478bc510fa9fc792d98efbc80dc547b97a13b99' in submodule path 'components/bt/host/nimble/nimble'
Unable to checkout 'eeecc49ce8af123cf8ad40efdb9673e37b56230f' in submodule path 'components/cmock/CMock'
Unable to checkout '998e05636344949ba219e8ac56feb733e27bf5d2' in submodule path 'components/esp_phy/lib'
Unable to checkout 'da0306da9b882c4bbd4c6ed5ee497c5c0e9945c4' in submodule path 'components/esp_wifi/lib'
Unable to checkout 'd348621ca93571343a56862df7de4ff3bc9b5667' in submodule path 'components/json/cJSON'
Unable to checkout '269a58f18597aedec31c439e34cdaea47229dda3' in submodule path 'components/lwip/lwip'
Unable to checkout '65b3c0857c9f207dcb0b646f2b4ebd8e5edf83c1' in submodule path 'components/mbedtls/mbedtls'
Unable to checkout 'dffabb067fb3c39f486033d2e47eb4b1416f0c82' in submodule path 'components/mqtt/esp-mqtt'
Unable to checkout '785e946222f6c6ab453d7d47c15c551d4a181f79' in submodule path 'components/openthread/lib'
Unable to checkout '5beae143700db54c6e9bd4b15a568abe2f305723' in submodule path 'components/openthread/openthread'
Unable to checkout '0dbb3f71c5f6fae3747a9d935372773762baf852' in submodule path 'components/spiffs/spiffs'
Unable to checkout '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25' in submodule path 'components/unity/unity'

I don't understand 'unable to checkout' errors.
checkout IDF with branch 'release/v5.1'
make sure there are no 'modified' and repo is clean in 'git status'

you will have to check git by yourself.
once idf is clean, 

$ ./install.sh
$ . ./export.sh
$ cd ~/esp_slave_hosted/esp_hosted_fg/esp/esp_driver/network_adapter
$ rm -rf sdkconfig build
<Do your changes if not already present for sdio timing>
$ idf.py flash monitor

mantriyogesh, 2023-08-03:
1. Are there external pull-ups as mentioned in porting guide?
2. What is the timing of sdio used at ESP32?
3. run **bash** rpi_init.sh sdio
4. is it release binary or flashed from source? if release which release?
5. Which code base (git commit) was used for rpi kernel module building?

gauthamikosanam, 2023-08-03:
Hi,
1.yes mentioned in porting guide for esp32.
2.I took the release fg 0.0.5 version and flashed the binaries to board.
3.Master branch

gauthamikosanam, 2023-08-03:
Hi ,
Used FG 0.0.5 version but still getting read error at esp side.
[rasp_host.txt](https://github.com/espressif/esp-hosted/files/12251147/rasp_host.txt)

gauthamikosanam, 2023-08-04:
Hi Yogesh,
when I do idf.py flash monitor, getting error.Attaching the log,please let me know the solution.Could not able to tune the sdio timing. 
[esp .txt](https://github.com/espressif/esp-hosted/files/12257525/esp.txt)
.

mantriyogesh, 2023-08-04:
https://github.com/espressif/esp-hosted/issues/242#issuecomment-1663809513

Please check with ESP-IDF team if you have any troubles setting up IDF.
Without IDF correctly setup, ESP-Hosted firmware cannot be built.

gauthamik9, 2023-08-04:
Hi Yogesh, can you please provide the binary files w.r.t sdio timing to flash into esp32 board.
#if defined(CONFIG_IDF_TARGET_ESP32)
                .timing             = SDIO_SLAVE_TIMING_NSEND_PSAMPLE,
#endif

gauthamik9, 2023-08-08:
Hi Yogesh,
I have compiled the esp-idf but if i do idf.py build in network adaptor path getting cmake error.Attaching the log, please let me know the solution.

[esp driver log.docx](https://github.com/espressif/esp-hosted/files/12288571/esp.driver.log.docx)

mantriyogesh, 2023-08-08:
```diff
diff --git a/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c b/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c
index 5087d15..96b9f02 100644
--- a/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c
+++ b/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c
@@ -202,9 +202,7 @@ static interface_handle_t * sdio_init(void)
                 * In these cases, Please tune timing below using value from
                 * https://github.com/espressif/esp-idf/blob/release/v5.0/components/hal/include/hal/sdio_slave_types.h#L26-L38
                 * */
-#if defined(CONFIG_IDF_TARGET_ESP32C6)
                .timing             = SDIO_SLAVE_TIMING_NSEND_PSAMPLE,
-#endif
        };
        config.flags |= SDIO_SLAVE_FLAG_DEFAULT_SPEED;
```
[2023.08.08_gh242_sdio_timingchange_ESP32_sdio.tgz](https://github.com/espressif/esp-hosted/files/12288782/2023.08.08_gh242_sdio_timingchange_ESP32_sdio.tgz)

IDF: 79b1379662b547f6eb0e5fed33df70587b03d99c
ESP-Hosted: 8cb1ccb4da87da952b3b28e5544c849c1dcd7b67

Host side git commit **must be** 8cb1ccb4da87da952b3b28e5544c849c1dcd7b67

gauthamik9, 2023-08-17:
Hi Yogesh,
I have git commited 8cb1ccb in both host and esp board. In raspberry pi 3 model B+ as host and esp32 as slave it is working . This is the log.
[rasp_log.txt](https://github.com/espressif/esp-hosted/files/12366690/rasp_log.txt)
Same setup configured in arm processor as host ,esp32 as slave it is not going into esp probe function.This is the log.
root@rugged-board-a5d2x-sd1:~# modprobe esp32_sdio.ko resetpin=101
esp_sdio: probe of mmc0:0001:1 failed with error -110
esp_sdio: probe of mmc0:0001:2 failed with error -110
CIS tuple is happening
mmc0: queuing unknown CIS tuple 0x01 (3 bytes)
mmc0: queuing unknown CIS tuple 0x1a (5 bytes)
mmc0: queuing unknown CIS tuple 0x1b (8 bytes)
mmc1: SDHCI controller on b0000000.sdio-host [b0000000.sdio-host] using ADMA
ledtrig-cpu: registered to indicate activity on CPUs
atmel_aes f002c000.aes: version: 0x500
mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
atmel_aes f002c000.aes: Atmel AES - Using dma0chan0, dma0chan1 for DMA transfers
atmel_sha f0028000.sha: version: 0x510
atmel_sha f0028000.sha: using dma0chan2 for DMA transfers
mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
mmc0: new SDIO card at address 0001

Reset also hapenning
esp_reset, ESP32: Resetpin of Host is 101
esp_reset, ESP32: Triggering ESP reset.

Can you please suggest on this issue.

mantriyogesh, 2023-08-17:
But you had ng working on the same setup earlier. Correct?

gauthamik9, 2023-08-17:
Yes,NG driver is working with the same setup. I want soft ap mode support so shifted to FG dirver.

mantriyogesh, 2023-08-17:
Can you please try the ng again?
There is no difference in transport later for fg and ng.

gauthamik9, 2023-08-18:
Hi Yogesh,
I tried again with NG driver , it is working.
This is the log of host.
root@rugged-board-a5d2x-sd1:~# modprobe esp32_sdio.ko
esp_probe: ESP network device detected
Rx Pre ====== 0
Rx Pos ======  0
Tx Pre ======  0
Tx Pos ======  10
esp_probe:793
esp_sdio: probe of mmc0:0001:2 failed with error -22
root@rugged-board-a5d2x-sd1:~# 
Received ESP bootup event
EVENT: 3
ESP chipset detected [esp32]
EVENT: 0
ESP peripheral capabilities: 0x1d
ESP Bluetooth init
Capabilities: 0x1d. Features supported are:
         * WLAN on SDIO
         * BT/BLE
           - HCI over SDIO
           - BT/BLE dual mode
EVENT: 4
EVENT: 1
POWERON_RESET
esp32: ESP Firmware version: 1.0.2

root@rugged-board-a5d2x-sd1:~# ifconfig -a
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:48 

espsta0   Link encap:Ethernet  HWaddr C4:DE:E2:63:56:8C  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 04:91:62:D3:7A:C6  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:28 Base address:0x8000 

lo        Link encap:Local Loopback  
          LOOPBACK  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

mantriyogesh, 2023-08-18:
[2023.03.16_NG_timingNP_1.0.2.zip](https://github.com/espressif/esp-hosted/files/12377392/2023.03.16_NG_timingNP_1.0.2.zip)

Please try with NP binaries with new host side code from master: commit: 5910e160d20e99d6ea9ab5a45c0bbdb6afa65629 -> https://github.com/espressif/esp-hosted/tree/5910e160d20e99d6ea9ab5a45c0bbdb6afa65629/esp_hosted_ng/host

let me know If this works, will create fg bins.

Please stick to binaries and host side git commit. no other commit should be used.

gauthamik9, 2023-08-18:
Hi Yogesh,
For Fg driver, driver is loaded and created nodes in raspberry pi . How can I test for both station and ap mode.
If I give this command getting error.
wpa_supplicant -D nl80211 -i ethsta0 -c /etc/wpa_supplicant.conf &
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=ethsta0 disabled_11b_rates=0
ethsta0: Failed to initialize driver interface

gauthamik9, 2023-08-18:
I raspberry pi, for soft ap mode to test I am using c_support.
In this I am getting protobuf-c.c no such file or directory.
In protobuf-c folder .c file is not there. 
I tried with https://github.com/protobuf-c/protobuf-c.git but getting No such file or directory.

mantriyogesh, 2023-08-18:
> wpa_supplicant -D nl80211 -i ethsta0 -c /etc/wpa_supplicant.conf &

This is not the way to use FG driver, please check FG documentation.

mantriyogesh, 2023-08-18:
> I raspberry pi, for soft ap mode to test I am using c_support.
> In this I am getting protobuf-c.c no such file or directory.

If you follow the documentation correctly, you should not face this issue.

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Linux_based_readme.md#12-host-setup -> 
`Clone ESP-Hosted code repository`

gauthamik9, 2023-08-21:
In raspberrypi 3 model B+ , I am testing the soft ap mode,getting this error.
pi@raspberrypi:~/Documents/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support $ sudo ./test.out set_softap_vendor_ie
Enable heartbeat with duration 20
Heartbeat operation successful
Freeing 0x0x350730
Success in set vendor specific ie

Requested operation complete
Sleeping for some time just to showcase heartbeat
2023-08-18 17:48:12 >  App EVENT: Heartbeat event [0]
2023-08-18 17:48:37 >  App EVENT: Heartbeat event [1]
Disable Heartbeat
Heartbeat operation successful
pi@raspberrypi:~/Documents/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support $ sudo ./test.out reset_softap_vendor_ie
Enable heartbeat with duration 20
Heartbeat operation successful
Success in set vendor specific ie

Requested operation complete
Sleeping for some time just to showcase heartbeat
2023-08-18 17:49:15 >  App EVENT: Heartbeat event [0]
2023-08-18 17:49:40 >  App EVENT: Heartbeat event [1]
Disable Heartbeat
Heartbeat operation successful
pi@raspberrypi:~/Documents/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support $ sudo ./test.out softap_start
Enable heartbeat with duration 20
Heartbeat operation successful
esp32 softAP started 
ethap0 interface down
MAC address 30:ae:a4:10:95:59 set to ethap0 interface
ethap0 interface up

Requested operation complete
Sleeping for some time just to showcase heartbeat
2023-08-18 17:50:19 >  App EVENT: Heartbeat event [0]
2023-08-18 17:50:44 >  App EVENT: Heartbeat event [1]
Disable Heartbeat
Heartbeat operation successful

daemon --no-resolv --no-poll --dhcp-script=/system/bin/dhcp_announce --dhcp-range=192.168.4.1,192.168.4.20,1h

dnsmasq: failed to create listening socket for port 53: Address already in use

any suggestion.

mantriyogesh, 2023-08-21:
It is illustrated just as an example to show how higher layer **can** use the dhcp server.
dnsmasq issues are out of scope for ESP-Hosted.

gauthamik9, 2023-08-21:
If I give this command ,getting the error.
sudo python3 test.py start_softap --ssid ESP_WiFi --pwd ESP_WiFi@123 --channel 6 --sec_prot wpa_wpa2_psk --max_conn 9 --hide_ssid False --bw 40
control response timed out after 30 sec
Response not received.

mantriyogesh, 2023-08-21:
Is this same observed always in python and not in C?
Or also fails in c some times?

Can you please run the raw throughput test and tell the spi level performance?

mantriyogesh, 2023-08-21:
Most likely this is spi level timing issue.
Please run raw throughput test both Tx and Rx) check [raw TP doc](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md)

Which esp chipset is being used?

From porting guide at:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#24-peripheral-configurations

Check all things ( smaller wires, smaller clk, timing tuning etc step by step)

gauthamik9, 2023-08-22:
Hi Yogesh,
I am using esp32 wroom 32E module.
In raspberry pi 3 model B + I am trying to connect station mode,but ip is not pinging.
Attaching the logs of both host and esp.
[rasp_log.txt](https://github.com/espressif/esp-hosted/files/12404824/rasp_log.txt)
[esp_log.txt](https://github.com/espressif/esp-hosted/files/12404828/esp_log.txt)

mantriyogesh, 2023-08-22:
As seen from logs, you can send the packets from host to esp. but esp to host packet are not sent some times.
Also with combination,
`raspberry pi 3B+ <--sdio--> esp32`
you do not need timing adjustment.
so test with sdio timing =0;

mantriyogesh, 2023-08-22:
Make sure that you have very small wires and external pull-ups in place. if you do not do these two things, you may still face issues.

[109 more replies]

closed 2023-09-29 by mantriyogesh