espressif/esp-hosted #244: checksum failed

zhou88mang, 2023-07-19:
i have ported the esp-hosted with lwip , when in the STA mode, i start dhcp , i found that checksum failed ( UDP (or UDP Lite) datagram discarded due to failing checksum) errors occur frequently ,
like this:
[13.510975] Connected to OPPOzm
[13.511463] dhcp_start(netif=801497cc) st2
[13.512096] dhcp_start(): mallocing new DHCP client
[13.512795] dhcp_start(): allocated dhcp
[13.513379] dhcp_start(): starting DHCP configuration
[13.514168] udp_bind(ipaddr = 0.0.0.0, port = 68)
[13.514859] udp_bind: bound to 0.0.0.0, port 68)
[13.515507] udp_connect: connected to 0.0.0.0, port 67)
[13.516300] dhcp_discover()
[13.516721] transaction id xid(5851f42d)
[13.517263] dhcp_discover: making request
[13.517891] dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
[13.519120] udp_send: added header in given pbuf 8011336c
[13.519916] udp_send: sending datagram of length 316
[13.520616] udp_send: UDP packet length 316
[13.521254] udp_send: UDP checksum 0xeffa
[13.521837] udp_send: ip_output_if (,,,,0x11,)
[13.522480] ip4_output_if: st2
[13.522908] IP header:
[13.523230] +-------------------------------+
[13.523868] | 4 | 5 |  0x00 |       336     | (v, hl, tos, len)
[13.524726] +-------------------------------+
[13.525365] |        0      |000|       0   | (id, flags, offset)
[13.526271] +-------------------------------+
[13.526909] |  255  |   17  |    0xba9d     | (ttl, proto, chksum)
[13.527819] +-------------------------------+
[13.528459] |    0  |    0  |    0  |    0  | (src)
[13.529158] +-------------------------------+
[13.529795] |  255  |  255  |  255  |  255  | (dest)
[13.530539] +-------------------------------+
[13.531178] ip4_output_if: call netif->output()
[13.531919] dhcp_discover: deleting()
[13.532190] dhcp_discover: SELECTING
[13.532675] dhcp_discover(): set request timeout 2000 msecs
[13.571517] ip_input: iphdr->dest 0xd927a8c0 netif->ip_addr 0x0 (0x0, 0x0, 0xd927a8c0)
[13.572736] ip_input: iphdr->dest 0xd927a8c0 netif->ip_addr 0x106a8c0 (0x27a8c0, 0x6a8c0, 0xd9000000)
[13.574122] ip_input: iphdr->dest 0xd927a8c0 netif->ip_addr 0x105a8c0 (0x27a8c0, 0x5a8c0, 0xd9000000)
[13.575506] ip4_input: UDP packet to DHCP client port 68
[13.576260] ip4_input: DHCP packet accepted.
[13.576894] ip4_input:
[13.577217] IP header:
[13.577580] +-------------------------------+
[13.578219] | 4 | 5 |  0x00 |       338     | (v, hl, tos, len)
[13.579076] +-------------------------------+
[13.579713] |    41089      |010|       0   | (id, flags, offset)
[13.580618] +-------------------------------+
[13.581258] |   64  |   17  |    0xc896     | (ttl, proto, chksum)
[13.582167] +-------------------------------+
[13.582805] |  192  |  168  |   39  |   89  | (src)
[13.583503] +-------------------------------+
[13.584142] |  192  |  168  |   39  |  217  | (dest)
[13.584842] +-------------------------------+
[13.585523] ip4_input: p->len 338 p->tot_len 338
[13.586214] udp_input: received datagram of length 318
[13.586961] UDP header:
[13.587286] +-------------------------------+
[13.587923] |        67     |        68     | (src port, dest port)
[13.588833] +-------------------------------+
[13.589471] |       318     |     0xea33    | (len, chksum)
[13.590275] +-------------------------------+
[13.590912] udp (192.168.39.217, 68) <-- (192.168.39.89, 67)
[13.591719] pcb (0.0.0.0, 68) <-- (0.0.0.0, 67)
[13.592408] udp_input: calculating checksum
[13.592998] udp_input: UDP (or UDP Lite) datagram discarded due to failing checksum

so , i set the board to static IP ,  under the circumstances , when i ping packets of different bytes, there will be different results, for example:
when i ping 1000 bytes, i can ping through the board successfully ,  if i ping 700 bytes , it will timeout
and the get the error : checksum failed for received ICMP echo 
the log like this:
ip_input: iphdr->dest 0xd927a8c0 netif->ip_addr 0xd927a8c0 (0x27a8c0, 0x27a8c0, 0xd9000000)
[333.428646] ip4_input: packet accepted on interface st
[333.429441] ip4_input:
[333.429767] IP header:
[333.430135] +-------------------------------+
[333.430774] | 4 | 5 |  0x00 |       728     | (v, hl, tos, len)
[333.431634] +-------------------------------+
[333.432318] |    39722      |000|       0   | (id, flags, offset)
[333.433226] +-------------------------------+
[333.433866] |  128  |    1  |    0xcc94     | (ttl, proto, chksum)
[333.434778] +-------------------------------+
[333.435419] |  192  |  168  |   39  |   60  | (src)
[333.436120] +-------------------------------+
[333.436761] |  192  |  168  |   39  |  217  | (dest)
[333.437508] +-------------------------------+
[333.438193] ip4_input: p->len 728 p->tot_len 728
[333.438842] icmp_input: ping
[333.439310] icmp_input: checksum failed for received ICMP echo

I don't know why it went wrong, can you give me some advice ?

mantriyogesh, 2023-07-19:
which code base you had used?

Can you please check https://github.com/espressif/esp-hosted/issues/186#issuecomment-1639235847?

mantriyogesh, 2023-07-19:
it might also be possible that your SPI timing is incorrect?
Please try to run raw throughput test Tx and Rx (better to do from https://github.com/espressif/esp-hosted/issues/186#issuecomment-1639235847 with two ESP boards connected)

zhou88mang, 2023-07-19:
> it might also be possible that your SPI timing is incorrect? Please try to run raw throughput test Tx and Rx (better to do from [#186 (comment)](https://github.com/espressif/esp-hosted/issues/186#issuecomment-1639235847) with two ESP boards connected)

can you tell me how to use ESP_as_Host_with_stable_LWIP.patch ?

mantriyogesh, 2023-07-19:
You can do connections as per https://github.com/espressif/esp-hosted/issues/186#issuecomment-1628491346
follow flashing procedure from https://github.com/espressif/esp-hosted/issues/186#issuecomment-1626859260 -> Flashing Procedure

in the host iperf example, once you get shell, you can connect to station using 

```
> sta <SSID> <Password>
```
Replace your AP's SSID and password here.

Once IP is assigned in host side logs, run the iperf.
For example, for iperf client,
```
> iperf -c <iperf_server_ip> -p <iperf_server_port> -i <reporting_duration> -t <running_duration>
```

these switches are iperf standard switches.
make sure iperf server is already running on the station (say laptop) connected to same AP, which Host ESP is connected to.

zhou88mang, 2023-07-19:
> You can do connections as per [#186 (comment)](https://github.com/espressif/esp-hosted/issues/186#issuecomment-1628491346) follow flashing procedure from [#186 (comment)](https://github.com/espressif/esp-hosted/issues/186#issuecomment-1626859260) -> Flashing Procedure
> 
> in the host iperf example, once you get shell, you can connect to station using
> 
> ```
> > sta <SSID> <Password>
> ```
> 
> Replace your AP's SSID and password here.
> 
> Once IP is assigned in host side logs, run the iperf. For example, for iperf client,
> 
> ```
> > iperf -c <iperf_server_ip> -p <iperf_server_port> -i <reporting_duration> -t <running_duration>
> ```
> 
> these switches are iperf standard switches. make sure iperf server is already running on the station (say laptop) connected to same AP, which Host ESP is connected to.

ok , thanks !