espressif/esp-hosted #245: Porting to ATSAMA5D over SDIO: wait_and_decode_cmd_resp: Command[4] timed out

guidogiunchi, 2023-07-19:
Hi! I'm trying to port the ng version to the **ATSAMA5D27**, specifically to the acme roadrunner SoC over **SDIO** mmc0.
I managed to get the hw part working fine using the commit `77aec01`, both on esp and linux, using **kernel 5.15.92**.

The only modifications done to this point are adding **kprint** to track the module while it runs its functions, not more.

The esp32 is working, I see all the SDIO regs being written and read (0x3ff55060, 0x3ff55058...) and the interface is brought up using 

`sudo ifconfig espsta0 up`

but on 

`sudo iw dev espsta0 scan`

The ESP32 gets the command and reports the start of the scan over uart.

it times out, as if the host is not receiving SDIO interrupts or similar via `wait_event_interruptible_timeout`

The **ESP32** gets the command and dutifully reports the scan start scan over **uart**.

The kernel log ends with 

```
> [...]
> cmd_scan_request
> prepare_command_request
> get_free_cmd_node
> esp_alloc_skb, len: 2048
> esp_alloc_skb, skb_ptr: 34030614
> queue_cmd_node
> wait_and_decode_cmd_resp
> esp_cmd_work
> esp_get_adapter
> esp_get_adapter
> esp_send_packet
> esp_send_packet: success, skb len 64
> esp32_sdio:wait_and_decode_cmd_resp: Command[4] timed out
> recycle_cmd_node
> reset_cmd_node
> esp32_sdio:cmd_scan_request: esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
```

I checked that the interrupts are coming through, via  `cat  /proc/interrupts`
running the scan function takes 
```
           CPU0
 21:        308  atmel-aic5  31 Level     mmc0
```
to
```
           CPU0
 21:        310  atmel-aic5  31 Level     mmc0
```
So the interrupts are being received from the SDIO..

I am trying to understant what the problem might be, it is _almost_ working, can you help me solve this issue please?

Thank you!
Guido

mantriyogesh, 2023-07-19:
Does it work when you use vanilla flavour?

guidogiunchi, 2023-07-25:
No, I added all the printk to get an idea when it was not working as is, I am tempted to bring out the oscilloscope and look ath the SDIO bus, if it has activity after the scan

guidogiunchi, 2023-07-25:
I tried also the release files and only compiling the host side, but it stops even earlier

guidogiunchi, 2023-07-26:
I am using the main branch code, compiled from the latest main branch push, not from the release.
The SDIO is working, as they exchange data and I can see "start data path" and "request scan" on the ESP32 debug UART.

IMHO all the hw side is working, I am missing just this timeout issue after the scan request.

mantriyogesh, 2023-07-26:
We will look into this problem.

guidogiunchi, 2023-09-04:
Hi, thank you and sorry for the wait, in Italy august is a weird vacation month and I lost access to my setup for most part of the month.

Back on track, since I was trying to diagnose the problem I only added a couple debug prints to
` esp/esp_driver/network_adapter/main/cmd.c`
Here is the git diff, I can remove them if you think it might be useful.

<details>
  <summary>git diff  esp/esp_driver/network_adapter/main/cmd.c</summary>

```
diff --git a/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c b/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c
index 35c921e..46048b8 100644
--- a/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c
+++ b/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c
@@ -708,6 +708,9 @@ int process_start_scan(uint8_t if_type, uint8_t *payload, uint16_t payload_len)
        interface_buffer_handle_t buf_handle = {0};
        uint8_t cmd_status = CMD_RESPONSE_SUCCESS;
        struct scan_request *scan_req;
+       
+       ESP_LOGI(TAG, "Slave -> Host: Start scan\n");
+
 
        /* Register to receive probe response and beacon frames */
        type = (1 << WLAN_FC_STYPE_BEACON) | (1 << WLAN_FC_STYPE_PROBE_RESP) |
@@ -747,6 +750,7 @@ int process_start_scan(uint8_t if_type, uint8_t *payload, uint16_t payload_len)
                        esp_wifi_register_mgmt_frame_internal(0, 0);
        }
 
+       ESP_LOGI(TAG, "Scan done\n");
        buf_handle.if_type = ESP_STA_IF;
        buf_handle.if_num = 0;
        buf_handle.payload_len = sizeof(struct command_header);
@@ -772,6 +776,7 @@ int process_start_scan(uint8_t if_type, uint8_t *payload, uint16_t payload_len)
                goto DONE;
        }
 
+       ESP_LOGI(TAG, "Slave -> Host: Sent command response\n");
        return ESP_OK;
 
 DONE:
@@ -779,7 +784,7 @@ DONE:
                free(buf_handle.payload);
                buf_handle.payload = NULL;
        }
-
+       ESP_LOGI(TAG, "Slave -> Host: Failed to send command response\n");
        return ret;
 }

```
</details>

The ESP32 prints out:

```
rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6668
ho 0 tail 12 room 4
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (28) boot: ESP-IDF 91dc99a3-dirty 2nd stage bootloader
I (28) boot: compile time 15:52:23
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
[... 83 lines cut ...]
I (180222) FW_MAIN: Start Data Path
I (180272) FW_SDIO_SLAVE: last reset cause: 1x
I (180376) FW_MAIN: Initial set up done
I (180852) FW_MAIN: INIT Interface command

I (180856) wifi:mode : sta (0c:b8:15:c3:0d:a8)
I (180856) wifi:enable tsf
I (180856) FW_CMD: Wifi Sta mode set

I (181430) FW_MAIN: Get MAC command

I (532216) FW_MAIN: Scan request

I (532216) FW_CMD: Slave -> Host: Start scan

I (532218) FW_CMD: Scan done

I (532218) FW_CMD: Slave -> Host: Sent command response
```

The host still prints out 

`esp32_sdio:wait_and_decode_cmd_resp: Command[4] timed out`

It sounds strange to me that all the handshakes are working but it fails at this moment

kapilkedawat, 2023-10-10:
Hi @guidogiunchi , apologies for delayed response, are you still facing this issue?