espressif/esp-hosted #246: SDIO Transfer Problem

zhou88mang, 2023-07-20:
project : esp-hosted-fg  
transport :  sdio

problem  1：
 in sdio_host.c 
function ： sdio_host_get_packet
 ```
stm_ret_t sdio_host_get_packet(void* out_data, size_t size,
		size_t* out_length, uint32_t wait_ms)
{
	stm_ret_t err = STM_OK;
	uint32_t len = 0, wait_time = 0, len_remain = 0;
	uint8_t* start_ptr = NULL;
	int len_to_send = 0, block_n = 0;

	if (size <= 0) {
		printf("Invalid size:%d\n\r", size);
		return STM_FAIL_INVALID_ARG;
	}

	for (;;) {
		err = esp_sdio_slave_get_rx_data_size(&len);

		if (err == STM_OK && len > 0) {
#if DEBUG_TRANSPORT
		//	printf("Expected length to be read %lu\n\n",len);
#endif
			break;
		}

		/* If no error and no data, retry */
		wait_time++;

		if (wait_time >= wait_ms) {
			return STM_FAIL_TIMEOUT;
		}

		hard_delay(1);
	}

	if (len > size) {
		printf("Pkt size to be read[%lu] > max sdio size supported[%u]\n\r",len, size);
		return STM_OK;
	}

	len_remain = len;
	start_ptr = (uint8_t*)out_data;

	do {
		/* currently driver supports only block size of 512 */

		block_n = len_remain / ESP_BLOCK_SIZE;

		if (block_n != 0) {
			len_to_send = ESP_BLOCK_SIZE;
#if DEBUG_TRANSPORT
		//	printf("block_n %u, len-to_send %lu\n\r",block_n,len_to_send);
#endif

			err = sdio_driver_read_blocks(SDIO_FUNC_1,
					ESP_SLAVE_CMD53_END_ADDR - len_remain,
					start_ptr, len_to_send, block_n);
		} else {
			len_to_send = len_remain;
			/* though the driver supports to split packet of unaligned size into length
			 * of 4x and 1~3, we still get aligned size of data to get higher
			 * efficiency. The length is determined by the SDIO address, and the
			 * remaining will be ignored by the slave hardware
			 */
			err = sdio_driver_read_bytes(SDIO_FUNC_1,
[... 530 lines cut ...]
45 45 45 45 45 45 45 45  45 45 45 45 45 45 45 45    EEEEEEEEEEEEEEEE
45 45 45 45 45 45 45 45  45 45 45 45 45 45 45 45    EEEEEEEEEEEEEEEE
45 45 45 45 45 45 45 45  45 45 45 45 45 45 45 45    EEEEEEEEEEEEEEEE
45 45 45 45 45 45 45 45  45 45 45 45 45 45 45 45    EEEEEEEEEEEEEEEE
45 45 45 45 45 45 45 45  45 45 45 45 45 45 45 45    EEEEEEEEEEEEEEEE
45 45 45 45 45 45 45 45  45 45 45 45 45 45 45 45    EEEEEEEEEEEEEEEE
45 45 45 45 46 46 46 46  46 46 46 46 46 46 46 46    EEEEFFFFFFFFFFFF
46 46 46 46 46 46 46 46  46 46 46 46 46 46 46 46    FFFFFFFFFFFFFFFF
46 46 46 46 46 46 46 46  46 46 46 46 46 46 46 46    FFFFFFFFFFFFFFFF
46 46 46 46 46 46 46 46  46 46 46 46 46 46 46 46    FFFFFFFFFFFFFFFF
46 46 46 46 46 46 46 46  46 46 46 46 46 46 46 46    FFFFFFFFFFFFFFFF
46 46 46 46 46 46 46 46  46 46 46 46 46 46 46 46    FFFFFFFFFFFFFFFF
46 46 46 46 46 46 46 46  47 47 47 47 47 47 47 47    FFFFFFFFGGGGGGGG
47 47 47 47 47 47 47 47  47 47 47 47 47 47 47 47    GGGGGGGGGGGGGGGG
47 47 47 47 47 47 47 47  47 47 47 47 47 47 47 47    GGGGGGGGGGGGGGGG
47 47 47 47 47 47 47 47  47 47 47 47 47 47 47 47    GGGGGGGGGGGGGGGG
47 47 47 47 47 47 47 47  47 47 47 47 47 47 47 47    GGGGGGGGGGGGGGGG
47 47 47 47 47 47 47 47  47 47 47 47 47 47 47 47    GGGGGGGGGGGGGGGG
47 47 47 47 47 47 47 47  47 47 47 47 48 48 48 48    GGGGGGGGGGGGHHHH
48 48 48 48 48 48 48 48  48 48 48 48 48 48 48 48    HHHHHHHHHHHHHHHH
48 48 48 48 48 48 48 48  48 48 48 48 48 48 48 48    HHHHHHHHHHHHHHHH
48 48 48 48 48 48                                   HHHHHH

```
if i send different number of bytes , it will produce different results.  in the example , i send 745 bytes data, and always it will lose 64bytes data.

mantriyogesh, 2023-07-23:
Hello,
Is this issue still observed at your end?
Had you referred https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md at your end?

zhou88mang, 2023-07-23:
it didn't appearance after i fixed my code

mantriyogesh, 2023-07-23:
Okay. I will check this tomorrow morning..

Thanks a lot for letting us know this.

mantriyogesh, 2023-07-24:
@zhou88mang 
we are checking in this issue and notify further.

### Dev notes:

1. Diff pointed by user, which resolves issues at their end
#### Diff
esp_hosted_fg/host/microcontrollers/drivers/transport/sdio/sdio_drv.c

```diff
55a56,57
> 			start_ptr += len_to_send * block_n ;
> 			len_remain -= len_to_send * block_n;
65a68,70
> 			start_ptr += len_to_send  ;
> 			len_remain -= len_to_send ;
>
75,76c80
<    	start_ptr += len_to_send;
<    	len_remain -= len_to_send;
---
>
79,80c83,84
<    *out_length = len;
<    sdio_esp_rx_bytes += len;
---
> //	*out_length = len;
> //	sdio_esp_rx_bytes += len;
```
#### Observations

2. We might need to understand why @zhou88mang has working only after this change.
3. possible that caller of this func not behaving as expected?

mantriyogesh, 2023-07-24:
@zhou88mang 
Can you please state the MCU you are using?
Also if you can point to the HAL layer oper source code for your MCU for SDIO communication will be great.

zhou88mang, 2023-07-24:
> @zhou88mang Can you please state the MCU you are using? Also if you can point to the HAL layer oper source code for your MCU for SDIO communication will be great.

my mcu is x2000
sorry , i cant provide all the code because of company regulations.
and the function "sdio_io_rw_extended" is this:
```
int sdio_io_rw_extended(struct mmc_card *card, int write, uint32_t fn,
    uint32_t addr, int incr_addr, uint8_t *buf, uint32_t blocks, uint32_t blksz)
{
    assert(card);

    struct mmc_host *mmc = card->host;
    struct mmc_cmd cmd = {0};
    struct mmc_data data = {0};

    assert(mmc);
    assert(fn <= SDIO_MAX_FUNCS);
    assert(blksz != 0);

    /* sanity check */
    if (addr & ~0x1FFFF)
        return -EINVAL;

    /* cmd */
    cmd.opcode      = SD_IO_RW_EXTENDED;
    cmd.arg         = write ? 0x80000000 : 0x00000000;
    cmd.arg         |= fn << 28;
    cmd.arg         |= incr_addr ? 0x04000000 : 0x00000000;
    cmd.arg         |= addr << 9;
    if (blocks == 0)
        cmd.arg     |= (blksz == 512) ? 0 : blksz;  /* byte mode */
    else
        cmd.arg     |= 0x08000000 | blocks;         /* block mode */

    cmd.resp_type   = MMC_RSP_R5 | MMC_CMD_ADTC;

    /* data */
    /* Code in host drivers/fwk assumes that "blocks" always is >=1 */
    data.blksz      = blksz;
    data.blocks     = blocks ? blocks : 1;

    if (write) {
        data.flags  = MMC_DATA_WRITE;
        data.src    = (char *)buf;
    } else {
        data.flags  = MMC_DATA_READ;
        data.dest   = (char *)buf;
    }

    mmc_send_cmd_data(mmc, &cmd, &data);

    if (cmd.error)
        return cmd.error;

    if (data.error)
        return data.error;

    return 0;
}
```

But I want to point this out :
I solved the lost data problem because I changed the function
```
"err = sdio_io_rw_extended(sdio_wifi_card,0,1,ESP_SLAVE_CMD53_END_ADDR - len_remain,1,start_ptr,block_n,(len_to_send + 3) & (~3));" 
```
  to 
```
 "err = sdio_io_rw_extended(sdio_wifi_card,0,1,ESP_SLAVE_CMD53_END_ADDR - len_remain,1,start_ptr,1,512);"
```

and why i changed the 
```
> 			start_ptr += len_to_send * block_n ;
> 			len_remain -= len_to_send * block_n;
65a68,70
> 			start_ptr += len_to_send  ;
> 			len_remain -= len_to_send ;
```
because if block_n > 1 ,  the next start_ptr should not be  start_ptr += len_to_send  , the next len_remain should not be len_remain -= len_to_send , they should be start_ptr += len_to_send * block_n  and len_remain -= len_to_send * block_n

SohKamYung-Espressif, 2023-08-04:
@zhou88mang To help verify the missing data issue you are seeing, could you try this:

In the slave ESP32, dump the whole SDIO buffer to be sent:

```
// print the payload data send
	// my_hex_dump(sendbuf + offset,buf_handle->payload_len);
	my_hex_dump(sendbuf, total_len);

	ret = sdio_slave_transmit(sendbuf, total_len);
	if (ret != ESP_OK) {
		ESP_LOGE(TAG , "sdio slave transmit error, ret : 0x%x\r\n", ret);
		sdio_buffer_free(sendbuf);
		return ESP_FAIL;
	}

	

	sdio_buffer_free(sendbuf);

	return buf_handle->payload_len;
```

In the host, dump the data being received by the sdio host driver:

```
		block_n = len_remain / ESP_BLOCK_SIZE;

		if (block_n != 0) {
			len_to_send = ESP_BLOCK_SIZE;
#if DEBUG_TRANSPORT
			printf("block_n %u, len-to_send %lu\n\r",block_n,len_to_send);
#endif
			// err = sdio_driver_read_blocks(SDIO_FUNC_1,
			// 		ESP_SLAVE_CMD53_END_ADDR - len_remain,
			// 		start_ptr, len_to_send, block_n);
			err = sdio_io_rw_extended(sdio_wifi_card,0,1,ESP_SLAVE_CMD53_END_ADDR - len_remain,1,start_ptr,block_n,512);

			my_hex_dump(start_ptr, 512 * block_n);

			start_ptr += 512*block_n ;
			len_remain -= 512*block_n ;
		} else {
			len_to_send = len_remain;
			/* though the driver supports to split packet of unaligned size into length
			 * of 4x and 1~3, we still get aligned size of data to get higher
			 * efficiency. The length is determined by the SDIO address, and the
			 * remaining will be ignored by the slave hardware
			 */
			// err = sdio_driver_read_bytes(SDIO_FUNC_1,
			// 		ESP_SLAVE_CMD53_END_ADDR - len_remain, start_ptr,
			// 		(len_to_send + 3) & (~3), block_n);
			err = sdio_io_rw_extended(sdio_wifi_card,0,1,ESP_SLAVE_CMD53_END_ADDR - len_remain,1,start_ptr,block_n,(len_to_send + 3) & (~3));

			my_hex_dump(start_ptr, len_remain);

			start_ptr += len_to_send;
			len_remain -= len_to_send;
		}
```

If the lost data still happens, modify the second part to your fix (where you read 512 bytes of data instead of len_to_send) and try again.

I have tested sending your test data and do not see lost data. Maybe it might be a sdio host related issue.

closed 2023-11-20 by zhou88mang