espressif/esp-hosted #252: Unable to connect ESP32-C3 with raspberry pi using Hosted-NG

Tenet420, 2023-08-08:
Problem: Error while running "bash rpi-init.sh spi"

Error while compiling:
>  Building for spi protocol
> rm: cannot remove 'spidev_disabler.dtbo': No such file or directory
> rm -rf *.o sdio/*.o spi/*.o *.ko
> make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /lib/modules/6.1.21-v8+/build M=/home/pleco-dev/esp-hosted/esp_hosted_ng/host clean
> make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
> make[1]: Entering directory '/home/pleco-dev/linux-0afb5e98488aed7017b9bf321b575d0177feb7ed'
> make[1]: Leaving directory '/home/pleco-dev/linux-0afb5e98488aed7017b9bf321b575d0177feb7ed'
> make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /lib/modules/6.1.21-v8+/build M=/home/pleco-dev/esp-hosted/esp_hosted_ng/host modules
> make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
> make[1]: Entering directory '/home/pleco-dev/linux-0afb5e98488aed7017b9bf321b575d0177feb7ed'
> warning: the compiler differs from the one used to build the kernel
>   The kernel was built by: gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
>   You are using:           arm-linux-gnueabihf-gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
>   CC [M]  /home/pleco-dev/esp-hosted/esp_hosted_ng/host/esp_bt.o
> In file included from ./include/linux/build_bug.h:5,
>                  from ./include/linux/container_of.h:5,
>                  from ./include/linux/list.h:5,
>                  from ./include/linux/timer.h:5,
>                  from ./include/linux/workqueue.h:9,
>                  from /home/pleco-dev/esp-hosted/esp_hosted_ng/host/include/esp.h:11,
>                  from /home/pleco-dev/esp-hosted/esp_hosted_ng/host/include/esp_api.h:11,
>                  from /home/pleco-dev/esp-hosted/esp_hosted_ng/host/esp_bt.c:9:
> ./include/linux/compiler.h:246:10: fatal error: asm/rwonce.h: No such file or directory
>   246 | #include <asm/rwonce.h>
>       |          ^~~~~~~~~~~~~~
> compilation terminated.
> make[2]: *** [scripts/Makefile.build:250: /home/pleco-dev/esp-hosted/esp_hosted_ng/host/esp_bt.o] Error 1
> make[1]: *** [Makefile:2012: /home/pleco-dev/esp-hosted/esp_hosted_ng/host] Error 2
> make[1]: Leaving directory '/home/pleco-dev/linux-0afb5e98488aed7017b9bf321b575d0177feb7ed'
> make: *** [Makefile:61: all] Error 2

configuration: 
ESP32C3 - flashed with "ESP-Hosted-NG release 1.0.2" binary file using esptool.py for SPI 

RaspberryPi: Using Hosted_ng mode with Only SPI connection for Wifi and Bluetooth. 

Possible error: 
the kernel that currently gets installed by "`sudo apt install raspberrypi-kernel-headers`" is different or has been upgraded to a version that is not compatible to run it by default.

mantriyogesh, 2023-08-09:
If in case headers mismatch, please check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Linux_based_readme.md#211-raspberry-pi-specific-setup

This is common problem at rpi to mismatch headers.
Please check rpi_source method from above link.

kapilkedawat, 2023-10-06:
@Tenet420 Please let us know if you are still facing this issue.