espressif/esp-hosted #253: Buildroot now supports esp-hosted

giuliobenetti, 2023-08-09:
FYI I've added esp-hostead package to buildroot:
https://gitlab.com/buildroot.org/buildroot/-/commit/7b2e5e6550aed4f79461919c6a59e00741eb4b11

Feel free to test it.

Best regards

ForgetCSX, 2024-07-03:
Hello @giuliobenetti , I made a Linux board, using the executable compiled by buildroot, now I am ready to use esp32 as wifi and bluetooth, but I don't know how to add esp-hosted as a package to buildroot, can you help make a tutorial for porting, please forgive me for being a buildroot newbie

giuliobenetti, 2024-07-03:
Hello @ForgetCSX, I think it's pretty easy to enable esp-hosted package in Buildroot.

After setting up the _defconfig you only need to issue `make menuconfig`, then you can search for esp-hosted that is under "Target Packages -> Hardware Handling -> esp-hosted". Select it with space-bar, Save while Exit and issue `make` again.

After that you should find .ko module under /lib/modules/...

Note that esp-hosted in Buildroot is only meant as the Linux Driver, so you have to load firmware separately. You can also load it using package python-esptool, so I suggest you to enable it in Buildroot for building.

Let me know if it works.

Best regards

ForgetCSX, 2024-07-04:
Hello, I followed your submission on buildroot, ported to my project, compilation error, my linux version is linux-5.4.99

ForgetCSX, 2024-07-04:
@giuliobenetti Here's my console info：
xiang@ubuntu:~/linux/f1c200s-sdk$ make -j2
/usr/bin/make -j1 O=/home/xiang/linux/f1c200s-sdk/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: warning: -j1 forced in submake: resetting jobserver mode.
>>> esp-hosted 11e90f185a5a9922343e73e44032816a0e3ddedb Downloading
--2024-07-04 01:50:49--  https://github.com/espressif/esp-hosted/archive/11e90f185a5a9922343e73e44032816a0e3ddedb/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb.tar.gz
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/espressif/esp-hosted/tar.gz/11e90f185a5a9922343e73e44032816a0e3ddedb [following]
--2024-07-04 01:50:50--  https://codeload.github.com/espressif/esp-hosted/tar.gz/11e90f185a5a9922343e73e44032816a0e3ddedb
Resolving codeload.github.com (codeload.github.com)... 20.205.243.165
Connecting to codeload.github.com (codeload.github.com)|20.205.243.165|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘/home/xiang/linux/f1c200s-sdk/output/build/.esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb.tar.gz.0Ra6kI/output’

/home/xiang/linux/f1c200s-sdk/output/build/.esp-hos     [       <=>                                                                                                        ]  15.30M  11.3MB/s    in 1.4s

2024-07-04 01:50:52 (11.3 MB/s) - ‘/home/xiang/linux/f1c200s-sdk/output/build/.esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb.tar.gz.0Ra6kI/output’ saved [16040282]

esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb.tar.gz: OK (sha256: 3310845b2324bda5774b91bcdd3f47621126f4a0011a0f61e31b0ad4a44db2f5)
>>> esp-hosted 11e90f185a5a9922343e73e44032816a0e3ddedb Extracting
gzip -d -c /home/xiang/linux/f1c200s-sdk/dl/esp-hosted/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb.tar.gz | /home/xiang/linux/f1c200s-sdk/output/host/bin/tar --strip-components=1 -C /home/xiang/linu                          x/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb   -xf -
>>> esp-hosted 11e90f185a5a9922343e73e44032816a0e3ddedb Patching
>>> esp-hosted 11e90f185a5a9922343e73e44032816a0e3ddedb Configuring
>>> esp-hosted 11e90f185a5a9922343e73e44032816a0e3ddedb Building
>>> esp-hosted 11e90f185a5a9922343e73e44032816a0e3ddedb Building kernel module(s)
PATH="/home/xiang/linux/f1c200s-sdk/output/host/bin:/home/xiang/linux/f1c200s-sdk/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" PKG_CON                          FIG="/home/xiang/linux/f1c200s-sdk/output/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/xiang/linux/f1c200s-sdk/out                          put/host/lib/pkgconfig:/home/xiang/linux/f1c200s-sdk/output/host/share/pkgconfig" BR_BINARIES_DIR=/home/xiang/linux/f1c200s-sdk/output/images KCFLAGS=-Wno-attribute-alias /usr/bin/make  -C /home/xiang/linux                          /f1c200s-sdk/output/build/linux-5.4.99 HOSTCC="/usr/bin/gcc -O2 -I/home/xiang/linux/f1c200s-sdk/output/host/include -L/home/xiang/linux/f1c200s-sdk/output/host/lib -Wl,-rpath,/home/xiang/linux/f1c200s-sdk/o                          utput/host/lib" ARCH=arm INSTALL_MOD_PATH=/home/xiang/linux/f1c200s-sdk/output/target CROSS_COMPILE="/home/xiang/linux/f1c200s-sdk/output/host/bin/arm-buildroot-linux-gnueabi-" DEPMOD=/home/xiang/linux/f1c2                          00s-sdk/output/host/sbin/depmod INSTALL_MOD_STRIP=1 target=spi PWD=/home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host M=/home/xiang/linux/f1c20                          0s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host modules
make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  CC [M]  /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/main.o
  CC [M]  /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/esp_wpa_utils.o
  CC [M]  /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/spi/esp_spi.o
  LD [M]  /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/esp32_spi.o
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "hci_register_dev" [/home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/esp32_spi.ko] undefined!
[... cut ...]
ERROR: "hci_free_dev" [/home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/esp_hosted_ng/host/esp32_spi.ko] undefined!
make[3]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[2]: *** [Makefile:1639: modules] Error 2
make[1]: *** [package/pkg-generic.mk:270: /home/xiang/linux/f1c200s-sdk/output/build/esp-hosted-11e90f185a5a9922343e73e44032816a0e3ddedb/.stamp_built] Error 2
make: *** [Makefile:84: _all] Error 2
xiang@ubuntu:~/linux/f1c200s-sdk$ ls package/esp-hosted/
Config.in  esp-hosted.hash  esp-hosted.mk

mantriyogesh, 2024-07-04:
Hello @ForgetCSX ,

Errors as hci_* definitions not found is not buildroot issue.
Do you have bluetooth kernel module or definitions allowed?

First of all do you need bluetooth?
If not, for now workaround is to comment these calls.

Proper fix is pending since long from us.
Fix would be to add new Makefile config or module param to disable bluetooth altogether.

ForgetCSX, 2024-07-04:
Hello@mantriyogesh, yes I want my system to support Bluetooth, I am currently working on an 86 panel, I expect to be able to use SPI to support both WiFi and Bluetooth, because my serial port is occupied by the flow control pin, but I still connect the serial port to ESP32, this as an alternative, should I add some HCI support to the kernel?
Here's my board：
[image]

Mr-Bossman, 2024-07-04:
@ForgetCSX I can take a look tomorrow (~12hr from now). Also Fun! an f1c200s.

ForgetCSX, 2024-07-04:
@Mr-Bossman Thank you for your reply,This chip is really fun,Now the support is relatively complete,USB,Audio input and output,LCD,IIC and SPI can be used normally,Now if you port WiFi and Bluetooth up, it will be more perfect,If I have enough ability, I will open source this software and hardware。

Mr-Bossman, 2024-07-04:
@ForgetCSX wifi should work i tried with a usb wifi adapter a while ago. run `make linux-menuconfig` an enable wifi and Bluetooth it should fix the error

giuliobenetti, 2024-07-04:
@ForgetCSX which Buildroot version are you using? It should automatically enable the correct drivers when you select esp-hosted. Because there you can choose which SoC and interface to use.

ForgetCSX, 2024-07-04:
@giuliobenetti https://github.com/ForgetCSX/f1c200s-sdk.git This is the buildroot I'm using now, which is an early version and doesn't join esp-hosted

Mr-Bossman, 2024-07-04:
@ForgetCSX Typically you use br2-external so you can easily update ur buildroot version. <https://buildroot.org/downloads/manual/manual.html#customize>

here are some examples
<https://github.com/Mr-Bossman/bpi-f3-buildroot> 
<https://github.com/Mr-Bossman/licheepi-nano-buildroot>

Mr-Bossman, 2024-07-04:
@ForgetCSX type `make linux-menuconfig` navigate to "Networking support -> Bluetooth subsystem support"  `CONFIG_BT`  Select it with space-bar, Save while Exit and run `make linux-rebuild`  then `make` that worked for me

ForgetCSX, 2024-07-06:
@Mr-Bossman @giuliobenetti @mantriyogesh Good afternoon, I opened "Bluetooth subsystem support" and the compilation was successful, but it seems that esp32_spi.ko was not compiled correctly.
[image]
[image]
In menuconfig, the interface I chose is spi, but the driver actually compiled is sdio. Is it because the parameters are not correctly passed into the Makefile?
Another point is, how should I write the device tree to make the kernel match this driver? Apart from reset, I did not find any other IO applications in the host driver. Is it okay for the device tree to be similar to the following?
`    esp32spi@0 {
        **compatible = "espressif,esp32-spi";**
        reg = <0>; /* CS #0 */
        spi-max-frequency = <24000000>;
		cs = <&pio 4 8 GPIO_ACTIVE_HIGH>; // PE7
		hl = <&pio 4 7 GPIO_ACTIVE_HIGH>; // PE6
		interrupt= <&pio 4 6 IRQ_TYPE_EDGE_FALLING>; // PE6
    };`

Mr-Bossman, 2024-07-06:
@ForgetCSX run `make esp-hosted-rebuild` to rebuild esp-hosted as the config changed. then `make`.
So I made a patch to get spi to match on the device tree quite a while ago but it never got merged. You will need to edit these lines for HANDSHAKE_PIN, SPI_DATA_READY_PIN, chip_select, and bus_num. you will probably also need to set the SPI speed with the kernel module parameter.
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/spi/esp_spi.c#L423
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/spi/esp_spi.h#L12

mantriyogesh, 2024-07-06:
Hello @Mr-Bossman , can you please point to the device tree change?
If is is generic change (platform agnostic), we can definitely take. If the change points to specific GPIO or some specific Linux, which is generally nature of Linux it could be added as documentation instead of code with some explanation.

giuliobenetti, 2024-07-06:
> it seems that esp32_spi.ko was not compiled correctly.

@ForgetCSX I've just successfully built esp32_spi.ko using Buildroot menuconfig by choosing SPI as interface and ESP32 as SoC. So it works for me. Linux version is 6.9.4

ForgetCSX, 2024-07-08:
@Mr-Bossman Hello, thanks for trying, but migrating to the new version is difficult for me at the moment, there is still a lot to learn, with the help of Mr-Bossman I executed "make esp-hosted-rebuild" can already generate the driver, given that the current esp-hosted has compiled normally, I will focus on getting the driver to work, if you want me to continue testing, I am also happy to try.

ForgetCSX, 2024-07-12:
@giuliobenetti @Mr-Bossman Good Friday,After the driver was compiled normally last time,I tried to load the driver,But it's very confusing.,This driver is directly solidified in the io-info driver.,So it's definitely me.,I got an error that the application for io failed.,It didn't use of_match_table or of_device_id this set to get io-info from the device tree.,Of course, this may not be necessary.,But it's becoming very unfamiliar to me.,Haha,But for the time being, I didn't care about that., I tried to replace the io solidified in the code directly with the io of the hardware connection, and then tried to load the driver again, obviously, this time I didn't get any errors, but unexpectedly there were no WiFi-related nodes generated under the dev directory, can you provide some help with testing, thank you very much.

Here's the information output from the console:
# ls
esp32_spi.ko
# modprobe esp32_spi.ko
[ 1921.571265] esp32_spi: spi_dev_init: esp32_spi: No dev using spi!
[ 1921.578032] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [135], SPI Clock [10]
# ls /dev/
audio            snd              tty31            tty59
block            stderr           tty32            tty6
cedar_dev        stdin            tty33            tty60
char             stdout           tty34            tty61
console          tty              tty35            tty62
cpu_dma_latency  tty0             tty36            tty63
disk             tty1             tty37            tty7
dsp              tty10            tty38            tty8
fb0              tty11            tty39            tty9
fd               tty12            tty4             ttyS0
ffs-mtp          tty13            tty40            ttyS1
full             tty14            tty41            ttyS2
gpiochip0        tty15            tty42            ttyS3
i2c-0            tty16            tty43            ttyS4
input            tty17            tty44            ttyS5
ion              tty18            tty45            ttyS6
kmsg             tty19            tty46            ttyS7
log              tty2             tty47            urandom
mem              tty20            tty48            vcs
mixer            tty21            tty49            vcs1
mmcblk0          tty22            tty5             vcsa
mmcblk0p1        tty23            tty50            vcsa1
mmcblk0p2        tty24            tty51            vcsu
mmcblk0p3        tty25            tty52            vcsu1
null             tty26            tty53            watchdog
ptmx             tty27            tty54            watchdog0
pts              tty28            tty55            zero
random           tty29            tty56
rfkill           tty3             tty57
shm              tty30            tty58
# rmmod esp32_spi.ko
# rmmod esp32_spi.ko
rmmod: can't unload module 'esp32_spi': No such file or directory
# modprobe esp32_spi.ko
[ 1943.871341] esp32_spi: spi_dev_init: esp32_spi: No dev using spi!
[ 1943.878115] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [135], SPI Clock [10]
# rmmod esp32_spi.ko
# modprobe esp32_spi.ko
[ 1950.571210] esp32_spi: spi_dev_init: esp32_spi: No dev using spi!
[ 1950.577975] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [135], SPI Clock [10]
#

ForgetCSX, 2024-07-12:
It's weird that after I manually reset the esp32, he outputs the following log，Looks like the load was successful.

[   78.104048] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   78.111073] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   78.117431] esp32_spi: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SPI
[   78.125071] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   78.131472] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   78.137817] esp32_spi: process_event_esp_bootup: Bootup Event tag: 4
[   78.144200] esp32_spi: process_event_esp_bootup: Unsupported tag=4 in bootup event
[   78.151791] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   78.158131] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   78.164789] esp32_spi: print_reset_reason: POWERON_RESET
[   78.170100] esp32_spi: check_esp_version: ESP Firmware version: 1.0.2
[   78.181153] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   83.280600] esp32_spi: wait_and_decode_cmd_resp: Command[0xF] timed out
[   83.287245] esp32_spi: cmd_get_tx_power: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   83.510683] esp32_spi: init_bt: ESP Bluetooth init
[   83.521059] esp32_spi: print_capabilities: Capabilities: 0xf8. Features supported are:
[   83.528980] esp32_spi: print_capabilities:    * WLAN on SPI
[   83.534570] esp32_spi: print_capabilities:    * BT/BLE
[   83.539621] esp32_spi: print_capabilities:      - HCI over SPI
[   83.545409] esp32_spi: print_capabilities:      - BT/BLE dual mode
[   94.973822] esp32_spi: esp_inetaddr_event: NETDEV_UP interface espsta0 ip changed to  169.254.226.202

mantriyogesh, 2024-07-12:
Hello @ForgetCSX please raise another issue instead of spilling into this one.

If your kernel module is inserted fine buildroot should be already working.
Please check common errors listed as part of [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md)

Mr-Bossman, 2024-07-12:
@ForgetCSX https://github.com/espressif/esp-hosted/issues/424 this person is also doing the same thing as you btw

mantriyogesh, 2024-07-12:
@Mr-Bossman,

It's important to note that there can be significant differences when it comes to lower layer drivers. I personally would not treat cases as the same unless they involve exactly the same hardware (SoC). Even with the same SoC, numerous other factors, such as issues with jumper cables, can come into play.

@ForgetCSX,

Since your build root is working fine, I suggest raising a new issue to handle this further. We're currently flooding this post, making it more complicated for someone to address later. You can reference this issue in the new post so that anyone who finds it relatable can easily refer to your new issue.